u-blox/ubxlib #111: How to create a DTLS socket

jraats, 2023-06-23:
Hi, I'm using your library to create a UDP socket, but I can't find any example to create a secure (DTLS) UDP socket. I'm using this library in zephyr (i've created my own network zephyr ublox driver). 
The calls are:
```
uNetworkInterfaceUp(ctx->devHandle, U_NETWORK_TYPE_CELL, &gNetworkCfg)
int32_t sock_id;

if (sock->ip_proto == IPPROTO_UDP || sock->ip_proto == IPPROTO_DTLS_1_2) {
sock_id = uSockCreate(g_ctx.devHandle, U_SOCK_TYPE_DGRAM, U_SOCK_PROTOCOL_UDP);
} else {
sock_id = uSockCreate(g_ctx.devHandle, U_SOCK_TYPE_STREAM, U_SOCK_PROTOCOL_TCP);
}
if (sock_id < 0) {
LOG_ERR("Failed to create network socket.\n");
modem_socket_put();
return -1;
}

sock->id = sock_id;
if (uSockSecurity(sock->id, &g_ctx.tls_settings) < 0) {
return -1;
}
```

In the tls_settings struct are the following fields set:
- Loaded it with: U_SECURITY_TLS_SETTINGS_DEFAULT
- tls_settings.psk.pBin -> "SomeSecretText..."
- tls_settings.psk.size = 16
- tls_settings.pskId.pBin -> "MyIdentity"
- tls_settings.pskId.size = 10

Below is the AT input/output:
```
AT+USOCR=17
+USOCR: 0
OK
U_SOCK: socket created, descriptor 0, network handle 0x2000d810, socket handle 0.
AT+USECPRF=0
OK
AT+USECPRF=0,11,0
OK
AT+USECPRF=0,8,"SomeSecretText.."
OK
AT+USECPRF=0,9,"MyIdentity"
OK
AT+USECPRF=0,0,0
OK
AT+USOSEC=0,1,0
OK
AT+USOST=0,"xxx.xxx.xxx.xxx",5684,67
<<THE ACTUAL DATA..............>>
+USOST: 0,67
```

In wireshark i see a DTLS connection, but the payload is not encrypted at all. Can you tell me which additional call I need to make?

Thanks!
Joost

RobMeades, 2023-06-23:
Hi there, and thanks for posting.

When you say "In wireshark I see a DTLS connection, but the payload is not encrypted at all", where are you looking with Wireshark?  Remember that the encryption is being carried out inside the cellular module itself, anything you see on the AT interface will be in plain-text (either on its way into the cellular module, which will DTLS-it and transmit it over cellular, or coming out of the cellular module after the cellular module has de-DLTS'ed it from the server-end).

For Wireshark to see encrypted data you would need to be looking on the server side, either before the server's IP stack gets the data from the cellular device to de-DTLS's it or, in the "return" direction, after the server's IP stack has DTLS'ed the data that the server-side-user has asked it to transmit to the cellular device.

RobMeades, 2023-06-23:
Thanks for the context, that's a lot clearer.  I'm no particular expert on DTLS operation myself: since `ubxlib` is just passing parameters to the module we don't tend to think about these things a great deal, the module should do what the module does, if you see what I mean.  So we may need to engage others here, but a few thoughts:

- when you say that you "see a DTLS connection", what makes you say that: I guess something visible in Wireshark, a DTLS negotiation phase of some form, but no actual data?
- how are you entering the PSK for the DTLS connection on the laptop-side?  I guess there must be a list of PSK IDs versus PSK's given to the SSL layer of the laptop that it can match with the PSK/PSK IDs you've put into `uSecurityTlsSettings_t`?
- concerning PSK, the AT manual says (section 22.3.3.1) "Pre-shared key used for connection: defines a pre-shared key and key-name (PSK), when a `TLS_PSK_*` cipher suite is used": might it be that you need to specifically select one of the cipher schemes that support PSK (it is in the name of the cipher suite) in order for a PSK to be usable (you'd provide one of these in the `cipherSuites` field of `uSecurityTlsSettings_t`)?

It seems most likely that the DTLS negotiation phase isn't actually succeeding, which is why you never see any data, but as with any security procedure, there are about a million ways to do it wrong and only one way to do it right, and it won't give hints.

jraats, 2023-06-27:
I've a little update. I understand that this library sends AT commands, but doesn't include the modem firmware. I've read alot in the datasheet and here are some findings.
- I didn't use `uSockConnect`, because I thought UDP was stateless and doesn't have an active connection, but for DTLS this is required. That was the reason I didn't saw a handshake and only UDP packages which didn't reach the server. (Your point nr 1)
- The next problem was that the handshake fails. Apparently the modem sends all its default cipher suits, but doesn't filter only the configured ones (so in my example PSK). So I do expect that it will send only the default PSK cipher suites, but in reality it sends also the certificate cipher suites. The server can choose one of the given cipher suites and the server always chooses a certificate cipher suite, whereby the modem response with an invalid (not configured) certificate.
- I've tried to select one of the `TLS_PSK_*` ciper suites, but this still gives some errors (can't really say what is going wrong, because sometimes the modem gives `error` back and sometimes the server tries to select a different cipher after the handshake).

I think I have to find the last pieces myself. Thanks for thinking/brainstorming with me!

RobMeades, 2023-06-27:
Glad you've made progress in understanding what's going on, always the long-haul of any debug process. These security issues are really difficult to crack, so good luck with it and do contact us again if there is more we can do.

> `uSockConnect()`, because I thought UDP was stateless and doesn't have an active connection, but for DTLS this is required

I didn't know that myself, good information!

jraats, 2023-06-29:
Hi @RobMeades 
I've finally made the connection working.
Below is the callstack in order to create a DTLS connection (to support others in reaching the same goal):
- `uNetworkInterfaceUp` -> In order to start the Cellular network.
- `uSockCreate(devHandle, U_SOCK_TYPE_DGRAM, U_SOCK_PROTOCOL_UDP);` -> to create the UDP socket
- Start of tls_settings:
```
uSecurityTlsSettings_t tls_settings = U_SECURITY_TLS_SETTINGS_DEFAULT
tls_settings.psk.pBin = psk->buf;
tls_settings.psk.pBin = psk->len;
tls_settings.pskId.pBin = pskId->buf;
tls_settings.pskId.size = pskId->len;
```
- Like I said in my previous message. The modem have some kind of bug where it will send all its default cipher suite instead of only the configured ciper suites. So I use this piece of code to filter the cipers (for now only one ciper is OK, because I also manage the server, but it would be more convenient to support all PSK cipher formats):
```
size_t i = 0;
tls_settings.cipherSuites.suite[i++] = U_SECURITY_TLS_CIPHER_SUITE_PSK_WITH_AES_128_CCM_8;
tls_settings.cipherSuites.num = i;
```
- `uSockSecurity(sock->id, tls_settings);` -> add security to the socket
- `uSockConnect(sock->id, &address);` -> don't forget. DTLS requires a connection
- And now sending and receiving requires AT+USORD and AT+USOWR. (I know this is normally for TCP), but see the following in the datasheet:
https://content.u-blox.com/sites/default/files/SARA-R5_ATCommands_UBX-19047455.pdf page 308:

> SARA-R5
• The +USORF AT command should not be used for a DTLS connection, that is when the connection
has been configured using the +USOCO and the +USOSEC AT commands. For DTLS connections the
+USOCO, +USOSEC, +USORD and +USOWR AT commands need to be used.

So that means I should use `uSockRead` instead of `uSockReceiveFrom`. However in your library you have a check https://github.com/u-blox/ubxlib/blob/66d4e7ac8de33b38d0a42a4b2afdc4fbdcd0a914/common/sock/src/u_sock.c#L1082 which will enforce us to use `USORF` instead of `USORD`. I've removed the check in my local repository and now the DTLS connection is working.

I don't know if this is only for the SARA-R5 and how you want to implement this. I think the cleanest way would be a flag per socket which will keep track if the socket is secure. Then you can create something like:
```
        if (pContainer->socket.protocol == U_SOCK_PROTOCOL_UDP && pContainer->socket.secure == false) {
            // UDP style
            if (devType == (int32_t) U_DEVICE_TYPE_CELL) {
                negErrnoOrSize = uCellSockReceiveFrom(devHandle,
                                                      sockHandle,
                                                      pRemoteAddress,
                                                      pData,
                                                      dataSizeBytes);
            } else if (devType == (int32_t) U_DEVICE_TYPE_SHORT_RANGE) {
                negErrnoOrSize = uWifiSockReceiveFrom(devHandle,
                                                      sockHandle,
                                                      pRemoteAddress,
                                                      pData,
                                                      dataSizeBytes);
            }
        } else {
            // TCP style (or UDP DTLS)
            if (devType == (int32_t) U_DEVICE_TYPE_CELL) {
                negErrnoOrSize = uCellSockRead(devHandle,
                                               sockHandle,
                                               pData,
                                               dataSizeBytes);
            } else if (devType == (int32_t) U_DEVICE_TYPE_SHORT_RANGE) {
                negErrnoOrSize = uWifiSockRead(devHandle,
                                               sockHandle,
                                               pData,
                                               dataSizeBytes);
            }
        }
```

What do you think?

RobMeades, 2023-06-29:
Well found indeed!  It was obviously a mistake not to have a specific test for DTLS operation, I'd entirely missed the subtlety of that note in the interface manual.  Agree that a per-socket flag is the right approach.

Next week I will see if I can set up a version of our echo server that supports DTLS echo, then I can make the change and test it; will report back here.

Thanks for doing this, and also apologies that you had to do this.

RobMeades, 2023-07-07:
Update: I am doing this but I'm having a bit of a fight with the Golang on the server-side of our test system since the DLTS module I'm using (`pion`), it turns out, doesn't support any cipher suites for use with RSA-signed certificates, so I will need to regenerate the certs with ECDSA signing etc. to find a hole on the Venn diagram, and I'm on holiday next week.  Sorry!

RobMeades, 2023-07-10:
Update: fix is now done, tested and in review, should be here shortly.

jraats, 2023-07-11:
Thanks for the quick implementation. This issue is not blocking, because I have a small workaround. But when the update is available I will update the lib :-) .
Thanks for all the good work on this library.

RobMeades, 2023-07-11:
Just waiting on a review from @philwareublox (hint, hint Phil :-)).

RobMeades, 2023-07-12:
Fix now pushed here in commit 702834421f80eb322108f2c244c638bc442bfa55.  A [new example](https://github.com/u-blox/ubxlib/blob/master/example/sockets/main_dtls.c) is also added to show the DTLS usage pattern.

Gonna close this issue: please re-open or start another if required.

closed 2023-07-12 by RobMeades