u-blox/ubxlib #112: ZED-F9P UART2

maleroy, 2023-07-03:
Hello,

Since FW 1.30 was released for the ZED-F9P, UBX messages can be exchanged via UART2. I've noticed however that ubxlib is still assuming only UART1 can be used to communicate with the chip. Indeed, the port number that is defined in [u_gnss.c](https://github.com/u-blox/ubxlib/blob/master/gnss/src/u_gnss.c#L316) is an alias to the UART1 defined in [u_gnss_type.h](https://github.com/u-blox/ubxlib/blob/master/gnss/api/u_gnss_type.h#L207).

The changes seem easy to integrate, but I didn't do a PR as I am relatively new to ubxlib, and wasn't sure where the best place to add the UART port setting would be.

Thanks in advance for your reply :)

RobMeades, 2023-07-03:
Hi there, and thanks for pointing this out.  I've pushed a change which should allow you to do what you want to do to a preview branch here:

https://github.com/u-blox/ubxlib/tree/preview_gnss_uart_2

I've tested that it doesn't break anything else but we don't have a ZED-F9P connected to an MCU via UART 2 in our test system, so please give it a try and let me know if it works for you.  If it does, I'll add this to `master` and here.

maleroy, 2023-07-04:
Hi Rob,

Thanks for the quick action! It works on my side, so if you would push that to master that would be awesome!

Thanks again and have a good one :)

RobMeades, 2023-07-04:
The change is now on `master` here: https://github.com/u-blox/ubxlib/commit/a417799e08617d1c565ce29665006f4adfc39106.

Closing this issue, let us know if you have more.

closed 2023-07-04 by RobMeades