u-blox/ubxlib #113: Unable to register with the network

alexmaron81, 2023-07-19:
Hello everyone,
I haven't worked on the project for a long time.
Now I wanted to establish a connection again and get "unable to register with the network"

AT
AT
AT
AT
AT
[00]U_CELL_PWR: powering on.
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

M0.10.00,A.02.14

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K3

OK
AT+UPSV=4

OK
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 0

OK
AT+CFUN=4

OK
Opened device with return code 0.
Bringing up the network...
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

901405102306300

OK
U_CELL_NET: user-specified APN is "iot.1nce.net".
AT+CGDCONT=1,"IP","iot.1nce.net"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
AT+CFUN=1

+CME ERROR: operation not supported
U_CELL_NET: unable to register with the network, is APN "iot.1nce.net" correct and is an antenna connected?
AT+CFUN=4

OK
U_CELL_NET: connection attempt stopped after 1 second(s).
Unable to bring up the network!
Done.

RobMeades, 2023-07-19:
Hi Alex: getting `+CME ERROR: operation not supported` back from `AT+CFUN=1` is quite bizarre: I assume this is consistent?

alexmaron81, 2023-07-19:
i just tried the ubxlib library from before (which i used 07.2022), the connection could be established immediately.

AT
AT
AT
AT
AT
[00]U_CELL_PWR: powering on.
[00]AT
AT
AT
AT
AT
[00]AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

M0.10.00,A.02.14

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K3

OK
AT+UPSV=4

OK
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+CFUN=4

OK
Opened device with return code 0.
Bringing up the network...
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

901405102306300

OK
U_CELL_NET: user-specified APN is "iot.1nce.net".
AT+CGDCONT=1,"IP","iot.1nce.net"

OK
AT+CFUN=1

OK
U_CELL_NET: registering on 26201...
AT+COPS=1,2,"26201"

OK

+CREG: 5,"FFFE","22A9B00", 7

+CGREG: 5,"FFFE","22A9B00",7

+CEREG: 5,"BCC","8","22A9B00",7,,,"00000000","01100000"
7: RegR
7: RegR
7: RegR
AT+CREG?

+CREG: 2,5,"FFFE","22A9B00",7

OK
7: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 1,0,"262 01 1nce.net",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
U_CELL_NET: connected after 6 second(s).

alexmaron81, 2023-07-19:
I have now tried the following again and the connection can be established again.
https://github.com/u-blox/ubxlib/issues/25#issuecomment-892903911

AT
AT

OK
U_CELL_PWR: powering on, module is already on.
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

M0.10.00,A.02.14

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K3

OK
AT+UPSV=4

OK
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 0

OK
AT+CFUN=4

OK
Opened device with return code 0.
Bringing up the network...
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

901405102306300

OK
U_CELL_NET: user-specified APN is "iot.1nce.net".
AT+CGDCONT=1,"IP","iot.1nce.net"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
0: NReg

+CREG: 2

+CGREG: 2
-1: Search
-1: Search

+CEREG: 2,,,,,,,"00000000","01100000"
-1: Search
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?

+CEREG: 4,2

OK
0: Search
AT+CREG?

+CREG: 2,2

OK
0: Search

+CREG: 5,"FFFE","22A9B00", 7

+CGREG: 5,"FFFE","22A9B00",7

+CEREG: 5,"BCC","8","22A9B00",-1: RegR
-1: RegR
7,,,"00000000","01100000"
7: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"262 01 1nce.net",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
U_CELL_NET: connected after 3 second(s).

alexmaron81, 2023-07-19:
now I keep getting "unable to register with the network" despite the change

RobMeades, 2023-07-19:
For clarification, does an AT log still show `+CME ERROR: operation not supported` in response to `AT+CFUN=1, or is the "unable to register with the network" occurring for other reasons?  Best post a new AT log showing your current status.

RobMeades, 2023-07-19:
One thing that our AE's have pointed out, looking at your log, is that it would be better if you were to switch to using MNO profile 100. You would do that with something like:

```
int32_t mnoProfile = uCellCfgGetMnoProfile(deviceHandle);
if (mnoProfile >= 0 && mnoProfile != 100) {
    uCellCfgSetMnoProfile(deviceHandle, 100);
}
```

alexmaron81, 2023-07-20:
I have set the MNO to 100, but unfortunately without success.

AT
AT
AT
AT
U_CELL_PWR: powering on.
[00]AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

M0.10.00,A.02.14

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K3

OK
AT+UPSV=4

OK
AT+CPSMS?

+CPSMS:1,,,"00011000","00001010"

OK
AT+CEDRXS?

+CEDRXS: 2,"0010"
+CEDRXS: 4,"0010"
+CEDRXS: 5,"0010"

OK
AT+CEDRXS=2,2

OK
AT+CEDRXS=2,4

OK
AT+CEDRXS=2,5

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+CFUN=4

OK
Opened device with return code 0.
Bringing up the network...
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

901405103271762

OK
U_CELL_NET: user-specified APN is "iot.1nce.net".
AT+CGDCONT=1,"IP","iot.1nce.net"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
AT+CFUN=1

+CME ERROR: operation not supported
U_CELL_NET: unable to register with the network, is APN "iot.1nce.net" correct and is an antenna connected?
AT+CFUN=4

OK
U_CELL_NET: connection attempt stopped after 1 second(s).
Unable to bring up the network!
Done.

RobMeades, 2023-07-20:
Thanks for the AT log: getting `+CME ERROR: operation not supported` back from `AT+CFUN=1` (i.e. powering  up the radio) is not something I've ever seen before and, unfortunately, not something `ubxlib` can help out with.  Do you have any other copies of the HW, in case this one is broken somehow, or is this the only one?

alexmaron81, 2023-07-20:
I have the same problem with other hardware.
Can I turn off "U_CELL_NET: setting automatic network selection mode..." somehow?
I don't remember having this with the previous ubxlib version.

RobMeades, 2023-07-20:
Nothing has changed in this area of course: note that it is not putting the module into automatic mode, just checking that it is _already_ in automatic mode, which is the same as it did in your success-case log:

```
U_CELL_NET: setting automatic network selection mode...
AT+COPS?
+COPS: 0
OK
AT+CFUN=1
OK
```

I'm trying to understand what is different between your success case and your fail case.

RobMeades, 2023-08-14:
@alexmaron81: I think you said that this is no longer an issue hence I'm going to close this - please re-open if you disagree.

closed 2023-08-14 by RobMeades