u-blox/ubxlib #116: uLocationGetContinuousStart doesn't throw errors

PetervdPerk-NXP, 2023-07-20:
When using `uLocationGetContinuousStart` function to get a fix using the callback we don't get errors such as timeout or I/O failures.

Device errors aren't coming through e.g. we can disconnect the antenna and don't get a timeout. And we can even disconnect the GNSS completely and we don't get an error.

RobMeades, 2023-07-20:
Hi there.  Yes, [checking the code](https://github.com/u-blox/ubxlib/blob/2fceea6e84731d99fb4d023efd7ad32d99d24987/gnss/src/u_gnss_pos.c#L181) the user callback is only called if a message is received from the GNSS device indicating that a successful-and-at-least-2D position fix has been achieved.

I guess it could be changed to call the user callback if it is able to parse the message from the GNSS device, irrespective whether a position fix has actually been achieved or not, leaving it up to the user to decide what to do with the results.  Would that be better?

This would constitute a [minor] interface change, so I would need to justify it with a good use-case.

PetervdPerk-NXP, 2023-07-20:
When using it in a control system we would also like to know when we get a poor fix or no fix at all.

More particularly if an errors occurs i.e. for some reason the GNSS gets power cycled we could hypothetically restart ubxlib to reconfigure the GNSS.

RobMeades, 2023-07-20:
Understood.  Find here:

https://github.com/u-blox/ubxlib/tree/preview_fix_gnss_streamed_error_case_rmea

...a preview branch which should report any navigation fix received from the GNSS device, irrespective of whether it contains a fix or not.  Let me know if is is closed to what you want and I will push the change to `master` here, after it has passed testing, and delete the preview branch.

Of course, this still only calls the callback when the GNSS device has sent something.  That's kind of the nature of thte continuous API: the GNSS chip is asked to send navigation fixes at a given periodicity and if it doesn't, well, it doesn't!  The `ubxlib` code is simply passing on what it gets, IYSWIM.

PetervdPerk-NXP, 2023-07-20:
Well this is actually what I was looking for, so I can get a signal when our fix isn't good enough and we if don't get anything it means something is wrong. Thanks!

RobMeades, 2023-07-20:
Good stuff; will let you know when it's here, gotta tweak a few tests to expect the new behaviour first.

RobMeades, 2023-07-21:
You can now find the change on `master` here: https://github.com/u-blox/ubxlib/commit/efdc592cb2aa840900e1c6591d00124f41eb8867, I will delete the [preview branch](https://github.com/u-blox/ubxlib/tree/preview_fix_gnss_streamed_error_case_rmea) later today.

closed 2023-07-21 by RobMeades