u-blox/ubxlib #117: Zephyr: UART vs. USART

potentialdiffer, 2023-07-24:
Hi together,

I just ran into an issue, where the initialization of the uart port fails, as some MCUs have `usart` labels instead of `uart`. E.g. configuring `usart1` is not possible because of the way it is currently implemented: https://github.com/u-blox/ubxlib/blob/master/port/platform/zephyr/src/u_port_uart.c#L341

This holds true e.g. for STM32U5 MCUs.

I don't know if this needs to be fixed, as it is planned to restructure the complete thing anyway (https://github.com/u-blox/ubxlib/issues/93). 

Kind regards

RobMeades, 2023-07-24:
Hi: yes, that is an issue, the code works for the labels on Nordic MCUs but because the strings in the device tree are essentially platform-specific you hit issues the moment you try to use `ubxlib` with Zephyr and non-Nordic MCUs.

We could probably do a temporary fix by allowing the names to be `#define`d with a `#ifdef`/`#define`/`endif` pattern, then you could change the name when you compile `ubxlib` without having to modify the `ubxlib` code, but the real solution is to find a better way to accommodate Zephyr users and the device tree, as you say #93.

Would such a temporary fix be helpful or would you prefer to wait for the proper fix (which I guess is at least a month away, given holidays usually taken around now)?

potentialdiffer, 2023-07-25:
Hi Rob, thanks for your quick reply. For now, I just fixed it in my code base and it would be ok for me to wait for the proper fix to be released. 

I think we can close the issue, as it will be solved with #93.

RobMeades, 2023-07-25:
OK, closing, will resolve under #93.

closed 2023-07-25 by RobMeades