u-blox/ubxlib #118: Supprt AT+CCLK= to set the modem's time

johannesmay, 2023-07-27:
We want to enable TLS certificate expiry date checking and need to set the modem time for that.
Since our network providers seem to not always provide the network time, we want to set the modem time from our host processor using different timesources.

Unfortunately I couldn't find a way to set the time through the ubxlib API.
Could this be added to the API?

RobMeades, 2023-07-27:
Hi Jonannes: certainly, let me know if [this works for you](https://github.com/u-blox/ubxlib/tree/preview_feature_time_rmea) and, if so, I will merge it to `master` here and delete the preview branch.

johannesmay, 2023-07-28:
Hi Rob, that was quick! Thank you so much.
My colleague just tested the new setTime function and it works as expected 👍

RobMeades, 2023-07-28:
Great; I had nothing else to do yesterday evening so thought I'd JFDI, since it is an obvious hole in our APIs.  Thanks to you and your colleague for testing that it is good, now pushed to `master` here:

https://github.com/u-blox/ubxlib/commit/31ccbb4f8244d8b4b49a86f2672fef146468d624

Will delete the preview branch and close this issue.  Please feel free to re-open or start another if there is more to discuss.

closed 2023-07-28 by RobMeades