u-blox/ubxlib #119: nrf52840dk stops reading location data momentarily ?

darkhmiza, 2023-08-01:
Hello,

I am trying to read location data from my drotek F9P using ubxlib and nrf52840dk.
Here is the code below :

```
#include <string.h>
#include <stdio.h>
#include <time.h>

#include "ubxlib.h"

static size_t gPositionCount = 0;

static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_GNSS,
    .deviceCfg = {
        .cfgGnss = {
            .moduleType = U_GNSS_MODULE_TYPE_M9,  // This is the module type
            .pinEnablePower = -1,
            .pinDataReady = -1
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_I2C,
    .transportCfg = {
        .cfgI2c = {
            .i2c = 1, // This is the I2C HW block - ubxlib knows nothing about the device tree so you need to tell it
            .pinSda = -1,
            .pinScl = -1
        },
    },
};

// Return longitude/latitude value as string
static char *locStr(int32_t loc)
{
    static char str[25];
    const char *sign = "";
    if (loc < 0) {
        loc = -loc;
        sign = "-";
    }
    snprintf(str, sizeof(str), "%s%d.%07d",
             sign, loc / 10000000, loc % 10000000);
    return str;
}

void main()
{
   
   uDeviceHandle_t devHandle = NULL;
    int32_t returnCode;
    int32_t errorCode=0;

    int32_t guardCount = 0;

    // Initialise the APIs we will need
    uPortInit();
    uPortI2cInit(); // You only need this if an I2C interface is used
    uDeviceInit();

    returnCode=uDeviceOpen(&gDeviceCfg, &devHandle);
 uLocation_t location;

while(1){
    if (returnCode == 0) {
        // Since we are not using the common APIs we do not need
        // to call uNetworkInteraceUp()/uNetworkInteraceDown().

        // Start to get position

       
        int tries=0;

                printf(".");
                errorCode = uLocationGet(devHandle, U_LOCATION_TYPE_GNSS,
                                        NULL, NULL, &location, NULL);

        printk("error code : %d",errorCode);
        if (errorCode == 0) {
                
                printf("Position: https://maps.google.com/?q=");
                printf("%s,", locStr(location.latitudeX1e7));
                printf("%s\n", locStr(location.longitudeX1e7));
                
                
            } else if (errorCode == U_ERROR_COMMON_TIMEOUT) {
                printf("* Timeout\n");
            }

    } else {
        uPortLog("Unable to open GNSS!\n");
    }
}

    // Close the device
    // Note: we don't power the device down here in order
    // to speed up testing; you may prefer to power it off
    // by setting the second parameter to true.
    uDeviceClose(devHandle, false);

    // Tidy up
    uDeviceDeinit();
    uPortSpiDeinit(); // You only need this if an SPI interface is used
    uPortDeinit();

    uPortLog("Done.\n");

}```

At first I start getting data and all seems to work fine but sometimes I stop getting data for a couple of seconds and it resumes after.

Here is the last message I see in the console before it stops.
`
error code : 0error code : 0error code : 0
`

I have configured my F9P rate to 10 hz. I don't seem to have this problem with lower frequency, It would be much appreciated if you can help me with this.

RobMeades, 2023-08-01:
Hi there. and thanks for posting.  To get visibility of what's happening, you'd need to enable logging in order that we can see the communications with the GNSS device.  You can do this by adding a call to `uGnssSetUbxMessagePrint(devHandle, true)` just before your `while()` loop.

That said, is there any reason why you are polling position?  For such a high position rate, polling is likely to be rather a heavy load, especially on the little nRF52.  [This example](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_continuous.c) shows how to just set the GNSS chip up to send position at the desired rate with `uLocationGetContinuousStart()`, no polling loop required.

darkhmiza, 2023-08-01:
Hello Rob thank you for the quick response.

I have already enabled logging, here is what I get :
```
.U_GNSS: sent command b5 62 01 07 00 00 08 19.
U_GNSS: decoded UBX response 0x01 0x07: 2c f5 e5 0c e7 07 08 01 0c 06 13 f7 ea 03 00 00 f0 87 fa 05 03 01 ea 08 eb af df 00 7f d8 f7 19 78 4b 03 00 d7 8d 02 00 f4 02 00 00 47 02 00 00 02 00 00 00 e5 ff ff ff f3 ff ff ff 1b 00 00 00 00 00 00 00 76 00 00 00 80 a8 12 01 46 01 00 00 6c 18 42 3e 00 00 00 00 00 00 00 00 [body 92 byte(s)].
U_GNSS_POS: UTC time = 1690891579.
U_GNSS_POS: 3D fix achieved.
U_GNSS_POS: satellite(s) = 8.
U_GNSS_POS: longitude = 14659563 (degrees * 10^7).
U_GNSS_POS: latitude = 435673215 (degrees * 10^7).
U_GNSS_POS: altitude = 167383 (mm).
U_GNSS_POS: radius = 756 (mm).
U_GNSS_POS: speed = 27 (mm/s).
Position: https://maps.google.com/?q=43.5673215,1.4659563
.error code : 0error code : 0error code : 0error code : 0error code : 0error code : 0U_GNSS
```

And the reason I am polling is because I want to use this code in another code where I am using a polling loop, is there a way to get only the position  whith a polling loop. thank you in advance

RobMeades, 2023-08-01:
Thanks for the log: I'm having some trouble making sense of it.

You call `uLocationGet()`, which communicates with the GNSS chip and obtains an answer once, which will give an error code of 0; all good.  The `printk()` suggests that on the next call `uLocationGet()` returns an error code of 0, i.e. success, and yet the function has done nothing (according to the logging prints), not even sent a command to the GNSS device.  I can't see how the code could do that: if the code has not even been able to send anything to the GNSS chip it must return an error, so the `printk()` would print a negative number.

> is there a way to get only the position whith a polling loop

Not sure I follow your meaning: could you expand on your question?

darkhmiza, 2023-08-01:
It appears to me that the `printk` is repeating itself because even when it is working I get this :
```
U_GNSS_POS: UTC time = 1690898044.
U_GNSS_POS: 3D fix achieved.
U_GNSS_POS: satellite(s) = 10.
U_GNSS_POS: longitude = 14659861 (degrees * 10^7).
U_GNSS_POS: latitude = 435673288 (degrees * 10^7).
U_GNSS_POS: altitude = 163643 (mm).
U_GNSS_POS: radius = 1135 (mm).
U_GNSS_POS: speed = 9 (mm/s).
Position: https://maps.google.com/?q=43.5673288,1.4659861
.error code : 0error code : 0error code : 0error code : 0error code : 0error code : 0U_GNSS: sent command b5 62 01 07 00 00 08 19.
U_GNSS: decoded UBX response 0x01 0x07: a4 9c 48 0d e7 07 08 01 0d 36 04 37 18 00 00 00 7e cd cc 1d 03 01 ea 0a 15 b1 df 00 c8 d8 f7 19 e1 3c 03 00 40 7f 02 00 70 04 00 00 23 05 00 00 e1 ff ff ff 10 00 00 00 f8 ff ff ff 23 00 00 00 ac bb 30 00 b8 00 00 00 80 a8 12 01 c6 00 00 00 6c 18 42 3e 00 00 00 00 00 00 00 00 [body 92 byte(s)].
U_GNSS_POS: UTC time = 1690898044.
U_GNSS_POS: 3D fix achieved.
U_GNSS_POS: satellite(s) = 10.
U_GNSS_POS: longitude = 14659861 (degrees * 10^7).
U_GNSS_POS: latitude = 435673288 (degrees * 10^7).
U_GNSS_POS: altitude = 163648 (mm).
U_GNSS_POS: radius = 1136 (mm).
U_GNSS_POS: speed = 35 (mm/s).
Position: https://maps.google.com/?q=43.5673288,1.4659861
```
notice that `error code=0` is repeated 4 times before the next message is sent, to be honest this makes no sense to me.

#Not sure I follow your meaning: could you expand on your question?

Sorry for not being clear enough, what I want to do is get the location once when I ask my module because my hope is to add this code to another where I poll some other data for example :

```
while(1){
get_msg // here I get my msg from a UWB receiver for example

get location //Here I get location
}
```
I hope this clarifies my objective.

RobMeades, 2023-08-01:
Very strange: I'd assumed that somehow you had `printf()` disabled because none of your `printf()s` appeared to be producing any output in the log, but looking at the output you have just posted I can see `.` appearing, which you `printf()` just before the call to `uLocationGet()`, yet, as you say, there are way more `error code : 0` appearing than should be possible.

Unless you have a good reason for using it, I'd switch from `printk()` to the rather more standard `printf()` throughout and maybe post the output here again; perhaps it will make more sense?

On your desired mode of operation, you can of course do exactly as you describe, calling `uLocationGet()` where you have written `get location //Here I get location`, provided you do the relevant `uPortInit()`/`uPortI2cInit()`/`uDeviceInit()` and `uDeviceOpen()` calls at startup and, similarly, `uDeviceClose()` followed by `uDeviceDeinit()`/`uPortI2cDeinit()`/`uPortDeinit()` calls at shut-down.  The only down-side of polling in this way is that it takes time, of course: if you use the [uLocationGetContinuousStart()](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/location/api/u_location.h#L456) with your desired rate then your callback will be called by the GNSS chip at exactly that rate, no need to poll anything and you can just ignore stuff you don't need. I only suggest this because if your real application needs position every 100 ms then polling is going to take up a significant amount of your processing time.

darkhmiza, 2023-08-02:
Hello again Rob sorry for the late response, so I replaced `printk` with `printf` and it seems that it fixed the print repetition problem.
I also decided to use `uLocationGetContinuousStart` for my project, here is the code : 

```
#include <string.h>
#include <stdio.h>
#include <time.h>
#include <kernel.h>
#include "ubxlib.h"

static size_t gLocationCount = 0;
static size_t gPositionCount = 0;

// NETWORK configuration for GNSS
static const uNetworkCfgGnss_t gNetworkCfg = {
    .type = U_NETWORK_TYPE_GNSS,
    .moduleType = U_GNSS_MODULE_TYPE_M9,
    .devicePinPwr = -1,
    .devicePinDataReady = -1
};
//uDeviceCfg_t gDeviceCfg;
//static const uDeviceCfg_t gDeviceCfg = {.deviceType = U_DEVICE_TYPE_GNSS, .transportType = U_DEVICE_TRANSPORT_TYPE_I2C};

static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_GNSS,
    .deviceCfg = {
        .cfgGnss = {
            .moduleType = U_GNSS_MODULE_TYPE_M9,  // This is the module type
            .pinEnablePower = -1,
            .pinDataReady = -1
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_I2C,
    .transportCfg = {
        .cfgI2c = {
            .i2c = 1, // This is the I2C HW block - ubxlib knows nothing about the device tree so you need to tell it
            .pinSda = -1,
            .pinScl = -1
        },
    },
};

// Return longitude/latitude value as string
static char *locStr(int32_t loc)
{
    static char str[25];
    const char *sign = "";
    if (loc < 0) {
        loc = -loc;
        sign = "-";
    }
    snprintf(str, sizeof(str), "%s%d.%07d",
             sign, loc / 10000000, loc % 10000000);
    return str;
}

static char latLongToBits(int32_t thingX1e7,
                          int32_t *pWhole,
                          int32_t *pFraction)
{
    char prefix = '+';

    // Deal with the sign
    if (thingX1e7 < 0) {
        thingX1e7 = -thingX1e7;
        prefix = '-';
    }
    *pWhole = thingX1e7 / 10000000;
    *pFraction = thingX1e7 % 10000000;

[... 127 lines cut ...]
return code 0
Bringing up the network...
Starting continuous location.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 00 01 00 00 07 21.
U_GNSS: got Nack for 0x06 0x00.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 ff ff 21 00 b8 cc.
U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 05 00 21 10 00 03 00 21 20 01 01 00 21 30 64 00 02 00 21 30 01 00 [body 26 byte(s)].
U_GNSS: sent command b5 62 06 01 02 00 01 07 11 3a.
U_GNSS: decoded UBX response 0x06 0x01: 01 07 00 00 00 00 00 00 [body 8 byte(s)].
U_GNSS: sent command b5 62 06 01 08 00 01 07 01 00 00 00 00 00 18 e2.
U_GNSS: decoded UBX response 0x05 0x01: 06 01 [body 2 byte(s)].
Waiting up to 60 seconds for 5 location fixes.
I am here: https://maps.google.com/?q=+53.6910008,+0.0060753
I am here: https://maps.google.com/?q=+53.6910008,+0.0060753
I am here: https://maps.google.com/?q=+53.6910008,+0.0060753
I am here: https://maps.google.com/?q=+53.6910008,+0.0060753
I am here: https://maps.google.com/?q=+53.6910008,+0.0060753
I am here: https://maps.google.com/?q=+53.6910008,+0.0060753
I am here: https://maps
```
 Perhaps something is wrong with my code ?

RobMeades, 2023-08-02:
You seem to have moved from Toulouse to Grimsby, which I guess is unlikely.

As point of debug, could you try printing the raw values of all of the parameters as received by `callback()`?

darkhmiza, 2023-08-02:
So after printing out the parameters I get this :

```
Latitude Raw : 536904152
Longitude raw : 42665
altitude raw : 1
radius raw : 435673193
speed raw : 14659771
svs raw : 162194
time raw : 0
I am here: https://maps.google.com/?q=+53.6904152,+0.0042665
```
Now I am a bit confused it seems that the reel coordinates are in `radiusMillimetres` and `speedMillimetresPerSecond` instead of `latitudeX1e7` and `longitudeX1e7` I confirmed that its giving the correct location. Do you have any idea why this would happen ?

Also I have a quick question where you don't have to go into details but does ubxlib use multithreading ?

RobMeades, 2023-08-02:
There's something very weird going in here.  If you look at the definition of [uLocationGetContinuousStart()](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/location/api/u_location.h#L520) you'll see that the type for the callback is:

```
void (*pCallback) (uDeviceHandle_t devHandle,
                    int32_t errorCode,
                    const uLocation_t *pLocation);
```

In your code above you are passing in `callback()`, which has the function signature:

```
void callback(uDeviceHandle_t gnssHandle,
              int32_t errorCode,
              int32_t latitudeX1e7,
              int32_t longitudeX1e7,
              int32_t altitudeMillimetres,
              int32_t radiusMillimetres,
              int32_t speedMillimetresPerSecond,
              int32_t svs,
              int64_t timeUtc);
```

...i.e. it is utterly different, yet your code compiles!  I didn't think that was possible.  You should probably look at which compiler warnings are being switched off in Zephyr and remove a few of the switching-offs, as not catching this kind of issue can be counter-productive.

I reckon if you change the function signature of `callback()` to match what is expected then things will work.

> does `ubxlib` use multithreading

Yes, it does, and conversely the APIs of `ubxlib` are thread-safe unless otherwise stated.

darkhmiza, 2023-08-03:
Hi again Rob before closing this issue I was hoping you can answer some questions I was curious about :

So to my understanding the rate I am setting in `uLocationGetContinuousStart` is the rate of the callback, so in my case I am calling the callback every 100ms (10 Hz), what would happen if I set my module's rate to 1 Hz? Will I just get 10 of the same values ? Now what if my module's rate is 10 hz and my callback rate is 1 hz what would happen then? 

Another question would be can I use ubxlib to get raw data such as Rinex messages ?

Thank you in advance.

RobMeades, 2023-08-03:
> I was hoping you can answer some questions I was curious about...

Certainly: in the case of `uLocationGetContinuousStart()`, you give it a callback and a rate at which the GNSS chip will be requested to send position fixes.  All the `ubxlib` code does then is tell the GNSS chip the required rate and set up a task that sits there listening for position fix messages.  When it gets one (success or failure) your callback will be called.

So provided the GNSS chip is behaving, and the comms link to it is solid, your callback should be called at the requested rate: you have no control over that, if you see what I mean, the rate at which your callback is called is going to be that rate.

`ubxlib` has a generic GNSS message API, [uGnssMsg](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/gnss/api/u_gnss_msg.h#L193), see example of how to use it [here](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/example/gnss/msg_main.c#L286),  through which you can send/receive any messages that a u-blox GNSS chip understands.  However, not that I'm a GNSS expert, but I've never heard of Rinex messages; u-blox GNSS chips understand [UBX, NMEA and RTCM](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/gnss/api/u_gnss_type.h#L222).

RobMeades, 2023-08-03:
That's correct.

darkhmiza, 2023-08-07:
Hello again Rob, I hope you are doing great.

I was trying to get the fixe mode, I am using `uGnssCfgGetFixMode()` to do that. When I am in RTK mode I am getting a 3 as a fix mode while it should be 4, I am sure that I am in RTK mode because my RTK LED in my ZED F9P module is fixed. 

After looking through the Drotek Doc I found that what I am actually looking for is the quality field I was wondering if there was already a function that gets me that.

Thank you in advance.

RobMeades, 2023-08-07:
Yup, all good here, not on holiday, which is good as the weather really isn't holidayish in the UK, even outside Grimsby.

Can you pin-point the quality field you are referring to more exactly, maybe refer to a section of the u-ZED F9P u-blox [interface manual](https://content.u-blox.com/sites/default/files/documents/u-blox-F9-HPG-1.32_InterfaceDescription_UBX-22008968.pdf)?

darkhmiza, 2023-08-07:
Yes certainly you can find it in page 25-26 of the interface manual its the GGA quality field if you see the last table in page 25 you'll see that RTK fixed corresponds to 4 in GGA quality field.

To clarify what I want to do, I would like to just find out if I am in RTK mode or not through software without looking at my module RTK LED.

RobMeades, 2023-08-07:
Hmmm.  `ubxlib` doesn't use NMEA, it only uses UBX messages (and in fact it switches NMEA messages off if the fix rate is to be high so that they don't get in the way).  The equivalent UBX message would seem to be `UBX-NAV-PVT` (see section 3.15.13 of the interface manual), which includes the `fixType` field but that field does not indicate whether the fix has employed RTK or not and I can't find any UBX message that does, unless it is called something else (I'm just searching the manual for "RTK").

You can set and read the RTK _mode_ with `CFG-NAVHPG`: would that give you the information you need, or do you need to know if RTK has actually been used in the case of each fix?  If so, can you see anywhere this information is fed-back in a UBX message of the F9P?

darkhmiza, 2023-08-07:
Indeed I could not find in ubx message that has RTK mode in its fields, If I am to use `CFG-NAVHPG` hwo would I go about it, do I use `uGnssMsg` to write that message ?

RobMeades, 2023-08-07:
Yes, that's a shame isn't it?  Not really a GNSS person myself, don't know why it is not exposed.

To read `CFG-NAVHPG` you would follow [this example](https://github.com/u-blox/ubxlib/blob/64ff3b71bd93091e46aa3a214643537b66979e87/example/gnss/cfg_val_main.c#L280); it would be something like:

```
uint8_t byteValue = 0;
int32_t x = uGnssCfgValGet(devHandle, U_GNSS_CFG_VAL_KEY_ITEM_NAVHPG_DGNSSMODE_E1,
                           (void *) &byteValue, sizeof(byteValue),
                           U_GNSS_CFG_VAL_LAYER_RAM);
printf("Attempt to read NAVHPG DGNSSMODE returned %d.\n", x);
if (x == 0) {
    printf("RTK_FLOAT is %d.\n", byteValue);
}
```

mazgch, 2023-08-07:
@RobMeades UBX-NAV-PVT flags.bits.carrSoln contains the RTK status 0-No, 1-Float, 2-Fixed you would still need the UBX-NAV-HPPOSLLH to get enough fractional digits of lat lon.

RobMeades, 2023-08-07:
Thanks Michael!  That was well hidden. The `lat` and `lon` fields of `UBX-NAV-HPPOSLLH` still seem to both be `I4` 1e7 values: are they just "better" values somehow than `UBX-NAV-PVT` when using a high precision device?  Seems like maybe we need some changes to support high precision properly in `ubxlib`.

cturvey, 2023-08-07:
@RobMeades there are secondary files adding precision to both lat/lon and altitude. You have to unpack and add them.
The 32-bit signed values don't have enough digits of precision, beyond +/-180.xxxxxxx

RobMeades, 2023-08-07:
Ah, thanks, I was looking at `UBX-NAV-POSLLH` and not `UBX-NAV-HPPOSLLH`, I see the fields now.

I will look at how we can properly add support for this.

darkhmiza, 2023-08-08:
Hi again I guess the discussion got livelier in my absence :)

Does ubxlib make it possible to retrieve the flags.bits.carrSoIn, or should I attempt to read th raw message and then retrieve the last 2 bits of the 21st byte of the message ?

RobMeades, 2023-08-08:
For now I think do it the raw-way yourself.

Having talked with Michael I think the way forward might be to have a kind of "decoder" API, which you can attach to the `uGnssMsg` API somehow or other and, if it understands the message, it will provide a decoded, C structure, version of it.

We could then add to these slowly over time.  Initially we'd do just `UBX-NAV-PVT` and `UBX-NAV-HPPOSLLH`, as those are the obvious holes.

Something like that anyway.  I might throw a preview branch of it back here for feedback when I've got my head around it.

darkhmiza, 2023-08-08:
So I tried to do just that I used this [example](https://github.com/u-blox/ubxlib/blob/master/example/gnss/msg_main.c), here is the modified code :

```
/*
 * Copyright 2019-2023 u-blox
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

/** @brief This example demonstrates how to exchange message of your
 * choice with a GNSS device that is directly connected to this MCU;
 * this mechanism does not currently work if your GNSS device is
 * connected via an intermediate [cellular] module.
 *
 * The choice of module and the choice of platform on which this
 * code runs is made at build time, see the README.md for
 * instructions.
 */

// Bring in all of the ubxlib public header files
# include "ubxlib.h"

/* ----------------------------------------------------------------
 * COMPILE-TIME MACROS
 * -------------------------------------------------------------- */

/** The size of message buffer we need: enough room for a UBX-NAV-PVT
 * message, which has a body of length 92 bytes, and any NMEA message,
 * which have a maximum size of 82 bytes.
 */
# define MY_MESSAGE_BUFFER_LENGTH  (92 + U_UBX_PROTOCOL_OVERHEAD_LENGTH_BYTES)

static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_GNSS,
    .deviceCfg = {
        .cfgGnss = {
            .moduleType = U_GNSS_MODULE_TYPE_M9,  // This is the module type
            .pinEnablePower = -1,
            .pinDataReady = -1
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_I2C,
    .transportCfg = {
        .cfgI2c = {
            .i2c = 1, // This is the I2C HW block - ubxlib knows nothing about the device tree so you need to tell it
            .pinSda = -1,
            .pinScl = -1
        },
    },
};

// Count of messages received
static size_t gMessageCount = 0;

/* ----------------------------------------------------------------
 * STATIC FUNCTIONS
 * -------------------------------------------------------------- */

// Convert a lat/long into a whole number and a bit-after-the-decimal-point
// that can be printed by a version of printf() that does not support
[... 260 lines cut ...]
byte 80 : 12
byte 81 : 1
byte 82 : f1
byte 83 : 0
byte 84 : 0
byte 85 : 0
byte 86 : 6c
byte 87 : 18
byte 88 : 42
byte 89 : 3e
byte 90 : 0
byte 91 : 0
byte 92 : 0
byte 93 : 0
byte 94 : 0
byte 95 : 0
byte 96 : 0
byte 97 : 0
byte 98 : 92
byte 99 : de
I am here: https://maps.google.com/?q=+103.9993066,+112.4277704
0 NMEA message(s) received.
Done.
```
As you can see I am getting an absurd location, I am in RTK float mode but if you see the 21st byte I am getting a 0, I was thinking there might be an offset in the bytes order, what do you think ?

mazgch, 2023-08-08:
> Ah, thanks, I was looking at `UBX-NAV-POSLLH` and not `UBX-NAV-HPPOSLLH`, I see the fields now.
> 
> I will look at how we can properly add support for this.

The are I1s with lonHp and LatHp components in the HPPOS messages that you just add. these are scaled 1e-9 deg

RobMeades, 2023-08-08:
@darkhmiza: parsing the data by hand, I get:

```
b5 62 01 07 5c 00 7c f1 34 0d e7 07 08 08 0d 20 23 37 15 00 00 00 c5 6d c8 1d 03 43 
| Header (len 92)|    iTow   | 2023| 8/8 |13:20:23|  |   15 ns   |499,674,565|3D|  | ...
|                |           |    Time/Date valid    | TAccuracy |    ns     |  |  |
                                                          ----------------------|  |----------------------------
                                                           Flags: 0100 0011 - gnssFixOK, diffSoln, carrSoln = 1
```

...which I think is correct.  Don't forget that you get the whole message this way, so there are [U_UBX_PROTOCOL_HEADER_BYTES](https://github.com/u-blox/ubxlib/blob/594f4857f909605a48525938a240cd58da4d32fb/common/ubx_protocol/api/u_ubx_protocol.h#L46) (6) of UBX message header on the front.

RobMeades, 2023-08-11:
@darkhmiza: in case it is of interest, I have pushed here a [preview branch](https://github.com/u-blox/ubxlib/tree/preview_feature_gnss_decode_rmea) of a new `ubxlib` GNSS API, [uGnssDec](https://github.com/u-blox/ubxlib/blob/f9abe61fdbe3f5eda28a2ef8cb87a643f1d8e096/gnss/api/u_gnss_dec.h#L152), which allows you to decode messages generically from a GNSS device.

Currently only `UBX-NAV-PVT` and `UBX-NAV-HPPOSLLH` (your use case) are implemented but a user-callback is also included so you could add your own decoders at run-time.  The API should be reasonably obvious, though note that this has NOT yet passed internal review and hence may change.  When it has been reviewed and pushed to `master` here I will update this thread and delete the preview branch.

For your use case you would set the navigation rate [epoch] you want using [uGnssCfgSetRate()](https://github.com/u-blox/ubxlib/blob/f9abe61fdbe3f5eda28a2ef8cb87a643f1d8e096/gnss/api/u_gnss_cfg.h#L231), if necessary ensure that the messages you are after are being emitted once per epoch using [uGnssCfgSetMsgRate()](https://github.com/u-blox/ubxlib/blob/f9abe61fdbe3f5eda28a2ef8cb87a643f1d8e096/gnss/api/u_gnss_cfg.h#L284) and then, in the callback that you have set up using [uGnssMsgReceiveStart()](https://github.com/u-blox/ubxlib/blob/f9abe61fdbe3f5eda28a2ef8cb87a643f1d8e096/gnss/api/u_gnss_msg.h#L363), you would have something like:

```
char buffer[512];

void myCallback(uDeviceHandle_t gnssHandle,
                const uGnssMessageId_t *pMessageId,
                int32_t errorCodeOrLength,
                void *pCallbackParam)
{
    (void) pMessageId;

    if (errorCodeOrLength > 0) {
        if (errorCodeOrLength > sizeof(buffer)) {
            errorCodeOrLength = sizeof(buffer);
        }
        errorCodeOrLength = uGnssMsgReceiveCallbackRead(gnssHandle,
                                                        buffer,
                                                        errorCodeOrLength);
        if (errorCodeOrLength > 0) {

            uGnssDec_t *pDec = pUGnssDecAlloc(buffer, errorCodeOrLength);
            if ((pDec != NULL) && (pDec->errorCode == 0)) {

                // Do stuff with pDec->pMsg, casting it to the correct message
                // type (see uGnssDecUnion_t in u_gnss_dec.h for the types)
                // by checking pDec->pId

            }
            // Must *always* free the memory that pUGnssDecAlloc() allocated
            uGnssDecFree(pDec);
        }
    }
}
```

darkhmiza, 2023-08-14:
Hello Rob, sorry for the late response.

Indeed I completely forgot about the header bytes, I was getting the correct data.

Thank you for the new branch it will be of great use to me.

RobMeades, 2023-08-14:
@darkhmiza: I'm going to close this now as I think we're done, at least with its original purpose.  Please feel free to re-open it if you disagree.

closed 2023-08-14 by RobMeades

RobMeades, 2023-08-20:
The `uGnssDec` API is now pushed to `master` here, see e2d0597b4b53efe227f74220562ebb4550a97428.

I will delete the [preview branch](https://github.com/u-blox/ubxlib/tree/preview_feature_gnss_decode_rmea) some time late next week.