u-blox/ubxlib #120: u_cell_sock.c - assumes always non-blocking socket behaviour...

PawelOstropolski, 2023-08-01:
uCellSockBlockingSet and uCellSockBlockingGet assume the cell sock is always non-blocking - there is currently no implementation behind these functions in u_cell_sock.c.

However u_sock.c uSockCreate function pSockContainerCreate:L638  pContainer->socket.blocking = true;  
As a result, cellular sockets can't be set to non-blocking with the current library implementation.  

Suggest putting implementation behind the uCellSockBlockingSet function so that the u_sock.c:L1208 intent is fulfilled.

RobMeades, 2023-08-01:
Hi there: it is a while since I did this so I will check properly in the morning but I _believe_ the `uCellSockBlockingSet()` stuff is only there to ensure that **non-blocking** behaviour is definitively set at the `uCell` level.  Then [uSockBlockingSet()](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/sock/src/u_sock.c#L1583) can be called to set the value of `pContainer->socket.blocking`, so that it is able to provide blocking behaviour over the top of the non-blocking `uCell` API, IYSWIM.

Provided all sockets stuff is done at the `uSock` API level, that code will then block, or not, having made sure the `uCell` API always _doesn't_ block.

If you see what I mean.

PawelOstropolski, 2023-08-01:
Hi, got the full concept now. I missed the uSockBlockingSet(). All good. :)

PawelOstropolski, 2023-08-01:
not an issue.

closed 2023-08-01 by PawelOstropolski