u-blox/ubxlib #121: Unable to POST

alexmaron81, 2023-08-03:
I'm using the example:
https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/example/http_client/http_main.c

The Module: SARA-R412M 02B-01 

Following error:

OK
U_CELL_NET: connected after 3 second(s).
AT+CGSN

359159971535480

OK
U_CELL_INFO: IMEI is 359159971535480.
AT+UHTTP=0

OK
AT+UHTTP=0,1,"ubxlib.redirectme.net"

OK
AT+UHTTP=0,5,8081

OK
AT+UHTTP=0,4,0

OK
AT+UHTTP=0,7,30

OK
AT+UHTTP=0,6,0

OK
Unable to POST file "/359159971535480.html" to ubxlib.redirectme.net:8081; status code was -4!
Taking down network...

RobMeades, 2023-08-03:
Hi Alex: [error code](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/error/api/u_error_common.h#L48) 4 is `U_ERROR_COMMON_NOT_SUPPORTED`, which seems to be what [uHttpClientPostRequest()](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/http_client/api/u_http_client.h#L303) is returning.  This is weird, because [pUHttpClientOpen()](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/http_client/api/u_http_client.h#L215) worked.  All I can think of is that, somehow, we are suffering again from the [known problem with the Espressif linker](https://www.esp32.com/viewtopic.php?f=13&t=8418&p=35899) where it drops the implementation of functions that have weak alternatives.

We worked around this by [adding dummy functions that are always linked](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/cell/src/u_cell.c#L159), see here in `uCellInit()`:

[image]

...where the critical line in this case is `uCellFilePrivateLink()`.  Can you check that the version of code you are using has this?

alexmaron81, 2023-08-03:
[image]

RobMeades, 2023-08-03:
Looks like you have old code: this issue was fixed back in March with https://github.com/u-blox/ubxlib/commit/80e76d49b97d8128202d5dd2a2fa9bfa2ecbc47a, raised by you over e-mail I think.

Can you move to a later version, which would fix this, or is https://github.com/u-blox/ubxlib/issues/113 preventing you doing that (in which case we need to address that first)?

alexmaron81, 2023-08-03:
So until now the problem "https://github.com/u-blox/ubxlib/issues/113" hasn't surfaced anymore. I haven't changed anything either.

RobMeades, 2023-08-03:
> until now the problem hasn't surfaced anymore. I haven't changed anything either.

Never good when that happens.  Anyway, I recommend you move to latest `master` (save what you have first) and try again; your problem with getting back `U_ERROR_COMMON_NOT_SUPPORTED` from `uHttpClientPostRequest()` should then go away.

alexmaron81, 2023-08-03:
I switched to the last master.
In fact, I no longer have an error.
Thank you Rob!

closed 2023-08-03 by alexmaron81

RobMeades, 2023-08-03:
Phew! Good news :-).