u-blox/ubxlib #123: Unable to GET file (status code -9)

alexmaron81, 2023-08-04:
What does status code -9 mean?

And where can I see the status codes and AT command errors and their meaning?

RobMeades, 2023-08-07:
The common ones, which you will find 99% of the time, can be found in [u_error_common.h](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/error/api/u_error_common.h#L48).

You may, very occasionally, get a higher-[negatively]-numbered one, which is technology-specific (`cell` or `gnss` or `wifi` or `ble`): those can be found in [u_cell.h](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/cell/api/u_cell.h#L151), [u_gnss.h](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/gnss/api/u_gnss.h#L53), [u_wifi.h](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/wifi/api/u_wifi.h#L84) and [u_ble.h](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/ble/api/u_ble.h#L67).

alexmaron81, 2023-08-07:
I have two files on the web server.
One where there is simply a "test" and a ".bin" file with the firmware for ESP32.

The simple one is loaded with GET over HTTP and the content is displayed.
The -9 status code occurs with the ".bin" file with 470kb

RobMeades, 2023-08-07:
470 kbytes might be a bit of a stretch!  This file is downloaded by the module over HTTP into its file system and then has to be read from the cellular module file system into your MCU, into a buffer which of course needs to be 470,000 bytes big.

First thing to find out is whether the HTTP operation succeeded, downloading the file into the cellular module's file system.  Assuming you are still using the [HTTP example](https://github.com/u-blox/ubxlib/blob/master/example/http_client/http_main.c) the  [uHttpClientPostRequest()](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/http_client/api/u_http_client.h#L360) there will have downloaded the file into [a file named](https://github.com/u-blox/ubxlib/blob/31ccbb4f8244d8b4b49a86f2672fef146468d624/common/http_client/src/u_http_client.c#L470) `ubxlibhttp_X_putpost`, where X is the HTTP instance number you happen to be using (likely 0).  In an AT log you should see something like:

```
AT+UHTTPC=0,4,"/somefile","ubxlibhttp_0","ubxlibhttp_0_putpost",6,"some-content-type"
OK

...time passes while 470 kbytes are downloaded...

+UUHTTPCR: 0,4,1
```

The 1 on the end of the `+UUHTTPCR` is the outcome of the POST operation: 1 is success, 0 is failure.  For a fail case,  the error code will be the next thing to be printed, something like:

```
AT+UHTTPER=0
UHTTPER: 0,3,X
OK
```

...where X is the error code and can be found in Appendix A.7.2 of the SARA-R4 AT manual.

What do you see?

alexmaron81, 2023-08-07:
Oh yes exactly (That's what I asked you via email), why "uHttpClientPostRequest()" and not "uHttpClientGetRequest()"?

RobMeades, 2023-08-07:
>  why "uHttpClientPostRequest()" and not "uHttpClientGetRequest()"

Apologies, forgot we had an e-mail thread going as well.  I only suggested POST because I thought that was what the Espressif example used.  Just use whatever they use, I don't really know how the OTA setup they have proposed works.  For a GET request you should see something like:

```
AT+UHTTPC=0,1,"/somefile","ubxlibhttp_0"
OK
+UUHTTPCR: 0,1,1
```

alexmaron81, 2023-08-08:
A question about the bin file size. ubxlib is bigger than anything else. can I somehow reduce?

[image]

RobMeades, 2023-08-08:
Assuming you are interested in cellular only, not GNSS or Wifi or BLE, then, if you haven't done so already, you can set the `UBXLIB_FEATURES` CMake variable to be just `cell`: see where we set it here:

https://github.com/u-blox/ubxlib/blob/master/port/platform/esp-idf/mcu/esp32/components/ubxlib/CMakeLists.txt

alexmaron81, 2023-08-08:
I get the following output when I try to download the file:

AT+UHTTP=0

OK
AT+UHTTP=0,0,"x.x.x.x"

OK
AT+UHTTP=0,5,9000

OK
AT+UHTTP=0,4,0

OK
AT+UHTTP=0,7,30

OK
AT+UHTTP=0,6,0

OK
I (19325) OTA: Writing to partition subtype 16 at offset 0x120000
AT+UHTTPC=0,1,"/CONNECT.bin","ubxlibhttp_0"

OK

+UUHTTPCR: 0,1,1
GET the data: it was "0<@�" (1024 byte(s)).
I (64855) OTA: New firmware version: 1.2  
I (64855) OTA: Running firmware version: 1
I (64855) OTA: esp_ota_begin succeeded    
AT+UHTTPC=0,1,"/CONNECT.bin","ubxlibhttp_0"

OK

+UUHTTPCR: 0,1,1
GET the data: it was "0<@�" (1024 byte(s)).
AT+UHTTPC=0,1,"/CONNECT.bin","ubxlibhttp_0"

OK

+UUHTTPCR: 0,1,1
GET the data: it was "0<@�" (1024 byte(s)).
AT+UHTTPC=0,1,"/CONNECT.bin","ubxlibhttp_0"

OK

+UUHTTPCR: 0,1,1
GET the data: it was "0<@�" (1024 byte(s)).
AT+UHTTPC=0,1,"/CONNECT.bin","ubxlibhttp_0"

OK

RobMeades, 2023-08-08:
That looks correct, from a `ubxlib` perspective at least: it says that it has successfully performed a GET on the file `CONNECT.bin`, the file should be in the module's file system.

Your print "Get the data: it was ... (1024 bytes()))", where does the "bytes" value come from?  1024 bytes looks like a buffer length rather than the file length.

alexmaron81, 2023-08-08:
Yes exactly, only when I try to set the buffer to the value does the ESP stop.

RobMeades, 2023-08-08:
After the HTTP GET call, could you add a call to [uCellFileGetSize()](https://github.com/u-blox/ubxlib/blob/64ff3b71bd93091e46aa3a214643537b66979e87/cell/api/u_cell_file.h#L165), giving it the file name `ubxlibhttp_0` and printing out the result?  Just to double-check that the whole file has been downloaded?

If it has, then the problem remains that the `ubxlib` HTTP interface requires the whole file to be downloaded from the module whereas you need a kind of a "chuncked"/"streamed" mechanism.

alexmaron81, 2023-08-08:
[image]

RobMeades, 2023-08-08:
Perfect.  Let me think for a moment as to the best way to get you what you want.

alexmaron81, 2023-08-08:
Can I read the file bit by bit and append it to a file on the ESP?

RobMeades, 2023-08-08:
One more thing: with this mechanism the file you read from them module file system is the entire HTTP GET response, with the headers on the front, which you will need to remove. The headers always end with "\r\n\r\n" so that should be easy enough to find.

RobMeades, 2023-08-09:
Not sure I'm really following.

`AT+URDBLOCK="ubxlibhttp_0",14336,1024` indicates that the code has asked the module to read from offset 14,336 bytes in the file and is asking for 1024 bytes and what is shown seems fairly likely to be the contents of a binary file.  I don't know what `ESP_LOGI` is: if it is intended to print strings (i.e. sequences of printable characters with the end marked by a null) then it might only print the first bit, up to the first control character; after all, what you have there is not a string, it is random binary content.

Just to confirm: are you parsing the file data that is read from the module to remove the HTTP GET header (i.e. the stuff up to and including the `\r\n\r\n` that marks the end of the HTTP GET header), before you write the stuff into the ESP/OTA world (which I guess wants the binary file and doesn't want the HTTP GET header)?

alexmaron81, 2023-08-09:
I write the buffer to a file:

FILE *f = NULL;
f = fopen("/spiffs/firmware.bin", "a");
fwrite(buffer, sizeof(char), strlen(buffer), f);
fclose(f);

When reading, I only get the beginning up to a certain character:
[image]

[image]

alexmaron81, 2023-08-10:
Oh OK, I already thought that the "printf" and "ESP_LOGI" do not show this in the cli.

I have now noticed that loading the file from sara to esp takes a long time. about 100kb in a minute.
How does it come about?
Can this be speeded up?

RobMeades, 2023-08-10:
Without knowing where the time is going it's difficult to say. I suggest you annotate your code with a few of these:

`printf("xxx: %d ms.\n", uPortGetTickTimeMs());`

...where `xxx` is replaced by some different meaningful phrase in each case, e.g. "starting blah",  "just done blah", that kind of thing.  Put them in useful places so that we get an idea of how time is passing and also put a few in [uCellFileBlockRead()](https://github.com/u-blox/ubxlib/blob/594f4857f909605a48525938a240cd58da4d32fb/cell/src/u_cell_file.c#L266).

Question: do you have the the UART flow control lines connected between the MCU and the module?  You will need these for data integrity if we are to speed anything up.

This is a *huge* file, and the module is likely reading it from flash each time to get your 1 kbyte chunk, so likely increasing that chunk size is the thing, if you have the memory for it, but we may hit upon limitations in the AT interface itself.

RobMeades, 2023-08-10:
The other possibility would be to do this over a plain socket, rather than using the `uCellHttp` interface, but let's see where the time is going first.

alexmaron81, 2023-08-10:
[image]

alexmaron81, 2023-08-10:
Flow control: if you mean CTS and RTS, then yes.

RobMeades, 2023-08-10:
OK, so 440 ms per 1 kbyte, 2.3 kbytes/s with a theoretical maximum (at 115200 bits/s) of ~11.5 kbytes/s. Question is, how much of that is the module reading a half megabyte file from the file system to give you your chunk and how much is just AT interface speed related.

We can do two things: up the UART speed or try using straight sockets instead, so no intermediate file. I'm inclined towards the latter, since the intermediate file on the module is not doing anything for us if you're storing it anyway on the MCU.

Thoughts?

alexmaron81, 2023-08-10:
We can try the second. Download directly to ESP

Where do you set the UART speed?
Does it make sense?

RobMeades, 2023-08-10:
We don't have a `ubxlib` API to change the baud rate 'cos it's kinda clumsy on SARA-R4; those modules does not support auto-bauding, so you can't just change the MCU's UART speed and it will follow, you have to use an AT command to change the baud rate, `AT+IPR`, then of course you have to close the MCU's UART ('cos it will no longer be speaking at the right rate), change the baud rate of the MCU's UART, start it up again and issue the command `AT&W` to store the new baud-rate in the NVRAM of the module (otherwise the module will revert to 115,200 next time around).

If you like you can connect to the module from a PC or some such and do the above, setting the module to run the UART at 230,400 or 460,800, and hoping that doesn't have any undesirable side-effects.  All you'd do then is set the `baudRate` field of the `uDeviceCfgUart_t` part of the `uDeviceCfg_t` structure you pass into `uDeviceOpen()` to the new baud rate.

I will post some suggested code for the sockets approach soon.

RobMeades, 2023-08-10:
Find below the code: it performs an HTTP GET over a secure socket, parses the header that is returned to get Content Length and then sits in a loop downloading that content until done or time-out.

I have tested this against our test server `ubxlib.com` using a secure connection (port 8081) and downloading a 512k byte binary file (`/test_binary_file_512k`) which I have left there for now (I will remove it next week): the #defines below are set such that you can repeat downloading exactly the same file if you wish, or you can change `MY_SERVER_URL`, `MY_SERVER_PORT` and `MY_PATH_ON_SERVER` to point to the file you want to download on your server.  For me, downloading the test file took about 2.5 minutes from start to finish, so about 27 kbits/s.

Note that the code switches off AT prints during the binary download to avoid debug splatter.

```
// Make sure to include these headers, if not done already
#include "stdlib.h"
#include "string.h"

#define MY_DOWNLOAD_TIMEOUT_SECONDS 240
#define MY_SERVER_URL               "ubxlib.com"
#define MY_SERVER_PORT              8081
#define MY_PATH_ON_SERVER           "/test_binary_file_512k"
#define HTTP_HEADER_END_MARKER      "\r\n\r\n"
#define HTTP_HEADER_CONTENT_LENGTH  "Content-Length: "

// Return a pointer to the END of the string "\r\n\r\n" in pBuffer, or NULL if not found.
static const char *pFindHttpHeaderEnd(const char *pBuffer, size_t size)
{
    const char *pTmp = NULL;
    size_t y = strlen(HTTP_HEADER_END_MARKER);

    for (size_t x = 0; (pTmp == NULL) && (x < size - y); x++) {
        if (memcmp(pBuffer + x, HTTP_HEADER_END_MARKER, y) == 0) {
            pTmp = pBuffer + x + y;
        }
    }

    return pTmp;
}

// Copy from pBufferFrom into pBufferTo, returning the number of bytes copied
static size_t copyBuffer(const char *pBufferFrom, size_t sizeBufferFrom,
                         char *pBufferTo, size_t sizeBufferTo)
{
    if (sizeBufferFrom > sizeBufferTo) {
        printf("WARNING: unable to copy %d character(s) into buffer.\n", sizeBufferFrom - sizeBufferTo);
        sizeBufferFrom = sizeBufferTo;
    }
    memcpy(pBufferTo, pBufferFrom, sizeBufferFrom);

    return sizeBufferFrom;
}

// Find "Content-Length: <length>" in pBuffer and return the value of <length>
static size_t getContentLength(const char *pBuffer, size_t size)
{
    size_t length = 0;
    const char *pTmp = NULL;
    size_t y = strlen(HTTP_HEADER_CONTENT_LENGTH);

    for (size_t x = 0; (pTmp == NULL) && (x < size - y); x++) {
        if (memcmp(pBuffer + x, HTTP_HEADER_CONTENT_LENGTH, y) == 0) {
            pTmp = pBuffer + x + y;
        }
    }

    if ((pTmp != NULL) && (pTmp < pBuffer + size)) {
        // Found the content length string, grab the value of <length> which should follow
        length = strtol(pTmp, NULL, 10);
    }

    return length;
}

...

// This part inside your code
[... 165 lines cut ...]
                } else {
                    printf("### %06d ERROR: uSockWrite() failed  after %d characters (%d).\n",
                           uPortGetTickTimeMs() - startTimeMs, pRead - buffer, errorCodeOrSize);
                }
                // Stop read/writing from/to the socket
                uSockShutdown(sockHandle, U_SOCK_SHUTDOWN_READ_WRITE);
            } else {
                printf("### %06d ERROR: uSockConnect() to %s failed (%d).\n",
                       uPortGetTickTimeMs() - startTimeMs, MY_SERVER_URL, errorCodeOrSize);
            }
        } else {
            printf("### %06d ERROR: uSockSecurity() failed (%d).\n",
                   uPortGetTickTimeMs() - startTimeMs, errorCodeOrSize);
        }

        // Close the socket
        uSockClose(sockHandle);
        uSockCleanUp();
    } else {
        printf("### %06d ERROR: uSockCreate() failed (%d).\n", uPortGetTickTimeMs() - startTimeMs, sockHandle);
    }
} else {
    printf("### %06d ERROR: uSockGetHostByName() failed (%d).\n", uPortGetTickTimeMs() - startTimeMs, errorCodeOrSize);
}
```

alexmaron81, 2023-08-11:
Many thanks Rob!
[image]

I still have one question: I already have an IPv4 address when I use inet_addr(MY_SERVER_IP); I get the IP in reverse order?

RobMeades, 2023-08-11:
Excellent: 34 kbits/s is quite a lot better than I was getting.

>  I get the IP in reverse order?

That will be an [endianness](https://en.wikipedia.org/wiki/Endianness) thing: the simple print I do in the code I gave you does not do an endianness check, it just prints an `int32_t` in whatever endianness the MCU has.  If your MCU happens to be little-endian, not big-endian or "network order", because it is what IP uses, then the order will be reversed.

RobMeades, 2023-08-14:
@alexmaron81: I'm going to close this now as I think we're done, please re-open if you disagree.

closed 2023-08-14 by RobMeades