u-blox/ubxlib #125: Error Setting Power Mode on MAX-M10S

TryingToGITthis, 2023-08-19:
Getting position updates just fine but when I try to set the power mode, I'm getting a -1025 error returned.
Am I missing something in setup or is the M10 not supported in this way?

```
errorcode = uDeviceOpen(&gDeviceCfg, &gnssHandle);
errorcode = uNetworkInterfaceUp(gnssHandle, U_NETWORK_TYPE_GNSS, &gnetworkcfg);
gOriginalMode = uGnssPwrGetMode(gnssHandle);
uGnssSetRetries(gnssHandle, 4);
errorcode = uGnssPwrSetMode(gnssHandle, U_GNSS_PWR_SAVING_MODE_ON_OFF );
if (errorcode != 0)
   printk("Error setting power saving: %d\n", errorcode);
errorcode = uGnssPwrSetMode(gnssHandle, U_GNSS_PWR_FLAG_EXTINT_INACTIVITY_ENABLE );
if (errorcode != 0)
   printk("Error setting power saving: %d\n", errorcode);
```

RobMeades, 2023-08-20:
Hi there: which one is returning [U_GNSS_ERROR_NACK](https://github.com/u-blox/ubxlib/blob/2d66485aca773f1a80872d4e3add0a1d81f73ead/gnss/api/u_gnss.h#L68) (-1025): `uGnssPwrSetMode(gnssHandle, U_GNSS_PWR_SAVING_MODE_ON_OFF )` or `uGnssPwrSetMode(gnssHandle, U_GNSS_PWR_FLAG_EXTINT_INACTIVITY_ENABLE )`?

I can't see why the first one would return an error but, for the second one, what you're passing into `uGnssPwrSetMode()` is not a member of the [uGnssPwrSavingMode_t](https://github.com/u-blox/ubxlib/blob/2d66485aca773f1a80872d4e3add0a1d81f73ead/gnss/api/u_gnss_pwr.h#L146) `enum`: you're passing in a member of the [uGnssPwrFlag_t](https://github.com/u-blox/ubxlib/blob/2d66485aca773f1a80872d4e3add0a1d81f73ead/gnss/api/u_gnss_pwr.h#L168) `enum` instead.

For that one what you probably meant was:

[uGnssPwrSetFlag](https://github.com/u-blox/ubxlib/blob/2d66485aca773f1a80872d4e3add0a1d81f73ead/gnss/api/u_gnss_pwr.h#L333)`(gnssHandle, 1UL << U_GNSS_PWR_FLAG_EXTINT_INACTIVITY_ENABLE)`

The compiler should have issued an error, or at least a waning, about this; you might want to increase your error levels.

TryingToGITthis, 2023-08-20:
uGnssPwrSetMode(gnssHandle, U_GNSS_PWR_SAVING_MODE_ON_OFF ) is returning the -1025. 

Thanks for flag setting issue, I'll fix that.

RobMeades, 2023-08-20:
Hmmm, not sure I can explain that.  Exactly which M10 device is this?

Here is a segment of log output from our test system [running through setting the power saving modes](https://github.com/u-blox/ubxlib/blob/94411b16bd6e8ee30a7ddbd86602f22c0aa71162/gnss/test/u_gnss_pwr_test.c#L389) in test `gnssPwrSaving()` on test instance 10.0 (which is a MAX-M10S, FW version 5.10, protocol version 34.10), where `U_GNSS_PWR_SAVING_MODE_ON_OFF` is mode 1:

```
2023-08-18 00:51:39,022 u_monitor_10.0 INFO    - progress update - item gnssPwrSaving() started on 00:51:39.022444.
2023-08-18 00:51:39,033 u_monitor_10.0 INFO    - U_GNSS_PWR_TEST: testing on transport I2C...
2023-08-18 00:51:39,043 u_monitor_10.0 INFO    - U_GNSS_TEST_PRIVATE: test preamble start.
2023-08-18 00:51:39,054 u_monitor_10.0 INFO    - U_GNSS_TEST_PRIVATE: opening GNSS I2C 0...
2023-08-18 00:51:39,065 u_monitor_10.0 INFO    - U_GNSS_TEST_PRIVATE: adding a GNSS instance...
2023-08-18 00:51:39,077 u_monitor_10.0 INFO    - U_GNSS: initialising with ENABLE_POWER pin not connected, transport type I2C.
2023-08-18 00:51:41,163 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 01 00 d0 20 8a 55.
2023-08-18 00:51:41,183 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 01 00 d0 20 00 [body 9 byte(s)].
2023-08-18 00:51:41,195 u_monitor_10.0 INFO    - U_GNSS_PWR_TEST: power-saving mode was 0.
2023-08-18 00:51:41,206 u_monitor_10.0 INFO    - U_GNSS_PWR_TEST: setting power-saving mode 2...
2023-08-18 00:51:41,282 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 01 00 d0 20 02 8e f1.
2023-08-18 00:51:41,293 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
2023-08-18 00:51:41,392 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 01 00 d0 20 8a 55.
2023-08-18 00:51:41,413 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 01 00 d0 20 02 [body 9 byte(s)].
2023-08-18 00:51:41,425 u_monitor_10.0 INFO    - U_GNSS_PWR_TEST: setting power-saving mode 1...
2023-08-18 00:51:41,524 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 01 00 d0 20 01 8d f0.
2023-08-18 00:51:41,536 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
2023-08-18 00:51:41,632 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 01 00 d0 20 8a 55.
2023-08-18 00:51:41,653 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 01 00 d0 20 01 [body 9 byte(s)].
2023-08-18 00:51:41,665 u_monitor_10.0 INFO    - U_GNSS_PWR_TEST: setting power-saving mode 0...
2023-08-18 00:51:41,762 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 01 00 d0 20 00 8c ef.
2023-08-18 00:51:41,774 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
2023-08-18 00:51:41,872 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 01 00 d0 20 8a 55.
2023-08-18 00:51:41,893 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 01 00 d0 20 00 [body 9 byte(s)].
2023-08-18 00:51:42,002 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 01 00 d0 20 00 8c ef.
2023-08-18 00:51:42,014 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
```

You might want to set [uGnssSetUbxMessagePrint(gnssHandle, true)](https://github.com/u-blox/ubxlib/blob/94411b16bd6e8ee30a7ddbd86602f22c0aa71162/gnss/api/u_gnss.h#L287) and capture the trace output to see if there is anything obvious in the UBX message exchange with the device.

TryingToGITthis, 2023-08-20:
Here is the output:

`U_GNSS: initialising with ENABLE_POWER pin not connected, transport type I2C.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 04 04 00 00 00 09 00 17 76.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 01 00 d0 20 8a 55.
U_GNSS: decoded UU_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 01 00 d0 20 01 8d f0.
U_GNSS: decoded UBX response 0x05 0x00: 06 8a [body 2 byte(s)].
Error setting power saving: -1025
Error setting power saving: -1025
U_GNSS: sent command b5 62 06 8b 18 00 00 00 00 00 02 00 d0 40 03 00 d0 40 05 00 d0 30 07 00 d0 20 06 00 d0 20 c0 35.
U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 02 00 d0 40 0a 00 00 00 03 00 d0 40 0a 00 00 00 05 00 d0 30 00 00 07 00 d0 20 00 06 00 d0 20 00 [body 36 byte(s)].`

RobMeades, 2023-08-20:
Well, that is pretty clearly a NACK:

```
U_GNSS: decoded UU_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 01 00 d0 20 01 8d f0.
U_GNSS: decoded UBX response 0x05 0x00: 06 8a [body 2 byte(s)]
```

You could maybe add a call to [uGnssInfoGetFirmwareVersionStr()](https://github.com/u-blox/ubxlib/blob/94411b16bd6e8ee30a7ddbd86602f22c0aa71162/gnss/api/u_gnss_info.h#L91) with, say, a 256 byte buffer and print the string that comes back, just so that we could get the exact device version details?

Is it possible that you [have one of the flavours of GNSS device that doesn't support power saving](https://github.com/u-blox/ubxlib/blob/94411b16bd6e8ee30a7ddbd86602f22c0aa71162/gnss/api/u_gnss_pwr.h#L37)?

TryingToGITthis, 2023-08-20:
MAX-M10S
Firmware Version: 537075204

RobMeades, 2023-08-20:
I don't understand that FW version but the MAX-M10S should be fine since it is exactly the same device as was test with in the output I pasted-in above (GNSS device FW version 5.10).

I have _very_ occasionally seen GNSS devices return a NACK if they are hit with many power-saving related requests in succession: maybe it is a timing thing and somehow the device is not ready to accept the power saving mode change?  For debug purposes, maybe try adding a delay before sending the power saving mode to see if that makes a difference?

TryingToGITthis, 2023-08-20:
Sorry, did a %d instead of %s. 
Firmware Version: ROM SPG 5.10 (7b202e)
Five second delay before the power saving mode change didn't help.

RobMeades, 2023-08-20:
So, we have the same GNSS device, same FW version.

In our test we happen to set the power saving modes in the order 2, 1, 0: for debug purposes, maybe try setting mode 2 ([U_GNSS_PWR_SAVING_MODE_CYCLIC_TRACKING](https://github.com/u-blox/ubxlib/blob/94411b16bd6e8ee30a7ddbd86602f22c0aa71162/gnss/api/u_gnss_pwr.h#L158C5-L158C43) instead, or perhaps try setting the mode that the device is already in, just to see if those succeed?

TryingToGITthis, 2023-08-20:
Same error. Any other thoughts? The device never enters sleep mode either.

This is all the code up to changing the mode:
`
   uPortLogOff();
   // initiate ubxlib
   uPortInit();
   uPortI2cInit(); // you only need this if an i2c interface is used
   uDeviceInit();

    int32_t errorcode;
    uDeviceHandle_t gnssHandle;
    printk("\ninitiating the module...\n");
    errorcode = uDeviceOpen(&gDeviceCfg, &gnssHandle);
    if (errorcode == 0) {
      // bring up the gnss
      errorcode = uNetworkInterfaceUp(gnssHandle, U_NETWORK_TYPE_GNSS, &gnetworkcfg);
      if (errorcode == 0) {
         char str[256];
         uGnssInfoGetFirmwareVersionStr(gnssHandle, str, 256);
         printk("Firmware Version: %s\n", str);
         int32_t gOriginalMode = uGnssPwrGetMode(gnssHandle);
         printk("Original power mode: %d\n", gOriginalMode);
         uGnssSetRetries(gnssHandle, 4);
      k_sleep(K_SECONDS(5));
         errorcode = uGnssPwrSetMode(gnssHandle, U_GNSS_PWR_SAVING_MODE_CYCLIC_TRACKING);
`

TryingToGITthis, 2023-08-20:
This is on a custom board if it matters.

[image]

RobMeades, 2023-08-20:
Maybe, for some reason, the various timers that the GNSS device uses to govern the durations of the different power saving phases have somehow gone screwy, so it is unable to power save?  Not that I can see how that would happen, but maybe call [uGnssPwrGetTiming()](https://github.com/u-blox/ubxlib/blob/94411b16bd6e8ee30a7ddbd86602f22c0aa71162/gnss/api/u_gnss_pwr.h#L474), print out all of the values to see if they make sense, and maybe write these back again to be completely sure, correcting any that don't look sensible, with [uGnssPwrSetTiming()](https://github.com/u-blox/ubxlib/blob/94411b16bd6e8ee30a7ddbd86602f22c0aa71162/gnss/api/u_gnss_pwr.h#L406)?

TryingToGITthis, 2023-08-20:
Here are the values:
`
Acq Period Secs: 10
Acq Retry Period Secs: 10
On Time Secs: 0
Max Acq Time Secs: 0
Min Acq Time Secs: 0
`

RobMeades, 2023-08-20:
Those look OK to me.

I'm not an expert in the GNSS device and/or HW, but I wonder if `V_BKUP` needs to be connected to VCC?  When the device enters power saving it enters back-up mode, which might rely on the supply from that pin?

RobMeades, 2023-08-20:
Another possibility: checking the integration manual, it says that RAM is erased during on/off power saving, restored from BBRAM on wake-up, with BBRAM maintained from VIO.

The code in `uGnssPwr` only sets the configuration in the RAM layer; maybe, somehow, in your case with `V_BKUP` not connected, the device is NACKing the request because it knows that it doesn't make sense, it will just be lost when power saving is entered/left?  This is just guess of course but, thinking about it, only setting the power saving settings in the RAM layer doesn't make sense for on/off power saving.

To debug if that has an impact, search [u_gnss_pwr.c](https://github.com/u-blox/ubxlib/blob/master/gnss/src/u_gnss_pwr.c) for all occurrences of `uGnssCfgPrivateValSetList` and replace `U_GNSS_CFG_VAL_LAYER_RAM` with `U_GNSS_CFG_VAL_LAYER_RAM | U_GNSS_CFG_VAL_LAYER_BBRAM`.

TryingToGITthis, 2023-08-20:
V_BKUP was never connected because main power is never removed. Unless the internal circuitry disconnects 
VCC and V_IO. 

By changing those occurrences, my uGnssPwrSetTiming() settings are now being saved between code updates:
Acq Period Secs: 3600
Acq Retry Period Secs: 120
On Time Secs: 10
Max Acq Time Secs: 240
Min Acq Time Secs: 12

No change to the error code though.

RobMeades, 2023-08-20:
OK, so I will make the change to add the BBRAM layer to the CFG-VAL save operations, that makes sense, but still getting a NACK back for the request to enter on/off power saving I can't explain.

I will see if I can find someone who knows more about this than I to give me some hint tomorrow.

TryingToGITthis, 2023-08-20:
Thanks for all your help. Tomorrow will try supplying power to V_BKUP and see if it makes a difference.

RobMeades, 2023-08-21:
Writing to BBRAM also is now pushed to `master` here: https://github.com/u-blox/ubxlib/commit/7446b97dd61d099c41a30a129511af3f1d04b221.  I've not found anyone who can explain the NACK yet though.

RobMeades, 2023-08-29:
Hi again: did you get any further with the "NACK" issue?  I have not found anyone here who is able to come up with a reason why you should get a NACK; I have shown them your schematic but that didn't raise any eyebrows.

TryingToGITthis, 2023-08-29:
We have made a modification to the board to enable V_BKUP. Should have that tested by the end of the week to see if it makes a difference.

RobMeades, 2024-01-04:
I'm going to close this as I guess you have resolved your issues, or maybe this is not relevant any more, etc.  Please re-open if there is more to discuss on this topic.

RobMeades, 2024-03-20:
@nautalert: apologies, just wanted to confirm that this is the same issue as it has been a little while now.  Let me restate (and re-open) the issue, please let me know if any of my assumptions are incorrect.

- you are using a `MAX-M10S-00B` GNSS chip, ROM SPG 5.10 (7b202e)
- whatever you do with it, the response to the UBX command `b5 62 06 8a 09 00 01 03 00 00 01 00 d0 20 02 90 01` is a 0x05 0x00, which is a NACK.

FYI, breaking down the `UBX` command:
```
    b5 62 06 8a 09 00 01 03 00 00 01 00 d0 20 02 90 01
    ^^^^^| UBX header|  |  |  |  |           |  |     |
          ^^^^^| Message class/ID: UBX_CFG_VALSET (see interface manual section 3.10.5.2)
                ^^^^^| Message body length: 9 bytes
                      ^^| Version: 1         |  |     |
                         ^^| Layers: RAM (bit 0) & BB RAM (bit 1)
                            ^^| Transaction: 0 (transactionless)
                               ^^| Reserved (hence 0)
                                  ^^^^^^^^^^^| Key ID: 0x20d00001 CFG-PM-OPERATEMODE (see interface manual section 4.9.1)
                                              ^^| Value for this key ID: 2 PSM cyclic tracking operation (see interface manual section 4.9.1, table 34)
                                                 ^^^^^| FCS (see working example below for comparison)
```

..and we send the same message to a MAX-M10S chip in our test system, where the exchange looks like this:

```
2024-03-17 11:18:29,385 u_monitor_10.0 INFO    - U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 01 00 d0 20 02 90 01.
2024-03-17 11:18:29,403 u_monitor_10.0 INFO    - U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
```

...and where the MAX-M10S chip reports the following version structure:

```
ROM SPG 5.10 (7b202e)
000A0000
FWVER=SPG 5.10
PROTVER=34.10
MOD=MAX-M10S
GPS;GLO;GAL;BDS
SBAS;QZSS
```

Hmph.

RobMeades, 2024-03-20:
Purely as an experiment, could you try compiling `ubxlib` with `U_GNSS_CFG_LAYERS_SET=U_GNSS_CFG_VAL_LAYER_RAM` (that or hand-modify [gnss/src/u_gnss_private.h](https://github.com/u-blox/ubxlib/blob/1b94ec51334a5a326328341ad1e5a078692c2f46/gnss/src/u_gnss_private.h#L108-L114) to make the change), just to see if that makes a darned bit of difference?

RobMeades, 2024-03-20:
FYI, find attached a complete log of our standard [regression test of GNSS power saving](https://github.com/u-blox/ubxlib/blob/1b94ec51334a5a326328341ad1e5a078692c2f46/gnss/test/u_gnss_pwr_test.c#L348-L663), run on the MAX-M10S device over I2C, where you can see the command sent at line 11 is identical to what you are sending.

[power_saving_test_m10.txt](https://github.com/u-blox/ubxlib/files/14675536/power_saving_test_m10.txt)

I have already asked our positioning AE team why this might happen and everyone looked at my blankly; we must be missing something here, some fact that is probably staring us collectively in the face.  I will ask again...

nautalert, 2024-03-25:
Making the change in u_gnss_private.h didn't help.

RobMeades, 2024-03-25:
No, didn't think it would.  I am trying to find someone internally who has any idea how the problem you are seeing could ever occur.  Out of interest, does this happen on more than one device, just in case it is something really crazy going wrong with one particular chip?

nautalert, 2024-03-25:
It's happening on all chips.

nautalert, 2024-04-03:
Any update on why this might be happening?

RobMeades, 2024-04-03:
Apologies, I will prod the experts again...

RobMeades, 2024-04-10:
The gentleman who knows what he's talking about has got back to me and there is a slim possibility: might you have enabled system BeuDou B1C (this is specifically B1C, not B1l) in the GNSS chip?  He says that "PSM does not support BeiDou B1C"; BeiDou B1C seems to be in the default set for SAM-M10Q but it is not for MAX-M10S, you would have to have enabled it specifically.

RobMeades, 2024-04-10:
Examples of how to do configuration using the `CFG_VAL` mechanism (that you must use for M10 and later GNSS modules), which I guess you've found, are in [cfg_val_main.c](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/example/gnss/cfg_val_main.c#L273).

For the system types, you want the configuration key IDs with `_SIGNAL` in the name, [see here](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/gnss/api/u_gnss_cfg_val_key.h#L3070).  So to read all of them at once, you could use [uGnssCfgValGetListAlloc()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/gnss/api/u_gnss_cfg.h#L468), something like:

```
    uint32_t keyId;
    uGnssCfgVal_t *pCfgValList = NULL;
    int32_t numValues;

    keyId = U_GNSS_CFG_VAL_KEY(U_GNSS_CFG_VAL_KEY_GROUP_ID_SIGNAL, U_GNSS_CFG_VAL_KEY_ITEM_ID_ALL, 0);
    numValues = uGnssCfgValGetAlloc(devHandle, keyId, &pCfgValList, U_GNSS_CFG_VAL_LAYER_RAM);
    for (size_t x = 0; x < numValues ; x++) {
        printf("%d keyId 0x%08x = ", x + 1, pCfgValList->keyId);
        if (pCfgVal->value) {
            printf("true");
        } else {
            printf("false");
        }
        printf("\n");
        pCfgValList++;
    }
    uPortFree(pCfgValList);
```

The exact key for Beidou B1C is missing in our list of key IDs (I will fix that) but, no matter, you can just write-in the key ID yourself as follows (this assuming you are just setting the key value in RAM):

```
if (uGnssCfgValSet(devHandle, 0x1031000f, 1, U_GNSS_CFG_VAL_TRANSACTION_NONE, U_GNSS_CFG_VAL_LAYER_RAM) == 0) {
    printf("Set 0x1031000f (AKA CFG-SIGNAL-BDS_B1C_ENA) to true.\n");
} else {
    printf("Unable to set 0x1031000f!\n");
}
```

...value taken from the [interface manual](https://content.u-blox.com/sites/default/files/u-blox-M10-SPG-5.10_InterfaceDescription_UBX-21035062.pdf), section 4.9.21:

[image]

RobMeades, 2024-04-10:
Update: you should probably replace `U_GNSS_CFG_VAL_LAYER_RAM` in my example code above with `U_GNSS_CFG_VAL_LAYER_RAM | U_GNSS_CFG_VAL_LAYER_BBRAM` since a change to the signal settings will reset the GNSS chip.

RobMeades, 2024-04-10:
D'oh, and of course you want a `0` in there to disable, so:

```
if (uGnssCfgValSet(devHandle, 0x1031000f, 0, U_GNSS_CFG_VAL_TRANSACTION_NONE, U_GNSS_CFG_VAL_LAYER_RAM) == 0) {
...
```

nautalert, 2024-04-11:
You were correct that BeuDou B1C was set. Power mode is now able to be set correctly. Thank you for your help with this.

RobMeades, 2024-04-11:
Phew, my apologies that it took so long.  I will add a note to the code somewhere about this in the hope that I remember this corner-case the next time someone asks.  Will close this issue [again :-)] now: feel free to re-open it, or open a new one, if there is more to discuss.

closed 2024-04-11 by RobMeades