u-blox/ubxlib #126: STM32U5 support feature

rickko40, 2023-08-21:
Is there any plan to support a version of the ubxlib for other STM32 families?  We started with the STM32F4 dev kit, but are looking to use STM32U5xx with CMSIS v2 api for our product.  Are there plans to support either STM32U5 with Azure RTOS or STM32U5 with CMSIS v2?  If ubxlib releases are planned with Azure RTOS we may consider moving our application to use this api and middleware since that appears to be STM's roadmap for it's newer devices.

hkro, 2023-08-22:
We have looked into supporting STM32U5. Currently the situation with RTOS support of STM32U5 and `ubxlib` looks like this:

## AzureRTOS 
[STM32CubeU5](https://github.com/STMicroelectronics/STM32CubeU5/tree/main) has Azure RTOS / ThreadX natively built-in under the middlewares/ST folder (interesting). The [STM32CubeU5 HAL Driver MCU Component](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Drivers/STM32U5xx_HAL_Driver) is copied in there, no git submodule is used, perhaps synced manually.

 `ubxlib` currently does not support Azure RTOS. If there will be enough demand we'll add it.

## FreeRTOS
There is a [STM32U5 reference repo vailable](https://github.com/FreeRTOS/iot-reference-stm32u5/) which includes the [STM32U5 HAL driver component](https://github.com/STMicroelectronics/stm32u5xx_hal_driver) as submodule. This model of driver publication/delivery is relatively new from ST as written in the last paragraph [here](https://github.com/STMicroelectronics/stm32u5xx_hal_driver).

 `ubxlib` supports the legacy (monolitic) STM32Cube package. Extending that to support the new STM32U5 `MCU components` we'll consider on the long run.

## Zephyr OS

Zephyr has hal_stm32 build-in under https://github.com/zephyrproject-rtos/hal_stm32.

`ubxlib` support if you run Zephyr on the STM32U5:
- Include `ubxlib` as Zephyr module as described [here](https://github.com/u-blox/ubxlib/tree/master/port/platform/zephyr).
- Then all left to do is changing the node names of the interfaces to which u-blox modules are attached (e.g., I2C) in the [Zephyr device tree](https://docs.zephyrproject.org/latest/build/dts/intro-syntax-structure.html#example) to match the values of STM32U5 (given by ST). This is currently the easiest way to use STM32U5 with `ubxlib`.

Hope this helps.

RobMeades, 2024-03-04:
We now include in our regression testing `ubxlib` running on an STM32U575ZI processor (Nucleo U575ZI board) under Zephyr, talking to a cellular module (over UART) and a GNSS device (over I2C), see e96516309e8a05e9ca910931e511b00770a0a0a4.  Given that works, I'm afraid it is unlikely we would do an STM32Firmware/CMSIS-based port of the same MCU, so we will close this issue.

closed 2024-03-04 by RobMeades