u-blox/ubxlib #128: Using uGnssDec API

darkhmiza, 2023-08-29:
Hello everyone ( Rob ) :).

I've been trying to use  the uGnssDec API, Here is my code :

```
#include <zephyr/kernel.h>
#include <sys/printk.h>
#include <stdio.h>
#include<string.h>
#define LOG_LEVEL 3
#include <logging/log.h>
#include <zephyr/drivers/uart.h>
#include<time.h>
#include "ubxlib.h"

static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_GNSS,
    .deviceCfg = {
        .cfgGnss = {
            .moduleType = U_GNSS_MODULE_TYPE_M9,  // This is the module type
            .pinEnablePower = -1,
            .pinDataReady = -1
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_I2C,
    .transportCfg = {
        .cfgI2c = {
            .i2c = 1, // This is the I2C HW block - ubxlib knows nothing about the device tree so you need to tell it
            .pinSda = -1,
            .pinScl = -1
        },
    },
};

static size_t gMessageCount = 0;
char buffer[512];

void myCallback(uDeviceHandle_t gnssHandle,
                const uGnssMessageId_t *pMessageId,
                int32_t errorCodeOrLength,
                void *pCallbackParam)
{
     uPortLog("Calback Working 1 ");
    (void) pMessageId;
    int j=0;
    char dist_str[30];
    if (errorCodeOrLength > 0) {
        if (errorCodeOrLength > sizeof(buffer)) {
            errorCodeOrLength = sizeof(buffer);
        }
        errorCodeOrLength = uGnssMsgReceiveCallbackRead(gnssHandle,
                                                        buffer,
                                                        errorCodeOrLength);
        if (errorCodeOrLength > 0) {

            uGnssDec_t *pDec = pUGnssDecAlloc(buffer, errorCodeOrLength);
            if ((pDec != NULL) && (pDec->errorCode == 0)) {

                   uPortLog("Calback Working 2 ");
            }
            // Must *always* free the memory that pUGnssDecAlloc() allocated
            uGnssDecFree(pDec);
        }
    }
}

void main()
{
    /* Display application name. */
    LOG_INF(APP_NAME);
    /* Hold copy of status register state here for reference so that it can be examined at a debug breakpoint. */
     uint32_t status_reg = 0;
     /* Hold copy of frame length of frame received (if good) so that it can be examined at a debug breakpoint. */
     uint16_t frame_len = 0;
    /*GPS handle*/
    uDeviceHandle_t devHandle = NULL;
    int32_t errorCode=0;
    // Initialise the APIs we will need
  
   
    uGnssMessageId_t messageId = {0};

    // Enough room for the UBX-NAV-PVT message, which has a body of length 92 bytes,
    // and any NMEA message (which have a maximum size of 82 bytes)
    int j=0;
    char dist_str[30]; 
    int32_t length = 0;
    int32_t returnCode;
    int32_t handle;
    uPortInit();
    uPortI2cInit(); // You only need this if an I2C interface is used
    uDeviceInit();
    
    while(returnCode!=0){
    returnCode=uDeviceOpen(&gDeviceCfg, &devHandle);
    

    }

int l=0;
   
        char *pBuffer = (char *) pUPortMalloc(MY_MESSAGE_BUFFER_LENGTH);
        uGnssCfgSetRate(devHandle,100,-1,-1);
        messageId.type = U_GNSS_PROTOCOL_UBX;
        messageId.id.ubx = 0x0107;
        handle = uGnssMsgReceiveStart(devHandle, &messageId, myCallback,pBuffer);
if (handle >= 0) {
            // Wait a while for some messages to arrive
            uPortTaskBlock(5000);
            // Stop the message receiver(s) once more
            uGnssMsgReceiveStopAll(devHandle);
        } else {
            uPortLog("Unable to start message receiver!\n");
        }

    uDeviceDeinit();
    uPortI2cDeinit(); 
    uPortDeinit();

    uPortLog("Done.\n");

    uPortFree(pBuffer);

}

```

As you can see I am simply logging a message when I go to the callback but nothing happens. Perhaps I am coding this the wrong way?

RobMeades, 2023-08-29:
Hi there: could you post the output you get, so that we can see what you're seeing?

RobMeades, 2024-01-04:
I'm going to close this now as I guess you either resolved your issue or it is no longer relevant etc.  Please re-open if there is more to discuss.

closed 2024-01-04 by RobMeades