u-blox/ubxlib #131: Trouble Porting to ESP-IDF

michaelboeding, 2023-09-06:
I've followed all the steps in https://github.com/u-blox/ubxlib/tree/master/port/platform/esp-idf/mcu/esp32 to setup the hello_world esp idf project with the required config and its not building correctly. I never get the build/esp-idf/ubxlib/libubxlib.a to appear under my build files. 

Heres my current code. 

### Project level CMakeLists.txt file 

```cmake 
# This is the project CMakeLists.txt file for the test subproject
cmake_minimum_required(VERSION 3.5)

# Include the body of stuff in the directory
# above, main and the tests component from below
set(EXTRA_COMPONENT_DIRS "/Users/michaelboeding/Documents/Boeding Technologies/ToughTail/Code/ubxlib/port/platform/esp-idf/mcu/esp32/components"
                         "main")

set(SDKCONFIG "sdkconfig" CACHE STRING "IDF sdkconfig file path")

# Reduce the component list to a minimum
set(COMPONENTS "driver" "esptool_py" "unity" "ubxlib" "main" "heap")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(ubxlib)

```

### Main/CMakeLists.txt file 

```cmake

idf_component_register(SRCS "hello_world_main.c"
                    INCLUDE_DIRS "")

target_compile_options(${COMPONENT_TARGET} PUBLIC -DU_CFG_TEST_CELL_MODULE_TYPE=U_CELL_MODULE_TYPE_SARA_R5 -Wno-missing-field-initializers)

```

### Build Error 

[image]

### Build folder (missing the build/esp-idf/ubxlib/libubxlib.a file) 

[image]

Am I doing something wrong here or are the examples just not working anymore/outdated for the esp-idf. Any help would be appreciated thanks. It almost seems like the CMakeLists.txt file is never linking since it doesn't show up in the build.

michaelboeding, 2023-09-06:
Im using version 4.4 of the esp-idf

michaelboeding, 2023-09-06:
> Im using version 4.4 of the esp-idf

More specificly v4.4.5-343-gd89b1e45fa

RobMeades, 2023-09-06:
We've tested with 4.3, 4.4 and 5 so I wouldn't think there is a version issue but I'll try doing exactly what you have done with v4.4.5 to see what gives.

Only thought at the moment is the space in the path name "/Users/michaelboeding/Documents/Boeding Technologies/ToughTail/Code/ubxlib/port/platform/esp-idf/mcu/esp32/components", which of course _shouldn't_ be a problem in the 21st century but it is something we would generally avoid doing in case of subtle issues in whatever bit of whatever toolchain.

michaelboeding, 2023-09-06:
I'll also try to move it into a different directory and see if that helps. Really appreciate the help and quick response.

michaelboeding, 2023-09-06:
I think that was the issue. It seems to be working now...continuing to test to make sure everything runs correctly but its now building.

RobMeades, 2023-09-06:
Phew!  I'm still downloading version 4.4.5 so I'll let that complete in case I need to do more.  Please close this issue when you're completely happy and I'll recover some disk space :-).

michaelboeding, 2023-09-06:
Haha, I was able to run the example and got the output below. Which seems to be correct but it was incredibly fast which makes me a little suspicious. Is that normal? And does that output looks correct? I was previously writing my own AT commands to the device over serial and it was way slower than that...like this completed in maybe 5 seconds. 

```bash 

U_CELL_PWR: powering on.
AT
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

02.06,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=1,1300

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 0,,,"10000101","00000011"

OK
AT+UMNOPROF?

+UMNOPROF: 90

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
Opened device with return code 0.
Bringing up the network...
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

[... 439 lines cut ...]
        }

        // Close the device
        // Note: we don't power the device down here in order
        // to speed up testing; you may prefer to power it off
        // by setting the second parameter to true.
        uDeviceClose(devHandle, false);

    } else {
        printf("Unable to bring up the device!\n");
    }

    // Tidy up
    uDeviceDeinit();
    uPortDeinit();

    printf("Done.\n");

}

// End of file

```

If that all looks correct and sounds right ill close this out! Again thanks for the help!

RobMeades, 2023-09-06:
Yup, that all looks correct to me.  With SARA-R5 on a Cat-M1 network the only thing that might lag would be the out-of-the-box network search with a global roaming SIM, which I guess the Hologram one is, but it looks like the search was pretty snappy (maybe only one cellular operator in view?) so it snapped-in pretty quickly.

michaelboeding, 2023-09-06:
I confirmed with holograms data viewer and it looks like it sees the connection and got the bytes....crazy fast im super impressed. My code was taking over 180 seconds to connect. Im using the SparkFun LTE GNSS Breakout - SARA-R5 https://www.sparkfun.com/products/18031 . And I was hesitant to believe it worked since the blue status light never came on. But must be something on their board. 

[image]

closed 2023-09-06 by michaelboeding