u-blox/ubxlib #132: Support for generic UBX-CFG APIs for GNSS M8 module.

minhnt97, 2023-09-07:
Hello everyone.

I've been looking for public APIs that support generic configuration of UBX-CFG for my ZOE-M8Q modules. For example, I want to enable concurrent GNSS using message UBX-CFG-GNSS (0x06 0x3E), but I can't find a suitable API for this purpose.

The related functions I found are located in 'u_gnss_private.h', which is dedicated to internal usage:
https://github.com/u-blox/ubxlib/blob/c4e77621393f92b9ee5b3daed3ac356354e37c29/gnss/src/u_gnss_private.h#L872-L907

Any suggestion on this problem? Thank you in advance.

minhnt97, 2023-09-08:
@RobMeades Thanks for the explanation.

RobMeades, 2024-01-04:
I'm going to assume this is done, please re-open if that is not the case.

closed 2024-01-04 by RobMeades