u-blox/ubxlib #133: Issue with uGnssPosGetStreamedStart()

michaelboeding, 2023-09-08:
It seems like I never get coordinates via the stream when using uGnssPosGetStreamedStart(). I can make calls directly using the uGnssPosGetStart() which will then return correctly in the callback with the coordinates but the stream never sends any data it seems. Below is my code, maybe Im just doing something incorrectly. This is all using the esp-idf port. 

### main.cpp
```c++

//create an instance of the MAXM10S class
MAXM10S* maxm10s = new MAXM10S();
if (maxm10s->setup() == true ){
      printf("Setup successful!\n");
      maxm10s->startGpsStream();
      maxm10s->getCoordinates();
}else{
      printf("Setup failed!\n");
}
    
 ```
 
 
 
 ### Setup function 
 
 ```c++
 
 //setup the gps module
bool MAXM10S::setup(){
    int32_t returnCode;
    int32_t guardCount = 0;
    uPortInit();
    uDeviceInit();
    // Open the device
    returnCode = uDeviceOpen(&gDeviceCfg, &devHandle);
    printf("Opened device with return code %d.\n", returnCode);
    if (returnCode == 0) {
        //configure the protocols we want to use
        uGnssCfgSetProtocolOut(devHandle, U_GNSS_PROTOCOL_UBX, true);
        uGnssCfgSetProtocolOut(devHandle, U_GNSS_PROTOCOL_NMEA, false);
        uGnssCfgSetProtocolOut(devHandle, U_GNSS_PROTOCOL_RTCM, false);
        // Start to get position
        printf("Starting position stream.\n");
        return true;

        // Close the device
        // Note: we don't power the device down here in order
        // to speed up testing; you may prefer to power it off
        // by setting the second parameter to true.
       // uDeviceClose(devHandle, false);

    } else {
        printf("Unable to open GNSS!\n");
        return false;
    }
}

```

### Devic config 

```c++

//the gps device 
        uDeviceCfg_t gDeviceCfg = {
            .deviceType = U_DEVICE_TYPE_GNSS,
            .deviceCfg = {
                .cfgGnss = {
                    .moduleType = U_GNSS_MODULE_TYPE_M10,
                    .pinEnablePower = U_CFG_APP_PIN_GNSS_ENABLE_POWER,
                    .pinDataReady = -1,
                },
            },
            .transportType = U_DEVICE_TRANSPORT_TYPE_UART,
            .transportCfg = {
                .cfgUart = {
                    .uart = UART_NUM,
                    .baudRate = BAUD_RATE,
                    .pinTxd = TX_PIN,
                    .pinRxd = RX_PIN,
                    .pinCts = CTS_PIN,
                    .pinRts = RTS_PIN,
                    .pPrefix = NULL
                },
            },
        };
        
```
 
 ### Start GPS stream function 

 ```c++
 //start the stream of data
bool MAXM10S::startGpsStream() {
    int32_t returnCode;
    uPortLog("Starting position stream.\n");
    returnCode = uGnssPosGetStreamedStart(devHandle,1000,callback);
    if (returnCode == 0) {
        printf("Waiting for position...\n");
        return true;
    } else {
        printf("Unable to start position stream!\n");
        return false;
    }
}
```

michaelboeding, 2023-09-11:
In the meantime just to add some more information to this I am just calling uGnssPosGetStart() at set intervals instead of streaming the locations using the uGnssPosGetStreamedStart since that doesn't seem to return any data ever. It may also be important to note that in order to get results I always have to call uGnssPosGetStop() before I call uGnssPosGetStart().

```c++
bool MAXM10S::getCoordinates() {
    int32_t returnCode;
    //stop any previous 
    uGnssPosGetStop(devHandle);
    returnCode = uGnssPosGetStart(devHandle,callback);
    //check the return call 
    if (returnCode == 0) {
        printf("Waiting for position...\n");
        return true;
    } else {
        printf("Error code is %d\n", returnCode);
        printf("Unable to get posostion\n");
        return false;
    }
}

```

RobMeades, 2023-09-11:
Thanks for the additional information: our test code likely always calls `uGnssPostGetStop()` after `uGnssPosGetStart()`, so if there were an issue with this we wouldn't spot it.  I will modify our tests to call `uGnssPosGetStart()` twice and see if that shows the same problem.

On the original issue, on the face of it your code seems little different to that of [our example](https://github.com/u-blox/ubxlib/blob/20a01be538d8ed8a552243870b57111c9e6548d9/example/gnss/pos_main.c#L249): `uPortInit()`, `uDeviceInit()`, `uDeviceOpen()` then `uGnssPosGetStreamedStart()`.  You also call `uGnssCfgSetProtocolOut()` but since all you're doing is enabling UBX and disabling NMEA and RTCM that shouldn't matter.

Would you be able to add a call to `uGnssSetUbxMessagePrint(devHandle, true)` somewhere after you have the device handle populated and post what that prints out here, in case there is anything of interest?

RobMeades, 2023-09-11:
FYI, it was indeed always necessary to call `uGnssPosGetStop()` after `uGnssPosGetStart()`, thanks for pointing this out, it is now fixed, so you can just call `uGnssPosGetStart()` whenever you like and any current attempt will be replaced by the new one.  Just putting that change through testing before it gets here.

Anyway, this is all beside the point, `uGnssPosGetStreamedXxx()` _should_ be the right thing for you to use, let's find out why that's not working for you.

michaelboeding, 2023-09-11:
Hey sorry for the delay see below. 

```bash

U_GNSS: initialising with ENABLE_POWER pin not connected, transport type UART.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 04 04 00 00 00 09 00 17 76.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
Opened device with return code 0.
Setting ubx...
Finished setting ubx...
U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 01 00 74 10 01 23 cc.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 02 00 74 10 00 23 d0.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 04 00 74 10 00 25 da.
U_GNSS: decoded UBX response 0x05 0x00: 06 8a [body 2 byte(s)].
Starting position stream.
Setup successful!
Waiting for position...
Task is running
U_GNSS: sent command b5 62 01 07 00 00 08 19.
U_GNSS: decoded UBX response 0x01 0x07: f8 f1 0b 0a e7 07 09 0b 16 30 39 37 1c 00 00 00 5d df fe ff 03 01 aa 08 17 f5 f0 c8 3d b0 25 17 4e df 02 00 7f 56 03 00 80 0d 00 00 b1 0e 00 00 fe ff ff ff 10 00 00 00 10 00 00 00 10 00 00 00 00 00 00 00 b7 00 00 00 a6 3b da 00 82 01 00 00 5c 40 56 2f 00 00 00 00 00 00 00 00 [body 92 byte(s)].
Got gps data in callback function fired!!!!!

```

RobMeades, 2023-09-11:
Thanks: just for context, this is with the call to `uGnssPosGetStreamedStart()`?  What I can see above looks like `uGnssPosGetStart()` is being called, rather than the streamed one, since it is only polling once, it isn't setting up  the rates etc. in the GNSS device.

RobMeades, 2023-09-11:
FYI, below is a log from the setup phase of running the [pos_main.c](https://github.com/u-blox/ubxlib/blob/20a01be538d8ed8a552243870b57111c9e6548d9/example/gnss/pos_main.c#L263) example on an M10S connected over I2C.

You can see in the the last line that it sends a `UBX-CFG-VALSET` message (0x068a) with key ID 0x20910006 (the bytes are reversed) followed by the value 0x01, and if you search for 0x20910006 in the M10 interface manual you will see that it is the key ID of `CFG-MSGOUT-UBX_NAV_PVT_I2C`  which sets the output rate of the `UBX-NAV-PVT` message on the I2C port, so the output rate of `UBX-NAV-PVT` is every second.

What might be happening in your case, since you're connecting over UART, is somehow we're not getting the key ID correct for the UART port (the message rate is interface-specific).

```
2023-09-11 18:39:51,456 U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 ff ff 21 00 b8 cc.
2023-09-11 18:39:51,480 U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 03 00 21 20 01 01 00 21 30 e8 03 02 00 21 30 01 00 [body 21 byte(s)].
2023-09-11 18:39:51,586 U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 06 00 91 20 50 eb.
2023-09-11 18:39:51,606 U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 06 00 91 20 00 [body 9 byte(s)].
2023-09-11 18:39:51,717 U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 06 00 91 20 01 55 5c.
2023-09-11 18:39:51,728 U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
```

RobMeades, 2023-09-11:
Aha, yes, that's it: our regression testing of the streamed position code is done over I2C (on M8, M9 and M10) and SPI (on M8 and M9), but we don't test UART on M10 and, [checking the code](https://github.com/u-blox/ubxlib/blob/20a01be538d8ed8a552243870b57111c9e6548d9/gnss/src/u_gnss_pos.c#L592), I don't modify the key ID based on the port.

I will fix and post a preview branch of the fix here for you to try.

RobMeades, 2023-09-11:
I've pushed a preview version of the fix to a branch here: https://github.com/u-blox/ubxlib/tree/preview_gnss_m10_uart_rmea.

Can you try it and see if you now get streamed position after calling `uGnssPosGetStreamedStart()`?  If so, I will merge the change to our internal code tree and push it out here hopefully some time tomorrow (Tuesday) and then delete the preview branch.

If not, please send a log of your application code calling `uGnssPosGetStreamedStart()` for position on this codebase and I will investigate further.

michaelboeding, 2023-09-12:
Let me try that quickly. And just for context I think I sent you the one with just the call to uGnssPosGetStart(). Heres a log with the call to uGnssPosGetStreamedXxx(). 

```C++

U_GNSS: initialising with ENABLE_POWER pin not connected, transport type UART.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 04 04 00 00 00 09 00 17 76.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
Opened device with return code 0.
Setting ubx...
Finished setting ubx...
U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 01 00 74 10 01 23 cc.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 02 00 74 10 00 23 d0.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 04 00 74 10 00 25 da.
U_GNSS: decoded UBX response 0x05 0x00: 06 8a [body 2 byte(s)].
Starting position stream.
Setup successful!
Starting position stream.
U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 ff ff 21 00 b8 cc.
U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 03 00 21 20 01 01 00 21 30 e8 03 02 00 21 30 01 00 [body 21 byte(s)].
Failed to get the message rate, error code: -5
Navigation Rate: 1000 ms
Measurement Period: 1000 ms
Navigation Count: 1
U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 ff ff 21 00 b8 cc.
U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 03 00 21 20 01 01 00 21 30 e8 03 02 00 21 30 01 00 [body 21 byte(s)].
U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 06 00 91 20 50 eb.
U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 06 00 91 20 00 [body 9 byte(s)].
U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 06 00 91 20 01 55 5c.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
Waiting for position...
Task is running
Waiting for position...
U_GNSS: sent command b5 62 01 07 00 00 08 19.
Got gps data in callback function fired!!!!!
Error code: -9

```

RobMeades, 2023-09-12:
Ah, great, that confirms it:

`U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 06 00 91 20 50 eb` << read I2C message rate for `UBX-NAV-PVT` (0x20910006) on I2C.
`U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 06 00 91 20 00 [body 9 byte(s)].` << value read is 0x00.
`U_GNSS: sent command b5 62 06 8a 09 00 01 03 00 00 06 00 91 20 01 55 5c.` << set I2C message rate for `UBX-NAV-PVT` to 0x01 but, stupidly, on I2C and not UART (i.e. still key ID 0x20910006)
`U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].` << Ack for that, but it was a bit pointless.

michaelboeding, 2023-09-12:
Yep it worked!!! Awesome thank you. So second question that you might have answered already but for the single coordinate call will I still always have to make a call to  uGnssPosGetStop(devHandle) before requesting new coordinates or was that a bug also? Just want to make sure I update my code accordingly

RobMeades, 2023-09-12:
Phew.  I will push the fix to `master` here tomorrow, update this issue when I have done so and delete the preview branch later in the week.  To your question, yes, the preview branch has (and `master here sometime tomorrow will also have) the fix so that you should be able to call `uGnssPosGetStart()` repeatedly, any new call simply replacing the last,

michaelboeding, 2023-09-12:
Awesome thank you very much! Worth noting I also call the uGnssPosGetStreamedStop(devHandle); before calling uGnssPosGetStreamedStart(). Not sure if that's needed but I had it in there based on the work around I did for the uGnssPosGetStart().

RobMeades, 2023-09-12:
I will check that out also, clarify/fix if required and update this issue with the outcome.

RobMeades, 2023-09-12:
On the last point, I have now modified the code so that `uGnssPosGetStreamedStart()` can be called multiple times without an intervening `uGnssPosGetStreamedStop()`.  [That change](https://github.com/u-blox/ubxlib/commit/21c2360be3db4a079c10c75c27c0fda302d714b9) and the changes for the original issue of [not being able to get streamed position when connecting to an M10 device over UART](https://github.com/u-blox/ubxlib/commit/132f338ef49285b6c40d070b38fe558acb6d3030) and [removing the need to call `uGnssPosGetStop()` before calling `uGnssPosGetStart()` once more](https://github.com/u-blox/ubxlib/commit/795285c026c10f72b3120d736a858efdd13a8295) are now pushed to `master` here,

I will delete the [preview branch](https://github.com/u-blox/ubxlib/tree/preview_gnss_m10_uart_rmea) later this week.

Closing this issue now: please feel free to re-open it or start another if you have any problems.

closed 2023-09-12 by RobMeades

michaelboeding, 2023-09-12:
Thanks I will move to the master.