u-blox/ubxlib #134: Issue with uMqttClientGetUnread and uMqttClientMessageRead

michaelboeding, 2023-09-08:
### Callback code 
```c++

// Callback for unread message indications.
static void messageIndicationCallback(int32_t numUnread, void *pParam){
    //cast the parameter to an instance of the SARAR5 class
    SARAR5* sarar5 = (SARAR5*) pParam;
    char topic[U_SECURITY_SERIAL_NUMBER_MAX_LENGTH_BYTES];
    const char message[] = "{ \"state\": { \"reported\": { \"isOnline\": \"true\" } } }";
    char buffer[64];
    size_t bufferSize;
    // It is important to keep stack usage in this callback
    // to a minimum.  If you want to do more than set a flag
    // (e.g. you want to call into another ubxlib API) then send
    // an event to one of your own tasks, where you have allocated
    // sufficient stack, and do those things there.
    uPortLog("The broker says there are %d message(s) unread.\n", numUnread);
    // Read the new message from the broker
        bufferSize = sizeof(buffer);
        // If you were using MQTT-SN, you would call
        // uMqttClientSnMessageRead() instead and, rather
        // than passing it the buffer "topic", you
        // would pass it a pointer to a variable of
        // type uMqttSnTopicName_t
        printf("Reading message from broker...\n");
        while (uMqttClientGetUnread(sarar5->pContext) > 0) {
            if (uMqttClientMessageRead(sarar5->pContext, topic,
                                        sizeof(topic),
                                        buffer, &bufferSize,
                                        0) == 0) {
                printf("New message in topic \"%s\" is %d"
                            " character(s): \"%.*s\".\n", topic,
                            bufferSize, bufferSize, buffer);
            }else{
                printf("Failed to read message from broker!\n");
            } 
        }
        printf("Finished reading message from broker.\n");

    //*pMessagesAvailable = true;
  //  sarar5->handleNewMessage();
}

```

Im running into an issue where I get the callback for a new message but then the uMqttClientGetUnread and uMqttClientMessageRead seem to block and never execute. 

### Log
[image]

michaelboeding, 2023-09-09:
Seems like when I don't call the uMqttClientGetUnread it doesn't seem to block and all of the notifications of a new message get sent to the callback but I have no way of actually getting the messages since it blocks and never returns when I call it.

michaelboeding, 2023-09-09:
I was able to fix this by running the messages in a separate task in the esp-idf, if anyone else has this issue.

closed 2023-09-09 by michaelboeding

RobMeades, 2023-09-11:
Apologies for the late response on this: for thread-safety, most `ubxlib` APIs have a mutex within them and so it is possible to "get stuck" if you're calling a `ubxlib` API from a callback that, itself, came out of `ubxlib`.  The [intention in this particular case](https://github.com/u-blox/ubxlib/blob/20a01be538d8ed8a552243870b57111c9e6548d9/common/mqtt_client/api/u_mqtt_client.h#L301) was that calling [uMqttClientGetUnread()](https://github.com/u-blox/ubxlib/blob/20a01be538d8ed8a552243870b57111c9e6548d9/common/mqtt_client/api/u_mqtt_client.h#L327) should work, though not [uMqttClientMessageRead()](https://github.com/u-blox/ubxlib/blob/20a01be538d8ed8a552243870b57111c9e6548d9/common/mqtt_client/api/u_mqtt_client.h#L464), but I notice that [our test doesn't call uMqttClientGetUnread()](https://github.com/u-blox/ubxlib/blob/20a01be538d8ed8a552243870b57111c9e6548d9/common/mqtt_client/test/u_mqtt_client_test.c#L234), so I will take a look at what happens if it does and see if I can fix that, or if not give better advice.

RobMeades, 2023-09-14:
FYI, I've updated `master` here (https://github.com/u-blox/ubxlib/commit/bd8a69efdecc6655fd2f6fcaebc9d63dac1f7571) so that it _is_ possible to call into the cellular (or MQTT Client when using cellular) API from the MQTT message indication and disconnect callbacks.

To be completely clear, the approach you have taken, i.e. to spin up a separate task for the message reads, is STILL VERY MUCH  THE RIGHT THING to do and the comment above [uMqttClientSetMessageCallback()](https://github.com/u-blox/ubxlib/blob/bd8a69efdecc6655fd2f6fcaebc9d63dac1f7571/common/mqtt_client/api/u_mqtt_client.h#L294) has been updated to make this "IMPORTANT" rather than just a note; sitting in that callback will block the single internal callback queue that all of `ubxlib` uses, which could have unintended consequence, e.g. when a lot of MQTT messages arrive at once on return from a coverage outage.

But the mutex protection we were applying around the callback from cellular was bit too heavy-handed and it didn't match what the Wi-Fi code does in the same circumstances; now the callback from cellular is just as thread-safe but it doesn't prevent the application from getting a bit frisky in the callback if it really wants to.