u-blox/ubxlib #138: Reducing UBXLIB_FEATURES fails to build

arnoutdekimo, 2023-09-22:
Hi,

I am evaluating the STM32 port of ubxlib.
To reduce code footprint, I attempted to remove short_range from the feature list, as explained here:

https://github.com/u-blox/ubxlib/tree/master/port#UBXLIB_FEATURES
The effect is though, that the project no longer compiles. 
In particular, #include "u_short_range_test_selector.h" is not found in e.g. example/sockets/main.c.

I am assuming this is a bug (define placement), since ideally one would just compile the required features without having to modify the lib / creating stub header files I think.

Thank for your feedback,
Arnout

RobMeades, 2023-09-22:
Hi there: we do test that all combinations compile but only for the core library code, i.e. not bringing in the examples; since there are technology-specific examples it is possible that building a short-range one (the sockets example is, by default, run on WiFi I think) without short-range will fail.

I'll take a look and see if we can workaround that without cluttering the example code too much.

arnoutdekimo, 2023-09-22:
(Just just to come back to the socket example - it seems it's possible to select either wifi, OR cellular. So I guess it does make sense that it would compile without both features being enabled)

RobMeades, 2023-09-22:
Just out of interest, which features are you intending to leave in?

arnoutdekimo, 2023-09-22:
cellular and gnss. (not wifi/BT aka short_range)

RobMeades, 2024-01-04:
Hi Arnout: I've just been trying to address this issue but I can't seem to make it happen, wanted to double-check the exact scenario.  I did this:

- CD to `port\platform\stm32cube\mcu\stm32f4\runner`
- set environment variables as follows:
  - the features, asking for cellular and GNSS, leaving out short range: `set UBXLIB_FEATURES=cell gnss`,
  - the environment variable `U_FLAGS` just to set the module types I want to use in the examples/tests: `set U_FLAGS=-DU_CFG_TEST_GNSS_MODULE_TYPE=U_GNSS_MODULE_TYPE_M10 -DU_CFG_TEST_CELL_MODULE_TYPE=U_CELL_MODULE_TYPE_LARA_R6`,
  - the path to the STM32Cube FW: `set STM32CUBE_FW_PATH=C:\Users\rmea\.ubxlibpkg\stm32cubef4-v1.26.1`,
  - the path to `unity`, required for the examples/tests: `set UNITY_PATH=c:\projects\ubxlib_priv\Unity` (a slightly strange place to put it, I just happened to have a copy lying around there),
  - the path to ARM GCC: `set ARM_GCC_TOOLCHAIN_PATH=C:\Users\rmea\.ubxlibpkg\arm_embedded_gcc-10-2020-q4-major`.
 - executed `make`.

Result is below, compiling the cellular, GNSS and common code, with the tests/examples, to an `.elf` file, leaving out the short range/BLE/Wi-Fi stuff.  What did you do differently?

```
UBXLIB_FEATURES is "cell gnss"
STM32CUBE_FW_PATH:   C:\Users\rmea\.ubxlibpkg\stm32cubef4-v1.26.1
UNITY_PATH:          c:\projects\ubxlib_priv\Unity
CMSIS_DEVICE_HEADER: stm32f437xx.h
CC C:/projects/ubxlib_priv/common/network/src/u_network.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_shared.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_private_ble_extmod_stub.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_private_cell_stub.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_private_gnss_stub.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_private_wifi_stub.c
CC C:/projects/ubxlib_priv/common/device/src/u_device.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_serial.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_shared.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_private.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_private_cell_stub.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_private_gnss_stub.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_private_short_range_stub.c
CC C:/projects/ubxlib_priv/port/u_port_heap.c
CC C:/projects/ubxlib_priv/port/u_port_timezone.c
CC C:/projects/ubxlib_priv/port/u_port_resource.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_private_ble_extmod_link.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_private_wifi_link.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_private_short_range_link.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_private_cell.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_private_cell.c
CC C:/projects/ubxlib_priv/common/network/src/u_network_private_gnss.c
CC C:/projects/ubxlib_priv/common/device/src/u_device_private_gnss.c
CC C:/projects/ubxlib_priv/common/geofence/src/dummy/u_geofence_geodesic.c
CC C:/projects/ubxlib_priv/gnss/src/lib_mga/u_lib_mga.c
CC C:/projects/ubxlib_priv/port/platform/common/event_queue/u_port_event_queue.c
CC C:/projects/ubxlib_priv/port/platform/common/mutex_debug/u_mutex_debug.c
CC C:/projects/ubxlib_priv/port/platform/common/log_ram/u_log_ram.c
CC C:/projects/ubxlib_priv/port/platform/common/log_ram/u_log_ram_string.c
CC C:/projects/ubxlib_priv/common/at_client/src/u_at_client.c
CC C:/projects/ubxlib_priv/common/at_client/src/u_at_client_stub_short_range.c
CC C:/projects/ubxlib_priv/common/assert/src/u_assert.c
CC C:/projects/ubxlib_priv/common/location/src/u_location.c
CC C:/projects/ubxlib_priv/common/location/src/u_location_private_cloud_locate.c
CC C:/projects/ubxlib_priv/common/location/src/u_location_shared.c
CC C:/projects/ubxlib_priv/common/location/src/u_location_stub_cell.c
CC C:/projects/ubxlib_priv/common/location/src/u_location_stub_gnss.c
CC C:/projects/ubxlib_priv/common/location/src/u_location_stub_wifi.c
CC C:/projects/ubxlib_priv/common/mqtt_client/src/u_mqtt_client.c
CC C:/projects/ubxlib_priv/common/mqtt_client/src/u_mqtt_client_stub_cell.c
CC C:/projects/ubxlib_priv/common/mqtt_client/src/u_mqtt_client_stub_wifi.c
CC C:/projects/ubxlib_priv/common/http_client/src/u_http_client.c
CC C:/projects/ubxlib_priv/common/http_client/src/u_http_client_stub_cell.c
CC C:/projects/ubxlib_priv/common/http_client/src/u_http_client_stub_wifi.c
CC C:/projects/ubxlib_priv/common/security/src/u_security.c
CC C:/projects/ubxlib_priv/common/security/src/u_security_credential.c
CC C:/projects/ubxlib_priv/common/security/src/u_security_credential_stub_cell.c
CC C:/projects/ubxlib_priv/common/security/src/u_security_credential_stub_short_range.c
CC C:/projects/ubxlib_priv/common/security/src/u_security_stub_cell.c
CC C:/projects/ubxlib_priv/common/security/src/u_security_stub_short_range.c
CC C:/projects/ubxlib_priv/common/security/src/u_security_tls.c
CC C:/projects/ubxlib_priv/common/security/src/u_security_tls_stub_cell.c
[... 126 lines cut ...]
CC C:/projects/ubxlib_priv/cell/test//u_cell_mux_test.c
CC C:/projects/ubxlib_priv/cell/test//u_cell_net_test.c
CC C:/projects/ubxlib_priv/cell/test//u_cell_pwr_test.c
CC C:/projects/ubxlib_priv/cell/test//u_cell_sec_tls_test.c
CC C:/projects/ubxlib_priv/cell/test//u_cell_sock_test.c
CC C:/projects/ubxlib_priv/cell/test//u_cell_test.c
CC C:/projects/ubxlib_priv/cell/test//u_cell_test_preamble.c
[... cut ...]
CC C:/projects/ubxlib_priv/gnss/test//u_gnss_pos_test.c
CC C:/projects/ubxlib_priv/gnss/test//u_gnss_private_test.c
CC C:/projects/ubxlib_priv/gnss/test//u_gnss_pwr_test.c
CC C:/projects/ubxlib_priv/gnss/test//u_gnss_test.c
CC C:/projects/ubxlib_priv/gnss/test//u_gnss_test_private.c
CC C:/projects/ubxlib_priv/gnss/test//u_gnss_util_test.c
CC C:/projects/ubxlib_priv/port/platform/common/heap_check/u_heap_check.c
CC C:/projects/ubxlib_priv/port/platform/stm32cube/app/u_main.c
Linking output/runner.elf
```

arnoutdekimo, 2024-01-08:
Hmm, to be honest I don't recall anymore. 
Since you clearly tested it and confirmed it works, I guess the issue must have been something on my side then. Feel free to close it then.

RobMeades, 2024-01-08:
OK, will do, do let us know if you find anything like this again of course.

closed 2024-01-08 by RobMeades