u-blox/ubxlib #140: Stack Overflow in gnssMsgRx task

michaelboeding, 2023-09-26:
Im running into an issue where I have a stack overflow occurring that is causing my system to restart (ran it over night with a restart counter to diagnose any runtime problems and came across this). It seems like this is located here : https://github.com/search?q=repo%3Au-blox%2Fubxlib%20gnssMsgRx&type=code . This definitely may be something I'm doing wrong in my application code but seems to be a ubxlib task so wanted to check. 

```bash 

***ERROR*** A stack overflow in task gnssMsgRx has been detected.

Backtrace: 0x40081d82:0x3ffd14f0 0x40092679:0x3ffd1510 0x40092ce2:0x3ffd1530 0x40093ca3:0x3ffd15b0 0x40092774:0x3ffd15d0 0x40092726:0x00000000 |<-CORRUPTED
0x40081d82: panic_abort at /Users/mb/esp-idf/components/esp_system/panic.c:408

0x40092679: esp_system_abort at /Users/mb/esp/esp-idf/components/esp_system/esp_system.c:137

0x40092ce2: vApplicationStackOverflowHook at /Users/mb/esp/esp-idf/components/freertos/port/xtensa/port.c:407

0x40093ca3: vTaskSwitchContext at /Users/mb/esp/esp-idf/components/freertos/tasks.c:3505

0x40092774: _frxt_dispatch at /Users/mb/esp/esp-idf/components/freertos/port/xtensa/portasm.S:436

0x40092726: _frxt_int_exit at /Users/mb/esp/esp-idf/components/freertos/port/xtensa/portasm.S:231

```

This is all im doing in the callback from the position stream

```c++

// Callback for position reception.
static void callback(uDeviceHandle_t gnssHandle,
                     int32_t errorCode,
                     int32_t latitudeX1e7,
                     int32_t longitudeX1e7,
                     int32_t altitudeMillimetres,
                     int32_t radiusMillimetres,
                     int32_t speedMillimetresPerSecond,
                     int32_t svs,
                     int64_t timeUtc)
{
   // printf("Got gps data in callback function fired!!!!!\n");
    char prefix[2] = {0};
    int32_t whole[2] = {0};
    int32_t fraction[2] = {0};

    // Not using these, just keep the compiler happy
    (void) gnssHandle;
    (void) altitudeMillimetres;
    (void) radiusMillimetres;
    (void) speedMillimetresPerSecond;
    (void) svs;
    (void) timeUtc;

    if (errorCode == 0) {
        prefix[0] = latLongToBits(longitudeX1e7, &(whole[0]), &(fraction[0]));
        prefix[1] = latLongToBits(latitudeX1e7, &(whole[1]), &(fraction[1]));
        printf("I am here: https://maps.google.com/?q=%c%d.%07d,%c%d.%07d\n",
                 prefix[1], whole[1], fraction[1], prefix[0], whole[0], fraction[0]);

        // we want to print all of the other values as well
       // printf("Error code: %d\n", errorCode);
        //make the altitude be in meters

        /*
        printf("Altitude: %d\n", altitudeMillimetres / 1000);
        printf("Radius: %d\n", radiusMillimetres / 1000);
        printf("Speed: %d\n", speedMillimetresPerSecond / 1000);
        printf("SVS: %d\n", svs);
        printf("Time: %lld\n", timeUtc);
        */

        
        gPositionCount++;
    }else{
        printf("Error code: %d\n", errorCode);
    }
    //send the data to the callback function in the class
    if (globalClassInstance) {
        globalClassInstance->memberCallback(gnssHandle, errorCode, latitudeX1e7, longitudeX1e7, altitudeMillimetres, radiusMillimetres, speedMillimetresPerSecond, svs, timeUtc);
    }
}

```

RobMeades, 2023-09-26:
Always difficult, stack stuff.  For callback tasks we set a stack size that survives all of our tests and ratchet it up when they don't; we do thrash GNSS position reception pretty heavily so there shouldn't be a basic issue there.

In the case above, I think it would mostly depend what `memberCallback()` does, since it will be running in the same task context, any stack that `memberCallback()` consumes will fall within that of the callback task `gnssMsgRx`. 

If at all possible `memberCallback()` should simply transfer the data to another RTOS entity for processing, which usually means it would push the data onto some form of RTOS queue and come straight back.

michaelboeding, 2023-09-26:
Heres the callback - basically just creates a struct and forwards the message to another object. Which I guess actually has its own task running but not technically for that. So it calls that method in another class which is maybe causing the issue but its not in the callback?

```c++

 //use this callback to notify the class that data has been received and send it via the callbacks 
void MAXM10S::memberCallback(uDeviceHandle_t gnssHandle,
                        int32_t errorCode,
                        int32_t latitudeX1e7,
                        int32_t longitudeX1e7,
                        int32_t altitudeMillimetres,
                        int32_t radiusMillimetres,
                        int32_t speedMillimetresPerSecond,
                        int32_t svs,
                        int64_t timeUtc){
                            //printf("Got gps data in callback function!\n");
    //convert the data into the format we want before we send it 
    GnssPositionData positionData = {
        .latitude = latitudeX1e7,  // Keeping the original scale for latitude
        .longitude = longitudeX1e7, // Keeping the original scale for longitude
        .altitudeMeters = altitudeMillimetres / 1000,  // Convert from mm to meters
        .radiusMeters = radiusMillimetres / 1000, // Convert from mm to meters
        .speedMetersPerSecond = speedMillimetresPerSecond / 1000, // Convert from mm/s to m/s
        .satellitesCount = svs,
        .timeUtc = timeUtc
    };
    //send the data to the callback function
    if (this->callbacks != nullptr) {
        this->callbacks->locationChanged(this, &positionData);
    }
}

```

RobMeades, 2023-09-26:
I think the issue is that all of the C++ object stuff is still in the same task, just things calling things but in a C++ey way, as far as I can tell at least: there is no RTOS structure (basically a queue) interposed, that I can see.  Without an RTOS structure there can be no "taskiness", if you see what I mean, it's all in the one task, functions calling functions, using that one task's stack.

`ubxlib` has its own [u_port_event_queue.h](https://github.com/u-blox/ubxlib/blob/bd8a69efdecc6655fd2f6fcaebc9d63dac1f7571/port/api/u_port_event_queue.h#L29) pattern, which allows the creation of a task that can be fed by a queue, and we use this all over to move activity from one task context to another, taking advantage of RTOS asynchronous behaviours; the `gnssMsgRx` task is at the end of just such a queue.

There is an ESP-IDF tutorial on how to use their \[FreeRTOS\] queues here: https://esp32tutorials.com/esp32-esp-idf-freertos-queue-tutorial/.  When we run on ESP-IDF our `u_port_event_queue.h` pattern uses exactly these queues.

michaelboeding, 2023-09-26:
Yes - I think this may be my lack of experience with tasks. I will look into implementing this also. Thanks

closed 2023-09-26 by michaelboeding

philwareublox, 2023-09-27:
Hi Michael, 
We have a ubxlib cellular tracker application for our XPLR-IoT-1 development kit which uses a Task type design for each application "task". Registration Task, MQTT broker connection Task, Location Task, Signal Quality Monitoring Task...
https://github.com/u-blox/ubxlib_cellular_applications_xplr_iot/tree/development
Hopefully the structure of each "appTask" is simple enough to understand how to use ubxlib Queues and Tasks.