u-blox/ubxlib #141: Firmware using BLE: HELP !!!!

dringjessegomes, 2023-09-27:
Hey guys!!!!

I have the 672-EVK-NORA-B106-ND and already downloaded the files available at https://github.com/u-blox/ubxlib, but I am unable to start building the firmware that collects data from the following sensors:
  - Flammable gas and smoke (MQ2),
  - Carbon Monoxide (CH4),
  - Ammonia gas (NH3)
  - Ethanol gas C2H6O.

I need a step by step!!!!!

Who can please help me?????

plerup, 2023-09-27:
Hi,
We need some more information, like what are your build and host platforms and what specific problems do you have.

dringjessegomes, 2023-09-27:
Hello,

I am using VS Code, nRF Connect for Desktop 4.2.0, Toolchains/SDK v2.3.0. I downloadade the files from https://github.com/u-blox/ubxlib, but I am not able to put together the appropriate scripts to build a firmware that manages the collection of data from the sensors below and then sends the collected data via Bluetooth to an Android app.

Managed sensors to be insert on the board NORA B106.

- Flammable gas and smoke (MQ2),
- Carbon Monoxide (CH4),
- Ammonia gas (NH3)
- Ethanol gas C2H6O.

I need step-by-step guidance to start this firmware.

perssonmagnus, 2023-09-28:
Hi,

The main functionality of ubxlib is to connect a host with u-blox modules running AT commands (Cellular or Bluetooth/Wi-Fi modules) or NMEA/UBX protocol (positioning modules).

For reading sensor values you are recommended to use the SDK, Zephyr/nRF Connect SDK in your case, directly to read the connected sensors and send them over Bluetooth.

There are plenty of examples in the nRF Connect SDK for similar applications. 

Best Regards,
Magnus

closed 2023-10-29 by RobMeades