u-blox/ubxlib #142: Extra GPS Data

michaelboeding, 2023-10-05:
Is there a way to get extra gps data while using the uGnssPosGetStart, such as the information listed in this structure?

```c++
typedef struct {
    uint32_t iTOW;                       /**< GPS time of week of the
                                              navigation epoch
                                              in milliseconds. */
    uint16_t year;                       /**< year (UTC); to obtain this and
                                              the other time-related fields
                                              in this structure as a Unix-based
                                              UTC timestamp, see
                                              uGnssDecUbxNavPvtGetTimeUtc(). */
    uint8_t month;                       /**< month, range 1 to 12 (UTC). */
    uint8_t day;                         /**< day of month, range 1 to 31 (UTC). */
    uint8_t hour;                        /**< hour of day, range 0 to 23 (UTC). */
    uint8_t min;                         /**< minute of hour, range 0 to 59 (UTC). */
    uint8_t sec;                         /**< seconds of minute, range 0 to 60 (UTC). */
    uint8_t valid;                       /**< validity flags, see
                                              #uGnssDecUbxNavPvtValid_t. */
    uint32_t tAcc;                       /**< time accuracy estimate in
                                              nanoseconds. */
    int32_t nano;                        /**< fractional seconds part of UTC
                                              time in nanoseconds. */
    uGnssDecUbxNavPvtFixType_t fixType;  /**< the fix type achieved. */
    uint8_t flags;                       /**< see #uGnssDecUbxNavPvtFlags_t. */
    uint8_t flags2;                      /**< see #uGnssDecUbxNavPvtFlags2_t. */
    uint8_t numSV;                       /**< the number of satellites used. */
    int32_t lon;                         /**< longitude in degrees times 1e7. */
    int32_t lat;                         /**< latitude in degrees times 1e7. */
    int32_t height;                      /**< height above ellipsoid in mm. */
    int32_t hMSL;                        /**< height above mean sea level in mm. */
    uint32_t hAcc;                       /**< horizontal accuracy estimate in mm. */
    uint32_t vAcc;                       /**< vertical accuracy estimate in mm. */
    int32_t velN;                        /**< NED north velocity in mm/second. */
    int32_t velE;                        /**< NED east velocity in mm/second. */
    int32_t velD;                        /**< NED down velocity in mm/second. */
    int32_t gSpeed;                      /**< 2D ground speed in mm/second. */
    int32_t headMot;                     /**< 2D heading of motion in degrees times 1e5. */
    uint32_t sAcc;                       /**< speed accuracy estimate in mm/second. */
    uint32_t headAcc;                    /**< heading accuracy estimate (motion and
                                              vehicle) in degrees times 1e5. */
    uint16_t pDOP;                       /**< position DOP times 100. */
    uint16_t flags3;                     /**< see #uGnssDecUbxNavPvtFlags3_t. */
    int32_t headVeh;                     /**< if the #U_GNSS_DEC_UBX_NAV_PVT_FLAGS_HEAD_VEH_VALID
                                              bit of the flags field is set
                                              then this is the 2D vehicle
                                              heading in degrees times 1e5,
                                              else it is set to the same
                                              value as headMot. */
    int16_t magDec;                      /**< if the #U_GNSS_DEC_UBX_NAV_PVT_VALID_MAG
                                              bit of the valid field is set then this
                                              is the magnetic declination in degrees
                                              times 100; only supported on ADR 4.10
                                              and later. */
    uint16_t magAcc;                     /**< if the #U_GNSS_DEC_UBX_NAV_PVT_VALID_MAG
                                              bit of the valid field is set then this
                                              is the accuracy of the magnetic
                                              declination in degrees times 100; only
                                              supported on ADR 4.10 and later. */
} uGnssDecUbxNavPvt_t;

```

RobMeades, 2023-10-05:
Hi Michael: as more people are asking about this we should probably extend the `uGnssPos` API but until we've done that the simplest/quickest way for you to get at the level of detail you would like is to use the `uGnssMsg` and `uGnssDec` APIs instead of the `uGnssPos` API.  

Basically you would follow the [msg_main](https://github.com/u-blox/ubxlib/blob/c3c8c03d86fe1c90e2785f4e11a108d1c0e75a5b/example/gnss/msg_main.c#L319) example and call to `pUGnssDecAlloc()` in the callback.  The code might look something like this (I've NOT compiled this code, just a sketch to give you an idea):

```
char buffer[512];

//  Asynchronous message callback.
void callback(uDeviceHandle_t devHandle,
              uGnssMessageId_t *pMessageId,
              int32_t errorCodeOrLength,
              void *pCallbackParam)
{
    // Don't need the message ID since we asked specifically for UBX-NAV-PVT
    (void) pMessageId;
    (void) pCallbackParam;

    if (errorCodeOrLength > 0) {
        if (errorCodeOrLength > sizeof(buffer)) {
            errorCodeOrLength = sizeof(buffer);
        }
        errorCodeOrLength = uGnssMsgReceiveCallbackRead(devHandle,
                                                        buffer,
                                                        errorCodeOrLength);
        if (errorCodeOrLength > 0) {
            uGnssDec_t *pDec = pUGnssDecAlloc(buffer, errorCodeOrLength);
            if ((pDec != NULL) && (pDec->errorCode == 0)) {
                // Not checking pDec->pId since we only asked for UBX-NAV-PVT,
                // checking for NULL and a zero error code is sufficient

                // The decoded UBX-NAV-PVT message is now hooked
                // into the memory allocated at pDec->pMsg, so for instance: 
                uGnssDecUbxNavPvt_t *pUbxNavPvt = pDec->pBody.ubxNavPvt;
                print("iTow is %d.\n", pUbxNavPvt->iTOW);

                // The same rules apply here as to other callbacks:
                // do as little as possible, push any processing off
                // onto an RTOS queue, to somewhere that your
                // application controls the stack-size of
            }

            // IMPORTANT: must *always* free the memory that
            // pUGnssDecAlloc() allocated when done (you can
            // do this at the other end of your RTOS queue if
            // you like).
            uGnssDecFree(pDec);
        }
    }
}
```

...then in your main code:

```
int32_t x;
uGnssMessageId_t messageId;

// devHandle is assumed to be set up somewhere above

// Set up the message ID we want to capture: in this case just UBX-NAV-PVT
messageId.type = U_GNSS_PROTOCOL_UBX;
messageId.id.ubx = 0x0107;

// This sets up the message rate for our message ID to be once per "epoch".
// The default epoch of the GNSS chip is one second (you can alter it with
// uGnssCfgSetRate() if you feel a need, but it is usually clearer to leave it
// alone and increase the per-epoch rate for a given message as necessary)
x = uGnssCfgSetMsgRate(devHandle, &messageId, 1);
if (x == 0) {

    // Set up an asynchronous message receiver for this message ID,
    // giving it the callback above.  If you like you can also give it
    // a void * parameter (NULL in this example)
    x= uGnssMsgReceiveStart(devHandle, &messageId, callback, NULL);
    if (x < 0) {
        printf("Unable to start message receiver (%d)!\n", x);
    }

   // callback() will now be called when UBX-NAV-PVT messages arrive,
   // which should be once a second

    ...

    // Call this when you want to stop
    uGnssMsgReceiveStopAll(devHandle);

} else {
    printf("Unable to set UBX-NAV-PVT message rate (%d)!\n", x);
}
```

michaelboeding, 2023-10-07:
Thanks for this Rob, I was able to get it working. There were only a few small things I had to fix. One thing that I don't seem to have working though is the rate setting (just commented it out for now). But I seem to be getting a -4 error.

```c++
returnCode = uGnssCfgSetMsgRate(this->devHandle, &messageId, 1);
```

RobMeades, 2023-10-07:
Ah, yes, sorry, my bad, I should have [read my own comments](https://github.com/u-blox/ubxlib/blob/6821628ede243f9901f44c8d316e3973785bb209/gnss/api/u_gnss_cfg.h#L267): -4 is [U_ERROR_COMMON_NOT_SUPPORTED](https://github.com/u-blox/ubxlib/blob/6821628ede243f9901f44c8d316e3973785bb209/common/error/api/u_error_common.h#L59C5-L59C33); M10 has the new configuration API, so you need to use that.  The message rate setting is interface specific (UART1 v UART2 v SPI v I2C v USB), not sure which interface you are using, but you would choose from one of the following:

`returnCode = U_GNSS_CFG_SET_VAL_RAM(this->devHandle, MSGOUT_UBX_NAV_PVT_I2C_U1, 1);`
`returnCode = U_GNSS_CFG_SET_VAL_RAM(this->devHandle, MSGOUT_UBX_NAV_PVT_SPI_U1, 1);`
`returnCode = U_GNSS_CFG_SET_VAL_RAM(this->devHandle, MSGOUT_UBX_NAV_PVT_UART1_U1, 1);`
`returnCode = U_GNSS_CFG_SET_VAL_RAM(this->devHandle, MSGOUT_UBX_NAV_PVT_UART2_U1, 1);`
`returnCode = U_GNSS_CFG_SET_VAL_RAM(this->devHandle, MSGOUT_UBX_NAV_PVT_USB_U1, 1);`

michaelboeding, 2023-10-07:
Tried that with the correct UART and seem to get the following error, which I don't see in the common errors. 

```c++
returnCode = U_GNSS_CFG_SET_VAL_RAM(this->devHandle, MSGOUT_UBX_NAV_PVT_UART2_U1, 1);
```
```bash 
Unable to set UBX-NAV-PVT message rate (-1025)!
```

RobMeades, 2023-10-07:
Hmm, that's a NACK from the GNSS chip (see https://github.com/u-blox/ubxlib/blob/master/DEBUG_HINTS.md#reading-error-return-codes).  Are you sure you're on UART2?

michaelboeding, 2023-10-07:
Switched to UART1 and it worked but I definitely thought this was configured for UART2.

Heres my config

```c++
 //the gps device 
        uDeviceCfg_t gDeviceCfg = {
            .deviceType = U_DEVICE_TYPE_GNSS,
            .deviceCfg = {
                .cfgGnss = {
                    .moduleType = U_GNSS_MODULE_TYPE_M10,
                    .pinEnablePower = U_CFG_APP_PIN_GNSS_ENABLE_POWER,
                    .pinDataReady = -1,
                },
            },
            .transportType = U_DEVICE_TRANSPORT_TYPE_UART,
            .transportCfg = {
                .cfgUart = {
                    .uart = UART_NUM,
                    .baudRate = BAUD_RATE,
                    .pinTxd = TX_PIN,
                    .pinRxd = RX_PIN,
                    .pinCts = CTS_PIN,
                    .pinRts = RTS_PIN,
                    .pPrefix = NULL
                },
            },
        };
```

Where the defined UART_NUM is 
#define UART_NUM UART_NUM_2

RobMeades, 2023-10-07:
Ah, sorry, it is confusing: in this case UART2 means from the GNSS chip's internal point of view, rather than the UART interface of the MCU.  Various of the GNSS chip's settings, of which the message rate is one, it remembers on a "per interface" basis, and some of the GNSS chips have two UARTs.

Where they have only one, it is UART1, only a very few have a second UART but UART2 is present in the API signaling of course, you'll just get a NACK from the GNSS chip if you try to use it.  So unless you have a GNSS chip that has two UARTs, or if you have and you are using the first of them, the message rate settings should all be the "UART1" settings.

IYSWIM.

michaelboeding, 2023-10-07:
Makes sense! I think I have it all working now. Thanks! It would be nice to have this in the generic API. At least some of the items.

closed 2023-10-07 by michaelboeding

RobMeades, 2023-10-07:
Agreed, I will put it on the list.  Would really not be very difficult to do, since I wrote it just above :-).  Just needs adding.

michaelboeding, 2023-10-07:
As always thanks again for the help!