u-blox/ubxlib #143: Minimum firmware requirements for ublox modules to run with ubxlib

arnoutdekimo, 2023-10-06:
Hi,

I am using the SARA-R4 (R410M) and was testing ubxlib with the exampleHttpClient.
The example is giving errors with:

```
AT+CEDRXS=2,2

ERROR
```

and if I work around that

```
AT+CEREG=4

ERROR
```

The firmware I am using is 

[image]

Perhaps there are minimum requirements for the FW in order to run ubxlib? If so, this would be useful to list as requirement / auto-detect in FW.

Thanks for your feedback.

FYI, full logs in attachment [fulllog.txt](https://github.com/u-blox/ubxlib/files/12830576/fulllog.txt)

RobMeades, 2023-10-06:
Interesting: the SARA-R410M-02B we have in the test system reads FW version `M0.11.01,A.02.17`, which I think is because we had to use an internal FW release that was just about to become an official external at the time.

I will try to find out how the two map but we've not touched the FW version on these modules in two or three years now so it is certainly not a recent version.  I'd _guess_ the last version number in the string is the important one, which would put the one we are testing after your initial one (`A.02.01`) and before your newest one (`A.02.21`).  Frankly I'm surprised there are still modules out there with FW as early as (`A.02.01`), must have been quite a few years in stock.

One of the things on the TODO list is to, at least attempt, allow `U_CELL_MODULE_TYPE_ANY` to be specified and for `ubxlib` to figure out what cellular module is connected (similar for short range and GNSS modules).  As part of that it will likely read the module FW version string, though quite what it will do with that remains to be seen.

But you're right that a list of the FW versions that we test for each module type would be a good idea, I will add that to the TODO list also.

RobMeades, 2023-11-04:
Done in 612397849dbb5667e82fafc62381f4607a140563.  Closing this: please feel free to re-open or open a new issue if there is more to do.

closed 2023-11-04 by RobMeades