u-blox/ubxlib #155: Getting GNSS-data from GNSS-integrated modules like SARA-R5510M8S using only 1 main UART (CMUX usage?)

elektron314, 2023-11-02:
Hello!
Could anybody please help me set up SARA-R510M8S so that I could interact with the GNSS module via multiplexed main UART (RX/TX only)?

What's the channel of interaction of the cell module with the internal GNSS module? Should I consider it? Is it AT-command-based as well?

Example from main_loc_gnss_cell_continuous.c doesn't work because `uDeviceOpen(&gDeviceCfg, &devHandle);` returns -5 error, which means that some parameters are wrong. But which parameters can be wrong if devHandle is NULL and `gDeviceCfg` is just static const `uDeviceCfg_t gDeviceCfg = {.deviceType = U_DEVICE_TYPE_NONE};` ?

I have also tried to set up CMUX, so multiplexer is enabled successfully but the position is not coming. So I first set up the cell network, tested it after connection, and tried to add the GNSS part with CMUX. CMUX is enabled. `uCellMuxAddChannel` is ok. `uGnssAdd` is ok. Then something strange is happening. I don't understand why before uGnssPosGet everything is sending AT-commands and when the `uGnssPosGet` is executing, it is sending some data somewhere.
like
`U_GNSS: sent command b5 62 01 07 00 00 08 19`
Why is it sending some data and where? I'm trying to get the location. It should use AT commands, no?
Before that, I could see a lot of AT commands setting up the cellular module and testing it and then some strange hex data instead of AT commands. As a result - no position data.

RobMeades, 2023-11-02:
Hi there.

Starting at the start, when the GNSS chip is inside the cellular module and you are using `ubxlib` to access it, that access will always be through the single UART of the cellular module, the one that AT commands travel across.

There are two ways that `ubxlib` can do this: with the `AT+UGUBX` AT command, or by opening a CMUX channel so that AT commands can go over one channel and direct communication with the GNSS chip can go over another.  The latter is _always_ better (you will not be able to do continuous position without it) and so `ubxlib` will always try this first.  SARA-R5 supports such CMUX access, so that is what you will end up using.  All of the CMUX stuff is handled internally within `ubxlib` when you use the [location API](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/common/location/api/u_location.h#L479), you don't need to worry about it at application level.

As you have already found, [main_log_gnss_cell_continuous.c](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_cell_continuous.c) is the correct example to use for this: it opens the cellular device and then opens a GNSS "network" on that cellular device and, when you call [uLocationGetContinuousStart()](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/common/location/api/u_location.h#L479) that will, internally to `ubxlib`, open the necessary CMUX channels etc,, so there is no need for AT commands when talking to GNSS.  We just need to get your device configuration set up correctly.

Remember that the device is a _cellular_ device, which just happens to have a GNSS chip inside it also (the GNSS device being slave to the cellular system).  So it is a cellular device that you must open; the configuration you need will be something like this:

```
static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_CELL,
    .deviceCfg = {
        .cfgCell = {
            .moduleType = U_CELL_MODULE_TYPE_SARA_R5,
            .pSimPinCode = NULL,
            .pinEnablePower = -1,  // Set this to the MCU pin that enables the power supply to the cellular module, if you have one
            .pinPwrOn = xxx, // Set this to the MCU pin that is connected to the PWR_ON input pin of of the cellular module
            .pinVInt = -1, // Set this to the MCU pin that is connected to the VINT output pin of the cellular module, if you have one
            .pinDtrPowerSaving = -1
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_UART,
    .transportCfg = {
        .cfgUart = {
            .uart = xxx, // Set this to the HW block in your MCU that you are going to use for the UART
            .baudRate = 115200,
            .pinTxd = xxx, // Set this to the pin of your MCU that is connected to the TXD _input_  of the cellular module (it uses the DCE convention, hence the TXD pin of the cellular module is an input, data _to_be_transmitted_)
            .pinRxd = xxx, // Set this to the pin of your MCU that is connected to the RXD _output_  of the cellular module
            .pinCts = xxx, // Set this to the input pin of your MCU that is connected to the CTS output of the cellular module, -1 if you do not intend to connect flow control (though that is NOT RECOMMENDED when the CMUX is in use since a single loss of character may result in the whole thing falling apart as it will destroy the CMUX protocol)
            .pinRts = xxx, // Set this to the output pin of your MCU that is connected to the RTS input of the cellular module, -1 if you do not intend to connect flow control
            .pPrefix = NULL // I'm assuming you're not using Linux so this can be left at NULL
        }
```

The example code will [open the cellular device](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/example/location/main_loc_gnss_cell_continuous.c#L185) and then open a GNSS network on it.  Once the device is open and the GNSS network is up you can call any of the `ubxlib` APIs with the handle of the cellular device, including the [location ones](https://github.com/u-blox/ubxlib/blob/master/common/location/api/u_location.h) and any of the [GNSS ones](https://github.com/u-blox/ubxlib/tree/master/gnss/api), and they should do what you want.

FYI, you will see many `U_GNSS: sent command b5 62...`-type prints from the `ubxlib` code that are nothing to do with the activity of your application: this is because such commands are used by `ubxlib` to manage the GNSS device (power it up and configure it etc.).

elektron314, 2023-11-02:
The example [main_log_gnss_cell_continuous.c](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_cell_continuous.c) uses
`static const uNetworkCfgGnss_t gNetworkCfg = {.type = U_NETWORK_TYPE_NONE};`
Should I change it as well? How?

If I use the same parameters and order of functions then `uNetworkInterfaceUp` returns "-5" and uLocationGetContinuousStart returns "-5" as well.

RobMeades, 2023-11-02:
> The example [main_log_gnss_cell_continuous.c](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_cell_continuous.c) uses
static const uNetworkCfgGnss_t gNetworkCfg = {.type = U_NETWORK_TYPE_NONE};
Should I change it as well? How?

That bit of the example is on the `#else` side of the `#ifdef`, it is only there to keep the compiler happy when a GNSS module type has not been set.  These examples are built and tested by the `ubxlib` test system, so they are constructed to work based on #defines passed-in to the build (e.g. `U_CFG_TEST_CELL_MODULE_TYPE` which the test system might set to `U_CELL_MODULE_TYPE_SARA_R5` or `U_CELL_MODULE_TYPE_LARA-R6`, etc. and `U_CFG_TEST_GNSS_MODULE_TYPE`, which the test system might set to `U_GNSS_MODULE_TYPE_M8`/`U_GNSS_MODULE_TYPE_M9`, `U_GNSS_MODULE_TYPE_M10`); of course you don't have to do this.

You need to have code like the _other_ side of the `#ifdef`, just above where you pasted from, where a GNSS module type has been defined, something like:

```
// NETWORK configuration for GNSS
static const uNetworkCfgGnss_t gNetworkCfg = {
    .type = U_NETWORK_TYPE_GNSS,
    .moduleType = U_GNSS_MODULE_TYPE_M10, 
    .devicePinPwr = -1, // You can leave this at -1: it is only needed for the case where an external GNSS module is connected to the cellular module
    .devicePinDataReady = -1 // You can leave this at -1: it is only needed for the case where an external GNSS module is connected to the cellular module
};
```

elektron314, 2023-11-02:
Do you have any ideas why the location is not established? 
`errorCode` inside of the callback returns "-9"
The PCB with the antenna is on the windowsill right now.

That's my log:
```
Starting continuous location.
U_GNSS: sent command b5 62 06 08 00 00 0e 30.
U_GNSS: decoded UBX response 0x06 0x08: e8 03 01 00 01 00 [body 6 byte(s)].
U_GNSS: sent command b5 62 06 01 02 00 01 07 11 3a.
U_GNSS: decoded UBX response 0x06 0x01: 01 07 00 00 00 00 00 00 [body 8 byte(s)].
U_GNSS: sent command b5 62 06 01 08 00 01 07 01 00 00 00 00 00 18 e2.
U_GNSS: decoded UBX response 0x05 0x01: 06 3e [body 2 byte(s)].
Waiting up to 60 seconds for 5 location fixes.
rtt:~$ rtt:~$ U_GNSS: sent command b5 62 06 01 08 00 01 07 00 00 00 00 00 00 17 dc.
U_GNSS: decoded UBX response 0x05 0x01: 06 01 [body 2 byte(s)].
```

RobMeades, 2023-11-02:
General advice on interpreting error codes can be found [here](https://github.com/u-blox/ubxlib/blob/master/DEBUG_HINTS.md#reading-error-return-codes) but, as I think you may have now found, the vast majority of them are from [u_error_common.h](https://github.com/u-blox/ubxlib/blob/master/common/error/api/u_error_common.h) and -9 is, as you might expect, [U_ERROR_COMMON_TIMEOUT](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/common/error/api/u_error_common.h#L64C5-L64C27).  -8, as you have spotted is [U_ERROR_COMMON_PLATFORM](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/common/error/api/u_error_common.h#L63), which means an error **from** the platform, so something like UART or IO or other issues, likely to be a emitted by ESP-IDF, it doesn't mean that the platform is wrong, IYSWIM.

The only reason I've seen for not getting a GNSS fix is simply that the GNSS chip has not been able to achieve one.  The callback will be called whenever a navigation message arrives from the GNSS chip, even if it has _not_ established a position, so you could modify the `callback()` function of the example to, say, always print `errorCode` to check that it is at least receiving navigation messages with a non-zero error code.

Assuming you are, you could then temporarily modify the [messageCallback()](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/gnss/src/u_gnss_pos.c#L384) function so that the last parameter passed to `posDecode()` is `true` instead of `false`: this will cause the decode function to print out what it sees; this should _not_ be left on for normal operation as it can be quite a load on the system, but it could be interesting while you're trying to find out what's happening.

elektron314, 2023-11-03:
I wouldn't say that there are any changes in the behavior of the program on changing the passing parameter printIt from false to true. I didn't mention any.

There is some ambiguity I would say. In the description of `uLocationGetContinuousStart` it says that `pCallback` "_will be called when location has been determined_" but at the same time `errorCode` returns `-9` which means timeout. So it seems like the location is determined because we see this callback fired but what is the timeout? timeout of what?

For example, in my case coordinates that I read during this timeouted callback are +0.0000000,+53.7112400. Weird.

RobMeades, 2023-11-03:
So that I can understand better what you are seeing, in your original log above you had:

```
Waiting up to 60 seconds for 5 location fixes.
```

...and then nothing, by which I understood that the print marked in red below:

[image]

...was not occurring, `errorCode` was never zero.

You say "it seems like the location is determined because we see this callback fired": I guess you have added a print in the callback so that we can see if it is being called with a negative `errorCode`.  If that is the case, how many times do you see it being called, versus what you would expect from the rates you passed to [uLocationGetContinuousStart()](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/common/location/api/u_location.h#L479)?

If changing the last parameter in the call to [posDecode()](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/gnss/src/u_gnss_pos.c#L141) to `true` results in nothing more being printed out, then either no navigation messages are arriving (hence my question above about whether the callback is being called at all) or navigation messages are arriving and none of the fields contain a fix: no time fix, no location fix.

> There is some ambiguity I would say. In the description of uLocationGetContinuousStart...

Sorry, yes, as a result of issue #116 the code was changed in commit efdc592cb2aa840900e1c6591d00124f41eb8867 to call the callback even when there is an error; the description is out of data, I will fix it.

If the `errorCode` is not zero then none of the location fields will have been populated and so will have random contents.  I might change that to set them to the "empty" values that are usually employed elsewhere in `ubxlib` (e.g. `INT_MIN` for lat/long etc.).

elektron314, 2023-11-03:
You are right in everything.

how many times do you see it being called, versus what you would expect from the rates you passed to [uLocationGetContinuousStart()](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/common/location/api/u_location.h#L479)?

5 of 5

Do you have any idea why isn't there any useful load in GNSS internal messages?

Can I test anything besides those messages? Is there anything else I can get from the GNSS part? Could it be a problem with the antenna path? I'm worrying now as it has to send me anything because the antenna's amp is powered on, and a filter is in place. But this is the first prototype of the U-blox SARA-R510M8S+nRF5340 system of mine.

RobMeades, 2023-11-03:
Understood: to be clear then, you have a SARA-R510M8S module, you are setting the `.moduleType` in the `uNetworkCfgGnss_t` structure to `U_GNSS_MODULE_TYPE_M8`, you are receiving navigation messages at the expected rate but the prints in [posDecode()](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/gnss/src/u_gnss_pos.c#L1410) emit nothing.

I think maybe it is time to see the whole navigation message.  There are other ways to do this, but for the sake of simplicity in this "point debug" situation, I suggest you modify [posDecode()](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/gnss/src/u_gnss_pos.c#L1410) to add the following at the top of it:

```
    int32_t y;
    int64_t t = -1;   // Existing stack variable declarations

    uPortLog("### UBX-NAV-PVT: 0x");
    for (size_t x = 0; x < 92; x++) {
        uPortLog("%02x ", *(pMessage + x));
    }
    uPortLog("\n");

    if ((*(pMessage + 11) & 0x03) == 0x03) { // Existing if() statement at the start of posDecode()
```

...noting that I've not compiled this code, just a sketch to get you working, please adjust for semicolons or other compilation errors as necessary.

Post the output here and we can pick through it.

RobMeades, 2023-11-04:
> Can I test anything besides those messages? Is there anything else I can get from the GNSS part? 

We're probably running before we can walk here: [uLocationGetContinuousStart()](https://github.com/u-blox/ubxlib/blob/cd01364006c3bcf1476b7711ae43b5fbf7b4a79b/common/location/api/u_location.h#L479) doesn't print debug out because it is intended to be used, well, continuously, with messages coming at it at a relatively high rate.

Since you're just in the process of getting things working, it may be better to use the blocking [uLocationGet](https://github.com/u-blox/ubxlib/blob/0b98b7396f430e03c5cb03ef67520039d040e7a4/common/location/api/u_location.h#L316), i.e. as the [main_loc_gnss_cell.c](https://github.com/u-blox/ubxlib/blob/0b98b7396f430e03c5cb03ef67520039d040e7a4/example/location/main_loc_gnss_cell.c#L205) example.  This function returns the location directly into a variable, rather than to a callback, and, because it is just doing one blocking call, it also by default prints out debug, which will include the whole message from the GNSS chip.

Here's some sample output from when our test system runs that example in the exact same scenario as you, with a SARA-R510M8S module; the timings on the left are added by the test system, so you won't see those; I've left them in for your information:

```
2023-11-03 19:52:22,783  U_GNSS: sent command b5 62 01 07 00 00 08 19.
2023-11-03 19:52:24,013  U_GNSS: decoded UBX response 0x01 0x07: 40 b7 03 1e e7 07 0b 03 13 34 16 37 fc ff a7 00 30 00 00 00 00 00 06 00 00 00 00 00 00 00 00 00 f2 ed 01 00 8a ab 01 00 7a 0e b4 01 ac 56 34 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2023-11-03 19:52:24,025  U_GNSS_POS: UTC time = 1699041142.
2023-11-03 19:52:24,113  U_GNSS: sent command b5 62 01 07 00 00 08 19.
2023-11-03 19:52:25,014  U_GNSS: decoded UBX response 0x01 0x07: 28 bb 03 1e e7 07 0b 03 13 34 17 37 fc ff a7 00 d4 86 fd ff 00 00 06 00 00 00 00 00 00 00 00 00 f2 ed 01 00 8a ab 01 00 ac 0e b4 01 cc 56 34 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2023-11-03 19:52:25,025  U_GNSS_POS: UTC time = 1699041143.
2023-11-03 19:52:25,113  U_GNSS: sent command b5 62 01 07 00 00 08 19.
2023-11-03 19:52:26,014  U_GNSS: decoded UBX response 0x01 0x07: 10 bf 03 1e e7 07 0b 03 13 34 18 37 fc ff a7 00 fb 85 fd ff 00 00 06 00 00 00 00 00 00 00 00 00 f2 ed 01 00 8a ab 01 00 fa 0e b4 01 04 57 34 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2023-11-03 19:52:26,025  U_GNSS_POS: UTC time = 1699041144.
2023-11-03 19:52:26,113  U_GNSS: sent command b5 62 01 07 00 00 08 19.
2023-11-03 19:52:27,030  U_GNSS: decoded UBX response 0x01 0x07: f8 c2 03 1e e7 07 0b 03 13 34 19 37 fc ff a7 00 18 85 fd ff 00 00 06 00 00 00 00 00 00 00 00 00 f2 ed 01 00 8a ab 01 00 60 0f b4 01 4e 57 34 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2023-11-03 19:52:27,043  U_GNSS_POS: UTC time = 1699041145.
2023-11-03 19:52:27,124  U_GNSS: sent command b5 62 01 07 00 00 08 19.
2023-11-03 19:52:28,030  U_GNSS: decoded UBX response 0x01 0x07: e0 c6 03 1e e7 07 0b 03 13 34 1a 37 fc ff a7 00 34 84 fd ff 00 00 06 00 00 00 00 00 00 00 00 00 f2 ed 01 00 8a ab 01 00 e6 0f b4 01 ac 57 34 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2023-11-03 19:52:28,047  U_GNSS_POS: UTC time = 1699041146.
2023-11-03 19:52:28,125  U_GNSS: sent command b5 62 01 07 00 00 08 19.
2023-11-03 19:52:29,039  U_GNSS: decoded UBX response 0x01 0x07: c8 ca 03 1e e7 07 0b 03 13 34 1b 37 fc ff a7 00 4b 83 fd ff 00 00 06 00 00 00 00 00 00 00 00 00 f2 ed 01 00 8a ab 01 00 88 10 b4 01 1e 58 34 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2023-11-03 19:52:29,051  U_GNSS_POS: UTC time = 1699041147.
2023-11-03 19:52:29,134  U_GNSS: sent command b5 62 01 07 00 00 08 19.
2023-11-03 19:52:30,039  U_GNSS: decoded UBX response 0x01 0x07: b0 ce 03 1e e7 07 0b 03 13 34 1c 37 fc ff a7 00 67 82 fd ff 00 00 06 00 00 00 00 00 00 00 00 00 f2 ed 01 00 8a ab 01 00 44 11 b4 01 a4 58 34 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2023-11-03 19:52:30,051  U_GNSS_POS: UTC time = 1699041148.
2023-11-03 19:52:30,134  U_GNSS: sent command b5 62 01 07 00 00 08 19.
2023-11-03 19:52:31,040  U_GNSS: decoded UBX response 0x01 0x07: 98 d2 03 1e e7 07 0b 03 13 34 1d 37 fc ff a7 00 37 fc fd ff 00 00 06 00 00 00 00 00 00 00 00 00 f2 ed 01 00 8a ab 01 00 1c 12 b4 01 3e 59 34 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2023-11-03 19:52:31,053  U_GNSS_POS: UTC time = 1699041149.
2023-11-03 19:52:31,134  U_GNSS: sent command b5 62 01 07 00 00 08 19.
[... cut ...] ault/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf) (section 32.17.17), and the hex bytes that follow (`45 de 03...`) is the [92 byte] payload of the `UBX-NAV-PVT` message, the start of which you can see highlighted in red below in the interface manual.

[image]

You should then be able to study the detail of the message to find out what the M8 chip is seeing.  For instance, as you can see, the value at offset 23 is the number of satellites the GNSS device could see.

RobMeades, 2023-11-06:
> Where does the module get the date and time and why is it so constant?

I'd guess that is just the earliest date the FW has, some time when the code was written, effectively an "I've not nothing" answer, which would be the case if no satellites whatsoever are visible; there is nowhere to get the time.

GNSS antennas, grounding around them, power supply decoupling, matching, etc. all needs relatively careful design since the signal levels are very low indeed.  This is not something I know a lot about I'm afraid, can only suggest you look at the integration manual:

https://content.u-blox.com/sites/default/files/SARA-R5_SysIntegrManual_UBX-19041356.pdf

...section 2.4.3, and of course make sure that the antenna can see open sky.

RobMeades, 2023-11-13:
Yes, that's mostly right: you don't need the `uAtClientInit()` or the `uCellInit()`, `uDeviceOpen()` will do those as necessary, you just need `uPortInit()` so that there is a porting layer underneath you and `uDeviceInit()` to initialise the device layer, then `uDeviceOpen()` the cellular module and `uNetworkInterfaceUp()` with type `U_NETWORK_TYPE_GNSS` to bring into operation the GNSS device that is inside the cellular module; in other words, see the [main_loc_gnss_cell.c](https://github.com/u-blox/ubxlib/blob/cba2456bf025d18b2dd0af763fe21e1164ae0310/example/location/main_loc_gnss_cell.c#L169) example.

I'm trying to think if there's anything, aside from antenna issues, that might make a GNSS module see nothing _whatsoever_.  Nothing has occurred to me so far, will continue thinking.

RobMeades, 2023-11-13:
One possibility: if you have a _very_ old SARA-R510M8S module (manufactured mid 2020, production week (printed on the module label, upper right) earlier than 20/27), a configuration change was needed for the LNA to switch on inside the GNSS device. The code that does this is still in the `ubxlib` code-base, just define `U_CFG_SARA_R5_M8_WORKAROUND` for your build.

RobMeades, 2023-11-16:
One thing you could do to get an idea of what your antenna can see is to read the `UBX-NAV-SAT` message from the GNSS chip, see section 32.17.20 of the [M8 receiver description](https://www.u-blox.com/docs/UBX-13003221).  You would do this something like this (code not compiled by me, please correct for semi-colons etc., intention is to point you in the right direction):

```
char buffer[1024];

// Callback for asynchronous message reception.
static void callback(uDeviceHandle_t devHandle, const uGnssMessageId_t *pMessageId,
                     int32_t errorCodeOrLength, void *pCallbackParam)
{
    int32_t length;

    (void) pMessageId;
    (void) pCallbackParam;

    if (errorCodeOrLength >= 0) {
        if (errorCodeOrLength > sizeof(buffer)) {
            errorCodeOrLength = sizeof(buffer)
        }
        // Read the message into our buffer and print it
        length = uGnssMsgReceiveCallbackRead(devHandle, buffer, errorCodeOrLength);
        if (length >= 6 + 8) {

            // YOU SHOULD ANALYZE THE CONTENTS OF BUFFER HERE

            // It will contain the entire UBX-NAV-SAT message, including
            // 6 byte UBX message header, see section 32.17.20 of the M8
            // receiver description manual.

            // For instance the number of satellites your GNSS chip
            // can see would be (reversed 'cos the messages are little-endian):
            printf("Number of satellites visible: 0x%02x%02x.\n", buffer[6 + 7], buffer[6 + 6]);

        } else {
            printf("Empty or bad message received.\n");
        }
    }
}
```

...then in your main code:

```
int32_t x;
uGnssMessageId_t messageId;

// devHandle is assumed to be set up somewhere above

// Set up the message ID we want to capture: in this case just UBX-NAV-SAT
messageId.type = U_GNSS_PROTOCOL_UBX;
messageId.id.ubx = 0x0135; // message class/ID of UBX-NAV-SAT

// Set the rate for this message ID to once per second
x = uGnssCfgSetMsgRate(devHandle, &messageId, 1);
if (x == 0) {
    // Set up an asynchronous message receiver for this message ID,
    // giving it the callback above.
    x = uGnssMsgReceiveStart(devHandle, &messageId, callback, NULL);
    if (x < 0) {
        printf("Unable to start message receiver (%d)!\n", x);
    }

   // callback() will now be called when UBX-NAV-SAT messages arrive,
   // which should be once a second

    ...

    // Call this when you want to stop
    uGnssMsgReceiveStopAll(devHandle);
} else {
    printf("Unable to set UBX-NAV-SAT message rate (%d)!\n", x);
}
```

RobMeades, 2023-11-28:
That's a shame, sorry to hear you're having HW troubles, good luck figuring it out (rather you than me :-)).

I'll leave it up to you whether you want to keep this issue open or close it (and open a new one when necessary).  If you want to leave it open I will close it eventually, in a month or two, when it seems there is no more to add.

elektron314, 2023-12-07:
And hello again @RobMeades ,

I have found a strange thing.
ublox EVK-R5 sends this message when trying to get a location fix:
```
U_GNSS_POS: UTC time = 1701712958.
ublox: GNSS message callback returnCode -9.
```
but my custom board with the problem modem+antenna sends just this
`ublox: GNSS message callback returnCode -9.`

Do you have any ideas why my board doesn't even come into this part of the code that sends the UTC time?
And where does it get that time from?

RobMeades, 2023-12-07:
Hi again: the GNSS chip will be able to return the time if it has been able to decode the data from at least one satellite; that's where it is getting the time from, the satellites, and then it is returning -9 (timeout) on getting a position fix, which of course involves decoding data from, several satellites to triangulate position.

If it is not even returning a time then, unfortunately, that means that the GNSS chip is not managing to receive a signal from even one satellite.

elektron314, 2023-12-07:
Wow, that's interesting that EVK is getting time even without antenna

RobMeades, 2023-12-07:
That is indeed interesting: it might be that the time is sent most heavily coded (I would guess) and is therefore going to be the easiest thing to pull out of the noise.

elektron314, 2023-12-11:
I finally made it! After some rewiring my board can connect to GNSS.

But I have a question. Is it possible to get the location continuously and get data about C/N0?
uLocationGetContinuousStart vs periodical UBX-NAV-SAT

I'm not sure that I set it up properly but this is the output after I set them up together:

```
U_GNSS_POS: UTC time = 1702326932.
U_GNSS_POS: 3D fix achieved.
U_GNSS_POS: satellite(s) = 6.
U_GNSS_POS: longitude = 247438612 (degrees * 10^7).
U_GNSS_POS: latitude = 594248226 (degrees * 10^7).
U_GNSS_POS: altitude = -89256 (mm).
U_GNSS_POS: radius = 15577 (mm).
U_GNSS_POS: speed = 122 (mm/s).
iTow: 0x40 0x01 0x70 0xCC
version: 0x91
numSvs: 0x09
gnssId: 0x1A
svId: 0x00
cno: 0x00
```

Where starting from "iTow" comes messages from callback for UBX-NAV-SAT.
I don't understand what's wrong with "cno" it is constantly 0.
And if you see `numSvs: 0x09` and `U_GNSS_POS: satellite(s) = 6` differ.

RobMeades, 2023-12-11:
Woohoo!  Nicely done.

If you want to receive both UBX-NAV-PVT and UBX-NAV-SAT messages, the best way to do that is to extend what you already have with the asynchronous message reception API and not bother with the `uLocation` API.  The code would go something like this (code not compiled, this to point you in the right direction):

```
char buffer[1024];

// Callback for asynchronous message reception.
static void callback(uDeviceHandle_t devHandle, const uGnssMessageId_t *pMessageId,
                     int32_t errorCodeOrLength, void *pCallbackParam)
{
    int32_t length;
    uGnssDec_t *pDec;
    uGnssDecUbxNavPvt_t *pUbxNavPvt;

    (void) pCallbackParam;

    if (errorCodeOrLength >= 0) {
        if (errorCodeOrLength > sizeof(buffer)) {
            errorCodeOrLength = sizeof(buffer)
        }
        if (pMessageId->type == U_GNSS_PROTOCOL_UBX) {
            // Read the message into our buffer
            length = uGnssMsgReceiveCallbackRead(devHandle, buffer, errorCodeOrLength);
            if (pMessageId->id == 0x0107) {
                // It is a UBX-NAV-PVT message: ask the uGnssDec API to
                // decode it (it can currently ony decode this message and
                // the high precision version)
                pDec = pUGnssDecAlloc(buffer, length);
                if ((pDec != NULL) && (pDec->errorCode == 0)) {
                    pUbxNavPvt = pDec->pBody.ubxNavPvt;
                    // The decoded UBX-NAV-PVT message is now hooked
                    // into the memory allocated at pDec->pMsg, do what you
                    // need with it here but don't block this call-back for
                    // any length of time, return as quickly as possible and
                    // bare in mind that the stack for this call-back will
                    // be relatively limited

                    // Just as an example:
                    print("iTow is %d.\n", pUbxNavPvt->iTOW);
                }
                // IMPORTANT: must *always* free the memory that
                // pUGnssDecAlloc() allocated when done
                uGnssDecFree(pDec);
            } else if (pMessageId->id == 0x0107) {
                // It is a UBX-NAV-SAT message; process that "manually" here
                // as you like, skipping past the 6 byte UBX message header
                // For instance the number of satellites your GNSS chip
                // can see would be (reversed 'cos the messages are little-endian):
                if (length >= 8) {
                    printf("Number of satellites visible: 0x%02x%02x.\n", buffer[6 + 7], buffer[6 + 6]);
                }
            } else {
                printf("Unknown UBX message received (0x%04x).\n", pMessageId->id);
            }
        }
    }
}
```

...then in your main code:

```
int32_t x;
uGnssMessageId_t messageId;

// devHandle is assumed to be set up somewhere above

// Set up the message ID of UBX-NAV-SAT
messageId.type = U_GNSS_PROTOCOL_UBX;
messageId.id.ubx = 0x0135;

// Set the rate for UBX-NAV-SAT to once per second
x = uGnssCfgSetMsgRate(devHandle, &messageId, 1);
if (x == 0) {
   // Set up the message ID of UBX-NAV-PVT
    messageId.id.ubx = 0x0107;
    // Set the rate for UBX-NAV-PVT to once per second
    x = uGnssCfgSetMsgRate(devHandle, &messageId, 1);
    if (x == 0) {
        // Set up an asynchronous message receiver all UBX-NAV message IDs,
        // which will capture the two messages we have now enabled
        messageId.id.ubx = 0x01FF; // message class of UBX-NAV with ALL IDs
        x = uGnssMsgReceiveStart(devHandle, &messageId, callback, NULL);
        if (x < 0) {
            printf("Unable to start message receiver (%d)!\n", x);
        }

       // callback() will now be called when any UBX-NAV message arrives,
       // which should be two each second

        ...

        // Call this when you want to stop
        uGnssMsgReceiveStopAll(devHandle);
    } else {
        printf("Unable to set UBX-NAV-PVT message rate (%d)!\n", x);
    }
} else {
    printf("Unable to set UBX-NAV-SAT message rate (%d)!\n", x);
}
```

elektron314, 2023-12-12:
Why do you @RobMeades  indicate that numSV is buffer[6 + 7] and buffer[6 + 6] if it's 5th byte with 4 bytes offset?
Table 32.17.20.1 Satellite information

RobMeades, 2023-12-12:
Sorry, my mistake not sure where I got that from, you're quite right that it is a single byte at offset 5, so:

```
if (length >= 6 + 5) {
    printf("Number of satellites visible: %d.\n", buffer[6 + 5]);
}
```

elektron314, 2023-12-12:
but could you please explain why 6+ and not 4+?
In the beginning there are only 4 bytes constantly.

RobMeades, 2023-12-12:
Ah, no it's like this:

[image]

cturvey, 2023-12-12:
Ok, but I get @elektron314 point here.

What's "length" variable here? The entire packet with headers and sums? Or the value read from Length field of message structure.

The Payload starts at offset 6, the length of the payload for NAV-SAT is at least 8 bytes, then 12 more for each satellite described.

Assuming "length" variable is entire packet, there's further confusion as the "buffer" array is zero offset based so  (length > 6 + 5) not (length >= 6 + 5), you're touching element 11, the length of the array being at least 12 bytes

-Clive

RobMeades, 2023-12-12:
It is the length of the whole message, including any headers and check-sums.  I will make a note to update the description of `uGnssMsgReceiveCallbackRead()` to be clearer about this.

RobMeades, 2023-12-12:
> so (length > 6 + 5)

Yes, my mistake, that is correct.

cturvey, 2023-12-12:
Ok, that was my assumption based on a skim of the code and logic here.
So the minimum acceptable packet length is 16 bytes for zero satellites. That would be a fixed payload of 8 bytes
A test of (length >= 16) would be more appropriate for something complete with valid sums.
The width of the "Length" field being 2-bytes is a point of repeated confusion amongst readers of the u-Blox documentation.

Ok, saw you update in the feed. I think you've got a couple of those scattered in the code, see the odd things as I lurk in the email feed. Not trying to nit-pick, because I know I hate getting those myself, but it can confuse people when trying parse the code, ie intent vs actual behaviour

-Clive
https://portal.u-blox.com/s/profile/0052p000008HYMw

RobMeades, 2023-12-12:
No worries, keep nit-picking, especially if you come across any in the code.

elektron314, 2023-12-20:
Hello @RobMeades, so I implemented your proposal and I have these results and questions.
The first bytes of the UBX-NAV-SAT message and some information extracted from UBX-NAV-PVT are next:
```
numSV is 0
iTOW is 306447000
valid is 55
year is 2023
message: 0xB5 62 01 35 C8 00 98 02 44 12 01 10 00 00 00 02 0D A5 00 00 00 00
```
 As you can see UBX-NAV-SAT says that the number of satellites is 0x10 but the UBX-NAV-PVT have 0 there.

And I have a question about the Carrier to noise ratio (signal strength). it behaves like seconds increasing by 1 every second to when it gets some number it drops to zero again.

RobMeades, 2023-12-20:
> As you can see `UBX-NAV-SAT` says that the number of satellites is 0x10 but the `UBX-NAV-PVT` have 0 there.

My guess would be that the difference is down to the number in `UBX-NAV-SAT` being the number of satellites "visible" while the `UBX-NAV-PVT` is the number of satellites "used in the NAV solution": did the `UBX-NAV-PVT` contain a solution (i.e. was the `fixType` field 2, 3 or maybe 5)?

On the carrier to noise ratio, I'm not really expert in enough in GNSS to know: the general advice I have been given is "with a decent antenna, C/No of strongest signals at least 40 dBHz, good antenna up to 50 dBHz. Below 30 dBHz decoding data (i.e. ephemeris) is challenging."

cturvey, 2023-12-20:
Unfortunately there's really not enough meat in the NAV-SAT data reported to make much of a determination externally.
The payload here is 200 bytes, and could be more or less depending on the constellations and total number of satellites tracked.
There is a separate "used" flag to designate if particular satellites are usable, and the overall list could be reasonably expansive depending on what almanac data is on hand to drive a list of possibly visible satellites from your vantage point.
The count in NAV-SAT is the super-set of potentially available / tracked satellites, where as NAV-PVT reports how many were used in a solution. Zero suggesting that the antenna is placed or functioning very poorly. Be outside with the element facing upward.
Sporadic signals can be tracked into the 10's and 20's dBHz, The data demodulation threshold being around 32 dBHz in the designs I'm familiar with, although sub-frames may be accumulated through persistence or external aiding.
A system with 4-5 satellites in the 32-35 dBHz range should be able to pull in a solution within a couple of minutes.

RobMeades, 2024-01-30:
Hi @elektron314: do you have enough feedback on this issue, i.e. so that we can close it now?

elektron314, 2024-01-31:
Yes, thanks a lot in this topic. You helped me very much.

RobMeades, 2024-01-31:
Good stuff, thanks for letting me know.