u-blox/ubxlib #158: Faster connection - avoiding of calling AT#CFUN=4 every cellular device open

arnoutdekimo, 2023-11-08:
Hi,

For the usecase I am considering using ubxlib for, my embedded device would wake up the modem and want to send data to the cloud asap. Normally I think, when the (SARA-R4 in my case) device boots, it can automatically try to reassociate.
When using ubxlib though, when calling uDeviceOpen, at some point it calls `uCellPwrPrivateOn`, which then calls `moduleConfigure`.
In moduleConfigure, the argument **andRadioOff** is checked to see if the code should first unpower the radio or not. It -always- does this, since the caller function fills this in by `!uCellPrivateIsRegistered(pInstance)` . This function just checks a flag, that is never filled in when first calling the code, and I don't see a way for the lib to just poll the device to check if it is first registered or not.

The result is that upon boot, the library -always- sends AT#CFUN=4, then needs to reenable the radio and then needs to reregister. This looses some precious time (and battery power), which I'd like to avoid.

Is my conclusion correct here? And what would be the recommended way to use ubxlib to get actual (e.g. socket API's) up asap?

Thanks for your feedback!

RobMeades, 2023-11-08:
Hi there: yes, your assessment is correct.  The design of `ubxlib` is to keep the module in its lowest power state when no connection is required, hence the behaviour you see. The other aspects are that, in order to make certain choices (APN and associated fields in the [uNetworkInterfaceUp()](https://github.com/u-blox/ubxlib/blob/99b6f24580192d36ae75ae0c2e3390678915f32e/common/network/api/u_network.h#L199) case, also automatic network selection mode versus a specific MCC/MNC in the [uCellNetConnect()](https://github.com/u-blox/ubxlib/blob/99b6f24580192d36ae75ae0c2e3390678915f32e/cell/api/u_cell_net.h#L282) case) an API would usually be called to activate the connection.

A few questions to understand your use case:

1. Is it that you are _only_ interested in optimizing speed, optimizing power is not an aim, or is it that you are interested in optimizing both speed and power and doing things quickly works best for you?
2. Do you need to choose a specific APN or will the default assigned by the network always be fine for you (i.e. you would specify `NULL` or any empty string in the `pApn` field of [uNetworkCfgCell_t](https://github.com/u-blox/ubxlib/blob/99b6f24580192d36ae75ae0c2e3390678915f32e/common/network/api/u_network_config_cell.h#L42)?

To be clear about `AT+CFUN=4`, it should not cause de-registration with the network, it just switches the radio off, the network will still believe you are connected, like you are in a tunnel, until its periodic timers time (a few minutes) time out.  However, you are correct that it will cause a delay because the module will stop any initial search occurring.  It is bigger than just `andRadioOff` of course, since `ubxlib` always puts the module into its lowest power state when it shuts down, so the module will likely always come up in airplane mode in any case.  In other words, I suspect if you just hack `andRadioOff` to `false` it would not improve matters.

I'll think about what we might do to add a kind of "most connected" mode behaviour.

arnoutdekimo, 2023-11-08:
Hi!

- For my usecase, my device will be not doing anything for extended periods of time (talking **days** here). Hence the modem will be fully powered off, since that's the most optimal power situation. When it powers on, it needs to do so asap. So ultimately the amount of energy needed to send some data to the cloud is what I want to minimize, but I figured speeding things up only help, so I quickly looked into some things that I found strange. 

- The scenario I envisioned was that upon powerloss, the modem still "knows" upon it should try to register asap, and that when ubxlib starts talking to it, it does not interrupt this process (and hence lose more time). I did not know that the lib put it into airplane mode upon shutdown, so indeed in this case it shouldn't have been consuming alot of power upfront after powerup trying to register, and then be cut short when ubxlib tried to interrupt it. Instead upon powerup it should be +- idle, and wait until the radio is actually put back on a bit later. Correct? 

- The APN will be known upfront, and we will probably need to specify it. 

Anyway, thanks already for the feedback.

RobMeades, 2023-11-08:
Thanks for the clarification, very useful.

First a correction to what I said above: `AT+CFUN=4` _does_ deregister from the network, just like `AT+CFUN=0` would, the difference is that `AT+CFUN=4` persists across power cycles/reboots (there is an `AT+CFUN=10` that does the "but don't deregister" thing, but that is not available on SARA-R4).

So yes, you interpretation is correct that the design of `ubxlib` is that the module will already be in "radio off" state at boot, though we don't, for instance, test with a power meter connected so there could always be a bug.

That said, with this behaviour there will be a time short time at start-up when the module hardware will be powered-on but the module _radio_ hardware will not yet be powered on, because it has not been told by the application to do so.  It won't be very long but it will be there.  If you need to specify an APN then something, at some point, needs to pass that APN to the module and make it stay; I'm not sure off the top of my head if something needs to be done for the APN to persist (as a matter of policy `ubxlib` tries not to rely on persistence in the module, making control/configuration a matter for the application).

The other thing we have to be careful of is that, on some modules, certain commands must be issued for the internal clients of the module (IP stack, MQTT, HTTP) to operate correctly after a network connection has been achieved, and these commands are sent in the code that performs the connection, hence wouldn't be done if the `ubxlib` code had not caused the connection, IYSWIM. If you look in that big `struct` in [u_cell_private.c](https://github.com/u-blox/ubxlib/blob/99b6f24580192d36ae75ae0c2e3390678915f32e/cell/src/u_cell_private.c#L97) the "feature" in question is `U_CELL_PRIVATE_FEATURE_CONTEXT_MAPPING_REQUIRED` and it is, unfortunately, required in the SARA-R422 case; without this sockets won't work because the IP stack inside the module is unaware of the connection.

I think what I'm saying is that it _might_ be possible to optimize that bit of extra power consumption out but it would be a relatively large change; introduce a kind of "don't go off" mode and make sure that all of our cellular modules behave as expected, using the values once passed for APN etc.  If you think it is worth it I can put it on the "TODO" list.

arnoutdekimo, 2023-11-08:
And perhaps for completeness - hacking the andRadioOff (to ignore) turned out to be a bad idea
The module apparently reports all is well, but name resolving etc does always work anymore 

This is the output of a socket example

> 0:00:00:053 - AT
> 0:00:00:154 - 
> 0:00:00:193 - OK
> 0:00:00:193 - U_CELL_PWR: powering on, module is already on.
> 0:00:00:194 - ATE0
> 0:00:00:294 - 
> 0:00:00:333 - OK
> 0:00:00:333 - AT+CMEE=2
> 0:00:00:434 - 
> 0:00:00:474 - OK
> 0:00:00:474 - AT+UDCONF=1,0
> 0:00:00:576 - 
> 0:00:00:615 - OK
> 0:00:00:615 - ATI9
> 0:00:00:716 - 
> 0:00:00:755 - 01.24,A01.10
> 0:00:00:756 - 
> 0:00:00:756 - OK
> 0:00:00:756 - AT&C1
> 0:00:00:857 - 
> 0:00:00:897 - OK
> 0:00:00:897 - AT&D0
> 0:00:00:998 - 
> 0:00:01:038 - OK
> 0:00:01:038 - AT+UCGED=2
> 0:00:01:140 - 
> 0:00:01:179 - OK
> 0:00:01:179 - AT&K3
> 0:00:01:280 - 
> 0:00:01:320 - OK
> 0:00:01:320 - AT+UPSV=0
> 0:00:01:421 - 
> 0:00:01:461 - OK
> 0:00:01:461 - AT+UPSMR=1
> 0:00:01:563 - 
> 0:00:01:602 - OK
> 0:00:01:602 - AT+CPSMS?
> 0:00:01:703 - 
> 0:00:01:743 - +CPSMS:0,,,"01100000","00000000"
> 0:00:01:744 - 
> 0:00:01:744 - OK
> 0:00:01:745 - AT+CEDRXS?
> 0:00:01:847 - 
> 0:00:01:886 - +CEDRXS:
> 0:00:01:887 - 
> 0:00:01:887 - OK
> 0:00:01:887 - AT+UMNOPROF?
> 0:00:01:989 - 
> 0:00:02:028 - +UMNOPROF: 90
> 0:00:02:029 - 
> 0:00:02:029 - OK
> 0:00:02:029 - AT+UGPRF?
> 0:00:02:130 - 
> 0:00:02:170 - +UGPRF: 2,0,""
> 0:00:02:171 - 
> 0:00:02:171 - 
> 0:00:02:171 - 
> 0:00:02:171 - OK
> 0:00:02:171 - Opened device with return code 0.
> 0:00:02:172 - Bringing up the network...
> 0:00:02:172 - U_CELL_NET: preparing to register/connect...
> 0:00:02:172 - AT+CREG=2
> 0:00:02:274 - 
> 0:00:02:313 - OK
> 0:00:02:313 - AT+CGREG=2
[... 634 lines cut ...]
> 0:00:13:692 - 
> 0:00:13:693 - OK
> 0:00:13:693 - AT+UPSD=0,0,0
> 0:00:13:795 - 
> 0:00:13:835 - OK
> 0:00:13:835 - AT+UPSD=0,100,1
> 0:00:13:937 - 
> 0:00:13:977 - OK
> 0:00:13:977 - U_CELL_NET: connected after 11 second(s).
> 0:00:13:977 - Looking up server address...
> 0:00:13:978 - U_CELL_SOCK: looking up IP address of "ubxlib.com".
> 0:00:13:978 - AT+CGDCONT?
> 0:00:14:479 - 
> 0:00:14:519 - +CGDCONT: 1,"IP","web.be","100.71.33.56",0,0,0
> 0:00:14:520 - 
> 0:00:14:521 - OK
> 0:00:14:521 - AT+UDNSRN=0,"ubxlib.com"
> 0:00:14:624 - 
> 0:00:14:954 - +UDNSRN: "18.133.144.142"
> 0:00:15:104 - 
> 0:00:15:105 - OK
> 0:00:15:105 - U_CELL_SOCK: found it at "18.133.144.142".
> 0:00:15:105 - Address is: IPV4 18.133.144.142

But so long story short - cutting corners to skip AT#CFUN=4 gave me lots of issues. I'm keeping it in:)

RobMeades, 2023-11-08:
Ah, yes, I do recall there being issues with one or other SARA-R4 version with the need to do the step-by-step process, you can see that I invoked @philwareublox [in a comment](https://github.com/u-blox/ubxlib/blob/99b6f24580192d36ae75ae0c2e3390678915f32e/cell/src/u_cell_net.c#L2225): 

> This step _shouldn't_ be necessary.  However, for reasons I don't understand, SARA-R4 can be registered but not attached (i.e. `AT+CGATT` returns 0) on both RATs (unh?).  Phil Ware, who knows about these things, always goes through (a) register, (b) wait for `AT+CGATT` to return 1 and then (c) check that a context is active with `AT+CGACT` or using `AT+UPSD` (even for EUTRAN).  Since this sequence works for both RANs, it is best to be consistent.

Generally speaking, though, it should be `ubxlib`'s aim to accommodate the kind of performance tuning you need in your final application, preferably without you having to modify the code, so please do raise these things as you reach them.

arnoutdekimo, 2023-11-08:
By the way, even with the correct APN AND AT+CFUN=4 I am now getting that issue: :(

Not sure what I am supposed to do with this? I am not sure what is the cause, but in any case, the "ubxlib should absract all quirks away" mentality does not hold in this case.

(fyi the output, which remains the same if I reboot the module)

> 0:00:00:193 - U_CELL_PWR: powering on, module is already on.
> 0:00:00:193 - ATE0
> 0:00:00:293 - 
> 0:00:00:333 - OK
> 0:00:00:333 - AT+CMEE=2
> 0:00:00:434 - 
> 0:00:00:474 - OK
> 0:00:00:474 - AT+UDCONF=1,0
> 0:00:00:575 - 
> 0:00:00:615 - OK
> 0:00:00:615 - ATI9
> 0:00:00:715 - 
> 0:00:00:755 - 01.24,A01.10
> 0:00:00:755 - 
> 0:00:00:755 - OK
> 0:00:00:755 - AT&C1
> 0:00:00:856 - 
> 0:00:00:897 - OK
> 0:00:00:897 - AT&D0
> 0:00:00:997 - 
> 0:00:01:038 - OK
> 0:00:01:038 - AT+UCGED=2
> 0:00:01:139 - 
> 0:00:01:179 - OK
> 0:00:01:179 - AT&K3
> 0:00:01:279 - 
> 0:00:01:320 - OK
> 0:00:01:320 - AT+UPSV=0
> 0:00:01:421 - 
> 0:00:01:461 - OK
> 0:00:01:461 - AT+UPSMR=1
> 0:00:01:562 - 
> 0:00:01:602 - OK
> 0:00:01:602 - AT+CPSMS?
> 0:00:01:703 - 
> 0:00:01:743 - +CPSMS:0,,,"01100000","00000000"
> 0:00:01:744 - 
> 0:00:01:744 - OK
> 0:00:01:744 - AT+CEDRXS?
> 0:00:01:846 - 
> 0:00:01:886 - +CEDRXS:
> 0:00:01:886 - 
> 0:00:01:886 - OK
> 0:00:01:886 - AT+UMNOPROF?
> 0:00:01:988 - 
> 0:00:02:028 - +UMNOPROF: 90
> 0:00:02:028 - 
> 0:00:02:028 - OK
> 0:00:02:028 - AT+UGPRF?
> 0:00:02:130 - 
> 0:00:02:170 - +UGPRF: 2,0,""
> 0:00:02:170 - 
> 0:00:02:170 - 
> 0:00:02:170 - 
> 0:00:02:171 - OK
> 0:00:02:171 - AT+CFUN=4
> 0:00:02:272 - 
> 0:00:02:602 - OK
> 0:00:02:602 - DEVICE UPAT+CMUX=0,0,,128
> 0:00:02:705 - 
> 0:00:02:925 - OK
> 0:00:02:925 - ATE0
> 0:00:03:558 - 
> 0:00:03:598 - OK
[... 162 lines cut ...]
> 0:00:10:688 - OK
> 0:00:10:688 - AT+UHTTP=0,6,1,0
> 0:00:10:795 - 
> 0:00:10:835 - 
> 0:00:10:835 - 
> 0:00:10:835 - OK
> 0:00:10:835 - AT+UHTTP=0,5,443
> 0:00:10:941 - 
> 0:00:10:981 - 
> 0:00:10:981 - 
> 0:00:10:981 - OK
> 0:00:10:981 - HTTP GET request will be "/GetOfflineData.ashx?token=HCJx9y-_RKepara6bQJ3pw;gnss=gps;days=1;resolution=1".
> 0:00:10:981 - AT+UHTTPC=0,1,"/GetOfflineData.ashx?token=HCJx9y-_RKepara6bQJ3pw;gnss=gps;days=1;resolution=1","ubxlibhttp_0"
> 0:00:11:103 - 
> 0:00:11:143 - 
> 0:00:11:143 - 
> 0:00:11:143 - OK
> 0:00:11:143 - 
> 0:00:11:143 - +UUHTTPCR: 0,1,0
> 0:00:11:144 - AT+UHTTPER=0
> 0:00:11:259 - 
> 0:00:11:299 - +UHTTPER: 0,10,22
> 0:00:11:299 - 
> 0:00:11:299 - 
> 0:00:11:300 -

RobMeades, 2023-11-08:
Very strange indeed.  As you say, from the error codes in the back of the [AT manual](https://content.u-blox.com/sites/default/files/SARA-R4_ATCommands_UBX-17003787.pdf), section A.7, the "22" of your `+UHTTPER` means "PSD or CSD connection not established" when, from the AT command sequence, it clearly has been established as you have `+CEREG: 1,"C3D","2BF6B05",7,,,,` and the two `AT+UPSD` have been issued and OK'ed.

Let me see if I can engage the help of our San Diego office, who should be up and about now and who know SARA-R422.

RobMeades, 2023-11-08:
Something to try, just to see what it does, could you modify the function [activateContext()](https://github.com/u-blox/ubxlib/blob/99b6f24580192d36ae75ae0c2e3390678915f32e/cell/src/u_cell_net.c#L1446) so that it sends `AT+CGACT` even if 1 has been returned, something like this:

```
static int32_t activateContext(const uCellPrivateInstance_t *pInstance,
                               int32_t contextId, int32_t profileId)
{
    int32_t errorCode = (int32_t) U_CELL_ERROR_CONTEXT_ACTIVATION_FAILURE;
    uAtClientHandle_t atHandle = pInstance->atHandle;
    uAtClientDeviceError_t deviceError;
    bool activated = false;
    bool ours;

    deviceError.type = U_AT_CLIENT_DEVICE_ERROR_TYPE_NO_ERROR;
    uAtClientLock(atHandle);
    for (size_t x = 5; (x > 0) && keepGoingLocalCb(pInstance) &&
         (errorCode != 0) &&
         ((deviceError.type == U_AT_CLIENT_DEVICE_ERROR_TYPE_NO_ERROR) ||
          (deviceError.type == U_AT_CLIENT_DEVICE_ERROR_TYPE_ERROR)); x--) {
        uAtClientLockExtend(atHandle);

        /// CODE BLOCK MOVED UP FROM if() STATEMENT BELOW
        // Help it on its way.
        uAtClientLockExtend(atHandle);
        uAtClientCommandStart(atHandle, "AT+CGACT=");
        uAtClientWriteInt(atHandle, 1);
        uAtClientWriteInt(atHandle, contextId);
        uAtClientCommandStopReadResponse(atHandle);
        // If we get back ERROR then the module wasn't
        // ready, if we get back CMS/CME error then
        // likely the network has actively rejected us,
        // e.g. due to an invalid APN
        uAtClientDeviceErrorGet(atHandle, &deviceError);
        // END OF CODE BLOCK MOVED UP FROM if() STATEMENT BELOW

        uAtClientTimeoutSet(atHandle,
                            pInstance->pModule->responseMaxWaitMs);
        uAtClientCommandStart(atHandle, "AT+CGACT?");
        uAtClientCommandStop(atHandle);
        ours = false;
        for (size_t y = 0; (y < U_CELL_NET_MAX_NUM_CONTEXTS) &&
             !ours; y++) {
            uAtClientResponseStart(atHandle, "+CGACT:");
            // Check if this is our context ID
            if (uAtClientReadInt(atHandle) == contextId) {
                ours = true;
                // If it is, 1 means activated
                activated = (uAtClientReadInt(atHandle) == 1);
            }
        }
        uAtClientResponseStop(atHandle);
        // Do NOT unlock the AT client here
        if (activated) {
            errorCode = uCellPrivateActivateProfileNoAtLock(pInstance, contextId,
                                                            profileId, 5, keepGoingLocalCb);
        } else {
            uPortTaskBlock(2000);
        }
    }
    uAtClientUnlock(atHandle);

    return errorCode;
}
```

RobMeades, 2023-11-08:
Another thing to try: before doing your HTTP stuff, could you call [uCellNetGetIpAddressStr()](https://github.com/u-blox/ubxlib/blob/99b6f24580192d36ae75ae0c2e3390678915f32e/cell/api/u_cell_net.h#L759())?  Our man in San Diego is wondering if you might have been given an IPV6 address only, which is not something we[I] have seen before and may require different handling.

arnoutdekimo, 2023-11-09:
Hi Rob,

Left the module off for the night, and tested again this morning. Same thing. 
[httpnotworking.txt](https://github.com/u-blox/ubxlib/files/13305801/httpnotworking.txt)
[ipaddr.txt](https://github.com/u-blox/ubxlib/files/13305802/ipaddr.txt)

Then I tried to get the IP address before and after calling uNetworkInterfaceUp. The first time it failed (which is normal), the second time it just gave a nice IPv4 address:

> 0:00:18:894 - U_CELL_NET: IP address "100.64.145.89".

Regarding your other post, I was indeed thinking of trying to send some AT+CGACT commands to stop and restart the PSD, but that's not so simple with the existing API's. `uCellNetDeactivate` and friend check if the network is registered, but they don't actually check this through AT commands, but read out a cached variable in `uCellPrivateIsRegistered ` which is never filled in. So even though the module -is- registered, the API's dont support deregistering unless you first explicitly try to register.

Anyway, will try you redit and come back to you!

arnoutdekimo, 2023-11-09:
Hi,

So, the code change in activateContext did fix it. The next attempt, I got a clear "delay" when doing the HTTP command, and I am (sometimes) getting the correct data in.

There is still an issue with the HTTP parsing, which I think I figured out what it relates to (I'll make another ticket about that in a minute, to avoid spawning a 3rd issue in this thread :), but indeed, your code change with the AT+CGACT clearly fixed the "HTTP 22 error". => So that's good news!

> ..
> 0:00:08:749 - 
> 0:00:08:789 - 
> 0:00:08:789 - 
> 0:00:08:789 - OK
> 0:00:08:789 - HTTP GET request will be "/GetOfflineData.ashx?token=HCJx9y-_RKepara6bQJ3pw;gnss=gps;days=1;resolution=1".
> 0:00:08:789 - AT+UHTTPC=0,1,"/GetOfflineData.ashx?token=HCJx9y-_RKepara6bQJ3pw;gnss=gps;days=1;resolution=1","ubxlibhttp_0"
> 0:00:08:904 - 
> 0:00:08:944 - 
> 0:00:08:944 - 
> 0:00:08:944 - OK
> 0:00:08:944 - 
> 0:00:10:402 - +UUHTTPCR: 0,1,1
> 0:00:10:402 - HTTP GET response received, 2604 byte(s).
> 0:00:12:243 - AT+CCLK?
> 0:00:12:345 - 
> 0:00:12:385 - +CCLK: "23/11/09,10:19:15+04"
> 0:00:12:386 - 
> 0:00:12:386 - OK
> 0:00:12:386 - U_CELL_INFO: time is 23/11/09,10:19:15+04.
> 0:00:12:387 - U_CELL_INFO: local time is 1699525155, timezone offset 3600 seconds, hence UTC time is 1699521555.
> 0:00:12:388 - UTC timestamp according to cellular is 1699521555.
> 0:00:12:388 - Unable to download to the GNSS device (-5)!
> 0:00:12:388 - AT+ULSTFILE=0
> 0:00:12:490 - 
> 0:00:12:660 - 
> 0:00:12:660 - +ULSTFILE: "ubxlibhttp_0"
> 0:00:12:661 - 
> 0:00:12:661 - OK
> 0:00:12:662 - AT+UDELFILE="ubxlibhttp_0"
> 0:00:12:765 - 
> 0:00:12:765 - 
> 0:00:12:805 - OK
> ...

RobMeades, 2023-11-09:
>  the code change in `activateContext()` did fix it.

Fascinating: this code has been actively used on SARA-R422 for years, very difficult to understand why you should have a problem with it; maybe others have but have not reported it and fixed it themselves?

I suggested the change because we had determined that our Thingstream guys take the "blind AT+CGACT=" approach in their driver, but obviously it should not be necessary.  That said, I can't see a way it could do any harm (ignoring the fact that  I also can't see why it is necessary), so I will test the change here and if all is OK will push it here

RobMeades, 2023-11-10:
Commit cba2456bf025d18b2dd0af763fe21e1164ae0310 has the SARA-R422-force-CGACT code in it: in the end it had to be done just for SARA-R422 'cos SARA-R412M-02B/SARA-R410M-02B objected to the change.  Gonna close this one, please-reopen or start a new one if there is more to do on this.

closed 2023-11-10 by RobMeades