u-blox/ubxlib #164: sara-r412m-02b Unable to bring up the device!

alexmaron81, 2023-11-10:
Sometimes it happens that the module no longer works. If it doesn't work once, the next attempts will also be unsuccessful until the device is disconnected from the power supply.

AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
Opened device with return code -7.
Unable to bring up the device!

RobMeades, 2023-11-10:
Hi Alex: so is it the case that holding `RESET_N` low (e.g. by calling [uCellPwrResetHard()](https://github.com/u-blox/ubxlib/blob/cba2456bf025d18b2dd0af763fe21e1164ae0310/cell/api/u_cell_pwr.h#L446)) does not recover the device either?  From the [integration manual](https://content.u-blox.com/sites/default/files/SARA-R4_SysIntegrManual_UBX-16029218.pdf), section 1.6.3:

[image]

...and, [from the data sheet](https://content.u-blox.com/sites/default/files/SARA-R4_DataSheet_UBX-16024152.pdf), section 4.2.11:

[image]

`ubxlib` is aware of this timing and should hold the pin low for the correct amount of time.

alexmaron81, 2023-11-13:
Hi Rob,

yes the U_CFG_APP_PIN_CELL_RESET pin is defined for me and if "err = uDeviceOpen(&gDeviceCfg, &devHandle)" doesn't work, I call "uCellPwrResetHard(devHandle, U_CFG_APP_PIN_CELL_RESET);" on.
Unfortunately the same thing happens the next time you connect.
Can I check whether the "uCellPwrResetHard" is actually carried out?

alexmaron81, 2023-11-13:
Is the following way correct?

esp_err_t err = ESP_FAIL;

    uPortInit();
    uDeviceInit();

    err = uDeviceOpen(&gDeviceCfg, &devHandle);
    uPortLog("Opened device with return code %d.\n", err);

    if (err == 0)
    {
             // Connected
    }
    else
    {

           uCellPwrResetHard(devHandle, U_CFG_APP_PIN_CELL_RESET);
}

alexmaron81, 2023-11-13:
u_cell_pwr.c -> if ((pInstance != NULL) && (pinReset >= 0)) {

the "pInstance" is NULL

RobMeades, 2023-11-13:
Ah, sorry, I'd thought that in your application you would be using a module, so `uDeviceOpen()` would have returned a valid handle, you would find that the module had become unresponsive: then you would call `uCellPwrResetHard()`.

None of the APIs in `ubxlib` will work if the module was _already_ unresponsive at the point that `uDeviceOpen()` was called.  For that case you just need to drive the GPIO pin of your MCU that is connected to the `RESET_N` pin of the module low for the requisite amount of time, doing this directly using your platform APIs.  I believe you are using ESP-IDF so something like:

```
#include "driver/gpio.h"
#include "driver/rtc_io.h"

gpio_config_t config;
int32_t x;

#define MY_RESET_N_PIN (U_CFG_APP_PIN_CELL_RESET & 0xFF)

// Only required if MY_RESET_N_PIN is an RTC pin but harm in doing this in all cases
rtc_gpio_deinit(MY_RESET_N_PIN);

// Configure the pin
config.intr_type = GPIO_INTR_DISABLE;
config.mode = GPIO_MODE_OUTPUT_OD;
config.pull_down_en = GPIO_PULLDOWN_DISABLE;
config.pull_up_en = GPIO_PULLUP_DISABLE;
config.pin_bit_mask = 1ULL << MY_RESET_N_PIN;
x = gpio_config(&config);
if (x == ESP_OK) {
    // Set the pin low
    x =gpio_set_level((gpio_num_t) MY_RESET_N_PIN, 0); 
    if (x == ESP_OK) {
        // Wait for 15 seconds (only 10 required, being generous)
        vTaskDelay(15000/ portTICK_PERIOD_MS);
        // Set the lin high once more
        gpio_set_level((gpio_num_t) MY_RESET_N_PIN, 1);
    } else {
        printf("Unable to set level of pin %d (%d)!", MY_RESET_N_PIN, x);
    }
} else {
    printf("Unable to configure pin %d (%d)!", MY_RESET_N_PIN, x);
}
```

alexmaron81, 2023-11-13:
I'm relieved that it worked.
Thanks Rob!

closed 2023-11-13 by alexmaron81

RobMeades, 2023-11-13:
Good stuff!