u-blox/ubxlib #167: Module crashing after AT+CFUN=4?

arnoutdekimo, 2023-11-17:
Hi,

I've seen a few times now that after the module was unpowered, and I power it, and run ubxlib, that after it executes the CFUN=4 command, the module seems to crash?

[atcfun4reset.zip](https://github.com/u-blox/ubxlib/files/13391864/atcfun4reset.zip)
[consolelog.txt](https://github.com/u-blox/ubxlib/files/13391888/consolelog.txt)

I assume this is because somehow the uart RX line goes low for a long time.
Ubxlib then decides to toggle the pwron pin, but afterwards just keeps sending AT for a long time, before finally recovering

I'm guessing the root issue seems to be a crash of the module during AT+CFUN=4?

Kind regards,
Arnout

RobMeades, 2023-11-17:
When you say "the module crashes", I guess you are referring to about here in your console log:

[image]

If I've lined them up correctly, at the same point the Saleae log shows this:

[image]

So it looks like the OK did come back, it just didn't arrive at `ubxlib` for some reason.  Have I misunderstood?

RobMeades, 2023-11-17:
Ah, no I see the Saleae log starts later, hence what I'm seeing above is the second `AT+CFUN=4` not the first.

RobMeades, 2023-11-17:
So, just to line us up, the repeated `AT`s, which is `ubxlib` trying to get a response back from the module, are here:

[image]

...where, as you say, the RX line goes low for some time.

arnoutdekimo, 2023-11-17:
Ah..

The capture looks like this:

[image]
[image]

The RX line only goes low a few seconds after the last AT command. It does not match your screenshot

RobMeades, 2023-11-17:
Ah, yes, sorry, too many `.sal` files in too many places :-).

RobMeades, 2023-11-17:
I wondered, from the timing, if it could be UART power saving but you have `AT+UPSV=0` so it can't be that.  If you are not doing so already it is probably worth monitoring VINT, which may go low momentarily if the module is truly reset.  These modules usually tend to spit something out on the RXD line when they restart, but there's no sign of that in this case.

I'm afraid for this one I would have to refer you to our usual module support route: where did you source our module from?

arnoutdekimo, 2023-11-17:
Hi, 

I can indeed monitor the VINT pin.
I source the modules through the local distributor - batenburg. I can indeed get in contact with them directly as well regarding this issue

RobMeades, 2023-11-17:
Please try that and, if you get nowhere, ping me via e-mail and I'll find a way.

arnoutdekimo, 2023-11-24:
Presumably related to glitch on PWRON as well (https://github.com/u-blox/ubxlib/issues/163)

RobMeades, 2024-01-04:
Hi Arnout: did you get any resolution to this through your local distributor?  If you think there is still stuff to discuss, you know where to find me, otherwise I will close this now (please re-open it if you wish).

closed 2024-01-04 by RobMeades