u-blox/ubxlib #168: uSockRead blocks even if socket is closed

arnoutdekimo, 2023-11-17:
Hi,

Just a comment on the workings of the function uSockRead.
If we set a read timeout of 10s `(uSockOptionSet(sock, U_SOCK_OPT_LEVEL_SOCK, U_SOCK_OPT_RCVTIMEO),` and poll uSockRead in a loop, we expected indeed after 10s of no data received, that the functions returns.

However, if the socket is closed within that time, the module does report this:

> 0:00:37:859 - OK
> 0:00:37:859 - 
> 0:00:37:915 - +UUSOCL: 0
> 0:00:37:915 - 
> 0:00:37:915 - OK
> 0:00:37:916 - 
> 0:00:37:916 - +UUSOCL: 0
> 0:00:37:985 - AT+USORD=0,0
> 0:00:38:036 - 
> 0:00:38:073 - +CME ERROR: Operation not allowed
> 0:00:38:074 - 
> 0:00:38:074 - AT+USORD=0,0
> 0:00:38:191 - 
> 0:00:38:231 - +CME ERROR: Operation not allowed
> 0:00:38:232 - 
> 0:00:38:232 - AT+USORD=0,0
> 0:00:38:349 - 
> 0:00:38:389 - +CME ERROR: Operation not allowed
> 0:00:38:390 - 
> 0:00:38:390 - AT+USORD=0,0
> 0:00:38:507 - 

But ubxlib does not use this information and keeps blocking for the timeout setting. (loop in `receive `function)

Ideally, a posix socket would directly return in this case, indicating that the connection has closed.

It's possible to work around this behavior, but just letting it know that it is unexpected.

RobMeades, 2023-11-17:
Yes, I've wondered about capturing this and reporting it somehow, I will add it to the TODO list.

arnoutdekimo, 2023-11-17:
(FYI, my current workaround is using small socket timeouts (using my own big one externally) + `uSockRegisterCallbackClosed`, and letting it set a boolean, so I can check after `uSockRead ` returns -1 whether the socket is still alive or not. )

RobMeades, 2024-01-18:
Hi Arnout: I've pushed a preview branch of the change here:

https://github.com/u-blox/ubxlib/tree/preview_cell_sock_remote_close_rmea

Is that the kind of change you were expecting?  If so I'll push it to `master` here (then delete the preview branch).

arnoutdekimo, 2024-01-19:
Hi, yup, sounds like what was missing, thanks!

RobMeades, 2024-01-23:
The fix is now pushed to `master` here in commit c315b4ffad17e6581185e9234cbe86263ce90275; I will delete the preview branch.

closed 2024-01-23 by RobMeades