u-blox/ubxlib #172: URC bundled with other data - split after lock release?

arnoutdekimo, 2023-11-24:
Hi,

Perhaps more of a question: 

I am fetching data through a socket, and when the last piece of data comes in, the module reports the socket being closed immeidately afterwards:
[image]
[image]

Note how there is +- no delay between the URC and the last piece of data

A question - If the data is pushed into the code's ringbuffers, and then triggers and event `uPortEventQueueSendIrq`
=> That causes `urcCallback ` to be called, and attempting to check if there is an URC in.

BUT, the lock is then taken by the uSocketRead call, which wants to consume the data.

It will only consume the right amount of data (indicated in the +USORD: 0,988,"[c0]?[b, .... number), BUT

Now the real question:

- Does ubxlib, when the lock is released, again check whether there was remaining data in the buffer, and parse any remaining URC's that were in there (but were not yet parsed due to this lock)

Hopefully this is a much easier question for you, than for me to figure out..

Thank you for the help!

RobMeades, 2023-11-26:
In the case of a socket read we are in [uCellSockRead()](https://github.com/u-blox/ubxlib/blob/731256a47d18423b0819893c62de5678b6256e97/cell/src/u_cell_sock.c#L1669).  There are bytes pending and we will have sent `AT+USORD=blah` around here:

[image]

...then we read the response, the bit between the quotes, and then wait for the `\r\nOKr\n` on the end with `uAtClientResponseStop()`.

[image]

The AT Client is then unlocked.  In [uAtClientUnlock()](https://github.com/u-blox/ubxlib/blob/731256a47d18423b0819893c62de5678b6256e97/common/at_client/src/u_at_client.c#L3301) there is this bit of code:

[image]

...which checks if there is still stuff left in the buffer to process (in this case the `+UUSOCL: 0`) and, if there is it re-triggers UART event so the stuff _should_ get processed at that point.

That's the theory: not to say there isn't a bug of course, but that's how it is meant to work.

arnoutdekimo, 2023-11-27:
That's the perfect reply. Thanks! Super clear.

I'm still struggling with something in the socket that's going wrong, but I'll debug it further and make a more relevant post if it turns out to be an issue on the ubxlib side. I've finally got thread-aware debugging working on my stm, so hopefully that'll speed tings up

[image]

Thanks again!

closed 2023-11-27 by arnoutdekimo

RobMeades, 2023-11-27:
Very nice!  Always found getting the STM32Cube tools to play ball a bit of an uphill battle, but once there it is likely worth the effort.

arnoutdekimo, 2023-11-27:
Ok, major issue fixed on my end, causing very rare corruptions in UART data in the port folder of the lib.. That's gonna explain quite a few of those weird things (side from the issues due to the PWRON glitch thing).  Fingers  crossed, but I think this was a big one, and I see blue skies coming!
Anyway many thanks for all the support!

RobMeades, 2023-11-27:
Very well done on finding it!  Intermittent streamed-data-oriented things are awful to track down.

Blue skies on a crisp autumn morning are very welcome but, should the drizzle from a grey sky come back, don't be afraid to post an issue; at a minimum we can give you an alternative view of the problem.