u-blox/ubxlib #174: undefined reference to `k_thread_system_pool_assign'

mabroens, 2023-12-01:
Hi when building zephyr 3.4.0

I end up with:
**[build] C:/zephyr/repo/ubxlib/port/platform/zephyr/src/u_port.c:201: undefined reference to `k_thread_system_pool_assign'**

does any one have any suggestions on how to solve this?

RobMeades, 2023-12-01:
It looks like, somehow, your build is not picking up the `ubxlib` default configuration from [port/platform/zephyrdefault.conf](https://github.com/u-blox/ubxlib/blob/62b7a203af7f9c809b7cec784e4bca7ac31b7dc6/port/platform/zephyr/default.conf#L27); it is in that file that `CONFIG_HEAP_MEM_POOL_SIZE` is defined, which is required by Zephyr to bring in [k_thread_system_pool_assign()](https://github.com/zephyrproject-rtos/zephyr/blob/356c8cbe63ae01b3ab438382639d25bb418a0213/include/zephyr/kernel.h#L409C6-L409C32).

[That default.conf]([port/platform/zephyrdefault.conf](https://github.com/u-blox/ubxlib/blob/62b7a203af7f9c809b7cec784e4bca7ac31b7dc6/port/platform/zephyr/default.conf#L27)) should be brought in when you include `ubxlib` as a Zephyr module because the [ubxlib.cmake](https://github.com/u-blox/ubxlib/blob/62b7a203af7f9c809b7cec784e4bca7ac31b7dc6/zephyr/ubxlib.cmake#L7) file in the `ubxlib` [Zephyr module directory](https://github.com/u-blox/ubxlib/tree/master/zephyr) pulls it in.

mabroens, 2023-12-02:
Hi Rob, is theer someway that i can have the ubxlib locally in my oot project folder? i guess that makes configuration much easier and i prefer to have everything checked in in one local project folder/repo rather than doing an update or not having the proper verion over multiple machines.

RobMeades, 2023-12-02:
I'm not really sure: Zephyr is not something I build with myself I'm afraid.

[All that _should_ matter](https://github.com/u-blox/ubxlib/tree/master/zephyr#why-this-directory-is-here), from the point of view of `ubxlib`, is that you pull the `ubxlib` CMake file [zephyr/ubxlib.cmake](https://github.com/u-blox/ubxlib/blob/master/zephyr/ubxlib.cmake) into your application's `CMakeLists.txt`.

Provided you do that it shouldn't matter where the `ubxlib` directory is.

RobMeades, 2024-01-04:
Did you resolve your issue?

mabroens, 2024-01-11:
Hi Rob,

so sorry for the delay and thanks for the remainder.
i tested it today and yes after adding follwoing to my proj.conf i am now able to compile and start testing ubxlib.
```
CONFIG_HEAP_MEM_POOL_SIZE=4096
CONFIG_MINIMAL_LIBC_MALLOC=n
```

I am now as we type trying to implement the msg_main.c [https://github.com/u-blox/ubxlib/blob/master/example/gnss/msg_main.c](url)  where at the moment i get an error on: `if (uGnssMsgSend(devHandle, pBuffer, length) == length)` but i'm now debugging that.

Thanks so much for the pointer

Br Martijn

RobMeades, 2024-01-11:
Good stuff, glad to hear things are moving in a forwardly-kinda direction.  Please close this issue when you think appropriate; feeling free to raise more as you go of course.

closed 2024-01-11 by mabroens