u-blox/ubxlib #175: Socket read race condition with URC

arnoutdekimo, 2023-12-04:
Hi,

This is somewhat related to https://github.com/u-blox/ubxlib/issues/168, but in short, I am seeing the following race condition:

**Scenario**:
The scenario is that the following happens on AT:

[image]

Meaning that we write to a socket, and immeidiately afterwards, the server sends an URC with new data length + socket closes (because server has sent all the data)

**Issue**:
This results in the following race condition:

In `uCellSockRead`: 

[image]

The read code, first checks if the URC has executed, and has a length. 
But as you can see, during the lock, the URC parsing came in, and only then determined the length.

What's worse:

[image]

The SOCL URC is parsed immeidately afterwards. 

Which then executes, still while the other thread is yielding in atcallbacks

[image]

The result of this though, is that 

[image]

The socket handle is set to -1, causing the library to poll an invalid socket for data (AT+USORD=-1,0), and the socket API fails to deliver the data.

(FYI, I noticed this while testing on the LENA-R8, while the SARA-R4 did not show this behavior. Presumably because the reply timing is sutbly different)

Kind regards,
Arnout

arnoutdekimo, 2023-12-04:
Sigh, I quickly tried to put a return statement in `sockFree`, which then still allows the lib to poll the data, but..

[image]

The modem does not support reading from a socket that was already closed.

Meaning, the actual race condition is in the modem.. The MCU side must somehow read out the data faster than the modem receives the socket close from the server?

[image]

Yet on AT, this is even at the -same time-, meaning it is impossible to read out this data (unless the server does not close the connection immediately afterwards?)

RobMeades, 2023-12-04:
While I'm checking that, I _guess_ this is what the "shutdown" state of a TCP socket is for, but the AT interface does not export the shutdown state of a socket out as far as the MCU.  So, if you do NOT have control over the speed at which the remote server is closing the TCP connection, it might be that you must use PPP mode.

I can't recall which MCU platform you are using but, if it is ESP-IDF, there is a preview branch of our proposed integration with PPP mode, on ESP-IDF, which passes internal testing, here:

https://github.com/u-blox/ubxlib/tree/preview_cell_ppp_espidf_rmea

If you are using ESP-IDF and you fancy going that way, instructions on how to build it can be found here:

https://github.com/u-blox/ubxlib/tree/preview_cell_ppp_espidf_rmea/port/platform/esp-idf#ppp-level-integration-with-cellular

...and an example of using it can be found here:

https://github.com/u-blox/ubxlib/blob/8549e3eb5caffff4972b8d7a6880b02f0bd4ee2a/example/sockets/main_ppp_espidf.c#L189

Note that we've not release this yet [i.e. it remains subject to change, and further testing] as we want to get it working with Zephyr first, and Zephyr PPP support is still experimental (it seems to be fighting back).

RobMeades, 2023-12-04:
> If the module supports sockets over AT, I would expect them to be usable :)

Understood: I'm trying to determine how this is meant to work within the module.

arnoutdekimo, 2023-12-04:
Yes, I think so. I now swapped hardware so I can double check later if I swap back, but if you look at older posts I made, like https://github.com/u-blox/ubxlib/issues/165#issuecomment-1810151256

You can see in the logs

> 0:02:00:917 - 
> 0:02:00:918 - OK
> 0:02:00:918 - 
> 0:02:00:918 - +UUSORD: 0,136
> 0:02:00:919 - ATclientreadbytes-result 1024 bytes
> 0:02:00:919 - AT+USORD=0,136
> 0:02:01:033 - 
> 0:02:01:073 - 
> 0:02:01:073 - +USORD: 0,136,"[00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][bc][02][00] [00][00][00][00][01][00][00][00][00][00][00][00]J[00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00]J[00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00]"
> 0:02:01:079 - 
> 0:02:01:079 - OK
> 0:02:01:079 - ATclientreadbytes-READrequest 1 bytes
> 0:02:01:080 - ATclientreadbytes-result 1 bytes
> 0:02:01:080 - ATclientreadbytes-READrequest 136 bytes
> 0:02:01:080 - ATclientreadbytes-result 136 bytes
> 0:02:01:080 - AT+CSQ
> 0:02:01:182 - 
> 0:02:01:192 - +UUSOCL: 0

It shows how the SOCL indeed comes just after the last outstanding byte was read + the connection was closed by the remote in the meantime.

arnoutdekimo, 2023-12-04:
Yes, I think this is indeed a modem issue, and please do report it:) I already noticed something in your release notes that the LENA-R8 did not support the httplib correctly. I think now manually doing HTTP over sockets won't be that trivial either, at least when the server closes the socket upon replying.

But indeed, I also figured that waiting for the release was not going to be a fast path.
I think changing the server behavior is probably going to be path of least resistance. (But I still have to test that that works, as well as the SSL stack etc). But if I want to have something working asap, that'll probably be the way to go.

RobMeades, 2024-01-04:
Hi Arnout: I can confirm that this has been reported as a LENA-R8 bug and a fix has been made.  When a release with that fix in might be public I've no idea I'm afraid, it will depend on approval cycles and the like: would you like me to leave this open until then?

arnoutdekimo, 2024-01-08:
Thanks for the feedback!
In the meantime, I changed the server behavior to not autoclose the connection as a workaround. You don't have to keep the issue open just for me.

RobMeades, 2024-01-08:
OK: I will close this issue but, should I find out that a release is scheduled, I will update this issue anyway.

closed 2024-01-08 by RobMeades