u-blox/ubxlib #176: Question for assistnow example and LENA-R8

arnoutdekimo, 2023-12-13:
Hi,

I've been looking into getting the MGA-based offline/online assistnow data to the LENA-R8 through the ubxlib. So far no luck.
As far as I understand:

- In `uGnssAdd`, we check if `pInstance->transportType != U_GNSS_TRANSPORT_AT`
- Only then, are the ringbuffers created u`RingBufferCreateWithReadHandle(&(pInstance->ringBuffer) ..`
- In case of the LENA-R8, the transport type is AT for GNSS, CMUX is not used, so no ringbuffers are created

- During the actual `uGnssMsgPrivateReceiveStart` call (to put the assistnow data into the modem), it is attempted to take a ringbuffer `(uRingBufferTakeReadHandle(&(pInstance->ringBuffer));)`
- This fails because of the reasons above

So quick question: Is there a way that we can push the assitnow data manually (using the MGA lib) into the module?

By the way, I was actually initially just using the AT+UGPS commands manually, to let it do the download and pushing of the data to the modem itself internally, but for some reason this works considerably worse/slower than on the SARA-R4. Hence, why I quickly looked into pushing the data myself (what in the end is also just done by ucenter)

Thanks for your feedback!

RobMeades, 2023-12-13:
> By the way, I was actually initially just using the AT+UGPS commands manually, to let it do the download and pushing of the data to the modem itself internally, but for some reason this works [...]

Re-reading your post, I see that you've already tried what I was suggesting.

RobMeades, 2023-12-13:
Ugh.

Whatever we do here will be a complete hack I'm afraid.  The thing is that the `libMga` code was designed with the expectation of an asynchronous send/receive interface with the GNSS device, hence the same approach was taken when integrating it into `ubxlib`.  And an asynchronous send/receive world is exactly what we **don't** have when forced to use the essentially uni-directional, command/response-style, `AT+UGUBX` interface.

The hack (see branch including this below) is as follows:

- **only** use [uGnssMgaResponseSend()](https://github.com/u-blox/ubxlib/blob/e686051b70ca44c6321258215a9484f1f183859c/gnss/api/u_gnss_mga.h#L562) with flow control type `U_GNSS_MGA_FLOW_CONTROL_WAIT`; this sends the MGA messages one after another, not waiting for any acks.
- Then, in the implementation of [uGnssMgaResponseSend()](https://github.com/u-blox/ubxlib/blob/e686051b70ca44c6321258215a9484f1f183859c/gnss/src/u_gnss_mga.c#L752), comment out the [calls to uGnssMsgPrivateReceiveStart()](https://github.com/u-blox/ubxlib/blob/e686051b70ca44c6321258215a9484f1f183859c/gnss/src/u_gnss_mga.c#L816) (just set `errorCode` to 0 there) [and uGnssMsgPrivateReceiveStop()](https://github.com/u-blox/ubxlib/blob/e686051b70ca44c6321258215a9484f1f183859c/gnss/src/u_gnss_mga.c#L895) and modify the implementation of the `static` function [writeDeviceCallback()](https://github.com/u-blox/ubxlib/blob/e686051b70ca44c6321258215a9484f1f183859c/gnss/src/u_gnss_mga.c#L321) so that, rather than calling `GnssPrivateSendOnlyStreamRaw()`, it instead calls `sendReceiveUbxMessageAt()`, which is currently `static` over in `u_gnss_private.c`.
- You'll need to remove the `static` there and either `extern` `sendReceiveUbxMessageAt()` in `u_gnss_mga.c` or make a wrapper function for it which you can expose a prototype for in `u_gnss_private.h`).
- You will also need to move the [typedef for uGnssPrivateUbxReceiveMessage_t](https://github.com/u-blox/ubxlib/blob/e686051b70ca44c6321258215a9484f1f183859c/gnss/src/u_gnss_private.c#L128) out of `u_gnss_private.c` and into `u_gnss_private.h`.

`sendReceiveUbxMessageAt()` requires a non-NULL `pResponse` pointer but you can set the contents of the pointed-to-variable to be all zeroes, so something like this:

```
static void writeDeviceCallback(const void *pContext, const UBX_U1 *pData, UBX_I4 iSize)
{
    uGnssPrivateUbxReceiveMessage_t response = {0};
    uGnssPrivateInstance_t *pInstance = (uGnssPrivateInstance_t *) pContext;

    sendReceiveUbxMessageAt((const uAtClientHandle_t) pInstance->transportHandle.pAt,
                            (const char *) pData, iSize, &response, pInstance->timeoutMs,
                            pInstance->printUbxMessages);
}
```

I have made these changes on a branch which I have pushed here:

https://github.com/u-blox/ubxlib/tree/hack_hack_hack_for_arnout_nothing_to_see_here_rmea

...and run it on a SARA-R510M8S (which should be sufficiently similar for our purposes) and it appears to work, see logged output from running a hacked version of the [assist_now_main.c](https://github.com/u-blox/ubxlib/blob/hack_hack_hack_for_arnout_nothing_to_see_here_rmea/example/gnss/assist_now_main.c) example below (with suitably modified AssistNow authentication token, don't worry).

If this works for you, and means that LENA-R8001M10 becomes a usable beast, I can see if I can find some acceptable way to bring the hack into the `ubxlib` codebase.

```
20:12:36 ATE0
20:12:36 OK
20:12:36 AT+CMEE=2
20:12:36 OK
20:12:36 AT+UDCONF=1,0
20:12:36 OK
20:12:36 ATI9
20:12:36 03.15,A00.01
20:12:36 OK
20:12:36 AT&C1
20:12:36 OK
20:12:37 AT&D0
20:12:37 OK
20:12:37 AT&K3
20:12:37 OK
20:12:37 AT+UPSV=0
20:12:37 OK
20:12:37 AT+UPSMR=1
20:12:37 OK
20:12:37 AT+CPSMS?
20:12:37 +CPSMS: 0,,,"00010011","00000011"
20:12:37 OK
20:12:37 AT+UMNOPROF?
20:12:37 +UMNOPROF: 100
20:12:37 OK
20:12:37 AT+UGPRF?
20:12:37 +UGPRF: 18,0,""
20:12:37 OK
20:12:37 AT+CFUN=4
20:12:37 OK
20:12:37 Opened cellular device with return code 0.
20:12:37 Bringing up the cellular network...
20:12:37 U_CELL_NET: preparing to register/connect...
20:12:37 AT+CREG=2
20:12:37 OK
20:12:37 AT+CGREG=2
[... 254 lines cut ...]
20:12:51 OK
20:12:51 Download 93% complete.
20:12:51 AT+UGUBX="B56213204C0000001F00170C0D0CF40850FD6060C0BFECFF0FE4D95142F79B2BE3A3A65D631BB11308095927F1151F18970F5C4560E0BBE3D5DBB86DC8A39FB5DBFA204388C78D8567EF6822190D000000005AA9"
20:12:51 U_GNSS: sent UBX command b5 62 13 20 4c 00 00 00 1f 00 17 0c 0d 0c f4 08 50 fd 60 60 c0 bf ec ff 0f e4 d9 51 42 f7 9b 2b e3 a3 a6 5d 63 1b b1 13 08 09 59 27 f1 15 1f 18 97 0f 5c 45 60 e0 bb e3 d5 db b8 6d c8 a3 9f b5 db fa 20 43 88 c7 8d 85 67 ef 68 22 19 0d 00 00 00 00 5a a9.
20:12:51 +UGUBX: "no message"
20:12:51 OK
20:12:51 Download 96% complete.
20:12:51 AT+UGUBX="B56213204C0000002000170C0D0CF40850FD6060C0BFF7FE0FE4ADF2BE069B2BE323A65DE32247CE08096FF8D8031F18970F5C4560E0C934FF26A746E898440F4A2D5923C6EF261F44DE92081E2F0000000080E7"
20:12:51 U_GNSS: sent UBX command b5 62 13 20 4c 00 00 00 20 00 17 0c 0d 0c f4 08 50 fd 60 60 c0 bf f7 fe 0f e4 ad f2 be 06 9b 2b e3 23 a6 5d e3 22 47 ce 08 09 6f f8 d8 03 1f 18 97 0f 5c 45 60 e0 c9 34 ff 26 a7 46 e8 98 44 0f 4a 2d 59 23 c6 ef 26 1f 44 de 92 08 1e 2f 00 00 00 00 80 e7.
20:12:51 +UGUBX: "no message"
20:12:51 OK
20:12:51 Download 100% complete.
20:12:51 AssistNow data downloaded to GNSS.
20:12:51 AT+ULSTFILE=0
20:12:51 +ULSTFILE: "ubxlib_test_http_response","ubxlibhttp_0","ubxlibhttp_1"
20:12:51 OK
20:12:51 AT+UDELFILE="ubxlibhttp_0"
20:12:51 OK
20:12:51 Taking down GNSS network...
20:12:51 AT+UGPS=0
20:12:57 OK
20:12:57 +UUGIND: 8,0
20:12:57 +UUGIND: 1,0
20:12:57 Done.
```

arnoutdekimo, 2023-12-14:
Cool! Thanks Rob!

I quickly tried your code, and indeed, I manage to transfer the assistnow data to the lena-r8 now.

Unfortunately, is does not make the fix perform better than manually doing the AGPS=1,5 command. (that of course also does local aiding). 
I don't know why, but with the same GPS anteanna connected, the SARA-R4 takes only a few seconds from cold boot to lock using assistnow online, while the LENA-R8 takes easily 30s.a few minutes.  Hence I thought trying to push assistnow data directly .. but that does not make things faster. 

Even though it is interesting in trying to figure out how the assistnow works internally, and I can play with ucenter and manually try to upload assistnow data, I think I am just going to leave it up to the module, and for now have to live with the slower lock times on the lena-r8?

Anyway, many thanks for the branch! At least if I ever want to go back to manual control of the assitnow data, I can now use it this way, also on the lena!

RobMeades, 2023-12-14:
I think there's something wrong with the LENA-R8001M10 behaviour: the M10 module module it contains is our best yet, there is no way it should take 30 seconds or more (which is the usual cold-start time) when it has been given assistance information.  I will raise this as a bug internally.

If you have a copy of the branch now, is it OK if I delete it from here?  Thinking about it overnight, it wouldn't be _so_ bad if I modified the `ubxlib` code to allow the download of the assistance data over the AT interface, but LENA-R8001M10 is the only thing that this would be useful for and if this download method doesn't actually improve things over the "let the module do it by itself" approach, things are probably best left as they are.

I will get back to you with any feedback I receive over what LENA-R8001M10 is up to ref. assistance data.

arnoutdekimo, 2023-12-14:
Hi,

Ok thanks!
I have the branch locally, so feel free to delete it now.

Regarding the LENA, I think there's something wrong in how it internally handles the AGPS:

- First of all, it handles the UGSRV command differently than the SARA, and is not doing what the datasheet says it should
(See my post at https://portal.u-blox.com/s/question/0D5Oj000003LMyMKAW/lenar8001m10s-gps-assistnow-servers, although the reply I got isn't exactly very useful)
- Secondly, the internal assist now servers to be used are apparently -different- from the ones that can be used in ucenter/your lib code. AND differently from the ones supported by the sara.. 
- Thirdly, the UGIND indication 4,0 (saying the assistnow data is processed) either comes much later on the lena, .. or just never comes at all (see example snippet below)

So all these things combined, make the LENA-R8001M10 pretty inferior to the SARA-R4 for our application unfortunately.
Feel very much free to raise these things internally. My local FAE support isn't budging much (also due to illness/too much work I think), and the online forum isn't given much feedback either. I can raise a support case, but according to the local FAE, that will just circle back to them, and they say just to mail them (which isn't getting me far). I'm glad you're here though!

> 
> 0:00:20:038 - AT+UGPS=1,5,503
> 0:00:20:059 - 
> 0:00:20:109 - OK
> 0:00:20:110 - 
> 0:00:20:381 - +UUGIND: 1,0
> 0:00:20:382 - 
> 0:00:21:034 - +UUGIND: 0,1
> 0:00:21:035 - 
> 0:00:22:199 - +UUGIND: 0,503
> 0:00:22:200 - 
> 0:00:25:166 - +UUGIND: 1,0
> 0:00:25:167 
- Timeout waiting for assistnow result!
> 0:00:35:117 - 

(more than 10 seconds for the reply, while this is usually very fast on the sara)

RobMeades, 2023-12-14:
Sorry about this; internal bug raised, Jira number `LS_EL_R80-703` if you need to wield it in any conversations with u-blox support in general.  But, as indicated before, unless there's an AT-level workaround this is unlikely to be something where any fix will come around quickly, as is the nature of cellular and, who knows, we may get CMUX in it anyway.

closed 2024-11-11 by manemannen