u-blox/ubxlib #178: MCC/MNC list

alexmaron81, 2023-12-18:
I would like to use the module in Thailand and have received the following list of providers from my operator.
What is the smartest way to test the dial-in until the module has dialed in successfully?

True Move Company Ltd. (MCC: 520, MNC: 00, 04, 99):
2G: 900 und 1800 MHz
3G: Band 1 und 5
4G: Band 1, 3, 8 und 41

Advanced Wireless Network Company (AIS) (MCC: 520, MNC: 01, 03) 
2G: 900 and 1800 MHz
3G: Band 1 and 8
4G: Band 1, 3, 8 und 41

True Move H Universal Communication  (MCC: 520, MNC: 04)
2G: 900 und 1800 MHz
3G: Band 1
4G: Band 1, 3, 8 und 41

dtac TriNet Co., Ltd. (MCC: 520, MNC: 05)
3G: Band 1 und 5
4G: Band 1, 3, 28 und 40

RobMeades, 2023-12-19:
Hi Alex.  The simplest approach would be to call `uDeviceOpen()` on the cellular device as normal and then, rather than calling `uNetworkInterfaceUp()` (which uses automatic network selection), you would call [uCellNetConnect()](https://github.com/u-blox/ubxlib_priv/blob/959a531341c511ea62b2964aea03378726a4e803/cell/api/u_cell_net.h#L316) with the `pMccMnc` field set to point to the string giving the MCC/MNC of one of the networks (e.g. "52000" for the first TrueMove Company Ltd MCC/MNC), wait for the connection to succeed or fail, if it succeeds then call [uCellNetDisconnect()](https://github.com/u-blox/ubxlib_priv/blob/959a531341c511ea62b2964aea03378726a4e803/cell/api/u_cell_net.h#L500), then repeat for the next MCC/MNC (e.g. "52004" for the next True Move Company Ltd. MCC/MNC).

It might be worth doing a network scan first to see what networks are actually available at that location (see the [lte_cfg_main.c](https://github.com/u-blox/ubxlib_priv/blob/959a531341c511ea62b2964aea03378726a4e803/example/cell/lte_cfg/lte_cfg_main.c#L415) for how to do that, which also shows you how to configure the supported network bands, should you need it) in order to avoid wasting time trying to connect to networks which just aren't visible.

Since both `uCellNetScanGetFirst()` and `uCellNetScanGetNext()` return the MCC/MNC string anyway, you could just try each one in turn.

alexmaron81, 2023-12-20:
Hi Rob, I tried the lte_cfg_main.c example and get this:

AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK

+CEDRXP: 4,"0010"
U_CELL_CFG: getting band mask for RAT 10 (in module terms 0).
AT+UBANDMASK?

+UBANDMASK: 0,524420,1,524416

OK
U_CELL_CFG: band mask for RAT 10 (in module terms 7) is 0x0000000000000000 0000000000080084.
### Band mask for RAT CAT-M1 is 0x0000000000000000 0000000000080084.
### Setting band mask for RAT CAT-M1 to 0x0000000000000000 000000400b0f189f...
I (28269) CELL: >>>>>>>>>>SARA is not alive<<<<<<<<<<
U_CELL_CFG: setting band mask for RAT 10 (in module terms 0) to 0x0000000000000000 000000400b0f189f.
AT+UBANDMASK=0,275063445663,0

ERROR
### Unable to change band mask for RAT CAT-M1, it is likely your module does not support one of those bands.
U_CELL_CFG: getting band mask for RAT 10 (in module terms 0).
AT+UBANDMASK?

+UBANDMASK: 0,524420,1,524416

OK
U_CELL_CFG: band mask for RAT 10 (in module terms 7) is 0x0000000000000000 0000000000080084.
### Band mask for RAT CAT-M1 is 0x0000000000000000 0000000000080084.
AT+CFUN?

+CFUN: 0

OK
AT+CFUN=1

OK

+CEDRXP: 4,"0010","0010","0001"
AT+COPS=?

+COPS: (1,"262 01","262 01","26201",0),(2,"262 01","262 01","26201",7),(1,"262 03","262 03","26203",0),(1,"262 02","262 02","26202",0),(1,"262 02","262 02","26202",7),(1,"262 03","262 03","26203",7),,(0,1,2,3,4),(0,1,2)

OK
AT

OK
AT+CFUN=0

OK

+CEDRXP: 4,"0010"
### 6: network: 26201
I (63329) CELL: >>>>>>>>>>SARA is not alive<<<<<<<<<<
### 4: network: 26201
### 3: network: 26203
### 2: network: 26202
### 1: network: 26202
### 0: network: 26203
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK

RobMeades, 2023-12-20:
Different RATs I think:

`(1,"262 01","262 01","26201",0), (2,"262 01","262 01","26201",7)`
`                             ^                                ^`
`                            2G                               LTE`

RobMeades, 2024-01-04:
Hi Alex: are you OK with this now?

eeFLis, 2024-01-08:
Hi Rob
Sorry for the intermediate question but why not add `pMccMnc ` to `uNetworkCfgCell_t `so that `uNetworkInterfaceUp()` can be used? This way `pMccMnc ` can be passed to uCellNetConnect in `uNetworkPrivateChangeStateCell()`.

RobMeades, 2024-01-08:
Hi @eeFLis and happy new year to you:  that's an excellent idea, I will add it to the TODO list.

alexmaron81, 2024-01-08:
> Hi Alex: are you OK with this now?

Hi Rob, I'll tackle the task a little later. I'm currently working on another problem (not ubxlib).

eeFLis, 2024-01-08:
> Hi @eeFLis and happy new year to you: that's an excellent idea, I will add it to the TODO list.

Thank you, a happy new year to you too

RobMeades, 2024-01-10:
@eeFLis: now done in commit https://github.com/u-blox/ubxlib/commit/41349f324ed52de37ade97f8de638b4ab1549f9a.

RobMeades, 2024-02-22:
> Hi Rob, I'll tackle the task a little later. I'm currently working on another problem (not ubxlib).

Hi Alex: do you still need this one open?

RobMeades, 2024-03-05:
I'm going to assume this can be closed, please re-open (or open another issue) if there is more to discuss.

closed 2024-03-05 by RobMeades