u-blox/ubxlib #179: HTTP connectivity issue

jamal-mouline, 2023-12-18:
Test environment

we have a uBlox module mounted on a SparkFun board
https://www.sparkfun.com/products/14997

which itself is mounted on a nRF54840 devKit using the Arduino Header

we are able to successfully power up and register on the network
run some MQTT exchange with the test.mosquitto.org broker

but we are having issues with the HTTP test as provided with the example under
https://github.com/u-blox/ubxlib/blob/master/example/http_client/http_main.c
unable to create an HTTP instance (either secure or non-secure) using the **pUHttpClientOpen**() API

we are using a hologram SIM card with a NULL APN

copy / paste of output log below

uart:~$ modem_ctrl  power_up
Cell Modem Power Up
uart:~$ U_CELL: initialising with enable power pin not connected, PWR_ON pin 38 (0x26) (and is toggled from 1 to 0) and VInt pin not connected.
AT
AT
AT
AT

[00:00:11.093,505] <inf> CellModem: Opening a modem device instance...

[00:00:11.093,536] <inf> CellModem: [CM] MODEM_POWERED_DOWN->MODEM_POWERING_UP
uart:~$ AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT

OK
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

L0.0.00.00.05.12,A.02.19

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K0

OK
AT+UPSV=0

OK
AT+CPSMS?

+CPSMS:0,,,"00011000","00001010"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 0

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
AT+CGSN

350857124811117

OK
U_CELL_INFO: IMEI is 350857124811117.
[00:00:24.261,718] <inf> CellModem: Modem power up and ready to be configured

[00:00:24.261,779] <inf> CellModem: [CM] MODEM_POWERING_UP->MODEM_POWERED_UP
[00:00:24.412,078] <inf> CellModem: subTopic messages/350857124811117/devicebound/

[00:00:24.412,170] <inf> CellModem: httpPath /350857124811117.html

uart:~$ modem_ctrl  network_up
U_CELL_NET: preparing to register/connect...
Cell Modem Register
uart:~$ AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

234500007895849

OK
AT+CIMI

234500007895849

OK
U_CELL_NET: APN from database is "jtm2m".
AT+CGDCONT=1,"IP","jtm2m"

OK
AT+UAUTHREQ=1,0

OK
U_CELL_NET: setting automatic network selection mode...

[00:00:43.949,493] <inf> CellModem: Bringing up the network interface...

uart:~$ AT+COPS?

+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
-1: NReg

+CREG: 2

+CGREG: 2
-1: Search
-1: Search

+CEREG: 2,,,,,,,"00000000","01100000"
-1: Search
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,2

OK
-1: Search
AT+CREG?

+CREG: 5,"FFFE","105950D", 7

+CGREG: 5,"FFFE","105950D",7

+CEREG: 5,"7BDE","0","105950D",7,,,"00000000","01100000"

+CREG: 2,5,"FFFE","105950D",7

OK
3: RegR
7: RegR
7: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"T-Mobile Hologram",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
U_CELL_NET: connected after 2 second(s).
[00:00:47.338,745] <inf> CellModem: [CM] MODEM_POWERED_UP->MODEM_NETWORK_REGISTERED
uart:~$ modem_ctrl  http_test
Http test
uart:~$ AT+UHTTP=0

OK
AT+UHTTP=0,1,"ubxlib.com"

OK
AT+UHTTP=0,5,8080

OK
AT+UHTTP=0,4,0

OK
**AT+UHTTP=0,7,0
+CME ERROR: Operation not supported**

[00:01:06.278,930] <err> CellModem: Failed to create HTTP  instance!

RobMeades, 2023-12-19:
Hi there, sorry to hear your having problems.  The command `AT+UHTTP=0,7,0` should be supported by all of our module types:

[image]

...so my guess is that the SARA-R410M-02B module you have has a really old FW version (`L0.0.00.00.05.12,A.02.19`).

The best way forward would be to update it.

Take a look in the "Firmware Update" section of the [SARA-R4 pages on our website](https://www.u-blox.com/en/product/sara-r4-series?legacy=Current#Documentation-&-resources), and you should find the [SARA_R410M-02B one](https://content.u-blox.com/sites/default/files/2023-06/SARA-R410M-02B-04-P1-L0000000512A0221-000K00.zip) entry.  In that `.zip` file you will find a `.dof` file, which is the FW image, and a `.pdf` file.  Towards the end of the `.pdf` file you will find a link to the tool EasyFlash 13.03, which is the tool you use to perform the update (over the USB port of the SARA-R410-02B module).

Hopefully all of that will bring the module forward to a state where it will accept `AT+UHTTP=0,7,0`.  Let us know how that goes.

jamal-mouline, 2024-01-23:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hello again,
I was able to resolve the original problem with HTTP connectivity configuring the timeout to 180
So
AT+UHTTP=0,7,180
Did the job

My next task was to integrate the u-Blox Thingstream SDK
Where I encountered other issues, that I thought were related to the old firmware revision I had on my SARA-R4

(L0.0.00.00.05.12,A.02.19).
Then I took additional steps to upgrade to L0.0.00.00.05.12,A.02.21

Which I believe is the latest available

After upgrading the module firmware, I can still power up by using the uDeviceOpen() API from ubxlib

But I seem to have lost network registration after invoking uNetworkInterfaceUp()

See log output from ubxlib copy/paste below

Please advise how to proceed to restore successful registration

I am using a Hologram SIM. Setting APN to hologram. Also tried with NULL APN. But no registration success within the allocated 4 minute timer.

Thank you and best regards,

uart:~$ modem_ctrl   power_up
Cell Modem Power Up
uart:~$ U_CELL: initialising with enable power pin not connected, PWR_ON pin 38 (0x26) (and is toggled from 1 to 0) and VInt pin not connected.
AT
AT
AT
AT
[00:00:10.914,550] <dbg> CellModem: CellModemThread: _msgq_get message 1 field 200

[00:00:10.914,581] <inf> CellModem: Opening a modem device instance...

[00:00:10.914,611] <inf> CellModem: [CM] MODEM_POWERED_DOWN->MODEM_POWERING_UP
uart:~$ AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

L0.0.00.00.05.12,A.02.21

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K0

OK
AT+UPSV=0

OK
AT+CPSMS?

+CPSMS:0,,,"00011000","00001010"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 0

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
AT+CGSN

350857124811117

OK
U_CELL_INFO: IMEI is 350857124811117.
[00:00:22.527,343] <inf> CellModem: Modem power up and ready to be configured

[00:00:22.527,404] <inf> CellModem: [CM] MODEM_POWERING_UP->MODEM_POWERED_UP

uart:~$ modem_ctrl   network_up
U_CELL_NET: preparing to register/connect...
Cell Modem Register
uart:~$ AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

234500007895849

OK
U_CELL_NET: user-specified APN is "hologram".
AT+CGDCONT=1,"IP","hologram"

OK
AT+UAUTHREQ=1,0

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK

uart:~$ AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK

image.png (view on web)<https://github.com/u-blox/ubxlib/assets/6726745/f43dbe09-ef18-4a79-b0f8-5470c8ca8976>

CONFIDENTIALITY. This electronic mail and any files transmitted with it may contain information proprietary to Exponential Technology Group or one of its subsidiaries or affiliates, and are intended solely for the use of the individual or entity to whom they are addressed, shall be maintained in confidence and not disclosed to third parties without the written consent of the sender. If you are not the intended recipient or the person responsible for delivering the electronic mail to the intended recipient, be advised that you have received this electronic mail in error and that any use, dissemination, forwarding, printing, or copying of this electronic mail is strictly prohibited. If you have received this electronic mail in error, please immediately notify the sender by return mail. Thank you for your compliance.

RobMeades, 2024-01-24:
Hi again: the _usual_ reason for this is that the narrow-band (NB) RAT is active in SARA-R422; an initial search of all of the bands of this RAT, i.e. the kind of search that is performed when there is no previous registration information to use, for 3GPP reasons, can take up to 20 minutes.  We have [an example](https://github.com/u-blox/ubxlib/blob/bb93cdb2cfea08f6752e5aabe77a1733ebf5bbcf/example/cell/lte_cfg/lte_cfg_main.c#L358) which shows how you would read out the currently active RATs and modify them as you wish, or for a quick fix you could just call [uCellCfgSetRat(devHandle, U_CELL_NET_RAT_CATM1 )](https://github.com/u-blox/ubxlib/blob/bb93cdb2cfea08f6752e5aabe77a1733ebf5bbcf/cell/api/u_cell_cfg.h#L185) (as you seemed to be registered on Cat M1 in your earlier log).

jamal-mouline, 2024-01-24:
Hello and thanks for the feedback

I went ahead and integrated the piece of code in the linked example
Starting from U_PORT_TEST_FUNCTION("[example]", "exampleCellLteCfg")
Up to the point where the registration function is called uNetworkInterfaceUp()

Looks like my SARA-R410M does not support setting MNO profile?!? –see error copy/paste below

Thoughts?

I will go ahead and trim the MNO profile setting, just invoke the RAT setting, and see what happens
Will keep you posted here…
Thank you

Cell Modem Register
### This module does not support setting an MNO profile.
### unable to read bandmask!

RobMeades, 2024-01-24:
> This module does not support setting an MNO profile.

That's a bit weird: SARA-R410M does support MNO profile setting, not sure why you are getting that.  The segment of log you have posted above doesn't include any AT commands: by default (i.e. unless you have switched them off) the commands _should_ be being printed to the log.  I guess that the module is powered up at this point, i.e. you `uDeviceOpen()` has returned success (0)?

RobMeades, 2024-01-24:
Ah, it might be because you are connected, you have called `uNetworkInterfaceUp()` already - once you are connected you can't modify radio-related stuff.  Try putting the code _before_ the `uNetworkInterfaceUp()`.

jamal-mouline, 2024-01-24:
To your question,
I have included the configuration section BEFORE calling on uNetworkInterfaceUp()

I might have had an issue with the deviceHandler I was passing to those routines
After fixing that, here is a complete
At a high level,
1-the module power up completes successfully after calling uDeviceOpen
2-device configuration using the code snippet recommended this morning
3-registration attempt by calling on uNetworkInterfaceUp

U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

L0.0.00.00.05.12,A.02.21

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K0

OK
AT+UPSV=0

OK
AT+CPSMS?

+CPSMS:0,,,"00011000","00001010"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 0

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
AT+CGSN

350857124811117

OK
U_CELL_INFO: IMEI is 350857124811117.
[00:00:58.329,498] <inf> CellModem: Modem power up and ready to be configured

[00:00:58.329,528] <inf> CellModem: [CM] MODEM_POWERING_UP->MODEM_POWERED_UP

uart:~$ modem_ctrl network_up
Cell Modem Register
uart:~$ AT+UMNOPROF?

+UMNOPROF: 0

OK
U_CELL_CFG: MNO profile is 0.
AT+UMNOPROF=90

+CME ERROR: Operation not supported

U_CELL_CFG: unable to set MNO profile to 90.
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
### RAT[0] is unknown or not used.
### Setting RAT[0] to CAT-M1...
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
U_CELL_CFG: setting the RAT at rank 0 to 10 (in module terms 7).
U_CELL_CFG: RATs (removing duplicates) become:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
AT+URAT=7

+CME ERROR: Operation not supported

### Unable to set RAT[0] to CAT-M1.
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
### RAT[1] is unknown or not used.
### Unable to read RAT uCellCfgGetRat() return -5
U_CELL_CFG: getting band mask for RAT 10 (in module terms 0).
AT+UBANDMASK?
+CME ERROR: Operation not supported

### unable to read bandmask!
AT+CFUN?

+CFUN: 4

OK
AT+CFUN=1

OK
AT+COPS=?

+COPS: (1,"Verizon","Verizon","311480",7),(1,"T-Mobile","T-Mobile","310260",7),(1,"313 100","313 100","313100",7),(1,"AT&T","AT                                                                      &T","310410",7),(1,"Verizon","Verizon","311480",9),(1,"AT&T","AT&T","310410",9),(1,"T-Mobile","T-Mobile","310260",9),,(0,1,2,3,                                                                      4),(0,1,2)

OK
AT+CFUN=4

OK
### 7: network: 311480
### 5: network: 310260
### 4: network: 313100
### 3: network: 310410
### 2: network: 311480
### 1: network: 310410
### 0: network: 310260
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

234500007895849

OK
U_CELL_NET: user-specified APN is "hologram".
AT+CGDCONT=1,"IP","hologram"

OK
AT+UAUTHREQ=1,0

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?
+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK

RobMeades, 2024-01-24:
Something to try: I think we need to get the module out of MNO profile 0.  That particular module doesn't support MNO Profile 90 (later modules do), so can you try setting MNO profile 3 instead (which is Verizon)?  That might unlock it.

RobMeades, 2024-01-24:
...or, since you appeared to have been on T-Mobile before, you could use MNO profile 5, which is T-Mobile.

jamal-mouline, 2024-01-24:
I tried 2 instead –because I know we have good AT&T coverage in the building

Log below for the configuration set up
The key problem I see from the log is that URAT (both query and setting to 7 == CAT-M1) is not supported either?!?

Am I just unlucky here pursuing this proof of concept on the wrong type of modem?

Thanks,

uart:~$ modem_ctrl      network_up
Cell Modem Register
uart:~$ AT+UMNOPROF?

+UMNOPROF: 0

OK
U_CELL_CFG: MNO profile is 0.
AT+UMNOPROF=2

OK
U_CELL_CFG: MNO profile set to 2.
### MNO profile has been changed from 0 to 2.
### Re-booting the module to apply MNO profile change...
U_CELL_PWR: rebooting.
AT+CFUN=15

OK
[00:00:31.548,767] <dbg> CellModem: CellModemThread: _msgq_get message 3 field 300

uart:~$ AT
AT
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

L0.0.00.00.05.12,A.02.21

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K0

OK
AT+UPSV=0

OK
AT+CPSMS?

+CPSMS:0,,,"00011000","00001010"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 2

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
### RAT[0] is unknown or not used.
### Setting RAT[0] to CAT-M1...
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
U_CELL_CFG: setting the RAT at rank 0 to 10 (in module terms 7).
U_CELL_CFG: RATs (removing duplicates) become:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
AT+URAT=7

+CME ERROR: Operation not supported

### Unable to set RAT[0] to CAT-M1.
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
### RAT[1] is unknown or not used.
### Unable to read RAT uCellCfgGetRat() return -5
U_CELL_CFG: getting band mask for RAT 10 (in module terms 0).
AT+UBANDMASK?

+CME ERROR: Operation not supported

### unable to read bandmask!
AT+CFUN?

+CFUN: 4

OK
AT+CFUN=1

OK
AT+COPS=?

+COPS: (1,"T-Mobile","T-Mobile","310260",7),(1,"AT&T","AT&T","310410",7),(1,"313 100","313 100","313100",7),,(0,1,2,3,4),(0,1,2)

OK
AT+CFUN=4

OK
### 3: network: 310260
### 1: network: 310410
### 0: network: 313100
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

234500007895849

OK
U_CELL_NET: user-specified APN is "hologram".
AT+CGDCONT=1,"IP","hologram"

OK
AT+UAUTHREQ=1,0

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?
[00:00:56.093,200] <inf> CellModem: Bringing up the network interface...

uart:~$
+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK

jamal-mouline, 2024-01-24:
OK will try 5 next
In between, could you comment on the URAT being Not Supported

RobMeades, 2024-01-24:
Actually, no need 5, is fine, anything other than 0.

The thing I don't understand is that the module is returning not supported for things I know very well it supports:

```
AT+URAT?
+CME ERROR: Operation not supported
AT+UBANDMASK?
+CME ERROR: Operation not supported
```

I am enquiring with relevant support people internally as to how this is possible.

RobMeades, 2024-01-24:
Ah, checking the details, it looks as though, on the AT&T MNO profile, the `AT+URAT` and `AT+UBANDMASK` commands are locked:

[image]

However, that should not be a problem as the RAT is also locked to be LTE Cat M1 only, and in the list of networks you get back:

```
+COPS: (1,"T-Mobile","T-Mobile","310260",7),(1,"AT&T","AT&T","310410",7),(1,"313 100","313 100","313100",7),,(0,1,2,3,4),(0,1,2)
```

...all of them are Cat-M1 (the 7 on the end).  All of which doesn't explain why the module is not registering.  Let me think...

jamal-mouline, 2024-01-24:
Hold it
After multiple attempts, looks like I am registering again with the Hologram SIM –see fresh log capture below…
Stayed with profile 2

Ultimately my goal is to register with a THINGSTREAM SIM card
Do you have any specific recommendations for the lucky profile to use on this module?
Shall I remove the configuration code for URAT and band mask?

Thanks,

uart:~$ modem_ctrl  power_up
Cell Modem Power Up
uart:~$ U_CELL: initialising with enable power pin not connected, PWR_ON pin 38 (0x26) (and is toggled from 1 to 0) and VInt pin not connected.
AT
AT
AT
AT
[00:00:21.818,664] <dbg> CellModem: CellModemThread: _msgq_get message 1 field 200

[00:00:21.818,695] <inf> CellModem: Opening a modem device instance...

[00:00:21.818,756] <inf> CellModem: [CM] MODEM_POWERED_DOWN->MODEM_POWERING_UP
uart:~$ AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT

OK
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

L0.0.00.00.05.12,A.02.21

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K0

OK
AT+UPSV=0

OK
AT+CPSMS?

+CPSMS:0,,,"00011000","00001010"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 2

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
AT+CGSN

350857124811117

OK
U_CELL_INFO: IMEI is 350857124811117.
[00:00:33.818,084] <inf> CellModem: Modem power up and ready to be configured

[00:00:33.818,145] <inf> CellModem: [CM] MODEM_POWERING_UP->MODEM_POWERED_UP
[00:00:33.968,170] <inf> CellModem: subTopic messages/350857124811117/devicebound/

[00:00:33.968,231] <inf> CellModem: httpPath /350857124811117.html

uart:~$ modem_ctrl  network_up
Cell Modem Register
uart:~$ AT+UMNOPROF?

+UMNOPROF: 2

OK
U_CELL_CFG: MNO profile is 2.
### The MNO profile is already set to 2.
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
### RAT[0] is unknown or not used.
### Setting RAT[0] to CAT-M1...
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
U_CELL_CFG: setting the RAT at rank 0 to 10 (in module terms 7).
U_CELL_CFG: RATs (removing duplicates) become:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
AT+URAT=7

+CME ERROR: Operation not supported

### Unable to set RAT[0] to CAT-M1.
AT+URAT?

+CME ERROR: Operation not supported

U_CELL_CFG: RATs are:
  rank[0]: 0 (in module terms -1).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
### RAT[1] is unknown or not used.
### Unable to read RAT uCellCfgGetRat() return -5
U_CELL_CFG: getting band mask for RAT 10 (in module terms 0).
AT+UBANDMASK?

+CME ERROR: Operation not supported

### unable to read bandmask!
AT+CFUN?
[00:00:46.754,791] <dbg> CellModem: CellModemThread: _msgq_get message 3 field 300

uart:~$
+CFUN: 4

OK
AT+CFUN=1

OK
AT+COPS=?

+COPS: (1,"313 100","313 100","313100",7),(1,"T-Mobile","T-Mobile","310260",7),(1,"AT&T","AT&T","310410",7),,(0,1,2,3,4),(0,1,2)

OK
AT+CFUN=4

OK
### 3: network: 313100
### 1: network: 310260
### 0: network: 310410
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

234500007895849

OK
U_CELL_NET: user-specified APN is "hologram".
AT+CGDCONT=1,"IP","hologram"

OK
AT+UAUTHREQ=1,0

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?
[00:01:04.791,656] <inf> CellModem: Bringing up the network interface...

uart:~$
+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
-1: NReg

+CREG: 2

+CGREG: 2

-1: Search
-1: Search
+CEREG: 2,,,,,,,,
-1: Search
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,2

OK
-1: Search

+CREG: 5,"FFFE","BB69E0F", 7

+CGREG: 5,"FFFE","BB69E0F",7

+CEREG: 5,"270B","BB69E0F",7,,,,
0: RegR
3: RegR
8: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"AT&T Hologram",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1
+CGACT: 2,0

OK
U_CELL_NET: connected after 2 second(s).
[00:01:07.678,771] <inf> CellModem: [CM] MODEM_POWERED_UP->MODEM_NETWORK_REGISTERED
uart:~$

image.png (view on web)<https://github.com/u-blox/ubxlib/assets/6726745/44b207e4-25cc-4e66-908a-33f27d1bfbec>

RobMeades, 2024-01-24:
Good stuff: it might be just that, since the FW was updated, the module has ended up doing an "initial" search, with no prior knowledge of the carriers (because the previous network had been erased), a process which is mandated to behave in such a way that it will take longer.

Hopefully, now that it has registered, it should register again more quickly.

If you are going to stick with one of the MNO profiles that has locked the `AT+URAT` and `AT+UBANDMASK` commands (see appendix C.6 of the [AT manual](https://content.u-blox.com/sites/default/files/documents/LEXI-R4-SARA-R4_ATCommands_UBX-17003787.pdf) then yes, remove those commands.  Of course, you also don't need those commands to do normal stuff, it was just useful to explore the module settings with them.  If you move to an MNO profile that unlocks those two commands (e.g. MNO profile 100, which is Europe), then you could use them if you wish.

I'd say that if you are happy with AT&T then leave it on that MNO profile.

jamal-mouline, 2024-01-29:
Re: Hopefully, now that it has registered, it should register again more quickly.
>>> yes that it correct, while staying with MNO profile 2, future registration are few seconds away

Re: If you move to an MNO profile that unlocks those two commands (e.g. MNO profile 100, which is Europe), then you could use them if you wish.
Yes, I got a strong recommendation from a u-Blox FAE to switch to profile 100 for roaming SIMs
With that, I had found couple issues with ubxlib

  1.  I had set MY_RAT0 and disabled the secondary and tertiary ones

Yet, the URAT? Query returns 7,8

Is there a more restricted setting to URAT=7 only (no NB-IOT)

Please see log attached from executing sample code from https://github.com/u-blox/ubxlib/blob/master/example/cell/lte_cfg/lte_cfg_main.c

Thank you

// The RATs you want the module to use, in priority order.
// Set the value of MY_RAT0 to the RAT you want to use
// first (see the definition of uCellNetRat_t in cell/api/u_cell_net.h
// for the possibilities); for SARA-R41x you might chose U_CELL_NET_RAT_CATM1,
// for SARA-R412M you might chose U_CELL_NET_RAT_CATM1 or
// U_CELL_NET_RAT_GSM_GPRS_EGPRS and for SARA-R5 you might
// chose U_CELL_NET_RAT_CATM1.
// If your module supports more than one RAT at the same time
// (consult the data sheet for your module to find out how many
// it supports at the same time), add secondary and tertiary
// RATs by setting the values for MY_RAT1 and MY_RAT2 as required.
#ifndef MY_RAT0
#define MY_RAT0 U_CELL_NET_RAT_CATM1
#endif
#ifndef MY_RAT1
#define MY_RAT1 U_CELL_NET_RAT_UNKNOWN_OR_NOT_USED
#endif
#ifndef MY_RAT2
#define MY_RAT2 U_CELL_NET_RAT_UNKNOWN_OR_NOT_USED
#endif

AT+UMNOPROF=100

OK
U_CELL_CFG: MNO profile set to 100.
### MNO profile has been changed from 2 to 100.

+CEDRXS: 2,"0010"
+CEDRXS: 4,"0010"
+CEDRXS: 5,"0010"

OK
AT+CEDRXS=2,2

OK
AT+CEDRXS=2,4

OK
AT+CEDRXS=2,5

+UMNOPROF: 100

+URAT: 7,8

OK
U_CELL_CFG: RATs are:
  rank[1]: 11 (in module terms 8).
### RAT[0] is CAT-M1.

+URAT: 7,8

OK
U_CELL_CFG: RATs are:
  rank[1]: 11 (in module terms 8).
### RAT[1] is NB1.
### Setting RAT[1] to unknown or not used...

+URAT: 7,8

OK
U_CELL_CFG: RATs are:
  rank[1]: 11 (in module terms 8).

+URAT: 7,8

OK
U_CELL_CFG: RATs are:
  rank[1]: 11 (in module terms 8).

+URAT: 7,8

OK
U_CELL_CFG: RATs are:
  rank[1]: 11 (in module terms 8).
U_CELL_CFG: setting the RAT at rank 1 to 0 (in module terms -1).

OK
### Re-booting the module to apply RAT changes...

+CEDRXS: 2,"0010"
+CEDRXS: 4,"0010"
+CEDRXS: 5,"0010"

OK
AT+CEDRXS=2,2

OK
AT+CEDRXS=2,4

OK
AT+CEDRXS=2,5

+UMNOPROF: 100

+UBANDMASK: 0,524420,1,524420

OK
U_CELL_CFG: band mask for RAT 10 (in module terms 7) is 0x0000000000000000 0000000000080084.
### Band mask for RAT CAT-M1 is 0x0000000000000000 0000000000080084.
AT+CFUN?

+CFUN: 0

OK

+CEDRXP: 4,"0010"

jamal-mouline, 2024-01-29:
Ok, thanks for the feedback Re: using uCellCfgSetRat()<https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/cell/api/u_cell_cfg.h#L185>.

The good news: it works as now I am able to register with profile 100
The less good news is that it triggers the reboot API systematically
// Reboot the module if required

Follow up question:
Is there a way to query that the CAT-M1 is already set (and only that one)?
I’d like to add a logic similar to what is done on the MNO profile with the provided example

Would it make sense to use getRatSaraRx() for example
For rank 0, it should return 10
For rank 1, it should return 0

Thanks,

uart:~$ modem_ctrl  network_up
Cell Modem Register
uart:~$ AT+UMNOPROF?

+UMNOPROF: 100

OK
U_CELL_CFG: MNO profile is 100.
### The MNO profile is already set to 100.
U_CELL_CFG: setting sole RAT to 10 (in module terms 7).
AT+URAT=7

OK
AT+URAT?

+URAT: 7

OK
U_CELL_CFG: RATs are:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
### RAT[0] is 10 - CAT-M1.
### Re-booting the module to apply RAT changes...
U_CELL_PWR: rebooting.
AT+CFUN=15

OK
[02:23:35.282,440] <dbg> CellModem: CellModemThread: _msgq_get message 3 field 300

uart:~$ AT
AT
AT
AT

OK
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

L0.0.00.00.05.12,A.02.21

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K0

OK
AT+UPSV=0

OK
AT+CPSMS?

+CPSMS:0,,,"00011000","00001010"

OK
AT+CEDRXS?

+CEDRXS: 2,"0010"
+CEDRXS: 4,"0010"
+CEDRXS: 5,"0010"

OK
AT+CEDRXS=2,2

OK
AT+CEDRXS=2,4

OK
AT+CEDRXS=2,5

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK

+CEDRXP: 4,"0010"
U_CELL_CFG: getting band mask for RAT 10 (in module terms 0).
AT+UBANDMASK?

+UBANDMASK: 0,6170,1,524420

OK
U_CELL_CFG: band mask for RAT 10 (in module terms 7) is 0x0000000000000000 000000000000181a.
### Band mask for RAT CAT-M1 is 0x0000000000000000 000000000000181a.
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

234500091093307

OK
U_CELL_NET: user-specified APN is "tsiot".
AT+CGDCONT=1,"IP","tsiot"

OK
AT+UAUTHREQ=1,0

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?
[02:23:44.253,387] <inf> CellModem: Bringing up the network interface...

uart:~$
+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
-1: NReg

+CREG: 2

+CGREG: 2

+CEREG: 2,,,,,,,,
-1: Search
-1: Search
-1: Search
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,2

OK
-1: Search

+CEDRXP: 4,"0010","0010","0001"

+CREG: 0

+CGREG: 5,"FFFE","105950D",7

+CEREG: 5,,,,,,,,
-1: NReg
3: RegR
8: RegR
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"T-Mobile",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
U_CELL_NET: connected after 2 second(s).

Yes, to set just a single RAT and no others it is simplest to use uCellCfgSetRat()<https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/cell/api/u_cell_cfg.h#L185>.

RobMeades, 2024-01-29:
> The less good news is that it triggers the reboot API systematically

Indeed: if you have called [uCellCfgSetRat()](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/cell/api/u_cell_cfg.h#L185) to set the RAT then, internally `ubxlib` will flag that you need to reboot the cellular module for the change to take effect; `uCellRebootIsRequired()` will return `true`.

If you want to do all of this in your application as a matter of routine at every boot then you need to read the RATs and decide whether they need to be changed or not.  If you _always_ want just a single RAT then, as you suggest, it is even simpler: read out the RAT at 1 (i.e. the second one), if it is not `U_CELL_NET_RAT_UNKNOWN_OR_NOT_USED` (0) then call [uCellCfgSetRat()](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/cell/api/u_cell_cfg.h#L185) to set the single RAT that you want and reboot, job done.

RobMeades, 2024-01-29:
To be clear, If it were me I wouldn't try to implement my own version of the `ubxlib` internal static function `getRatSaraRx()`, I would use the [uCellCfg](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_cfg.h) APIs as presented; that way you are using the APIs of the library of code that is `ubxlib` as intended, in a tested/thread-safe way with clear boundaries, and you will always be able to update to later `ubxlib` versions without getting tangled up with our internal changes.

jamal-mouline, 2024-01-29:
Got it, thanks

I plan to use uCellCfgGetRat instead of getRatSaraRx

jamal-mouline, 2024-01-30:
OK, thanks for the confirmation.
After few trials, it all makes sense now, and no more systematic reboot after querying/setting the single RAT.

In summary:

1-I went ahead and restored the RAT setting loop from the example
        //------------------------- RAT -------------------------
        // Read out the existing RAT list and set the new ones

With one single change per your earlier suggestion today:

Replace uCellCfgSetRatRank with uCellCfgSetRatRat(handle, gMyRatList[0])
Where gMyRatList table is [U_CELL_NET_RAT_CATM1, U_CELL_NET_RAT_UNKNOWN_OR_NOT_USED, U_CELL_NET_RAT_UNKNOWN_OR_NOT_USED]

2-I altered the band setting section to the first RAT only.
        //----------------------- RF BANDS ----------------------
        for (x = 0; x < 1; x++) {
            if (gMyRatList[x] == U_CELL_NET_RAT_CATM1) {
                readAndSetBand(devHandle, gMyRatList[x], MY_CATM1_BANDMASK1, MY_CATM1_BANDMASK2);
        }

3-I disabled this last section, as somehow it was preventing the registration process to get started. I am assuming this is OK. Please confirm.
        //------------------ CONFIGURATION ENDS -----------------

        uint64_t readBandMask1;
        uint64_t readBandMask2;
        if (uCellCfgGetBandMask(devHandle, U_CELL_NET_RAT_CATM1,
                                &readBandMask1, &readBandMask2) == 0) {
            uPortLog("### Band mask for RAT %s is 0x%08x%08x %08x%08x.\n",
                     gpRatStr[U_CELL_NET_RAT_CATM1],
                     (uint32_t) (readBandMask2 >> 32), (uint32_t) readBandMask2,
                     (uint32_t) (readBandMask1 >> 32), (uint32_t) readBandMask1);
        } else {
            uPortLog("### unable to read bandmask!\n");
        }
        for (x = uCellNetScanGetFirst(devHandle, NULL, 0,
                                      buffer, NULL, NULL);
             x >= 0;
             x = uCellNetScanGetNext(devHandle, NULL, 0, buffer, NULL)) {
            uPortLog("### %d: network: %s\n", x, buffer);
        }

jamal-mouline, 2024-01-30:
Also, to close the loop on the original issue
Re: HTTP connectivity issue due to AT+UHTTP=0,7,0 returning ERROR

The modem firmware upgrade did not help
Had still to force the last parameter to 180, with a single line configuration addition, so no bid deal!

   pConnect->connection.timeoutSeconds = 180; // Temporary patch pending modem firmware upgrade!

   // Create an HTTPS instance for the server; to create an HTTP instance instead you would
   // replace &tlsSettings with NULL (and of course use port 8080 on the test HTTP server).
   pConnect->pContext = pUHttpClientOpen(pObj->deviceHandle, &pConnect->connection, &tlsSettings);

Let me know if you need a log showing the issue with the default timeout at 0

Thanks,

RobMeades, 2024-01-30:
Thats fine if it works for you: in the example, the bit after `//------------------ CONFIGURATION ENDS -----------------` is there to read out what you have set and perform a network scan to demonstrate that what went before worked, networks are visible.  You don't need these things in your application, though they shouldn't _stop_ anything working, just introduce a delay.

On the original issue, I hadn't spotted that you had:

```
AT+UHTTP=0,7,0
```

...i.e. the timeout is being set to zero seconds: the pattern used in the [http_main.c](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/example/http_client/http_main.c#L230) example is to always begin by assigning your [uHttpClientConnection_t](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/common/http_client/api/u_http_client.h#L103) structure  to [U_HTTP_CLIENT_CONNECTION_DEFAULT](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/common/http_client/api/u_http_client.h#L52), which will set the response timeout to [U_HTTP_CLIENT_RESPONSE_WAIT_SECONDS](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/common/http_client/api/u_http_client.h#L45) (30).  The [uCellHttpOpen()](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/cell/api/u_cell_http.h#L146) function, which [uHttpClientOpen()](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/common/http_client/api/u_http_client.h#L215) ultimately calls,  says that "the timeout in seconds when waiting for a response from the HTTP server, must be at least [U_CELL_HTTP_TIMEOUT_SECONDS_MIN](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/cell/api/u_cell_http.h#L63)"; I will update the description of [uHttpClientOpen()](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/common/http_client/api/u_http_client.h#L215) to mention that `pConnection` should be initially assigned to [U_HTTP_CLIENT_CONNECTION_DEFAULT](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/common/http_client/api/u_http_client.h#L52) and I will also modify [uCellHttpOpen()](https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/cell/src/u_cell_http.c#L454) to reject a zero value for the timeout to make this clearer.  Thanks for highlighting this!

jamal-mouline, 2024-01-30:
Ok, closing the loop on this one –original issue
My bet! I should have read through the comments more thoroughly!
I am now initializing the connection structure to U_HTTP_CLIENT_CONNECTION_DEFAULT<https://github.com/u-blox/ubxlib/blob/cd03192c93ea7fa4edd50168965d24160040f71f/common/http_client/api/u_http_client.h#L52> with the 30 second timer before calling uHttpClientOpen() and it’s all good
Thanks,

RobMeades, 2024-01-30:
Good stuff, I'm going to close this issue now; please feel free to re-open it, or open another one, as required.

closed 2024-01-30 by RobMeades

jamal-mouline, 2024-02-02:
@RobMeades
Please let me know if this would be the proper forum to discuss integration of the Thingstream SDK on top of ubxlib.
In which case I would be happy to open a new support issue as the http issue here is well wrapped up now.
Advice otherwise if other communication channels preferred.

Thingstream has currently couple SDK solutions, including one that runs on a Nordic nrf52840, using the nRF5.

Looking to port that Thingstream SDK on the same target hardware but using nRF Connect instead.

ubxlib is already very well integrated with nRF Connect, so the idea is to extend it to support Thingstream.
--For PoC, I am currently using a SARA-R4 mounted on a Sparkfun board, with an Arduino interface on top of a Nordic nrf52840 DevKit

Thingstream current design leaves it to the "integrating" application to handle modem power up / power down.
But requires the port of a "serial transport" to handle sending and receiving data stream.

To accomplish this, first thought is to pursue the following architecture approach,
where the "integrating" application would fully rely on the existing ubxlib / uDeviceOpen / uDeviceClose to manage modem power up / power down, create a UART instance, etc...
Once the modem is powered up, and the UART interface established, tap into the device handle obtained from the uDeviceOpen, in order to access uPortUartRead and uPortUartWrite, (and maybe others) from
https://github.com/u-blox/ubxlib/blob/master/port/platform/zephyr/src/u_port_uart.c
to implement the "serial transport" for Thingstream

An obvious challenge is to make sure that when there is ongoing “thingstream” operations, the UART interface would be exclusively dedicated to it. I am not sure whether the current usage of a gMutex lock/unlock was designed to provide just that?

Please let me know whether this is a sound approach to pursue with...
Otherwise, definitely open to better alternative approaches and suggestions...

Thank you

RobMeades, 2024-02-02:
Hi: yes, please open a new issue to discuss this.  We are aware of the Thingstream SDK but no-one on the `ubxlib` side has experience of actively using it, though we do know the right people to contact; we can involve them as necessary.

Whether you can use the `uDeviceOpen()`/`uDeviceClose()` APIs for this or not will depend upon whether the Thingstream SDK needs to react to URCs emitted by the cellular module: the `uDeviceOpen()`/`uDeviceClose()` APIs will start a `ubxlib` AT client on the UART, which will be the thing reacting to URCs from the module and the two will likely clash.  If the Thingstream SDK does not need to react to URCs then this won't matter; if it does then you will need to drop down to lower-level `ubxlib` APIs instead and things will be trickier.

Regarding protection against simultaneity, I suspect you will have to do that at application level as `ubxlib` will not be aware of anyone else using the same transport as it.  Basically, if you're doing Thingstream things, don't call `ubxlib` APIs.

When you open your issue, can you clarify if your aim, at application level, is to have Thingstream stuff that just happens to use the `ubxlib` porting layer for convenience, or do you actually want to use the `ubxlib` APIs as well as doing Thingstream stuff?  The former is trivially easy, the latter will be complex.

Note: FYI, when you reply to posts here I think you are doing it via e-mail and the e-mail you are responding to gets pasted on the end of your post (I've been deleting that bit from your posts); if you can, it is better to respond in the web interface dialogue boxes here.

jamal-mouline, 2024-02-02:
New ticket opened

https://github.com/u-blox/ubxlib/issues/195

will make sure to respond online going forward

Thank you

CONFIDENTIALITY. This electronic mail and any files transmitted with it may contain information proprietary to Exponential Technology Group or one of its subsidiaries or affiliates, and are intended solely for the use of the individual or entity to whom they are addressed, shall be maintained in confidence and not disclosed to third parties without the written consent of the sender. If you are not the intended recipient or the person responsible for delivering the electronic mail to the intended recipient, be advised that you have received this electronic mail in error and that any use, dissemination, forwarding, printing, or copying of this electronic mail is strictly prohibited. If you have received this electronic mail in error, please immediately notify the sender by return mail. Thank you for your compliance.