u-blox/ubxlib #182: Zephyr RTOS STM32F7 error, example request

hauserkristof, 2024-01-05:
Dear U-Blox Team!

I know that you only test the lib with Zephyr OS on nRF boards and Linux, but it would be great to supply a real example for (porting) STM32 series with Zephyr.

Hence, it is considered the easier and quite powerful way of porting the lib; it would be nice to have a bit more architecture samples for Zephyr, not just nRF, which has built-in support for it.

I currently have an issue when I set `CONFIG_UBXLIB=y` in prj.conf with the module installed. The code compiles and flashes but immediately catches a hard fault, even if I include the ubxlib.h in my main or not on my STM32F767BI board.

[image]
[sample_z_app.zip](https://github.com/u-blox/ubxlib/files/13847555/sample_z_app.zip)

Do you have any tips, please?

Thank you very much for the help!

hauserkristof, 2024-01-05:
[sample_z_app.zip](https://github.com/u-blox/ubxlib/files/13847571/sample_z_app.zip)

hauserkristof, 2024-01-05:
Also, if you could supply a sample project for STM32F4 native, with the correct directory / CMake / include tree that would be neat also.

Thank you very much!

RobMeades, 2024-01-06:
Hi there, and sorry your having trouble with this.  We've never used an STM32F7 ourselves, only STM32F4, though they appear to be pretty similar and Zephyr should support either of course.

If I've understood correctly, and looking at your `main.c`, you get a hard fault even without `ubxlib` in the build?  If that is the case, and maybe even if it is not, Zephyr-platform-related matters are best addressed to the STM Zephyr people: they hang out in the [STM32 section of the Zephyr Discord](https://discord.com/channels/720317445772017664/883445358821249084); Discord is undoubtedly the best way to get instant help on matters Zephyr, it is a very active place.

Concerning STM32F4-on-Zephyr, the directory layout etc. should be no different to that for any other Zephyr project, just follow their usual pattern and bring `ubxlib` in [as described here](https://github.com/u-blox/ubxlib/tree/master/port/platform/zephyr#integration) as a [Zephyr module](https://docs.zephyrproject.org/latest/develop/modules.html).  We don't actually use Zephyr much ourselves, aside from running our automated tests on it, so we aren't really experts on what the layout should be for development, we are "just a module" to Zephyr, so to speak.

The only issue I'm aware of with using ST devices in Zephyr with `ubxlib` is that, in the device tree, rather than having all UARTs named `uart0`, `uart1`, etc., some STM32F4 UARTs end up being named `usartx` (i.e. with an `s` in the name); if you happen to want to use a UART that is named `usartx` then you will need to include an **alias** in your `.overlay` file so that `ubxlib` can find it (see [here](https://github.com/u-blox/ubxlib/tree/master/port/platform/zephyr#device-tree) for how to do that).  But this has nothing to do with your hard fault.

We do have it on our list TODO list to run Zephyr on non-Nordic chips so I will see if I can get hold of an STM32F7 board but I can't promise how quickly we will get around to doing this I'm afraid: at a minimum it will be weeks rather than days.

cturvey, 2024-01-06:
The Nucleo 144-pin boards might be the most flexible choice. Cover the F4, F7, H7, U5, L5, etc
GPS/GNSS/MODEM shields via Arduino connection. ST-LINK VCP USART independent or the Arduino D0/D1 one.
Would suggest NUCLEO-F767ZI

RobMeades, 2024-01-06:
@cturvey: excellent, thanks for the tip.

hauserkristof, 2024-01-06:
Also @RobMeades as far I know you are in partnership with Mikroe, they have a click shield for the [Nucleo 144](https://www.mikroe.com/click-shield-for-nucleo-144) it maybe makes easier testing your own modules :)

RobMeades, 2024-01-06:
Yes, Mikroe and Sparkfun mainly.

RobMeades, 2024-01-11:
Not sure your're going to find this very helpful but...

- I now have a Nucleo-F767ZI board.
- I have built the default `ubxlib` Zephyr [runner](https://github.com/u-blox/ubxlib_priv/tree/master/port/platform/zephyr/runner) build for it (the build that just runs all our tests), using the Zephyr 3.4.99 version that comes with nrfConnect 2.5.0, with the addition of:
  - cloning and adding [hal_stm2](https://github.com/zephyrproject-rtos/hal_stm32) as a `ZEPHYR_EXTRA_MODULE`,
  - adding a [nucleo_f767zi.overlay](https://github.com/u-blox/ubxlib/files/13907954/nucleo_f767zi.overlay.txt) file to provide:
    ```
    / {
            chosen {
                    zephyr,bt_uart = &usart6;
        };
    };
    ```
    ...since the build wouldn't complete without a choice for `zephyr_bt_uart`.
  Attached find the [build output](https://github.com/u-blox/ubxlib/files/13908165/build.txt), for information.
- I downloaded the `.elf` file to the board with the STLink utility.
- I hung a terminal off the serial port that the board maps itself as, at 115200, but couldn't see anything; this is to be expected since our Zephyr porting layer will be sending debug output to a Nordic RTT port which doesn't exist on ST.
- So I hooked up OpenOCD by:
  - editing the [stm32cube_ide_openocd_swo.cfg](https://github.com/u-blox/ubxlib/files/13908012/stm32cube_ide_openocd_swo.cfg.txt) we use for STM32F4 to remove the STM32F4/SWD specifics and point it at STM32F7 instead (this copied to the OpenOCD `scripts` sub-directory),
  - running `c:\openocd-v0.11.0-3\scripts>..\bin\openocd.exe --file stm32cube_ide_openocd_swo.cfg` at a command prompt, which got me:
  ```
  xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-12-07-17:33)
  Licensed under GNU GPL v2
  For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
  Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
  Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
  Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
  Info : clock speed 2000 kHz
  Info : STLINK V2J39M27 (API v2) VID:PID 0483:374B
  Info : Target voltage: 3.229921
  Info : stm32f7x.cpu: Cortex-M7 r1p0 processor detected
  Info : stm32f7x.cpu: target has 8 breakpoints, 4 watchpoints
  Info : starting gdb server for stm32f7x.cpu on 3333
  Info : Listening on port 3333 for gdb connections
  Info : Unable to match requested speed 2000 kHz, using 1800 kHz
  Info : Unable to match requested speed 2000 kHz, using 1800 kHz
  target halted due to debug-request, current mode: Thread
  xPSR: 0x01000000 pc: 0x08042d18 msp: 0x20048000
  Info : Unable to match requested speed 8000 kHz, using 4000 kHz
  Info : Unable to match requested speed 8000 kHz, using 4000 kHz
  Info : Listening on port 6666 for tcl connections
  Info : Listening on port 4444 for telnet connections
  Info : accepting 'gdb' connection on tcp/3333
  target halted due to debug-request, current mode: Thread
  xPSR: 0x41000000 pc: 0x0804b130 psp: 0x20024dd8
  Info : device id = 0x10016451
  Info : flash size = 2048 kbytes
  Info : Single Bank 2048 kiB STM32F76x/77x found
  Info : flash size = 1024 bytes
  Info : Unable to match requested speed 2000 kHz, using 1800 kHz
  target halted due to debug-request, current mode: Thread
  xPSR: 0x01000000 pc: 0x08042d18 msp: 0x20048000
  Info : Unable to match requested speed 8000 kHz, using 4000 kHz
  Info : Padding image section 0 at 0x080001ac with 4 bytes
  Info : Unable to match requested speed 2000 kHz, using 1800 kHz
  target halted due to debug-request, current mode: Thread
  xPSR: 0x01000000 pc: 0x080762c0 msp: 0x20020000
  semihosting is enabled
  ```
  - starting up the STM32Cube IDE and configuring it to run GDB HW debugging [as described here](https://github.com/u-blox/ubxlib_priv/tree/master/port/platform/stm32cube/mcu/stm32f4#stm32cube-ide) and pointing it at the `zephry.elf` that was just built
  - start stepping through the code in the STM32Cube IDE.

With all of this I can see the target running through the porting tests happily enough:

[image]

No crashes yet, and we're certainly into "real" code, calling Zephyr RTOS functions, etc.  I will see if I can hack the debug output to send to the UART instead.

RobMeades, 2024-01-11:
Ah, yes, I edited the [prj.conf](https://github.com/u-blox/ubxlib_priv/blob/master/port/platform/zephyr/runner/prj.conf) file for our Zephyr `runner` build to comment out these lines:

```
#CONFIG_USE_SEGGER_RTT=y
#CONFIG_RTT_CONSOLE=y
```

...and set `CONFIG_UART_CONSOLE=y` (edited version of the file [attached](https://github.com/u-blox/ubxlib/files/13908390/prj.conf.txt)), built/downloaded to the board again and the debug logging immediately came streaming out of the serial port that the board maps itself as (@ 115200).  The full log output is [attached](https://github.com/u-blox/ubxlib/files/13908414/log.txt): you'll see plenty of failures since I have no cellular/GNSS/whatever HW attached to the STM32F7 board but the main thing is that all of the `port` tests pass, which means that all of the Zephyr/RTOS stuff is working as expected (these pasted in below).

I suppose the good news is that there's nothing _fundamental_ wrong but this doesn't really help you find out what is upsetting things for your case.

```
U_APP: Running portInitialisation...
U_PORT_TEST: main task stack had a minimum of 7524 byte(s) free (minimum is 5120).
U_PORT_TEST: resources are good (1 outstanding OS resource(s), as expected).
C:/projects/ubxlib_priv/port/test/u_port_test.c:1283:portInitialisation:PASS

U_APP: Running portRentrancy...
U_PORT_TEST_OS_REENT_TASK_1: 1 "4294967295 1 c".
U_PORT_TEST_OS_REENT_TASK_1: please wait while pUPortMalloc() is thrashed...
U_PORT_TEST_OS_REENT_TASK_2: 2 "4294967295 2 c".
U_PORT_TEST_OS_REENT_TASK_2: please wait while pUPortMalloc() is thrashed...
U_PORT_TEST_OS_REENT_TASK_3: 3 "4294967295 3 c".
U_PORT_TEST_OS_REENT_TASK_3: please wait while pUPortMalloc() is thrashed...
U_PORT_TEST_OS_REENT_TASK: instance 1 done, returning 0.
U_PORT_TEST_OS_REENT_TASK: instance 2 done, returning 0.
U_PORT_TEST_OS_REENT_TASK: instance 3 done, returning 0.
U_PORT_TEST: test task 1 had 1696 byte(s) free out of 2048.
U_PORT_TEST: test task 2 had 1696 byte(s) free out of 2048.
U_PORT_TEST: test task 3 had 1696 byte(s) free out of 2048.
U_PORT_TEST: reentrancy task(s) returned 0.
U_PORT_TEST: main task stack had a minimum of 7524 byte(s) free (minimum is 5120).
U_PORT_TEST: resources are good (1 outstanding OS resource(s), as expected).
U_PORT_TEST: we have leaked 0 resources(s).
C:/projects/ubxlib_priv/port/test/u_port_test.c:1311:portRentrancy:PASS

U_APP: Running portOs...
U_PORT_TEST: tick time now is 3612.
U_PORT_TEST: creating a mutex...
             returned error code 0, handle 0x20026930.
U_PORT_TEST: creating a data queue...
             returned error code 0, handle 0x20026950.
U_PORT_TEST: 20 entries free on data queue.
U_PORT_TEST: creating a control queue...
             returned error code 0, handle 0x200269e8.
U_PORT_TEST: 20 entries free on control queue.
U_PORT_TEST: locking mutex, preventing task from executing.
U_PORT_TEST: creating a test task with stack 2048 byte(s) and priority 5, passing it the pointer 0x20022c30 containing the string ""...
             returned erU_PORT_TEST_OS_TASK: task with handle 0x20026a80 started, received parameter pointer 0x20022c30 containing string "Boo!".
U_PORT_TEST_OS_TASK: uPortTaskGetHandle() returned 0x20026a80
U_PORT_TEST_OS_TASK: task trying to lock the mutex.
ror code 0, handle 0x20026a80.
U_PORT_TEST: time now 3690 ms.
U_PORT_TEST: unlocking mutex, allowing task to execute.
U_PORT_TEST_OS_TASK: task trying to lock the mutex again, should fail!.
U_PORT_TEST_OS_TASK: unlocking it again.
U_PORT_TEST_OS_TASK: locking it again (non-try version).
U_PORT_TEST_OS_TASK: task waiting on queue for data.
U_PORT_TEST: trying to lock the mutex, should fail...
U_PORT_TEST: sending stuff to task...
U_PORT_TEST_OS_TASK: task received 0.
                     item 1, expecting 0.
U_PORT_TEST_OS_TASK: queueItem 0.
U_PORT_TEST_OS_TASK: task received 100.
                     item 2, expecting 100.
U_PORT_TEST_OS_TASK: queueItem 100.
U_PORT_TEST_OS_TASK: task received 25.
                     item 3, expecting 25.
U_PORT_TEST_OS_TASK: queueItem 25.
U_PORT_TEST: test task had 1752 byte(s) free out of 2048.
[... 85 lines cut ...]
U_PORT_TEST: resources are good (1 outstanding OS resource(s), as expected).
U_PORT_TEST: we have leaked 0 resources(s).
C:/projects/ubxlib_priv/port/test/u_port_test.c:2351:portGetTimezoneOffsetSeconds:PASS

U_APP: Running portTimers...
U_PORT_TEST: testing timers...
U_PORT_TEST: at the end of the timer test:
U_PORT_TEST: timer 1 expired 0 time(s).
U_PORT_TEST: timer 2 expired 0 time(s).
U_PORT_TEST: timer 3 expired 2 time(s).
U_PORT_TEST: timer 4 expired 4 time(s).
U_PORT_TEST: main task stack had a minimum of 7524 byte(s) free (minimum is 5120).
U_PORT_TEST: resources are good (1 outstanding OS resource(s), as expected).
U_PORT_TEST: we have leaked 0 resources(s).
C:/projects/ubxlib_priv/port/test/u_port_test.c:3036:portTimers:PASS

U_APP: Running portCriticalSection...
U_PORT_TEST: testing critical sections, may take up to 10 second(s)...
U_PORT_TEST: error flag is 0x00000000.
U_PORT_TEST: main task stack had a minimum of 7524 byte(s) free (minimum is 5120).
U_PORT_TEST: resources are good (1 outstanding OS resource(s), as expected).
U_PORT_TEST: we have leaked 0 resources(s).
C:/projects/ubxlib_priv/port/test/u_port_test.c:3166:portCriticalSection:PASS
```
.

hauserkristof, 2024-01-15:
Hey @RobMeades !

Thanks for the really fast responses!

Can you please share the full project with me maybe? I would like to see the full project's structure.

RobMeades, 2024-01-16:
There isn't a "proper" project really, I just modified our `runner` build (see here https://github.com/u-blox/ubxlib/tree/master/port/platform/zephyr/runner), by adding a clone of [hal_stm32](https://github.com/zephyrproject-rtos/hal_stm32) somewhere on my hard disk, appended that location to the CMake variable `ZEPHYR_EXTRA_MODULE` in my `CMakeLists.txt` file, and added the [nucleo_f767zi.overlay](https://github.com/u-blox/ubxlib/files/13907954/nucleo_f767zi.overlay.txt) file in the `boards` directory 'cos we switch on BLE (you might not) and that caused the ST HAL code to want a choice for `zephyr_bt_uart`, it wouldn't build otherwise.

That built to completion and ran our tests as intended, it just needed the few modifications in the [prj.conf](https://github.com/u-blox/ubxlib/files/13908390/prj.conf.txt) file to send the log output to the right place.

But this isn't really an example of how a "normal" Zephyr project would be organised, if there is one: as indicated, we (I, at least) don't use Zephyr in our usual development environment so we don't know what a "normal" Zephyr project would look like.  Putting it another way: don't take our advice!  We are merely a submodule, in Zephyr terms, an add-on to a "normal" project.  Surely the Zephyr documentation indicates what a good structure might be?

RobMeades, 2024-01-18:
Sorry you didn't get anywhere with Zephyr.  Issue #3 is very old now, from the days when we used Eclipse-style projects with STM32F4; we moved away from trying to maintain the `.project` and `.cproject` files involved: there is a large amount of duplication inside the files themselves and they had a tendency to lose all your changes when opening Eclipse if you got one character wrong.

What we do instead is provide a `makefile` project that can be imported into Eclipse.  `Unity` is not required for the operation of `ubxlib` but our `runner` builds build and run all of our tests, and therefore _they_ _do_ require `Unity`, since we use `Unity` for testing.

The [README.md in the runner directory](https://github.com/u-blox/ubxlib/tree/master/port/platform/stm32cube/mcu/stm32f4/runner) describes how to pass conditional compilation flags into the `runner` build (if you need them) and where to obtain `Unity`, while the [README.md in the STM32F4 platform directory](https://github.com/u-blox/ubxlib/tree/master/port/platform/stm32cube/mcu/stm32f4#stm32cube-ide) describes how to set up the STM32Cube IDE.  Once you have a `runner` build working (and to do so would require `Unity`) you could replace the contents of our [u_main.c](https://github.com/u-blox/ubxlib/blob/master/port/platform/stm32cube/app/u_main.c) with your own, which will remove any linkage with `Unity`.  Of course you will need to adjust the main task stack and heap sizes etc. to your applications' needs as you do this.

RobMeades, 2024-02-25:
In commit b627d1ff8686631f5d6c64a81fa06ec4f0c9735 we have added to the `ubxlib` test farm testing of a Nucleo F767ZI board with native Zephyr (i.e. not through the nRFConnect SDK but direct from https://github.com/zephyrproject-rtos/zephyr).

With that I'm going to close this issue: feel free to open re-open it, or open a new issue, if there is more to discuss.

closed 2024-02-25 by RobMeades