u-blox/ubxlib #183: Sent message (MQTT)

alexmaron81, 2024-01-08:
What does “Sent message” mean?
Does it mean that the message has arrived at the MQTT broker?

[image]

RobMeades, 2024-01-08:
Hi Alex: it means that the module has returned `+UMQTTC: 2,1`.  Since MQTT runs on TCP, a reliable transport, I would guess that this means that, at the very least, the transmission of the MQTT message has been Ack'ed at transport level.  Whether it means that the MQTT broker has accepted your publish message I'm not sure; I would guess "yes", I could probably find out...?

alexmaron81, 2024-01-08:
I have to somehow ensure that the message arrives, and exactly once.
I understand that QoS 2 does this?

int32_t error = uMqttClientPublish(pMqttContext, logTopic, message, strlen(message), U_MQTT_QOS_EXACTLY_ONCE, false);

RobMeades, 2024-01-08:
Yes, that's correct.

alexmaron81, 2024-01-08:
But somehow not all messages arrive.
I don't know how I can make sure.
I use RabbitMQ, and some messages arrive in the queue multiple times. Also with QoS 2
As a test I brobed Mosquitto, not all messages arrive here.

RobMeades, 2024-01-08:
OK, I will raise the question internally.  To be clear, is it that some published messages in a sequence of published messages do not arrive at the broker or that some published messages arrive at the broker more than once even with QoS 2 (or both)?

alexmaron81, 2024-01-08:
I have now switched to (U_MQTT_QOS_AT_LEAST_ONCE), now all messages arrive at (Mosquitto).
But I don't know yet whether these will arrive twice in RabbitMQ, I still have to check.

RobMeades, 2024-02-22:
> But I don't know yet whether these will arrive twice in RabbitMQ, I still have to check.

Hi Alex: does this one need further attention or can it be closed?

RobMeades, 2024-03-05:
I'm going to assume this can be closed, please re-open (or open another issue) if there is more to discuss.

closed 2024-03-05 by RobMeades