u-blox/ubxlib #185: Automatic authentication failure for U260, U270, U280, among others

GioTB, 2024-01-18:
I was implementing the library with a SARA-U260-00S-02 and when trying to connect to the network it didn´t worked, diging a bit more realise that the library attempts to set the authentication type to "automatic" but this is not supported by some modules (according to the AT manual https://content.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_UBX-13002752.pdf, page 403 says: "SARA-U260-00S / SARA-U270-00S / SARA-U270-00X / SARA-U280-00S / LISA-U200-00S /
LISA-U200-01S / LISA-U200-02S / LISA-U200-52S / LISA-U200-62S / LISA-U230 / LISA-U260 /
LISA-U270
• If <param_tag>=6, <param_val>=3 (automatic selection of authentication type) is not supported.")

So would it be posible to add these modules type? the only SARA-U in the "uCellModuleType_t" is the SARA_U201, so it would be nice to have the others one too that doesn´t support this automatic authentication. Another idea would be to force that segment of the code (on the activateContextUpsd function), that set the param 6 value of the UPSD command would be taken from the authenticationMode parameter of the uNetworkCfgCell_t struct, this way the user has the option to use that parameter for that configuration, independantly the module type.

RobMeades, 2024-01-18:
Hi there, and thanks for posting.  Which version of `ubxlib` is this with?  Asking 'cos we changed the way it worked in commit https://github.com/u-blox/ubxlib/commit/f7d9d84804abdf207db20993adf42927a8420314.

RobMeades, 2024-01-18:
Understood, yes, it's hard coded for SARA-U201 but you are right that it needn't be. Will give it a try.

RobMeades, 2024-01-18:
A quick look suggest that changing this line:

https://github.com/u-blox/ubxlib/blob/41349f324ed52de37ade97f8de638b4ab1549f9a/cell/src/u_cell_net.c#L1693

...to be:

```
        uAtClientWriteInt(atHandle, pInstance->authenticationMode);
```

...would be sufficient to meet your needs.  Can you try this and let me know if it works?

GioTB, 2024-01-18:
yes!, that actually works, i tryed first hardcoding it to "0" but then i changed it exactly like you leave it and works right away.

RobMeades, 2024-01-18:
Great, will work up a change let you know when it has landed back here.

closed 2024-01-23 by RobMeades

RobMeades, 2024-01-23:
Fixed in commit f95208fc55abaa3448b510ba3342802099aab38e.