u-blox/ubxlib #186: Option to Exclude Geofence

FloriVeit, 2024-01-22:
Our project does not currently support C++, and it is not straightforward for us to add this support. We found that the Geofence feature specifically requires C++ support. Would it be possible to include an option in CMake to exclude the Geofence parts of ubxlib?
 
Additionally, we found that if we comment out line 143 in the `ubxlib.cmake` file, everything builds correctly.
`list(APPEND UBXLIB_SRC ${UBXLIB_BASE}/common/geofence/src/u_geofence_geodesic.cpp)`

RobMeades, 2024-01-22:
Ah, OK, let me find a way to do that...

RobMeades, 2024-01-22:
I've updated the existing preview branch to include a potential fix: could you give it a spin?

FloriVeit, 2024-01-23:
Thank you for the quick processing, the update works perfectly.

RobMeades, 2024-01-23:
The fix is now pushed to `master` here in commit 4d076219927cf6f5ed3951542a6b0b24a3228592.  Please let me know when you have moved there the preview branch and I will delete it.

RobMeades, 2024-02-07:
@FloriVeit: is it OK for me to delete https://github.com/u-blox/ubxlib/tree/preview_sock_sec_free_rmea now?

FloriVeit, 2024-02-07:
Apologies, I forgot to respond earlier. Yes, you can delete this branch. I’ve tested the master, and it works perfectly for us.

closed 2024-02-07 by RobMeades