u-blox/ubxlib #194: ESP-IDF with GNSS over I2C with bus already open

synologic, 2024-02-02:
Hi all,

i'm having an issue when trying to initialize a NEO9 via i2c, where my application has already initialized the I2C bus:

`    deviceConfig.deviceType = U_DEVICE_TYPE_GNSS;
    deviceConfig.deviceCfg.cfgGnss.moduleType = U_GNSS_MODULE_TYPE_M9;
    deviceConfig.deviceCfg.cfgGnss.pinDataReady = -1;
    deviceConfig.deviceCfg.cfgGnss.pinEnablePower = U_CFG_APP_PIN_GNSS_ENABLE_POWER;
    deviceConfig.deviceCfg.cfgGnss.i2cAddress = GPS_ADDRESS;
    deviceConfig.transportType = U_DEVICE_TRANSPORT_TYPE_I2C;
    deviceConfig.transportCfg.cfgI2c.alreadyOpen = true;
    deviceConfig.transportCfg.cfgI2c.i2c = 1;
    deviceConfig.transportCfg.cfgI2c.clockHertz = 400000;
    deviceConfig.transportCfg.cfgI2c.pinScl = I2C1_SCL;
    deviceConfig.transportCfg.cfgI2c.pinSda = I2C1_SDA;

    uPortInit();
    uPortI2cInit();
    uDeviceInit();

    int32_t returnCode = uDeviceOpen(&deviceConfig, &devHandle);
    ESP_LOGI(LOG_TAG,"GPS device opened with code %ld", returnCode);`

returnCode is U_ERROR_COMMON_PLATFORM

If i try the same but with alreadyOpen set to false, there's no problem

Any suggestions ?

Thanks

RobMeades, 2024-02-02:
Hi there, and sorry you're having problems with this.

The effect of setting `alreadyOpen` to `true` is that, in the internal function [uDevicePrivateI2cOpen()](https://github.com/u-blox/ubxlib/blob/9ce2d92bb594f7d4010964a57ca2a28d9cf902f9/common/device/src/u_device_private.c#L150), [uPortI2cAdopt()](https://github.com/u-blox/ubxlib/blob/9ce2d92bb594f7d4010964a57ca2a28d9cf902f9/port/api/u_port_i2c.h#L111) is called, instead of [uPortI2cOpen()](https://github.com/u-blox/ubxlib/blob/9ce2d92bb594f7d4010964a57ca2a28d9cf902f9/port/api/u_port_i2c.h#L78).

The [implementation of uPortI2cAdopt()](https://github.com/u-blox/ubxlib/blob/9ce2d92bb594f7d4010964a57ca2a28d9cf902f9/port/platform/esp-idf/src/u_port_i2c.c#L358) for ESP-IDF just calls the `static` function [openI2c()](https://github.com/u-blox/ubxlib/blob/9ce2d92bb594f7d4010964a57ca2a28d9cf902f9/port/platform/esp-idf/src/u_port_i2c.c#L263) with `adopt` set to `true`, and in that case `openI2c()` does not call any of the ESP-IDF functions ([i2c_param_config()](https://docs.espressif.com/projects/esp-idf/en/v5.0.5/esp32/api-reference/peripherals/i2c.html?highlight=i2c_param_config#_CPPv416i2c_param_config10i2c_port_tPK12i2c_config_t), [i2c_set_timeout()](https://docs.espressif.com/projects/esp-idf/en/v5.0.5/esp32/api-reference/peripherals/i2c.html?highlight=i2c_set_timeout#_CPPv415i2c_set_timeout10i2c_port_ti), or [i2c_driver_install()](https://docs.espressif.com/projects/esp-idf/en/v5.0.5/esp32/api-reference/peripherals/i2c.html?highlight=i2c_driver_install#_CPPv418i2c_driver_install10i2c_port_t10i2c_mode_t6size_t6size_ti)), as you will have already called these before you called `uDeviceOpen()`, it just remembers the `i2c` and `clockHertz` parameters, and that the I2C port was adopted (so the `ubxlib` code won't try to close it), and returns.

Can you confirm that at least [i2c_driver_install()](https://docs.espressif.com/projects/esp-idf/en/v5.0.5/esp32/api-reference/peripherals/i2c.html?highlight=i2c_driver_install#_CPPv418i2c_driver_install10i2c_port_t10i2c_mode_t6size_t6size_ti), and probably [i2c_param_config()](https://docs.espressif.com/projects/esp-idf/en/v5.0.5/esp32/api-reference/peripherals/i2c.html?highlight=i2c_param_config#_CPPv416i2c_param_config10i2c_port_tPK12i2c_config_t), have been called before `uDeviceOpen()`, and what parameters you called them with?

synologic, 2024-02-02:
I can confirm the i2c bus is correctly configured as GNSS shares the bus with 10 other peripherals which are working fine.

From tracing, i have managed to isolate the cause in the send() function from u_port_i2c.c, line 204, which is a bit weird (did not do bus sniffing yet).

In fact, several commands are sent over to the GNSS device via i2c_master_cmd_begin however only one returns ESP_TIMEOUT hence the write fails according to ESP-IDF

I made sure there are no other tasks competing for i2c bus, but afaik esp-idf has mutexes on i2c_master* functions and are thread safe.

In any case, my i2c init code:

    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = I2C0_SDA;
    conf.scl_io_num = I2C0_SCL;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = 400000; // 400 Khz
    conf.clk_flags = 0;

with looks the same as the uPortI2cOpen

When it's not working i get from the lib:

U_GNSS: sent command b5 62 0a 06 00 00 10 3a.

But when i let ubxlib handle the idc, i get an answer to that

U_GNSS: decoded UBX response 0x0a 0x06: 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].

RobMeades, 2024-02-02:
Hmph, that is very odd. When there's an I2C issue ESP-IDF usually spits out some log text of its own: do you see anything of that nature when this problem occurs?

RobMeades, 2024-02-02:
Maybe this is just a typo, but in your I2C init code that you pasted-in above you have:

```
conf.sda_io_num = I2C0_SDA;
conf.scl_io_num = I2C0_SCL;
```

...whereas in the `uDeviceCfg_t` initialisation you pasted in above you have:

```
deviceConfig.transportCfg.cfgI2c.pinScl = I2C1_SCL;
deviceConfig.transportCfg.cfgI2c.pinSda = I2C1_SDA;
```

synologic, 2024-02-02:
Not a typo, just wrong section of the code ... i use both i2c hardware ports with same settings but different pins, that's just the initial config object with the I2C0 pins, i have I2C1 pins later in the code ;-)

synologic, 2024-02-02:
Best i can do is Digital Discovery 2 :)
I'll keep you posted shortly

synologic, 2024-02-02:
On the DD2 trace, i see diferent behavior (device is on 0x42):

With I2C Port already initialized before:

Start, h85 [ h42 | RD ], Error! 
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], 
Restart, h84 [ h42 | WR ], 
Restart, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00 NAK, Error! 
Restart, h84 [ h42 | WR ], 
Restart, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, Error! 
Start, h84 [ h42 | WR ], Error! 
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00 NAK, Error! 
Restart, h84 [ h42 | WR ], 
Restart, h84 [ h42 | WR ], 

WIth ubxlib handling i2c initialization:
Restart, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hB5, h62, h0A, h06, h00, h00, h10, h3A, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
[... 253 lines cut ...]
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h80 NAK, Stop
Start, h85 [ h42 | RD ], hB5, h62, h0A, h06, h78, h00, h0F, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h00, h97, hF2 NAK, Stop
Start, h84 [ h42 | WR ], hFD, 
Restart, h85 [ h42 | RD ], h00, h00 NAK, Stop

synologic, 2024-02-02:
Looks like my calculation of the i2c_set_timeout value was wrong and not enough.
It works perfectly after setting it to your value of 10 * 80000

Sorry for the confusion :)

RobMeades, 2024-02-02:
Ah, is this the ESP32S?  If so Espressif [changed the way timing works on that chip](https://www.esp32.com/viewtopic.php?f=2&t=33583) but didn't document it...?

synologic, 2024-02-02:
Thanks for your time, closing issue

closed 2024-02-02 by synologic

synologic, 2024-02-03:
@RobMeades Looking at this issue further, but I cannot seem to find any useful info in any of the M9 datasheets, whith the higher timeout set, is it normal that M9 responds in about 1 second ?
I see this on the I2C Analyser:
[image]

Thanks

RobMeades, 2024-02-04:
Section 5.3 of the [NEO-M9N data-sheet](https://content.u-blox.com/sites/default/files/NEO-M9N-00B_DataSheet_UBX-19014285.pdf) would seem to be the reference for I2C timing, where the longest time value is the maximum of 20 ms clock stretching that the GNSS device might apply, so I wouldn't expect an I2C transaction to ever be longer than about that long.

Maybe it depends which timeout we are talking of?  Not sure I follow the picture, but the default "epoch" of the GNSS device, i.e. the interval at which it will try to make a navigation solution, emitting, say, a `UBX-NAV-PVT` message, is 1 second, so you wouldn't normally get a position more frequently than that?

synologic, 2024-02-04:
No need for faster location retrieval, however, the M9 shares the I2C bus with other devices which are read more often.

Put aside the fact that uGnssPosGet has a timeout of 240 seconds waiting for a proper nav solution from the device, i did bypass the timeout mechanism.

Long story short, out of cold start it takes some time for a proper solution, same happens when there's no sky visibility, antenna is disconnected, whatever.

Let's take UBX-NAV-STATUS for example, i send that with uGnssMsgSend and it takes 1s for the device to respond and uGnssMsgReceive to actually return from the call.

So it looks to me that at least both UBX-NAV-STATUS and UBX-NAV-PVT takes 1s, which makes sharing the bus a pain if i need other faster reads.

Either i'm doing something wrong (config, whateaver) or that's  how the device works, it's my first time with these devices and i have no spare pins on the CPU to have the M9 on it's own bus.

What you're seeing in the picture above is exactly that, the command is sent:

Start, h84 [ h42 | WR ], hB5, h62, h01, h07, h00, h00, h04, h0D, Stop

and then it takes almost a second for the answer to be sent by the M9

Thanks

RobMeades, 2024-02-04:
Usually the times when you are polling with a command and awaiting a response would be the exception rather than the rule; e.g. for configuration-type transactions.  Once done, the device would be set up to emit something, e.g. the location, periodically.

Since you are using I2C the master still needs to pull the data, of course; if you're concerned about bus occupancy you could configure the GNSS device to set a `TX_READY` pin when there is stuff to read (see 3.8.5 of the [integration manual](https://content.u-blox.com/sites/default/files/NEO-M9N_Integrationmanual_UBX-19014286.pdf)).  We've not had a use case where such signals are required before, so their use is not integrated into the operation of `ubxlib`, but we could do that if it would help?  Would need to think about how it would work.

synologic, 2024-02-05:
How would this work ? Send a 0x1 0x03 message and don't read anything until TX_READY is asserted (i guess i can setup EXTINT pin to act as TX_READY), and then send 0x01 0x03 message again and read back the answer immediately ?

RobMeades, 2024-02-05:
OK, I now have a MAX-M10S connected over I2C to an ESP32 and I'm calling [uGnssPosGet()](https://github.com/u-blox/ubxlib/blob/9ce2d92bb594f7d4010964a57ca2a28d9cf902f9/gnss/api/u_gnss_pos.h#L94) with no antenna connected.

The debug output from `ubxlib` looks like this for ever:

```
2024-02-05 11:06:41 u_monitor_27 - U_GNSS: decoded UBX response 0x01 0x07: e8 c7 09 00 e5 07 03 07 00 0a 29 f0 ff ff ff ff 00 00 00 00 00 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 98 bd ff ff ff ff ff ff 00 cb 84 df 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 5c 40 56 2f 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2024-02-05 11:06:41 u_monitor_27 - U_GNSS: sent command b5 62 01 07 00 00 08 19.
2024-02-05 11:06:42 u_monitor_27 - U_GNSS: decoded UBX response 0x01 0x07: d0 cb 09 00 e5 07 03 07 00 0a 2a f0 ff ff ff ff 00 00 00 00 00 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 98 bd ff ff ff ff ff ff 00 cb 84 df 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 5c 40 56 2f 00 00 00 00 00 00 00 00 [body 92 byte(s)].
2024-02-05 11:06:42 u_monitor_27 - U_GNSS: sent command b5 62 01 07 00 00 08 19.
2024-02-05 11:06:43 u_monitor_27 - U_GNSS: decoded UBX response 0x01 0x07: b8 cf 09 00 e5 07 03 07 00 0a 2b f0 ff ff ff ff 00 00 00 00 00 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 98 bd ff ff ff ff ff ff 00 cb 8
...
```

I've attached the [saleae capture](https://github.com/u-blox/ubxlib/files/14164748/i2c_maxm10s_esp322_no_signal.zip) (which you can read by downloading and installing their [application](https://www.saleae.com/downloads/) if you wish).  What you will see is patterns like this:

[image]

...i.e., since this is a blocking call and the underlying transport is I2C, there are polls to ask the GNSS device if it has any data to send and, about once a second it does, hence the thicker parts in this zoomed-out trace.  Zooming in to one of the polls, this is a "have you got any data to send?  No I havent." exchange:

[image]

...and this is a "have you got any data to send?  Yes I have." exchange, i.e. the start of one of the thicker blobs in the zoomed-out picture:

[image]

Between the polls, the I2C bus is not occupied, you should be able to do other stuff...?

synologic, 2024-02-05:
Your trace and my trace look similar, i just wonder if the polling actually holds the bus mutex or not, as esp-idf i2c_master* are thread safe and mutex enabled.

I did a bit of testing with having EXTINT pin assert when data is ready, i just can't figure out how to properly read that data over i2c, as if i send a receive message with 0x0107 it just polls the device for one second again before receiving the data (it's either that or i really screwed something up while testing this).

Long story short during that second needed for the polling, it's pretty much stopping my task.

Disabling NMEA and just sticking to UBX helps a very little, during data read, but the second it takes to actually receive the data is something i can't seem to find a good solution (yet) to get around.

RobMeades, 2024-02-05:
Understood: as an experiment, could you try hacking [the 10 ms here](https://github.com/u-blox/ubxlib/blob/6c99bdb7d1840e31bc76911cc74e9e296b770d15/gnss/src/u_gnss_private.c#L2469) to be, say 50 ms or 100 ms and see if that helps?

If it does, we know that it is the I2C "do you have any data" poll rate that is getting in your way and I will make it properly configurable.

synologic, 2024-02-05:
I'll try just to confirm, maybe in th is case, the workaround would be to have a separate task read the GNSS while another task holds the mutex and does all the dirty i2c jobs ... in the end, if it works, it works, just a matter of propagating a variable delay based on what's expected from the device, since not all messages take ~1s to get an answer

Keep you posted

nasihnazer, 2024-08-22:
Hi @synologic ,

I’m currently working on integrating ubxlib with ESP-IDF for my project, and I noticed your successful integration mentioned in issue #194. I would greatly appreciate it if you could share some insights on the integration process you followed.

Specifically, I’m interested in:
- The steps you took to adapt ubxlib for ESP-IDF.
- Any modifications or configurations you made in the CMakeLists.txt or other project files.
- How you handled dependencies and included the necessary headers.

Any additional tips or guidance you could provide would be extremely helpful.

synologic, 2024-08-22:
@nasihnazer ubxlib already works with ESP-IDF 5.1 (tested) it does not work with 5.3 due to i2c driver, not sure about 5.2 but i think it doesnt.

All i did was to copy ubxlib to my project root, then i just copied CMakeLists.txt and component.mk from ubxlib/port/platform/esp-idf/mcu/esp32/components/ubxlib/ to a folder called "ublox" in my project's components folder and required it in my main's CMakeFiles.txt ...

nothing else, it just works (tm)

nasihnazer, 2024-08-23:
Hi @synologic ,

Thank you for your guidance on integrating ubxlib. I followed your instructions and tried various methods to resolve the build issues, but I’m still encountering problems. Despite my efforts, I haven’t been able to get it to work.

Could you please provide detailed steps on integrating ubxlib? Any additional help or specific advice would be greatly appreciated.

Thank you!

synologic, 2024-08-23:
If you encounter issues with using ubxlib it may be a good idea to open a separate issue rather than polluting this issue

nasihnazer, 2024-08-23:
I’ve already opened a separate issue regarding the problems I’m facing with ubxlib. To better understand and resolve the issues, could you please share the detailed steps you followed while integrating ubxlib into your project? Any specific configurations or adjustments you made would be very helpful. is this the [steps ](https://github.com/u-blox/ubxlib/tree/master/port/platform/esp-idf/mcu/esp32) you followed to integrate ubxlib
Thank you!