u-blox/ubxlib #197: u_cell_mqtt module issue

AntonZyma, 2024-02-05:
Hi!

I've found an issue when working with MQTT over SARA-R5 modem.
When reading an MQTT message that doesn't fit into the buffer provided to the `uMqttClientMessageRead()`, the function will not return `U_ERROR_COMMON_TRUNCATED`, but the failure will be reported.

The issue is that `uAtClientReadBytes()` must be used with `standalone` parameter set to `true` in case `uAtClientIgnoreStopTag()` has been called.
So, the only function call that happens with "truncated" state is done wrong. This leads to the improper operation of the `readMessage()` function.

https://github.com/u-blox/ubxlib/blob/9ce2d92bb594f7d4010964a57ca2a28d9cf902f9/cell/src/u_cell_mqtt.c#L2068

Could you please check this and similar possible issues across the library?

Thanks,
Anton

RobMeades, 2024-02-05:
Hi, and thanks for spotting this.  I really should have [followed my own advice](https://github.com/u-blox/ubxlib/blob/140134b52047bd8d50d63e615cc1a1c13a8390fa/common/at_client/api/u_at_client.h#L1118) :-).

Searching for `uAtClientIgnoreStopTag()` throughout the code, as far as I can tell all other cases are fine, testing fix now.

RobMeades, 2024-02-05:
I've also pushed a preview branch with the fix here:

https://github.com/u-blox/ubxlib/tree/preview_fix_cell_mqtt_truncate_rmea

...if you would like to check that it does the trick yourself.

AntonZyma, 2024-02-05:
Thanks @RobMeades !

RobMeades, 2024-02-06:
Now pushed here in commit https://github.com/u-blox/ubxlib/commit/928ee28a6799bc7be2351c27d81bf88a7ed75335; I will close this issue and will delete the preview branch some time next week.

closed 2024-02-06 by RobMeades