u-blox/ubxlib #200: Using uSockWrite for UDP

smartme-dm, 2024-02-14:
What is the intenioned way to use the uSockWrite() for a UDP Packet to be sent?

```
if(sock < 0)
{
    sock = uSockCreate(devHandle,
                               U_SOCK_TYPE_DGRAM,
                               U_SOCK_PROTOCOL_UDP);
   uPortLog("Socket created: %u \n", sock);
   uSockConnect(sock, &address.ipAddress);
   uCellSockBlockingSet(devHandle, sock, false);
}

if(sock >= 0)
{
        uPortLog("Sending data...\n");
        int x = uSockWrite(sock, pBuffer, size);
        uPortLog("Sent %d byte(s) to server.\n", x);
}
```

Sending data with `uSockWrite()` only works once. After the first succescul sendint, uSockWrite returns always -1? why does the socket have to be "connected" even though I only want to send an udp packet?

Thanks for your support.

RobMeades, 2024-02-14:
Hi, and thanks for posting.

> why does the socket have to be "connected" even though I only want to send an udp packet?

You have created a UDP socket with [uSockCreate()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/sock/api/u_sock.h#L374); at that point this is just a local socket (let's call it A), it is not connected to anything.  With that done, you can either connect that socket to something (let's call this B) using [uSockConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/sock/api/u_sock.h#L391) and use [uSockWrite()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/sock/api/u_sock.h#L588) to send data over the connected socket to B, or you can not bother connecting the socket to B and instead call [uSockSendTo()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/sock/api/u_sock.h#L536), which takes the address of B as a parameter, and send it that way.  This is normal BSD sockets operation, I think.

> Sending data with `uSockWrite()` only works once. After the first successful sending, `uSockWrite()` returns always -1?

-1 is the standard BSD sockets "error" return value for everything: to see why a `uSockWrite()` is failing you could, first of all, print out `errno` as a quick diagnostic, or you could leave `ubxlib` diagnostic logging on (it would be on by default) and post the output at the time of the failure here.

RobMeades, 2024-02-14:
Not that this will be a cause of the -1 error but I thought I should point it out: the code you pasted in shows a call to [uCellSockBlockingSet()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/api/u_cell_sock.h#L206); it says in the function description "this function is provided for compatibility purposes only: this socket implementation is always non-blocking".  Blocking is done at the `uSock` API level, not at the `uCellSock` API level; if you want the socket to be blocking you should call [uSockBlockingSet()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/sock/api/u_sock.h#L451).

smartme-dm, 2024-02-15:
The second call of `uSockSendTo(sock, &address, pBuffer, size)` leads in `errno = -5`

RobMeades, 2024-02-15:
Hmm, odd, I was expecting a positive number but, whatever, I assume that is [U_SOCK_EIO](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/sock/api/u_sock_errno.h#L43C10-L43C20), and so code execution is probably about here:

https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_sock.c#L1284

...and the module has not accepted the `AT+USOST` which follows.  Could be various reasons for that; the debug output, if you can paste that here, would hopefully show us.

smartme-dm, 2024-02-16:
The debug output loosk as following when sending crashes:

``` log
E (30896) MOBILE: ACTUAL SOCKET: 0
IPV4 20.250.50.24:80
Sending data...
AT+USOST=0,"20.250.50.24",80,23,"026919000000007A613DD4FA6CE65A7A81EE3029781A06"
[00][00]Sent -1 byte(s) to server.
```

One step bevore, the sending of data worked finde:

``` log
E (30396) MOBILE: ACTUAL SOCKET: 0
IPV4 20.250.50.24:80
Sending data...
AT+USOST=0,"20.250.50.24",80,39,"02691900000000FCEB7550D3B9775B36C8669C462F9B2887993CF921C00E8DACABB94D40A4EB82"

+USOST: 0,39

OK
Sent 39 byte(s) to server.
```

Not much information.

RobMeades, 2024-02-16:
Could you post the whole `ubxlib` debug output?  I would like to see the initialisation and registration steps also.  Could you also indicate which module type this is with?

smartme-dm, 2024-02-16:
```
U_CELL: initialising with enable power pin not connected, PWR_ON pin 33 (0x21) (and is toggled from 1 to 0) and VInt pin not connected.
I (8906) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
AT
AT

OK
U_CELL_PWR: powering on, module is already on.
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
I (9136) TAG: Security parameters loaded for serial 6505
ATI9

00.12,A00.00

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=2

OK
AT&K0

OK
AT+UPSV=4

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 90

OK
AT+UGPRF?

+CME ERROR: Unknown

AT+CFUN=4

OK
Opened device with return code 0.
AT+COPS?

+COPS: 0

OK
AT+CFUN=4

OK
U_CELL_NET: disconnected.
U_CELL_CFG: setting sole RAT to 10 (in module terms 7).
AT+URAT=7

OK
U_CELL_PWR: rebooting.
AT+CFUN=15

OK
[00][00]I (18756) TAG: RX datalogger flash read pos:248

I (18756) TAG: Datalogger new value added. Size: 132, flash position: 248
I (18756) TAG: Datalength: 132

AT
AT

OK
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

00.12,A00.00

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=2

OK
AT&K0

OK
AT+UPSV=4

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 90

OK
AT+UGPRF?

+CME ERROR: Unknown

AT+CFUN=4

OK
Bringing up the network...
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

901405113256968

OK
U_CELL_NET: user-specified APN is "iot.1nce.net".
AT+CGDCONT=1,"IP","iot.1nce.net"

OK
AT+UAUTHREQ=1,0

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg

+CREG: 2
-1: Search

+CEREG: 2,,,,,,,,
-1: Search
AT+CREG?

+CREG: 2,2

OK
-1: Search
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,2

OK
-1: Search
AT+CREG?

+CREG: 2,2

OK
-1: Search

+CREG: 5,"9E34","01638106",7

+CEREG: 5,"9E34","1638106",7,,,,
7: RegR
7: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"Sunrise",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT=1,1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
U_CELL_NET: connected after 4 second(s).
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"10.241.65.156"

OK
U_CELL_NET: IP address "10.241.65.156".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"iot.1nce.net","10.241.65.156","","8.8.8.8","8.8.4.4"

OK
U_CELL_NET: primary DNS address:   "8.8.8.8".
U_CELL_NET: secondary DNS address: "8.8.4.4".
U_CELL_SOCK: looking up IP address of "testconnect.smart-me.com".
AT+CGDCONT?

+CGDCONT: 1,"IP","iot.1nce.net","10.241.65.156",0,0,0

OK
AT+UDNSRN=0,"testconnect.smart-me.com"

+UDNSRN: "20.250.50.24"

OK
U_CELL_SOCK: found it at "20.250.50.24".
testconnect.smart-me.com is: IPV4 20.250.50.24
Got IP for smart-me connect
AT+UDCONF=1,1

OK
I (28606) LOG: SendUdpPacket: UploadMeterData: 320, Action Type: 1
E (28606) MOBILE: ACTUAL SOCKET: -1
IPV4 20.250.50.24:80
Creating socket...
AT+USOCR=17

+USOCR: 0

OK
U_SOCK: socket created, descriptor 0, network handle 0x3ffc86f0, socket handle 0.
Socket created: 0
Sending data...
AT+USOST=0,"20.250.50.24",80,327,"0269190000000025987C15586A2A2D36EB77EA9F247C861A3DCFCA5EA931A9B8ECA933184F96EADA3F998852394E397707F067E3E69F432B5A92AAC7D28B595A51D9ABA63FF7860FBD7F642CFEE4439F52225AC7C12D8384B1E3D3248741D4746EC16731759522B42EAC746F7B30E7C5309C867B889DC3A13ED658042F612230CF2466454F455C821ABF42581ED712CA1F48C02FE34A8478223163F171EA9C230B4B7CF6D5C02BAEF912E6C83189DBCEE5FCA1A61EAAD875200F86391CAE75E784954E88622ED14C14F0B46D0FD2926734588F0ECF28A58B990C625F6FBAB98C84E19E2181E88610B76CAAB364836698FDAAE3937F386EC90E62AB8F63C6217D4334938174345370995091A2E8FF0B3983D78123D7880F5D4B557E00CF951CCCC3F08AB63A2E53389AE79E351409A3CBB91D2FFBCD70BA0D389E3C65B4C51D3F59FED1F246233C"

+USOST: 0,327

OK
Sent 327 byte(s) to server.

+UUSORF: 0,37
AT+USORF=0,512

+USORF: 0,"20.250.50.24",80,37,"02A653CF65915D7620613393D3C75DDCCC3F5DE354D0C41D3CF06FD049BC61A3AB78857633"
OK
I (29156) MOBILE: Mobile RX: 37
I (29836) LOG: SendUdpPacket: UploadMeterData: 16, Action Type: 3
E (29836) MOBILE: ACTUAL SOCKET: 0
IPV4 20.250.50.24:80
Sending data...
AT+USOST=0,"20.250.50.24",80,23,"0269190000000061AB45BA5C6ECACBF39238EF63E0AB36"

+USOST: 0,23

OK
Sent 23 byte(s) to server.

+UUSORF: 0,37
AT+USORF=0,512

+USORF: 0,"20.250.50.24",80,37,"02A753CF656BF7B2188CE51B874F182987FCACBDD91A62F4D71C342370B57DBE1E10D151FF"
OK
I (30396) MOBILE: Mobile RX: 37
E (30396) MOBILE: ACTUAL SOCKET: 0
IPV4 20.250.50.24:80
Sending data...
AT+USOST=0,"20.250.50.24",80,39,"02691900000000FCEB7550D3B9775B36C8669C462F9B2887993CF921C00E8DACABB94D40A4EB82"

+USOST: 0,39

OK
Sent 39 byte(s) to server.
I (30896) LOG: SendUdpPacket: UploadMeterData: 16, Action Type: 3
E (30896) MOBILE: ACTUAL SOCKET: 0
IPV4 20.250.50.24:80
Sending data...
AT+USOST=0,"20.250.50.24",80,23,"026919000000007A613DD4FA6CE65A7A81EE3029781A06"
[00][00]Sent -1 byte(s) to server.
I (40936) TAG: SmartMeDataUploadFailed
```

RobMeades, 2024-02-16:
Thanks, looks like this might be a SARA-R422?

How is the module being powered?  The interesting thing in the log is the `[00][00]` which you see at around the time of the problem: this usually indicates that the module has pulled `RXD` low for long enough to cause a break to be detected by the MCU and that usually happens when the module has powered up/down or browned-out.  To check this, could you hang a logic capture device (a Saleae probe or similar) off the `V_INT` output line of the module (maybe also the `TXD` and `RXD` lines if you can)?  `V_INT` _should_ remain high the whole time the module is on: if it glitches low that would indicate a brown-out. 

On a side note, you are requesting 3GPP power saving, though it is not being granted by the network as you have a roaming SIM, so that shouldn't be an issue, but if it _were_ granted you might, of course, _expect_ the module to power-off like this.

smartme-dm, 2024-02-16:
Yes it is a SARA-R422

i measured V_INT - somehow it glitches sometimes for about 30ms to low. very strange. when we are waiting for the response from the module, it always happens. 

i exchanged the power supply for 3.3V for the module with a powerfull external supply (able to deliver up to 6A) -> still the same issue

how to disable  3GPP power saving? actualy we dont need this.

RobMeades, 2024-02-16:
> I measured `V_INT` - somehow it glitches sometimes for about 30ms to low. very strange. when we are waiting for the response from the module, it always happens.

Very strange, seems likely to be transmit-peak-current-related, rather than average-current-related, IYSWIM.  I'm no HW guy, unfortunately, all I could suggest is having a good read through the quite lengthy section 1.5.1 of the [SARA-R4 system integration manual](https://www.u-blox.com/sites/default/files/SARA-R4_SysIntegrManual_UBX-16029218.pdf).

> how to disable 3GPP power saving? actually we don't need this.

Oops, sorry, I misread the debug output, those commands are actually just setting up a notification for 3GPP power saving events, they aren't actually switching 3GPP power saving on, so it is OK.

smartme-dm, 2024-02-21:
So far so good. Things are working. But - if we lose cellular connection and have to reconnnet we see sometimes the following 
```
U_CELL_SOCK: found it at "20.250.50.24".
testconnect.smart-me.com is: IPV4 20.250.50.24
Got IP for smart-me connect
[0;32mI (8510217) LOG: SendUdpPacket: UploadMeterData: 256, Action Type: 1[0m
Creating socket...
AT+USOCR=17

+USOCR: 0

OK
U_SOCK: socket created, descriptor 43, network handle 0x3ffca660, socket handle 43.
Socket created: 43
```
The returned socket in the `AT+USOCR=17` response is clearly 0. Why is `uSockCreate(`) returning 43?

Look like a discrepancy between the host memory and the acutal ip stack on the module itself..

`uSockDeinit()` is called before creating a new one..

```
uSockDeinit();
uPortLog("Creating socket...\n");
sock = uSockCreate(devHandle,
                   U_SOCK_TYPE_DGRAM,
                       U_SOCK_PROTOCOL_UDP);
uSockBlockingSet(sock, false);
uPortLog("Socket created: %u \n", sock)
```

RobMeades, 2024-02-21:
Hi again: the socket descriptor returned by [uSockCreate()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/sock/api/u_sock.h#L374) is the descriptor used by the `ubxlib` `uSock` API, not the value returned by the module: you can [see here](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/sock/src/u_sock.c#L1084) it is the next incrementing integer: the `uSock` API keeps a list of open sockets and maps them to the module's socket ID.

I will update the description of `uSockCreate()` to state this, hopefully avoiding any confusion.

RobMeades, 2024-03-04:
I'm going to close this one now: please feel free to re-open it if there is more to discuss, or open a new issue.

closed 2024-03-04 by RobMeades