u-blox/ubxlib #201: PSM and EDRX negotiating issues

elektron314, 2024-02-14:
Hello, I have such a system, SARA-R510M8S-01B + nRF5340 and I'm trying to negotiate PSM or E-DRX modes.
Let me explain what I get from the modem.

Almost all the required code to set uCellPwrSet3gppPowerSavingCallback and uCellPwrSetRequested3gppPowerSaving is copied from cell_power_saving_3gpp_main.c. These procedures come first in the code from booting up the OS, so there is no other stuff with the modem before them. I also have a timer of 30 seconds waiting for the callback from the network about this negotiation.

Running this code I can see this:
```
## Requesting 3GPP power saving with active time 6 seconds, periodic wake-up 120 seconds...
## 3GPP power saving is off, active time -1 seconds, periodic wake-up -1 seconds.
```

How can I understand the reason why the network almost immediately responds with the -1?

RobMeades, 2024-02-14:
Hi Aleksandr: could be many reasons - would it be possible to post the AT command exchange that `ubxlib` prints out here?  It might tell us more about the responses the module is getting from the network.

elektron314, 2024-02-14:
sure, here you go
```
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

03.15,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=1,1300

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 1,,,"00011000","00011110"

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
AT+URAT?

+URAT: 7

OK
U_CELL_CFG: RATs are:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
AT+CPSMS?

+CPSMS: 1,,,"00011000","00011110"

OK
AT+CPSMS=1,,,"10000100","00000011"

OK
U_CELL_PWR: requested PSM on, requested TAU time 120 second(s), requested active time 6 second(s).
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

234500024845556

OK
AT+CIMI

234500024845556

OK
U_CELL_NET: APN from database is "jtm2m".
AT+CGDCONT=1,"IP","jtm2m"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
<inf> ublox: ## Opened device with return code 0.

<inf> ublox: ## Requesting 3GPP power saving with active time 6 seconds, periodic wake-up 120 seconds...

<inf> ublox: ### Bringing up the network...

AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
0: NReg
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?

+CEREG: 4,0

OK
0: NReg

+CREG: 7,"2712","0311be5b",7

+CEREG: 5,"2712","0311be5b",7,,,"11100000",
7: RegS
7: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"Telia EE",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"10.11.110.179"
U_CELL_NET: connected after 2 second(s).
<inf> ublox: ## 3GPP power saving is off, active time -1 seconds, periodic wake-up -1 seconds.

U_CELL_SOCK: looking up IP address of "www.google.com".
AT
AT

OK
AT+CGDCONT?

+CGDCONT: 1,"IP","jtm2m.mnc050.mcc234.gprs","10.11.110.179",0,0,0,2,0,0,0,0,0,0,0

OK
AT+UDNSRN=0,"www.google.com"

+UDNSRN: "142.250.75.228"

OK
U_CELL_SOCK: found it at "142.250.75.228".
<inf> ublox: ### Unable to switch 3GPP power saving on!

<inf> ublox: ### Looking up server address...

<inf> ublox: ### www.google.com is:
<inf> ublox: IPV4
<inf> ublox:  
<inf> ublox: 142
<inf> ublox: .
<inf> ublox: 250
<inf> ublox: .
<inf> ublox: 75
<inf> ublox: .
<inf> ublox: 228
<inf> ublox: 

AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"Telia EE",7

OK
U_CELL_NET: operator is "Telia EE".
AT+COPS=3,2

OK
AT+COPS?

+COPS: 0,2,"24801",7

OK
U_CELL_NET: MCC/MNC is 248/1.
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"10.11.110.179"

OK
U_CELL_NET: IP address "10.11.110.179".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"jtm2m.mnc050.mcc234.gprs","10.11.110.179.255.0.0.0","10.0.0.1","8.8.8.8","8.8.4.4","","",0,0,1500,0,0

OK
U_CELL_NET: APN is "jtm2m.mnc050.mcc234.gprs".
AT+CGSN

352709570167338

OK
U_CELL_INFO: IMEI is 352709570167338.
AT+UPING="www.google.com",10,64,10000,64

OK

+UUPING: 1,64,"www.google.com","142.250.75.228",104,153

+UUPING: 2,64,"www.google.com","142.250.75.228",104,162

+UUPING: 3,64,"www.google.com","142.250.75.228",104,129
<inf> ublox: Registered on "Telia EE".

<inf> ublox: The MCC/MNC of the network is 2481.

<inf> ublox: Our IP address is "10.11.110.179".

<inf> ublox: The APN used was "jtm2m.mnc050.mcc234.gprs".

<inf> ublox: Get IMEI returned with return code 0.

<inf> ublox: Imei: 352709570167338.

+UUPING: 4,64,"www.google.com","142.250.75.228",104,176

+UUPING: 5,64,"www.google.com","142.250.75.228",104,137

+UUPING: 6,64,"www.google.com","142.250.75.228",104,177

+UUPING: 7,64,"www.google.com","142.250.75.228",104,144

+UUPING: 8,64,"www.google.com","142.250.75.228",104,148

+UUPING: 9,64,"www.google.com","142.250.75.228",104,165

+UUPING: 10,64,"www.google.com","142.250.75.228",104,162
<inf> ublox: ### Done.
```

RobMeades, 2024-02-14:
Thanks.  These are the commands which set the module up to request 3GPP power-saving:

```
AT+UPSMR=1
OK
AT+CPSMS?
+CPSMS: 1,,,"00011000","00011110"
OK
```

The answer from the network comes back on the end of the `+CEREG` URC (see section 14.19.3 of the SARA-R5 AT manual), for which you are getting:

```
+CEREG: 5,"2712","0311be5b",7,,,"11100000",
```

The "Assigned Active Time" is `11100000` (3GPP coding): the lower five bits are the time value, the upper three bits the unit that time value is in, where a "unit" of `111` (7), unfortunately, means "deactivated".

As to why the network is not letting you have 3GPP power saving, your first port of call should probably be to enquire with them. @philwareublox: do you know of any common reasons?

philwareublox, 2024-02-15:
+CEREG: 5 means ROAMING. I have found networks only allow eDRX and PSM features on Home SIMs. The MNO provide should be able to confirm this is the case.

elektron314, 2024-02-15:
Thanks for this detailed analysis! It REALLY helped. I talked with the SIM card service support.

I'm in Estonia and I'm using a 1oT SIM card. We have had their SIMs for a while and there is a fact that they don't support LTE-M / NB-IoT technologies that allow usage of PSM/eDRX.

So now need to find a solution that supports this technologies and then I hope I'll get things done.

I'll leave my comments about the progress anyway here.

elektron314, 2024-03-27:
I have just received new SIMs, and my provider says that they are now LTE-M and should be compatible with PSM/eDRX.

But the problem is that I still can't activate PSM/eDRX:

```
AT+CEDRXS=2,4,"1010","0000"

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"public4.m2minternet.com.mnc004.mcc204.gprs","10.22.255.142.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: already connected.
[00:01:33.048,950] <inf> ublox: ## Requesting E-DRX of 328 seconds...

rtt:~$ [00:01:33.176,422] <inf> ublox: ### Bringing up the network...

rtt:~$ rtt:~$ U_CELL_SOCK: looking up IP address of "www.google.com".
AT
AT

OK
AT+CGDCONT?

+CGDCONT: 1,"IP","public4.m2minternet.com.mnc004.mcc204.gprs","10.22.255.142",0,0,0,2,0,0,0,0,0,0,0

OK
AT+UDNSRN=0,"www.google.com"

+UDNSRN: "216.58.201.100"

OK
U_CELL_SOCK: found it at "216.58.201.100".
[00:02:03.306,518] <inf> ublox: ### Unable to switch E-DRX on!
```

This is his answer about the roaming for now:

> The 1oT SIM is a Roaming SIM, it operates with the roaming options enabled, regardless of the location to provide seamless global connectivity. 
> 
> The recent SIMs that you received are LTE-M compatible with Elisa, Estonia. For these SIMs to operate seamlessly, roaming needs to be enabled and the APN needs to be set as terminal.apn. 
> 
> ​These are the basic steps that need to be pre-configured for the SIM to operate.

RobMeades, 2024-03-27:
Hi again Aleks.

> But the problem is that I still can't activate PSM/eDRX:

The PSM activation commands, i.e.:

```
AT+UPSMR=1
OK
AT+CPSMS?
+CPSMS: 1,,,"00011000","00011110"
OK
```
...are not in the AT log fragment you have posted, it looks as though in this case you are requesting EDRX instead with:

```
AT+CEDRXS=2,4,"1010","0000"
OK
```

Once you are registered with the network, the EDRX values that the network has assigned can be read with [uCellPwrGetEDrx()](https://github.com/u-blox/ubxlib/blob/66627474fbb6435d71c73802ac1b829189280835/cell/api/u_cell_pwr.h#L832): could you post here the AT exchange that results when you call that API after having requesting EDRX?  You should get a URC from the module, `+CEDRXP`, when the network changes the assigned EDRX values (potentially on any cell change) but you might only get that on a change rather than on initial activation.

Did you try the PSM activation commands and then look for the response from the network on the end of the `+CEREG:` URC, as you did the first time?  You were previously getting `11100000`:

```
+CEREG: 5,"2712","0311be5b",7,,,"11100000",
```

...meaning deactivated: has that binary-coded number changed at all?

RobMeades, 2024-03-27:
> Any idea why the APN is changing to something more sophisticated and how to keep the original one?

This is just something networks do I'm afraid, just to confuse us I guess. It is nothing to worry about, the APN is fine, it is the one you asked for.

RobMeades, 2024-03-28:
Ah, yes, that might explain it.  See below for the parameter you want; manual network selection has always been possible down at the `uCellNet` API level, but we didn't export the ability up to the `uNetwork` API level until a couple of months ago; you might need to refresh your code:

https://github.com/u-blox/ubxlib/blob/b7e8edc93f4e1311a942a33b8dbedb77f2bfd2dd/common/network/api/u_network_config_cell.h#L102-L106

elektron314, 2024-03-28:
Thanks!

But now I'm really stuck. The code is updated, operator is M1 but something still keeps me from power saving.
Here is the excerpt from my log.

```
AT+URAT?

+URAT: 7

OK
U_CELL_CFG: RATs are:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
AT+UPSMVER?

+UPSMVER: 0

OK
AT+CEDRXS=2,4,"0100","0010"

OK
[00:01:18.784,881] <inf> ublox: ### The APN is "terminal.apn.mnc004.mcc204.gprs".

rtt:~$ [00:01:18.847,686] <inf> ublox: ## Requesting E-DRX of 60 seconds...

rtt:~$ rtt:~$ 
+CREG: 7,"07e5","0f469715",7

+CEREG: 5,"07e5","0f469715",7,,,"00001000",
7: RegS
7: RegR

+UUPSMR: 1
U_CELL_SOCK: looking up IP address of "www.google.com".
[00:01:48.977,020] <inf> ublox: ### Unable to switch E-DRX on!
```

RobMeades, 2024-03-28:
That is certainly a step forward; the response from the network no longer says "deactivated:

```
+CEREG: 5,"07e5","0f469715",7,,,"00001000",
```

The value `00001000` is made up of the "value" part in the top 5 bits and the "units" part in the bottom 3 bits.  So you have a value of `00001`, which is 16 (bit-reversed), and a "units" part of 0, which means 2-second units, so that's an active time of 32 seconds.

The example code is waiting for the module to send a URC, `+CEDRXP:`, which would result in a call to your `callback` function, but it might be that a data transfer has to happen or the like for that to be triggered (may depend on the network, whether the RRC connection had already been dropped at the time, etc.).

What you could do instead is call [uCellPwrGetEDrx()](https://github.com/u-blox/ubxlib/blob/b7e8edc93f4e1311a942a33b8dbedb77f2bfd2dd/cell/api/u_cell_pwr.h#L832-L858), that would be another way of confirming what has been agreed.  That or you could maybe stimulate some data transfer and then the URC might be emitted.

But it certainly looks to me as though you have probably succeeded.

RobMeades, 2024-03-28:
The example is now updated, see https://github.com/u-blox/ubxlib/commit/3da65fc6f19b1a5c410a3a8f0232462ab78b0ebe.

elektron314, 2024-04-02:
Hello again Rob!

I'm still struggling to get my eDRX working with no luck

I have added a piece of code checking eDRX parameters like from the sample you've updated with uCellPwrGetEDrx.
I have a connection with CAT-M1 network.
I have -01B-00 modem 22/13.
But still not set eDRX.

Below is a part of the log with the start of the eDRX-related stuff until the answer is that there's nothing about that.
This log is single continuous output, I just inserted my comments here to clarify some things.

```
AT+CEDRXS=2,4,"0001","0000"

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.249.26.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
AT+CFUN=4

OK

+CREG: 0

+CEREG: 0

+CGREG: 0

+UUPSDD: 0
-1: NReg
-1: NReg
-1: NReg

AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CGATT?

+CGATT: 0

OK
-1: NReg
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

204046996721715

OK
U_CELL_NET: user-specified APN is "terminal.apn".
AT+CGDCONT=1,"IP","terminal.apn"

OK
AT+UAUTHREQ=1,0,"",""

OK
AT+CFUN=1

OK
U_CELL_NET: registering on 24802...
AT+COPS=1,2,"24802"

OK

+CREG: 7,"07e5","0f469715",7

+CEREG: 5,"07e5"7: RegS
,"0f469715",7,,,"00001000",
7: RegR
AT+CREG?

+CREG: 2,7,"07e5","0f469715",7

OK
7: RegS
AT+COPS=3,0

OK
AT+COPS?

+COPS: 1,0,"Elisa EE",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"10.22.247.249"
AT+CEER

+CEER: "EMM cause",0,"No cause information available"

OK
U_CELL_NET: connected after 5 second(s).
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"10.22.247.249"

OK
U_CELL_NET: IP address "10.22.247.249".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.247.249.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: primary DNS address:   "195.10.102.11".
U_CELL_NET: secondary DNS address: "213.137.31.22".

+UUPSMR: 1
```
And here starts the loop waiting for eDRX to set
```
                    while (!gEDrxSet && (x < 30)) {
                        uPortTaskBlock(1000);
                        x++;
                    }
```
But I have a question to everything that is going on here.
1. What does mean **+UUPSMR: 1** just right before next output? Does it mean that the modem entered eDRX mode?
```
ATE0
ATE0
ATE0
```
2. Above is weird as well. What does it mean and why is it sending?
3. And below is more interesting. **+UUPSMR: 0** which means that the module is leaving eDRX mode, right?
4. Moreover, what is going on? Is it reset and loading again?
```
+UUPSMR: 0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

03.15,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=1,1300

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 1,,,"10000100","00000011"

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"10.22.247.249"
AT+CEDRXRDP

+CEDRXRDP: 0,"","",""

OK
U_CELL_PWR: for RAT 10 assigned E-DRX is off, -1 second(s), paging window -1 second(s).
### Unable to switch E-DRX on!
```
As wee see, no eDRX.

philwareublox, 2024-04-02:
Hi,

We see UBXLIB sending the AT+CEDRXS command to "request" eDRX to the network. Later we see UBXLIB checking what the network has configured with AT+CEDRXRDP.
AT+CEDRXRDP returns no settings, so I suspect this network/SIM configuration does not allow eDRX. 

If you are able to manually control the cellular module you can check this yourself by sending AT+CEDRXS and then checking the network's response with AT+CEDRXRDP. If you are doing this you can also enable +CSCON URC to check the module is sending the request to the network. 

AT+CSCON=1
OK

AT+CEDRXS=.....
OK

+CSCON: 1          # Module communicates edrx settings request to network.
+CSCON: 0

AT+CEDRXRDP
+CEDRXRDP: .....
OK

Regards,
Phil.

RobMeades, 2024-04-02:
`CSCON` URC will be on if you have set a [uCellNetSetBaseStationConnectionStatusCallback()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/api/u_cell_net.h#L704).

RobMeades, 2024-04-03:
Looking at the detail of the first part of your log, it looks pretty much the same as you had before (the print is broken up by the `7: RegS` print from the asynchronous URC handler task, which I have removed here):

```
+CEREG: 5,"07e5","0f469715",7,,,"00001000",
```

...indicates that E-DRX has been agreed with the network, active time 32 seconds.  Then you get `+UUPSMR: 1` which indicates that [the protocol stack in the module has prepared to go to sleep](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_pwr.c#L865) but the module _may_ not have entered deep sleep; you can never actually tell when it has from the AT interface.

For the remainder I will take you through what I see and where it seems the code is going and would recommend you poke around with some debug prints etc. to determine what is going on.

The sequence beginning `ATE0` seems to be the `ubxlib` code waking the module up from "cold", which I guess must be [when the loop has exited and uCellPwrGetEdrx() is being called](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/example/cell/power_saving/cell_power_saving_e_drx_main.c#L269); you could add a call to [uCellPwrGet3gppPowerSavingState()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/api/u_cell_pwr.h#L690) and print the result when you exit the loop to confirm this.  Since more than 6 seconds have passed, the `ubxlib` AT client knows that it has to wake the module up from 32 kHz (AKA "UART") sleep; usually at this point I would expect to see `AT` a few times as it prods the module into full wakefulness using on the UART.  However, in this case it seems to be sending `ATE0` which means that we are getting to [uCellPwrPrivateOn](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_pwr.c#L1421) which, logically, can only occur if `uCellPwrOn()` has been called (which it has not) or we got to [deepSleepWakeUp()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_private.c#L547).

Following the code through from the top, the AT client will call [uCellPrivateWakeUpCallback()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_private.c#L1145) (which is the only function that calls [deepSleepWakeUp()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_private.c#L547)).  [uCellPrivateWakeUpCallback()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_private.c#L1145) always calls [uCellPrivateIsDeepSleepActive()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_private.c#L1165) first to determine how sleepy we were.  [uCellPrivateIsDeepSleepActive()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_private.c#L1165) will only return `true` if (a) 3GPP power saving has been agreed and (b) VINT has been connected (i.e. you specified a non-negative `pinVint`  in [uDeviceCfgCell_t](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/common/device/api/u_device.h#L310)) and (c) `VINT` is low, indicating that the module has powered itself down, which should not be the case here.  However it does _seem_ to be the case, because that would result in  [deepSleepWakeUp()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/src/u_cell_private.c#L547) being called, which is what we are seeing.

So to find out what is going on, I recommend scattering debug prints around the code in the path mapped above; debug prints should not affect the module itself in this case and so should not, themselves, affect the problem.  In particular, assuming you have connected `VINT`, we should _never_ see it go low here, it would be interesting to know if that has actually gone low.

RobMeades, 2024-04-03:
Could you possible paste here the initial part of the log above?  I can see that it says `AT+CEDRXS=2,4,"0001","0000"`, which is setting up E-DRX, the thing that is confusing me is that the symptoms (`+UUPSMR: 1`) suggest that 3GPP Power Saving is being set _as_ _well_ as E-DRX.  That is not how the example configures the code, there should be an `AT+CPSMS=0` earlier on to indicate that 3GPP Power Saving is not being requested

E-DRX makes sense as an "occasional" sleep mechanism for a relatively active device that wants to save power, with no intervening deep-sleep that would lose all RAM contents (see [advice here](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/api/u_cell_pwr.h#L34)), 3GPP power saving on the other hand will lose all RAM/context in the module and the module will take longer to power back up again, it is intended for "wake up once a day" kind of devices.  I thought you were running the [E-DRX example](https://github.com/u-blox/ubxlib/blob/master/example/cell/power_saving/cell_power_saving_e_drx_main.c) but if you are enabling 3GPP Power Saving as well as E-DRX then you would get the behaviour above, i.e. the module really has gone into deep sleep and it needs to be woken up "from scratch".

elektron314, 2024-04-03:
This is the piece of log just right before the previous part
```
rtt:~$ *** Booting Zephyr OS build v3.2.99-ncs2 ***
uPortLog Test 314
U_CELL: initialising with enable power pin not connected, PWR_ON pin 2 (0x02) (and is toggled from 1 to 0) and VInt pin 24 (0x18) (and is 1 when module is on).
AT
AT
+UUPSMR: 0
03.15,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=1,1300

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 1,,,"10000100","00000011"

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK

+UUPSMR: 0
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

204046996721715

OK
U_CELL_NET: user-specified APN is "terminal.apn".
AT+CGDCONT=1,"IP","terminal.apn"

OK
AT+UAUTHREQ=1,0,"",""

OK
[00:00:02.351,470] <inf> ublox: ## Opened device with return code 0.

rtt:~$ rtt:~$ AT+CFUN=1

OK
U_CELL_NET: registering on 24802...
AT+COPS=1,2,"24802"

OK

+CREG: 7,"07e5","0f469715",7

+CEREG: 5,"07e5","0f469715",7,,,"00001000",
7: RegS
7: RegR
AT

OK
AT+CREG?

+CREG: 2,7,"07e5","0f469715",7

OK
7: RegS
AT+COPS=3,0

OK
AT+COPS?

+COPS: 1,0,"Elisa EE",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"10.22.249.26"
AT+CEER

+CEER: "EMM cause",0,"No cause information available"

OK
U_CELL_NET: connected after 23 second(s).
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"10.22.249.26"

OK
U_CELL_NET: IP address "10.22.249.26".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.249.26.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: primary DNS address:   "195.10.102.11".
U_CELL_NET: secondary DNS address: "213.137.31.22".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.249.26.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: APN is "terminal.apn.mnc004.mcc204.gprs".
AT+URAT?

+URAT: 7

OK
U_CELL_CFG: RATs are:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
AT+UPSMVER?

+UPSMVER: 0

OK
```

elektron314, 2024-04-03:
By means of uCellPwrGet3gppPowerSavingState I have found that exactly right after that 30-second loop my state is U_CELL_PWR_3GPP_POWER_SAVING_STATE_ACTIVE but I don't set the 3GPP before my e-DRX settings. As I explained already the code right now is just this: load the OS -> uNetworkInterfaceUp -> code from e-drx example.

How can it be possible?

RobMeades, 2024-04-03:
I think what might have happened is that the 3GPP Power Saving test was once run on this cellular module and the module has remembered that setting (which of course it has to since it switches all RAM off once it enters deep sleep).  You can see `ubxlib` query `AT+CPSMS` (the 3GPP Power Saving mode) near power-on:

```
AT+CPSMS?
+CPSMS: 1,,,"10000100","00000011"
OK
```

...where the `1` being returned means on, see section 16.2.1 of the SARA-R5 AT manual.

So effectively you have both 3GPP Power Saving _and_ E-DRX on; double the fun but probably not what you wanted.

Adding a call to [uCellPwrSetRequested3gppPowerSaving()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/api/u_cell_pwr.h#L525) with `onNotOff` set to `false` just after `uDeviceOpen()` returns, at least once, should revert the setting in the cellular module:

```
uCellPwrSetRequested3gppPowerSaving(devHandle, U_CELL_NET_RAT_CATM1, false, 0, 0);
```

elektron314, 2024-04-04:
Thank you. Yes, I was experimenting with both power-saving modes in the beginning.

Did it! That changed CPSMS to 0 but that's strange that the modem remembers this state after reset which seems unnatural because RAM should be cleared after reset.

But anyway, my device still struggling to get any positive answer from the network about settling eDRX parameters down. How can I help it now?

After that 30-seconds loop of checking if eDRX is set modem's power state is U_CELL_PWR_3GPP_POWER_SAVING_STATE_AVAILABLE.

RobMeades, 2024-04-04:
> Did it! That changed CPSMS to 0 but that's strange that the modem remembers this state after reset which seems unnatural because RAM should be cleared after reset.

Good stuff!  The issue with this particular command is that, for the kind of deep sleep the module can enter as a result of 3GPP power saving, only the RTC is left running, the entire module is otherwise effectively off, so it has to store the fact that it was asked to enter 3GPP power saving mode in NVRAM, which of course persists across any reset.

> After that 30-seconds loop of checking if eDRX is set modem's power state is `U_CELL_PWR_3GPP_POWER_SAVING_STATE_AVAILABLE`.

Could you paste the debug output into here, so that I can see what is happening now that 3GPP power saving is not getting in the way?

elektron314, 2024-04-04:
```

rtt:~$ *** Booting Zephyr OS build v3.2.99-ncs2 ***
uPortLog Test 314
U_CELL: initialising with enable power pin not connected, PWR_ON pin 2 (0x02) (and is toggled from 1 to 0) and VInt pin 24 (0x18) (and is 1 when module is on).
AT
AT

OK
U_CELL_PWR: powering on, module is already on.
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

03.15,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=1,1300

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 0,,,"01100000","00000000"

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK

+CREG: 0

+CEREG: 0

+CGREG: 0

+UUPSDD: 0

AT+CPSMS?

+CPSMS: 0,,,"01100000","00000000"

OK
AT+CPSMS=0,,,"01100000","00000000"

OK
U_CELL_PWR: requested PSM off, requested TAU time 0 second(s), requested active time 0 second(s).
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

204046996721715

OK
U_CELL_NET: user-specified APN is "terminal.apn".
AT+CGDCONT=1,"IP","terminal.apn"

OK
AT+UAUTHREQ=1,0,"",""

OK
AT+CFUN=1

OK
U_CELL_NET: registering on 24802...
AT+COPS=1,2,"24802"

OK

+CREG: 7,"07e5","0f469715",7

+CEREG: 5,"07e5","0f469715",7
7: RegS
7: RegR
AT+CREG?

+CREG: 2,7,"07e5","0f469715",7

OK
7: RegS
AT+COPS=3,0

OK
AT+COPS?

+COPS: 1,0,"Elisa EE",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"10.22.249.94"
AT+CEER

+CEER: "EMM cause",0,"No cause information available"

OK
U_CELL_NET: connected after 3 second(s).
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"10.22.249.94"

OK
U_CELL_NET: IP address "10.22.249.94".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.249.94.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: primary DNS address:   "195.10.102.11".
U_CELL_NET: secondary DNS address: "213.137.31.22".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.249.94.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: APN is "terminal.apn.mnc004.mcc204.gprs".
AT+URAT?

+URAT: 7

OK
U_CELL_CFG: RATs are:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
AT+UPSMVER?

+UPSMVER: 0

OK
AT+CEDRXS=2,4,"0001","0000"

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.249.94.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
AT+CFUN=4

OK

+CREG: 0

+CEREG: 0

+CGREG: 0

+UUPSDD: 0
-1: NReg
-1: NReg
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CGATT?

+CGATT: 0

OK
-1: NReg
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

204046996721715

OK
U_CELL_NET: user-specified APN is "terminal.apn".
AT+CGDCONT=1,"IP","terminal.apn"

OK
AT+UAUTHREQ=1,0,"",""

OK
AT+CFUN=1

OK
U_CELL_NET: registering on 24802...
AT+COPS=1,2,"24802"

OK

+CREG: 7,"07e5","0f469715",7

+CEREG: 5,"07e7: RegS
5","0f469715",7
7: RegR
AT+CREG?

+CREG: 2,7,"07e5","0f469715",7

OK
7: RegS
AT+COPS=3,0

OK
AT+COPS?

+COPS: 1,0,"Elisa EE",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"10.22.247.229"
AT+CEER

+CEER: "EMM cause",0,"No cause information available"

OK
U_CELL_NET: connected after 3 second(s).
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"10.22.247.229"

OK
U_CELL_NET: IP address "10.22.247.229".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.247.229.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: primary DNS address:   "195.10.102.11".
U_CELL_NET: secondary DNS address: "213.137.31.22".
AT
AT

OK
AT+CEDRXRDP

+CEDRXRDP: 0,"","",""

OK
U_CELL_PWR: for RAT 10 assigned E-DRX is off, -1 second(s), paging window -1 second(s).
### Unable to switch E-DRX on!

```

RobMeades, 2024-04-04:
Agreed, as far as I can see the cellular module is being configured to request E-DRX (`AT+CEDRXS=2,4,"0001","0000"`) but the network has not assigned E-DRX (`AT+CEDRXRDP` returns `0`).  Since E-DRX won't do anything anyway unless you have a base station connection active, you could, maybe, try stimulating some connected-mode activity, e.g. make a call to [uCellSockGetHostByName()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/api/u_cell_sock.h#L539) :

```
    uSockIpAddress_t ipAddress;
    uCellSockGetHostByName(devHandle, "1.2.3.4", &ipAddress);
```

...after `uNetworkInterfaceUp()` has returned and, maybe, before this, just after `uDeviceOpen()` returns, set up a [uCellNetSetBaseStationConnectionStatusCallback()](https://github.com/u-blox/ubxlib/blob/a2d986e2944141176eedaee6dc4be87be7eea57b/cell/api/u_cell_net.h#L704); it doesn't need to do anything, all we want to do is see the `+CSCON:` URCs indicating that the radio connection did go up and down again:

```
static void connectionStatusCallback(bool isConnected, void *pParam)
{
    (void) isConnected;
    (void) pParam;
}

void main()
{

    ...

    // After uDeviceOpen() has returned and before uNetworkInterfaceUp() is called
    uCellNetSetBaseStationConnectionStatusCallback(devHandle, connectionStatusCallback , NULL);

    '''
}
```

I guess it is _possible_ that the network supports 3GPP Power Saving (that was clearly working) but _doesn't_ support E-DRX?

Any other suggestions @philwareublox?

elektron314, 2024-04-05:
I perhaps didn't quite get you well but I put 5 instances of
```
uCellSockGetHostByName(dev_handle, "1.2.3.4", &ipAddress);
uCellNetSetBaseStationConnectionStatusCallback(dev_handle, connectionStatusCallback, NULL);
```
to the places you pointed out and 2 before setting eDRX as I have uNetworkInterfaceUp before the function from the example setting eDRX.

Anyway, here is the log. I don't know if it helps.
```
U_CELL: initialising with enable power pin not connected, PWR_ON pin 2 (0x02) (and is toggled from 1 to 0) and VInt pin 24 (0x18) (and is 1 when module is on).
AT
AT

OK
U_CELL

OK
AT+UDCONF=1,0

OK
ATI9

03.15,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=1,1300

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 0,,,"01100000","00000000"

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK

+CSCON: 1

+CSCON: 0

+CREG: 0

+CEREG: 0

+CGREG: 0

+UUPSDD: 0
U_CELL_SOCK: looking up IP address of "1.2.3.4".
AT+CGDCONT?

+CGDCONT: 1,"IP","terminal.apn","0.0.0.0",0,0,0,2,0,0,0,0,0,0,0

OK
AT+UDNSRN=0,"1.2.3.4"

+CME ERROR: operation not allowed
AT+USOER

+USOER: 1

OK

AT+CGDCONT?

+CGDCONT: 1,"IP","terminal.apn","0.0.0.0",0,0,0,2,0,0,0,0,0,0,0

OK
AT+UDNSRN=0,"1.2.3.4"

+CME ERROR: operation not allowed
AT+USOER

+USOER: 1

OK
U_CELL_SOCK: host not found.
AT+CSCON=1

OK
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

204046996721715

OK
U_CELL_NET: user-specified APN is "terminal.apn".
AT+CGDCONT=1,"IP","terminal.apn"

OK
AT+UAUTHREQ=1,0,"",""

OK
AT+CFUN=1

OK
U_CELL_NET: registering on 24802...
AT+COPS=1,2,"24802"

OK

+CSCON: 1

+CREG: 7,"07e5","0f469715",7

+CEREG: 5,"07e5","0f469715",7
7: RegS
7: RegR
AT+CREG?

+CREG: 2,7,"07e5","0f469715",7

OK
7: RegS
AT+COPS=3,0

OK
AT+COPS?

+COPS: 1,0,"Elisa EE",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"10.22.247.252"
AT+CEER

+CEER: "EMM cause",0,"No cause information available"

OK
U_CELL_NET: connected after 4 second(s).
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"10.22.247.252"

OK
U_CELL_NET: IP address "10.22.247.252".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.247.252.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: primary DNS address:   "195.10.102.11".
U_CELL_NET: secondary DNS address: "213.137.31.22".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.247.252.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: APN is "terminal.apn.mnc004.mcc204.gprs".
U_CELL_SOCK: looking up IP address of "1.2.3.4".
AT+CGDCONT?

+CGDCONT: 1,"IP","terminal.apn.mnc004.mcc204.gprs","10.22.247.252",0,0,0,2,0,0,0,0,0,0,0

OK
AT+UDNSRN=0,"1.2.3.4"

+UDNSRN: "1.2.3.4"

OK
U_CELL_SOCK: found it at "1.2.3.4".
AT+CSCON=1

OK
AT+URAT?

+URAT: 7

OK
U_CELL_CFG: RATs are:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
AT+UPSMVER?

+UPSMVER: 0

OK
AT+CEDRXS=2,4,"0001","0000"

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.247.252.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
AT+CFUN=4

OK

+CSCON: 0

+CREG: 0

+CEREG: 0

+CGREG: 0

+UUPSDD: 0
-1: NReg
-1: NReg
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg
AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg

AT+CGATT?

+CGATT: 0

OK
-1: NReg
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

204046996721715

OK
U_CELL_NET: user-specified APN is "terminal.apn".
AT+CGDCONT=1,"IP","terminal.apn"

OK
AT+UAUTHREQ=1,0,"",""

OK
AT+CFUN=1

OK
U_CELL_NET: registering on 24802...
AT+COPS=1,2,"24802"

OK

+CSCON: 1

+CREG: 7,"07e5","0f469715",7

+CEREG: 5,"07e5","0f469715",7
7: RegS
7: RegR
AT+CREG?

+CREG: 2,7,"07e5","0f469715",7

OK
7: RegS
AT+COPS=3,0

OK
AT+COPS?

+COPS: 1,0,"Elisa EE",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"10.22.245.123"
AT+CEER

+CEER: "EMM cause",0,"No cause information available"

OK
U_CELL_NET: connected after 4 second(s).
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"10.22.245.123"

OK
U_CELL_NET: IP address "10.22.245.123".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"terminal.apn.mnc004.mcc204.gprs","10.22.245.123.255.0.0.0","10.0.0.1","195.10.102.11","213.137.31.22","","",0,0,1500,0,0

OK
U_CELL_NET: primary DNS address:   "195.10.102.11".
U_CELL_NET: secondary DNS address: "213.137.31.22".

+CSCON: 0

AT
AT

OK
AT+CEDRXRDP

+CEDRXRDP: 0,"","",""

OK
U_CELL_PWR: for RAT 10 assigned E-DRX is off, -1 second(s), paging window -1 second(s).
### Unable to switch E-DRX on!
```
And I have asked my provider about the eDRX exactly. Waiting for his reply.

elektron314, 2024-04-09:
Thank you @RobMeades for all your support. It was super intense and really helpful!

My SIM card provider just told me, to their surprise as well, that eDRX is not supported by the mobile operator servicing my virtual operator and it seems that in Estonia there is no possibility to test this feature nowadays yet.

Although PSM 3GPP works.

elektron314, 2024-04-09:
We will elaborate on this question now

RobMeades, 2024-05-28:
Hi Aleksandr: do we still need to keep this issue open?

closed 2024-11-11 by manemannen