u-blox/ubxlib #202: Continuous UMQTTER 13,0 Error Loop and Improper Shutdown Behavior (SARA-R422M8S)

tommas0serafin, 2024-02-19:
Upon losing internet connection on my device, regardless of whether it's due to antenna disconnection or forced disconnection, the device starts throwing UMQTTER error code 13,0, which then transitions to 13,50 or 13,34, indicating a normal behavior. However, upon re-establishing the connection, the system enters an error loop where it fails to connect, consistently throwing UMQTTER error code 13,0. Occasionally I noticed a discrepancy between the actual connection status and the one identified by the internal callbacks (network is up but it is detected as down). This behavior is reproducible consistently.

Furthermore, I've observed that the module fails to shut down properly. After every shutdown attempt (with AT commands), the UART pins on the ESP32 side remain active, causing a current leak into the u-blox module. As a consequence the ublox module outputs 0.7v-1.1v not allowing a proper shutdown procedure. I patched this issue by resetting the gpios on the microcontroller's side but I need to understand if there's an hardware issue causing it or if it's a library bug.

- Device: [ESP32S3]
- Module: [SARA-R422M8S]
- Environment: [ESP-IDF]

Serial output upon reconnection:
```
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT

OK
ATE0
AT

OK
AT+CMEE=2
ATE0

OK

OK

OK

OK
AT+UDCONF=1,0

OK
ATI9

00.12,A00.00

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=2

OK
AT&K3

OK
AT+UPSV=4

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

295050903238865

OK
U_CELL_NET: user-specified APN is "em".
AT+CGDCONT=1,"IP","em"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

SARA-R422M8S-00B-00

ERROR

+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
0: NReg
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?

+CEREG: 4,0

OK
0: NReg
AT+CREG?

+CREG: 2,0

OK
0: NReg
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?

+CEREG: 4,0

OK
0: NReg
AT+CREG?

+CREG: 2,0

OK

+CREG: 2

+CEREG: 2,,,,,,,,
0: NReg
-1: Search
-1: Search

+CREG: 5,"4E62","00006BE6",3

3: RegR
AT+CGREG?

+CGREG: 2,2

OK
0: Search
AT+CEREG?

+CEREG: 4,4

OK
0: OoC
AT+CREG?

+CREG: 2,5,"4E62","00006BE6",3

OK
3: RegR
AT+CGREG?

+CGREG: 2,2

OK
3: Search
AT+CEREG?

+CEREG: 4,4

OK
3: OoC
AT+CREG?

+CREG: 2,5,"4E62","00006BE6",3

OK
3: RegR
AT+CGREG?

+CGREG: 2,2

OK
3: Search
AT+CEREG?

+CEREG: 4,4

OK
3: OoC
AT+CREG?

+CREG: 2,5,"4E62","00006BE6",3

OK
3: RegR
AT+CGREG?

+CGREG: 2,2

OK
3: Search
AT+CEREG?

+CEREG: 4,4

OK
3: OoC
AT+CREG?

+CREG: 2,5,"4E62","00006BE6",3

OK
3: RegR

+CGREG: 5,"4E62","00006BE6",3,"46"
3: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"vodafone IT",3

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

ERROR
AT+CGACT=1,1

+CGACT: 1,0

OK
AT+CGACT?

OK
AT+CGACT=1,1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
U_CELL_NET: connected after 15 second(s).
AT+UMQTT=2,"cssiot01.eurac.edu",1883

OK
AT+CGSN

355438111885391

OK
AT+UMQTT=0,"355438111885391"

OK
AT+UMQTT=10,120

OK
U_CELL_MQTT: trying to connect...
AT+UMQTTC=1

OK
U_CELL_MQTT: waiting for response for up to 120 second(s)...
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTER

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
```

RobMeades, 2024-02-19:
Hi there, and sorry to hear you're having so many problems.  Splitting them out:

> upon re-establishing the connection, the system enters an error loop where it fails to connect, consistently throwing UMQTTER error code 13,0.

From the trace you have pasted in above this looks as though it would be a module problem: at least, I can't see how the `ubxlib` code would cause this, unless you can see something?  If it _is_ a module problem we can try to help but it is kind of a "best effort" thing from a `ubxlib` standpoint and you might be better off raising it through the proper module support routes.

> Occasionally I noticed a discrepancy between the actual connection status and the one identified by the internal callbacks (network is up but it is detected as down). This behavior is reproducible consistently.

This may, of course, be a `ubxlib` issue - it would imply that we have missed a registration URC, which would be interesting.  if you are able to a log of the `ubxlib` debug output at the time that would be very interesting.

> Furthermore, I've observed that the module fails to shut down properly. After every shutdown attempt (with AT commands), the UART pins on the ESP32 side remain active, causing a current leak into [...]

When we close a UART we expect the platform code to put any pins into an "off" state but it might well be that not all platforms do this.  We could "grab" any pins that were once UART pins, making the, GPIOs and then put them into some form of "off" state (let them float high?).  This would be quite difficult to test of course, not something our automated system is built to do.  Or we could add a note somewhere that this is something to look out for at application level?  I'm always a little wary making a behavioural change to code that has been out there for several years.

tommas0serafin, 2024-02-19:
Hi Rob,
Thanks for your quick reply. For what concerns the connection status discrepancy the situation would be similar to the debug output I'm sharing. In this case the module is sending the mqtt messages but is not able to receive them. The defined callback in the struct uMqttClientConnection_t which checks uDeviceHandle_t detects the network as down and I receive these UMQTTER 13,0 while the connection seems to be up.
```
U_CELL: initialising with enable power pin not connected, PWR_ON pin 15 (0x0f) (and is toggled from 1 to 0) and VInt pin 7 (0x07) (and is 1 when module is on).
I (320600) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (320610) gpio: GPIO[7]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

00.12,A00.00

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=2

OK
AT&K3

OK
AT+UPSV=4

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

295050903238865

OK
U_CELL_NET: user-specified APN is "em".
AT+CGDCONT=1,"IP","em"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
0: NReg
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?

+CEREG: 4,0

OK
0: NReg
AT+CREG?

+CREG: 2,0

OK
0: NReg
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?

+CEREG: 4,0

OK
0: NReg
AT+CREG?

+CREG: 2,0

OK
0: NReg

+CREG: 2

+CEREG: 2,,,,,,,,
-1: Search
-1: Search

+CREG: 5,"4E62","00006BE6",3

3: RegR
AT+CGREG?

+CGREG: 2,2

OK
0: Search
AT+CEREG?

+CEREG: 4,4

OK
0: OoC
AT+CREG?

+CREG: 2,5,"4E62","00006BE6",3

OK
3: RegR
AT+CGREG?

+CGREG: 2,2

OK
3: Search
AT+CEREG?

+CEREG: 4,4

OK
3: OoC
AT+CREG?

+CREG: 2,5,"4E62","00006BE6",3

OK
3: RegR
AT+CGREG?

+CGREG: 2,2

OK
3: Search
AT+CEREG?

+CEREG: 4,4

OK
3: OoC
AT+CREG?

+CREG: 2,5,"4E62","00006BE6",3

OK
3: RegR
AT+CGREG?

+CGREG: 2,2

OK
3: Search
AT+CEREG?

+CEREG: 4,4

OK
3: OoC
AT+CREG?

+CREG: 2,5,"4E62","00006BE6",3

OK
3: RegR
AT+CGREG?

+CGREG: 2,2

OK
3: Search
AT+CEREG?

+CEREG: 4,4

OK
3: OoC

+CGREG: 5,"4E62","00006BE6",3,"46"
3: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"vodafone IT",3

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,0

OK
AT+CGACT=1,1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
U_CELL_NET: connected after 12 second(s).
AT+UMQTT=2,"xxxxxxxxx",1883

OK
AT+CGSN

355438111885391

OK
AT+UMQTT=0,"355438111885391"

OK
AT+UMQTT=10,120

OK
U_CELL_MQTT: trying to connect...
AT+UMQTTC=1

OK
U_CELL_MQTT: waiting for response for up to 120 second(s)...
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTER

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
I (340590) GNSS: Bringing up GNSS
AT+CMUX=0,0,,128

OK
ATE0

OK
AT+CMEE=2

OK
AT+UPSV?

+UPSV: 4 

OK
AT+UPSV=0

OK
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type Virtual Serial.
ATI

SARA-R422M8S-00B-00

OK
AT+UGIND=0

OK
AT+UGPS?

+UGPS: 0

OK

+UUMQTTC: 1,1
AT+UGPS=1,15,127

OK
I (343970) GNSS: Starting continuous location.

AT+UGPS?

+UGPS: 1,0

OK
AT+UGPS?

+UGPS: 1,0

OK
AT+UGPS=1,1

OK
E (344420) GNSS: Local Aid is off!
U_GNSS: sent command b5 62 06 08 00 00 0e 30.
U_GNSS: decoded UBX response 0x06 0x08: e8 03 01 00 01 00 [body 6 byte(s)].
U_GNSS: sent command b5 62 06 01 02 00 01 07 11 3a.
U_GNSS: decoded UBX response 0x06 0x01: 01 07 00 00 00 00 00 00 [body 8 byte(s)].
U_GNSS: sent command b5 62 06 01 08 00 01 07 01 00 00 00 00 00 18 e2.
U_GNSS: decoded UBX response 0x05 0x01: 06 3e [body 2 byte(s)].
I (345000) GNSS: Continuous location started!
AT+UMQTTC=4,1,"xxxxxxxxx"

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=9,1,0,"xxxxxxxxx",1

+UUMQTTC: 4,1,1,"xxxxxxxxx"

>[00]

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.

+UUMQTTC: 6,1

AT+UMQTTC=5,"xxxxxxxxx"

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=4,1,"xxxxxxxxx"

+CME ERROR: Operation not allowed

AT+UMQTTER

+UUMQTTC: 5,1

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=9,1,0,"xxxxxxxxx",1

>[00]

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.

+UUMQTTC: 9,1
AT+UMQTTC=5,"xxxxxxxxx"

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.

AT+UMQTTC=4,1,"xxxxxxxxx"

+CME ERROR: Operation not allowed

AT+UMQTTER

+UUMQTTC: 5,1

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=9,1,0,"xxxxxxxxx",1

>[00]

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
```
On the other hand, for what concerns the hardware access/control (GPIOs), I think the behavior differs according to the module used (at least that's what I understood from the datasheet). I do not exclude the chance of having some hardware issues on the board I'm currently working on, I was more interested in understanding if you were aware of this behavior.

As I said there might be some hardware (wiring/powering) problems leading to these errors, my biggest concern is the connection status one.

RobMeades, 2024-02-19:
> the situation would be similar to the debug output I'm sharing

Understood: I think it depends on what is meant by "up".  From this part of the AT trace:

```
AT+COPS?
+COPS: 0,0,"vodafone IT",3
OK
AT+CGATT?
+CGATT: 1
OK
AT+CGACT?
+CGACT: 1,0
OK
AT+CGACT=1,1
OK
AT+CGACT?
+CGACT: 1,1
OK
AT+UPSD=0,0,0
OK
AT+UPSD=0,100,1
OK
```

...the module is registered with the network, has a PDP context activated and the internal "profile" within the module has been connected to that context; hence all of the internal clients within the module (IP, MQTT, HTTP) should be able to connect.  For some reason the MQTT client within the module is stuck in a state where it does not.  Let me see if I can find out internally if this is a known issue and if there is some form of workaround that `ubxlib` could apply.

> for what concerns the hardware access/control (GPIOs) ... I was more interested in understanding if you were aware of this behavior.

This is not something we (where "we" means `ubxlib`) would notice, I think, power consumption in various states is not something that our test system addresses, for better or worse we leave that to the module.  As to what we should do in this particular case, I'm open to advice.

RobMeades, 2024-02-19:
A couple of questions for clarification:

I didn't read the detail of the second AT log above (the one concerning the "registration/connectedness discrepancy); you say:

> The defined callback in the struct uMqttClientConnection_t which checks uDeviceHandle_t detects the network as down and I receive these UMQTTER 13,0 while the connection seems to be up.

By this I guess you mean that, when you call [uMqttClientConnect](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L261) you set a `pKeepGoingCallback` in the [uMqttClientConnection_t](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L74) structure you pass to that function, and from that callback you call another `ubxlib` API to determine the network state - which one is that?  Obviously in this particular AT log an MQTT connection has successfully been made (`+UUMQTTC: 1,1`) , so the module is in the clear.

On the "MQTT client in the module does not connect" case, you say:

> Upon losing internet connection on my device, regardless of whether it's due to antenna disconnection or forced disconnection, the device starts throwing UMQTTER error code 13,0, which then [...]

I had assumed that the first AT log you pasted in above was an instance of this case, since there is no `+UUMQTTC: ` of any kind in it, yet in that AT log the module appears to be coming up from off-state: can you confirm that you believe the module has not connected in that AT log because when it was _previously_ powered-up, there was a coverage loss and, despite the module being powered off and on again, that loss of coverage resulted in the subsequent "lack of a +UUMQTTC` URC" behaviour of the MQTT client?

tommas0serafin, 2024-02-19:
Hey Rob,
This is the signature of the callback I'm using, it falls under uNetworkStatusCallback_t:
```
static void network_status_callback(uDeviceHandle_t devHandle,
                                    uNetworkType_t netType,
                                    bool isUp,
                                    uNetworkStatus_t *pStatus,
                                    void *pParameter)
```
Here I'm basically checking on the boolean value isUp.

For what concerns the second point, you're right this a module restart after a forced disconnection (I unplugged the antenna). This is for me the easiest way to force the connection issue. Basically after experiencing the loss of connection I close, power off and restart the module to start over. Since this problem is way less common if I restart the whole device (ublox + micro), I also though this might be a sync problem of the UART between the ublox module and the microcontroller.

Hope it helps,
Thanks in advance

tommas0serafin, 2024-02-19:
Unfortunately the hardware failed me, I'll come back tomorrow with more detailed logs (I'll add what you suggested). 

To clarify the last point, the antenna is reconnected while the module is powered off (but the micro isn't) so the whole log you read includes only the section with the antenna on.

RobMeades, 2024-02-19:
Darn.  More tomorrow then.

> To clarify the last point, 

Understood.  Makes it all the more difficult to understand though: if the antenna is reconnected before the module is powered back up that should not affect the MQTT client in the module, given that the module registers with the network.  If it is possible to take a log with `U_AT_CLIENT_PRINT_WITH_TIMESTAMP` defined for the `ubxlib` build then that might be interesting, having a view of time as well.

tommas0serafin, 2024-02-20:
Hi Rob,
I followed your advice and here's the log. I also noticed that most of the time the reason of the shutdown is related to uMqttClientIsConnected(pContext) and not the network callback. However I hope we can get something out of this log and sort things out. Hopefully for you it might be a hardware related issue.
```
U_CELL: initialising with enable power pin not connected, PWR_ON pin 15 (0x0f) (and is toggled from 1 to 0) and VInt pin 7 (0x07) (and is 1 when module is on).
I (783070) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (783080) gpio: GPIO[7]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
U_CELL_PWR: powering on.
0000/01/01 00:13:01.521: AT
0000/01/01 00:13:01.730: AT
0000/01/01 00:13:02.080: AT
0000/01/01 00:13:02.430: AT
0000/01/01 00:13:02.780: AT
0000/01/01 00:13:04.260: AT

0000/01/01 00:13:04.261: OK
0000/01/01 00:13:04.370: ATE0
0000/01/01 00:13:04.410: ATE0

0000/01/01 00:13:04.411: OK
0000/01/01 00:13:04.520: AT+CMEE=2
0000/01/01 00:13:04.560: 
0000/01/01 00:13:04.561: OK
0000/01/01 00:13:04.670: AT+UDCONF=1,0
0000/01/01 00:13:04.710: 
0000/01/01 00:13:04.711: OK
0000/01/01 00:13:04.820: ATI9
0000/01/01 00:13:04.860: 
0000/01/01 00:13:04.861: 00.12,A00.00
0000/01/01 00:13:04.861: 
0000/01/01 00:13:04.861: OK
0000/01/01 00:13:04.970: AT&C1
0000/01/01 00:13:05.010: 
0000/01/01 00:13:05.011: OK
0000/01/01 00:13:05.120: AT&D0
0000/01/01 00:13:05.160: 
0000/01/01 00:13:05.161: OK
0000/01/01 00:13:05.270: AT+UCGED=2
0000/01/01 00:13:05.310: 
0000/01/01 00:13:05.311: OK
0000/01/01 00:13:05.420: AT&K3
0000/01/01 00:13:05.470: 
0000/01/01 00:13:05.471: OK
0000/01/01 00:13:05.580: AT+UPSV=4
0000/01/01 00:13:05.620: 
0000/01/01 00:13:05.621: OK
0000/01/01 00:13:05.730: AT+UPSMR=1
0000/01/01 00:13:05.770: 
0000/01/01 00:13:05.771: OK
0000/01/01 00:13:05.880: AT+CPSMS?
0000/01/01 00:13:05.920: 
0000/01/01 00:13:05.921: +CPSMS:0,,,"01100000","00000000"
0000/01/01 00:13:05.921: 
0000/01/01 00:13:05.921: OK
0000/01/01 00:13:06.030: AT+CEDRXS?
0000/01/01 00:13:06.070: 
0000/01/01 00:13:06.071: +CEDRXS:
0000/01/01 00:13:06.071: 
0000/01/01 00:13:06.071: OK
0000/01/01 00:13:06.180: AT+UMNOPROF?
0000/01/01 00:13:06.220: 
0000/01/01 00:13:06.221: +UMNOPROF: 100
0000/01/01 00:13:06.221: 
0000/01/01 00:13:06.221: OK
0000/01/01 00:13:06.330: AT+UGPRF?
0000/01/01 00:13:06.370: 
0000/01/01 00:13:06.371: +UGPRF: 2,0,""
0000/01/01 00:13:06.371: 
0000/01/01 00:13:06.371: 
0000/01/01 00:13:06.371: 
0000/01/01 00:13:06.372: OK
[... 403 lines cut ...]
0000/01/01 00:14:36.011: 
0000/01/01 00:14:36.011: 
0000/01/01 00:14:36.011: OK
0000/01/01 00:14:36.120: AT+UPSV=4
0000/01/01 00:14:36.170: 
0000/01/01 00:14:36.171: OK
I (878270) GNSS: Continuous location stopped!
0000/01/01 00:14:37.401: AT
0000/01/01 00:14:37.440: 
0000/01/01 00:14:37.441: OK
0000/01/01 00:14:37.441: AT+UPSV=0
0000/01/01 00:14:37.490: 
0000/01/01 00:14:37.491: OK
U_CELL_PWR: powering off using the PWR_ON pin.
0000/01/01 00:14:39.530: 
0000/01/01 00:14:39.531: +UUMQTTC: 0,100
0000/01/01 00:14:40.720: 
0000/01/01 00:14:40.721: +CREG: 0
0000/01/01 00:14:40.721: 
0000/01/01 00:14:40.721: +CEREG: 0,,,,,,,,
-1: NReg
-1: NReg
Device powered off
```
Thanks for your time and patience

RobMeades, 2024-02-20:
Ah, great, that makes things a lot easier to follow.  I will explain in detail what I'm seeing, since I am inferring the behavour of your application and you may have ideas as to what's going on yourself.

Basically, the log says that ESP-IDF (which version, out of interest) mutexes are just not working in your case, which, if true, is quite bizarre.

- An application thread calls [uMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L261), which results in the MQTT connection request `AT+UMQTTC=1`, at `00:13:15.220`; at this point execution of that thread is in [uCellMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L2995) -> [connect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L1327) and the [U_CELL_MQTT_CLIENT_ENTRY_FUNCTION](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L95) macro at the start of [uCellMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L2995) will have called [entryFunction()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L839), which will have locked the cellular API mutex `gUCellPrivateMutex`; the thread is sitting there looping, with that mutex locked, waiting for the response.
- While [uMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L261) is running, another application thread calls [uNetworkInterfaceUp()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/network/api/u_network.h#L172) to bring up the internal GNSS chip within the SARA-R422M8S; doing this brings up the CMUX (see `AT+CMUX=0,0,,128` at `00:13:15.820`).  The effect of this is to swap-out the underlying AT client (by calling [uCellMuxEnable()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mux.c#L1949), which calls the internal function [uCellMuxPrivateEnable()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mux.c#L1601)); the original AT client will remain locked and it will be replaced with a new AT client that pushes commands down the CMUX channel instead, beside the GNSS commands running on another CMUX channel.  The moment this happens in the code is [just here](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mux.c#L1739): you can see that it replaces the `atHandle` variable inside the cellular `pInstance` pointer with the new one.  This is all fine, the `ubxlib` code is designed for thread-safety, that's what the mutexes are for.  But hang on a second...
- **Here's the bit I don't understand**: [uCellMuxEnable()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mux.c#L1949) won't run unless it can get a lock on `gUCellPrivateMutex`, which it should not be able to do because the other thread still has a lock on it.  So if mutexes are working, I don't see how the sequence above can ever happen.
- Anyway, at `00:13:17.251` the `+UUMQTTC: 1,1` comes back; do you have any diagnostic prints in your application that we should see if [uMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L261) has returned at this point?  It would be interesting to know if it has or if, for reasons to do with the CMUX thing above, the URC has been ignored.  My _guess_ is that it probably has returned because...
- At `00:13:20.111` there is an `AT+UMQTTC=4,1`, which would result from an application call to [uMqttClientSubscribe()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/src/u_mqtt_client.c#L499) -> [uCellMqttSubscribe()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L3245C9-L3245C27) -> [subscribe()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L1737); [uCellMqttSubscribe()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L3245C9-L3245C27) is **also** gated by [U_CELL_MQTT_CLIENT_ENTRY_FUNCTION](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L95) and so it could not be done unless `gUCellPrivateMutex` were lockable.
- **Here's the next bit I don't understand**: at `00:13:20.570` there is an `AT+UMQTTC=9,1,0`, which is a call to [uMqttClientPublish()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/src/u_mqtt_client.c#L461) -> [uCellMqttPublish()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L3216) -> [publish()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L1521) and the same thing applies: the thread should not be able to get here as [uMqttClientSubscribe()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/src/u_mqtt_client.c#L499) won't have returned ye [... cut ...] blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L3091) -> [getLastMqttErrorCode()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L318) is **also** guarded by a lock on `gUCellPrivateMutex`, so any thread calling it should be blocked, it should not be possible for all of the `AT+UMQTTER`'s to appear in the log like that.

Basically, all of the thread safety built into the `ubxlib` code is being ignored.

Unh?

tommas0serafin, 2024-02-21:
Hey Rob,
I added some extra prints in my log to clarify the situation. Some assumptions you made were correct but some others unfortunately weren't.

- _Anyway, at 00:13:17.251 the +UUMQTTC: 1,1 URC comes back; do you have any diagnostic prints in your application that we should see if [uMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L261) returns at this point? It would be interesting to know if it does or if, for reasons to do with the CMUX thing above, the URC is being ignored. My guess is that it probably does return because..._

[uMqttClientConnect()] is the clause of a conditional, hence I won't execute the other MQTT operations unless it succeeds

- _Here's the next bit I don't understand: at 00:13:20.570 there is an AT+UMQTTC=9,1,0, which is a call to [uMqttClientPublish()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/src/u_mqtt_client.c#L461) -> [uCellMqttPublish()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L3216) -> [publish()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L1521) and the same thing applies: the thread should not be able to get here as [uMqttClientSubscribe()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/src/u_mqtt_client.c#L499) won't have returned yet because the URC indicating the success of the subscribe has not come back from the module; you can see the module object to this with the +CME ERROR: Operation not allowed: it is still busy doing the subscribe._

[uMqttClientPublish()] basically never fails in terms of sending something, even when the connections seems to be absent. Additionaly it's not mandatory to be subscribed to a topic to use the function so I'm missing your point here.

- _There seem to be occurrences of AT+UMQTTER about every 150 ms: I guess the application is polling [uMqttClientGetLastErrorCode()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L337) in the background or some such? While it can do that, it doesn't mean a great deal until one of the uMqtt API calls has returned an error. Note that the ubxlib code will, internally, do an AT+UMQTTER when an error has occurred anyway, purely to get that information into the AT debug log.
The final bit I've just realised I don't understand: you won't be surprised to hear me say that [uMqttClientGetLastErrorCode()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L337) -> [uCellMqttGetLastErrorCode()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L3091) -> [getLastMqttErrorCode()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L318) is also guarded by a lock on gUCellPrivateMutex, so any thread calling it should be blocked, it should not be possible for all of the AT+UMQTTER's to appear in the log like that._

Here's the big news, I'm never explicitly calling [uMqttClientGetLastErrorCode()] in my whole firmware. I tried in the past but, since it never resets, I couldn't really find an application for it. The errors you see come 100% from some internal calls.

BTW, here I'm sharing two logs of the most common behaviors I experience after the network disconnection:

```
U_CELL: initialising with enable power pin not connected, PWR_ON pin 15 (0x0f) (and is toggled from 1 to 0) and VInt pin 7 (0x07) (and is 1 when module is on).
I (734644) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (734654) gpio: GPIO[7]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
U_CELL_PWR: powering on.
0000/01/01 00:12:11.740: [00]0000/01/01 00:12:11.850: AT
0000/01/01 00:12:12.060: AT
0000/01/01 00:12:12.410: AT
0000/01/01 00:12:12.760: AT
0000/01/01 00:12:13.110: AT
0000/01/01 00:12:14.280: AT

0000/01/01 00:12:14.280: OK
0000/01/01 00:12:14.380: ATE0
0000/01/01 00:12:14.420: ATE0

0000/01/01 00:12:14.420: OK
0000/01/01 00:12:14.520: AT+CMEE=2
0000/01/01 00:12:14.560: 
0000/01/01 00:12:14.560: OK
0000/01/01 00:12:14.660: AT+UDCONF=1,0
0000/01/01 00:12:14.700: 
0000/01/01 00:12:14.700: OK
0000/01/01 00:12:14.800: ATI9
0000/01/01 00:12:14.840: 
0000/01/01 00:12:14.840: 00.12,A00.00
0000/01/01 00:12:14.840: 
0000/01/01 00:12:14.841: OK
0000/01/01 00:12:14.950: AT&C1
0000/01/01 00:12:14.990: 
0000/01/01 00:12:14.990: OK
0000/01/01 00:12:15.090: AT&D0
0000/01/01 00:12:15.130: 
0000/01/01 00:12:15.130: OK
0000/01/01 00:12:15.230: AT+UCGED=2
0000/01/01 00:12:15.270: 
0000/01/01 00:12:15.270: OK
0000/01/01 00:12:15.370: AT&K3
0000/01/01 00:12:15.410: 
0000/01/01 00:12:15.410: OK
0000/01/01 00:12:15.510: AT+UPSV=4
0000/01/01 00:12:15.550: 
0000/01/01 00:12:15.550: OK
0000/01/01 00:12:15.650: AT+UPSMR=1
0000/01/01 00:12:15.690: 
0000/01/01 00:12:15.690: OK
0000/01/01 00:12:15.790: AT+CPSMS?
0000/01/01 00:12:15.830: 
0000/01/01 00:12:15.830: +CPSMS:0,,,"01100000","00000000"
0000/01/01 00:12:15.831: 
0000/01/01 00:12:15.831: OK
0000/01/01 00:12:15.940: AT+CEDRXS?
[... 774 lines cut ...]
U_CELL_MQTT: error codes 13, 0.

0000/01/01 00:13:53.910: 
0000/01/01 00:13:53.910: +UUMQTTC: 1,1

[... cut ...]
0000/01/01 00:15:38.670: AT+UMQTTER
U_CELL_MQTT: error codes -1, -1.
after uMqttClientSubscribe()
before uMqttClientPublish()
0000/01/01 00:15:48.680: AT+UMQTTC=9,1,0,"xxxxxxxxxxxx",1
0000/01/01 00:15:48.730: 9,1,0,"xxxxxxxxxxxxxxxx",10000/01/01 00:17:58.690: AT+UMQTTER
0000/01/01 00:18:08.700: AT+UMQTTER
U_CELL_MQTT: error codes -1, -1.
```

As I already mentioned these issues might be related to some hardware problem (first of all the AT-based power-off fail that forces me to use the PWR_ON pin).

RobMeades, 2024-02-21:
Hmmm, maybe my inference about the way your application is written is wrong, maybe something entirely different is going on here.  A fundamental question: is your application multi-threaded, i.e. are `ubxlib` API calls being made from more than one task, or only from a single task?

I had assumed it must be multi-threaded because what I see in that first log with timestamps, the one I commented on above, is this [removing superfluous line endings]:

```
U_CELL_MQTT: trying to connect...
0000/01/01 00:13:15.220: AT+UMQTTC=1
0000/01/01 00:13:15.261: OK
U_CELL_MQTT: waiting for response for up to 120 second(s)...
0000/01/01 00:13:15.370: AT+UMQTTER
0000/01/01 00:13:15.411: +UMQTTER: 13,0
0000/01/01 00:13:15.412: OK
U_CELL_MQTT: error codes 13, 0.
0000/01/01 00:13:15.520: AT+UMQTTER
0000/01/01 00:13:15.561: +UMQTTER: 13,0
0000/01/01 00:13:15.562: OK
0000/01/01 00:13:15.670: AT+UMQTTER
0000/01/01 00:13:15.711: +UMQTTER: 13,0
0000/01/01 00:13:15.712: OK
U_CELL_MQTT: error codes 13, 0.
I (797680) GNSS: Bringing up GNSS
0000/01/01 00:13:15.820: AT+CMUX=0,0,,128
```

a) On the face of it there has been an attempt to connect (`AT+UMQTTC=1`, which will be [uMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L261) -> [uCellMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L2995) meaning `gUCellPrivateMutex` is locked) and, while waiting for the connection to succeed or fail (i.e. to get the `+UUMQTTC: 1,x` URC), 600 millisecond later, GNSS is being brought up ([uNetworkInterfaceUp()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/network/api/u_network.h#L172) -> [uCellMuxEnable()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mux.c#L1949) [`AT+CMUX=0,0,,128`] and that **has** to have locked `gUCellPrivateMutex`).  It seemed to me that mutexes cannot be working if that sequence is to have happened.

(b) Same for the MQTT subscribe operation that follows: this is not about MQTT-protocol publish/subscribe rules, the `AT+UMQTTC=4,1` at `00:13:20.111` ([uMqttClientSubscribe()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/src/u_mqtt_client.c#L499) -> [uCellMqttSubscribe()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L3245C9-L3245C27) so `gUCellPrivateMutex` is now locked) is still running when there is an `AT+UMQTTC=9,1,0`, just 459 milliseconds later at `00:13:20.570` ([uMqttClientPublish()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/src/u_mqtt_client.c#L461) -> [uCellMqttPublish()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L3216), which **has** to have locked `gUCellPrivateMutex`).  You can see that the module says `+CME ERROR: Operation not allowed` because it is still processing the publish request, that transaction has not yet completed, so it is not as though we have lost the URC or something, the module has not sent it.

**HOWEVER**

If you are not calling [uMqttClientGetLastErrorCode()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/common/mqtt_client/api/u_mqtt_client.h#L337), the only reason the `ubxlib` would be calling [getLastMqttErrorCode()](https://github.com/u-blox/ubxlib/blob/a8ed25d09feb2f28584f753dc170f95dff4f8166/cell/src/u_cell_mqtt.c#L318) internally is if it thought the MQTT operation had failed, and pretty much immediately at that.  A reason it might think this is if the UART had _thought_ it had received something, effectively a phantom URC.  Thing is, the URC processing looks for the prefix string, so `+UUMQTTC:` the phantom URC would have to be quite specific.  And it should print it in the log.

You've been hinting at a possible HW problem.

> first of all the AT-based power-off fail that forces me to use the PWR_ON pin

Can you explain more about this please?  I'm guessing at `AT+CPWROFF` is failing?  That's not something I've ever heard of before, maybe if we focus on that it will turn up a clue.

RobMeades, 2024-02-21:
One other thing to confirm, in case of a misunderstanding: in the [uMqttClientConnection_t](https://github.com/u-blox/ubxlib/blob/9694ef9d70e6e7d90562174e8c4d52555d8730da/common/mqtt_client/api/u_mqtt_client.h#L74) structure you pass to [uMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/9694ef9d70e6e7d90562174e8c4d52555d8730da/common/mqtt_client/api/u_mqtt_client.h#L261), I believe you are setting a `pKeepGoingCallback` function; can you confirm that function is returning `true`?

tommas0serafin, 2024-02-21:
Hey Rob,
So the application has two threads, one for the MQTT tasks and one that handles the GNSS. 

- _Can you explain more about this please? I'm guessing that AT+CPWROFF is failing? That's not something I've ever heard of before, maybe if we focus on that it will turn up a clue._

The power-off with the AT commands (i.e. AT+CPWROFF) always fails. With that I mean the modules immediately powers down but, I guess because of the UART-related current leak, it stays in this in-between state where it can still be poked by the library checks. However when the power-off API tries with the hard power-off via PWR_ON pin instead of shutting down, the modules turns on again (V_INT comes back to 1.8v). 

- _One other thing to confirm, in case of a misunderstanding: in the [uMqttClientConnection_t](https://github.com/u-blox/ubxlib/blob/9694ef9d70e6e7d90562174e8c4d52555d8730da/common/mqtt_client/api/u_mqtt_client.h#L74) structure you pass to [uMqttClientConnect()](https://github.com/u-blox/ubxlib/blob/9694ef9d70e6e7d90562174e8c4d52555d8730da/common/mqtt_client/api/u_mqtt_client.h#L261), I believe you are setting a pKeepGoingCallback function; can you confirm that function is returning true?_

The callback function I wrote actually checks the status of the network from the devHandle and might return true or false accordingly. Is this a bad idea?

Thanks again for your support

tommas0serafin, 2024-02-21:
Hey Rob,

- _It is certainly not necessary: the module operation will fail anyway if the network status is such that the operation cannot succeed and calling into a ubxlib API function from a callback out of the ubxlib API can lead to problems: some API functions involve AT commands that have URC-style responses that get put into a callback queue, a queue which your callback may already be in, leading to the API locking up. Better just to use the callback as your application's timeout/watchdog/abandon-that-operation-please control based on information local to it._

The +UMQTTER: 13,0 was related to the callback, Now that I set it to false I'm not experiencing the issue anymore.

- _That is very strange indeed, seems as though the module is actively able to draw power from the UART IO lines. The pulse on the PWR_ON HW pin is a toggle (i.e. if the module is off it switches it on, if the module is on it switches it off), which is why the ubxlib code always checks if the module is up first by bouncing AT off it; if responsiveness to an AT check isn't reliable proof that the module is powered on or off, things are going to get very confused indeed. Not sure I can advise you how to proceed on how HW issues though, not really my area I'm afraid._

About this point, I think neither of us is actually supposed to solve it. 

Thanks again for your time,

Take care

closed 2024-02-21 by tommas0serafin

RobMeades, 2024-02-21:
That's excellent news, I leave you in the capable hands of your HW people :-).