u-blox/ubxlib #203: Set up PointPerfect with MQTT client using ubxlib

juncheng-dd, 2024-02-20:
Hi, I was playing with the ZED-F9P evaluation kit(EVK-F9P) and was able to build a host application which reads UBX-NAV-PVT message from the device using ubxlib. The next step I would like to achieve is to set up PointPerfect service for the RTK correction to achieve more accurate output.

Following the start up guide(https://developer.thingstream.io/guides/location-services/pointperfect-getting-started), I was able to configure a MQTT client using u-center. I am wondering whether we could use ubxlib to set up a MQTT client for the PointPefect service. If yes, would it be possible to share some examples to help me get started? Thanks

dbou-ublox, 2024-02-20:
Hi there,

maybe too complicated for your case, but there is c code that wraps around ubxlib and controls ZED-F9 and PointPerfect components (it works out of the box for the XPLR-HPG kits featuring ESP32 MCUs, but maybe you could look at that as a reference)
Here it is: https://github.com/u-blox/XPLR-HPG-software

cturvey, 2024-02-20:
Check also  
https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/tree/main/examples/NEO-D9S_and_NEO-D9C
https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/tree/main/examples/ZED-F9P/Example18_PointPerfectClient

juncheng-dd, 2024-02-20:
Hi @cturvey @dbou-ublox , thanks both for the examples! I will take a look

juncheng-dd, 2024-03-07:
I was able to use a third party MQTT library to achieve what I want: https://github.com/eclipse/paho.mqtt.c. Close this issue then

closed 2024-03-07 by juncheng-dd