u-blox/ubxlib #204: how to enable multiple selected messageId.id.pNmea msg types

mabroens, 2024-02-21:
Hi Rob,

i was looking for some more documentation to messageId.id.pNmea.

at the moment i have following:
```
messageId.type = U_GNSS_PROTOCOL_NMEA;
messageId.id.pNmea = "G?GGA";
```
but i would also like to have: `messageId.id.pNmea = "G?RMC";`

how can i enable this?
pNmea is not a list type and `messageId.id.pNmea = NULL; `does not give me the G?RMC.

what is the best way to get only specific msg types?

Br Martijn

RobMeades, 2024-02-21:
Hi Martijn: you could to this two ways:

(a) make another call to [uGnssMsgReceiveStart()](https://github.com/u-blox/ubxlib/blob/e46ddee8ff53f4fcd5acdd93d4abc6c4e9f4483d/gnss/api/u_gnss_msg.h#L324) for "G?RMC" (don't worry about resources, internally it will use the same callback task as your "G?GGA" is already using) and either do a [uGnssMsgReceiveStop()](https://github.com/u-blox/ubxlib/blob/e46ddee8ff53f4fcd5acdd93d4abc6c4e9f4483d/gnss/api/u_gnss_msg.h#L419) for both when done or just call [uGnssMsgReceiveStopAll()](https://github.com/u-blox/ubxlib/blob/e46ddee8ff53f4fcd5acdd93d4abc6c4e9f4483d/gnss/api/u_gnss_msg.h#L431) which will stop both of them,

(b) you could set the filter to `messageId.type = U_GNSS_PROTOCOL_NMEA; messageId.id.pNmea = "";` (i.e. all NMEA messages) and do the filtering yourself in your `uGnssMsgReceiveCallback_t` callback by checking the contents of the `pMessageId` pointer it is passed, e.g. with [uGnssMsgIdIsWanted()](https://github.com/u-blox/ubxlib/blob/e46ddee8ff53f4fcd5acdd93d4abc6c4e9f4483d/gnss/api/u_gnss_msg.h#L161) in a loop for a list of wanted message IDs, or by hand in your own way if you prefer: this might be a very _slightly_ higher processor load but, in the end, the filtering is always going to be in C code on the MCU and so whether you do it in your application or `ubxlib` does it is a moot point.

RobMeades, 2024-02-21:
I have added this advice to `master` here, see https://github.com/u-blox/ubxlib/commit/9694ef9d70e6e7d90562174e8c4d52555d8730da.

RobMeades, 2024-02-21:
Re-reading your issue:

> pNmea is not a list type and messageId.id.pNmea = NULL; does not give me the G?RMC.

Setting `messageId.id.pNmea = NULL`, rather than `messageId.id.pNmea = ""`, would cause the NMEA message type to be up the first five characters of whatever "string" is at address 0 on your MCU, which would explain why you don't get "G?RMC".

RobMeades, 2024-03-19:
Hi Martijn: did you resolve this issue or is there more to do?

closed 2024-03-19 by mabroens