u-blox/ubxlib #205: capture all UBX data from GNSS module

mabroens, 2024-02-23:
Hi Rob,

for my validation i woul dlike to printout all data in UBX format and prpint that to the console of my app.
im using zephyer and for NMEA this is working but ubx format should give me more information on SNR and spectrum analyser stuff.

i started by using your code example: dec_main.c
and was able to get that to work. it nicely prints:
```
Setting module type to M10
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type UART.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 00 00 00 00 00 00 00 00 00 00 00 00 00 0 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00[body 120 byte(s)].
U_GNSS: sent command b5 62 06 04 04 00 00 00 09 00 17 76.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 00 00 00 00 00 00 00 00 00 00 00 00 00 0 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00[body 120 byte(s)].
Device opened successfull.
U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 07 00 91 20 01 54 51.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
UTC time 1708697665000002466 nanoseconds.
UTC time 1708697668000003683 nanoseconds.
UTC time 1708697669000004087 nanoseconds.
UTC time 1708697669999800621 nanoseconds.
UTC time 1708697670999800997 nanoseconds.
UTC time 1708697671999801372 nanoseconds.
UTC time 1708697672999801748 nanoseconds.
UTC time 1708697673999802122 nanoseconds.
UTC time 1708697674999802497 nanoseconds.
UTC time 1708697675999802870 nanoseconds.
UTC time 1708697676999803243 nanoseconds.
UTC time 1708697677999803616 nanoseconds.
UTC time 1708697678999803990 nanoseconds.
13 UBX-NAV-PVT message(s) decoded.
Done.
```

but i would like it 
1.  to be streaming until i poweroff
2. to not only get PVT msg but all.

for 2) i changed:  `messageId.id.ubx = 0x0107; // The message class/ID of a UBX-NAV-PVT message`
to:   `messageId.id.ubx = U_GNSS_UBX_MESSAGE_ALL; // for all UBX format messages`

but i think i still only get PVT out. atleast when i seta breakpoint in the callback and watch `pDec`.
[image]

am i missing something?
and what is the best way to print the ubx to console such that u-Center will eat it?

Br Martijn

mabroens, 2024-02-23:
and is there a function that decodes the pvt content to human readebl for printing to log file?

RobMeades, 2024-02-23:
Hi again Martijn: the `ubxlib` code will only print out messages that the GNSS chip sends it and, by default, I don't believe any of the UBX messages are enabled in the GNSS chip.  The messages can be large and there are many of them; you will likely find that enabling all of them will lead to character loss on the communications link.

Which GNSS chip is this with?  The mechanism to enable the messages is different depending on whether you are using series 8, 9 or 10.

mabroens, 2024-02-23:
Hi again :-),

Fair point Rob, im using the M10.
but i think the pvt is enough. but i'll ask my FAE for this.I will also have a look at the EVK as ther i can see the spectrum analyser work.

mabroens, 2024-02-23:
cool thankx this will already help me out. let me try before you need to do anything.
and same here no idea what data uBlox unit spits out in UBX format. when i connect to the EVK i can see non human readable data. need to dive into this. but the goals is that.

other question where can i find more information on the MSGOUT types mentioned in [see the list in u_gnss_cfg_val_key.h](https://github.com/u-blox/ubxlib/blob/9694ef9d70e6e7d90562174e8c4d52555d8730da/gnss/api/u_gnss_cfg_val_key.h#L653).  ?

and if i would like the datarate to be 10Hz what should i do?
`uGnssCfgSetMsgRate(devHandle, &messageId, 10)`  or is there another function to do so?

mabroens, 2024-02-24:
hi Rob, 

again thnks for the feedback.

my M10 is running at 38400 so that should be fine. this speed is btw the def for the chip. perhaps that the modules use 9600 but the chip itself is using 38400.

and yes the epcoch need to look into that one. not fam. with it yet.

i'll dive into that.
thx

mabroens, 2024-03-03:
@RobMeades 
> 
> As to printing out messages so that u-Center will eat them, I'm not actually a u-Center user myself, so I don't even know what the format is! I guess such a thing could be provided though, I can put it on the list - how soon do you need it?

Would it be possible for you give this a try? i could not get any further then printing the decoded msg.

this is what i did so far:
```
static void callback(uDeviceHandle_t devHandle, const uGnssMessageId_t *pMessageId,
                     int32_t errorCodeOrLength, void *pCallbackParam)
{
    char *pBuffer = (char *) pCallbackParam;
    int32_t length;
    uGnssDec_t *pDec;
    char prefix[2] = {0};
    int32_t whole[2] = {0};
    int32_t fraction[2] = {0};

#ifndef U_CFG_TEST_USING_NRF5SDK
    uGnssDecUbxNavPvt_t *pUbxNavPvt;
    int64_t utcTimeNanoseconds;
#endif

    (void) pMessageId;

    if (errorCodeOrLength >= 0) {
        // Read the message into our buffer
        length = uGnssMsgReceiveCallbackRead(devHandle, pBuffer, errorCodeOrLength);
        if (length >= 0) {
            // Call the uGnssDec() API to decode the message
            pDec = pUGnssDecAlloc(pBuffer, length);
            if ((pDec != NULL) && (pDec->errorCode == 0)) {
                gDecCount++;
                // No need to check pDec->id (or pMessageId) here since we have
                // only asked for UBX-NAV-PVT messages.
#ifndef U_CFG_TEST_USING_NRF5SDK // NRF52 goes a bit crazy if you print here
                pUbxNavPvt = &(pDec->pBody->ubxNavPvt);
                // Do stuff with the contents
                utcTimeNanoseconds = uGnssDecUbxNavPvtGetTimeUtc(pUbxNavPvt);
                prefix[0] = latLongToBits(pUbxNavPvt->lon, &(whole[0]), &(fraction[0]));
                prefix[1] = latLongToBits(pUbxNavPvt->lat, &(whole[1]), &(fraction[1]));

                if (utcTimeNanoseconds >= 0) {
                    // This print will only do anything useful if you have
                    // a uPortLog() which supports 64-bit integers
                    // uPortLog("UTC time %lld nanoseconds.\n", utcTimeNanoseconds);

                   // Here we should just print teh binary data such that uCentre understands it.
                    printf("lon: %c%d.%07d, lat %c%d.%07d fixType: %dD, pDOP:%d, sAcc:%dm, vAcc:%dm\n",prefix[1], whole[1], fraction[1], prefix[0], whole[0], fraction[0], pUbxNavPvt->fixType, pUbxNavPvt->pDOP, pUbxNavPvt->sAcc,pUbxNavPvt->vAcc);
                } else {
                    uPortLog("UTC time not available.\n");
                }
#endif
            }
            // Must *always* free the memory that pUGnssDecAlloc() allocated
            uGnssDecFree(pDec);
        }
#ifndef U_CFG_TEST_USING_NRF5SDK // NRF52 goes a bit crazy if you print here
    } else {
        uPortLog("Empty or bad message received.\n");
#endif
    }
}
```

I was thinking. when i us uCenter to configure the M10 for UBX format then disconnect the uart port and open it with putty i can see binary data. So simply stated i would say when you just print the the binary data that could be a starting point.
otherwise it might just be a conversion that needs to be added or is this to simple??

Meanwhile i will also try to figure out if and where i can find information on the ubx format comming from the uBlox devcie.

RobMeades, 2024-03-03:
> Meanwhile i will also try to figure out if and where i can find information on the ubx format comming from the uBlox devcie.

The UBX-format messages are all defined in the [interface manual](https://www.u-blox.com/sites/default/files/u-blox-M10-SPG-5.10_InterfaceDescription_UBX-21035062.pdf), section 3.

I will capture a log from a device and see what it looks like: if it is similar to the `.ucf` format then it would appear to be some form of JSON, should be easy enough to print that out.

RobMeades, 2024-03-04:
Find here:

https://github.com/u-blox/ubxlib/tree/preview_feature_gnss_ucenter_script_rmea

...a preview branch which contains a Python (3) script [u_gnss_create_ucenter_ubx_file.py](https://github.com/u-blox/ubxlib/blob/preview_feature_gnss_ucenter_script_rmea/gnss/api/u_gnss_create_ucenter_ubx_file.py) which will take the `ubxlib` log output, find all of the `UBX` message exchanges in it and write these to a file which the u-blox [uCenter tool](https://www.u-blox.com/en/product/u-center) can open.

For instance, cut and paste any log output you have into a file `log.txt` and run the script with:

```
python u_gnss_create_ucenter_ubx_file.py log.txt log.ubx
```

...then open `log.ubx` in uCenter.  By default it will include commands sent to the GNSS device as well as the responses: add the command-line switch `-r` to include only the responses.

Give it a try and let me know if it works for you, all feedback welcomed.  When you are satisfied I will push it to `master` here and delete the preview branch.

RobMeades, 2024-03-18:
Hi again Martijn.  Further to our discussion I've updated the script on [the preview branch](https://github.com/u-blox/ubxlib/tree/preview_feature_gnss_ucenter_script_rmea), script now renamed to `u_gnss_ucenter_ubx.py`  ('cos it doesn't just do files anymore), so that it can also be given a serial device as input, instead of a saved file, and, on Linux, it can write its output to a `pty` (note: it needs an installation of the Python module `pyserial` to do this (`pip3 install pyserial`)); run the script with `-h` for help.

The problem, of course, is that Python **only** allows the creation of virtual serial ports (the `pty`) on Linux, there is no way to do this on Windows, and `uCenter` is a Windows application.  Do you have any way that your `uCenter` installation could be connected across to a Linux `pty`?

If not, there is one other possibility, which would be for you to install [com0com](https://com0com.sourceforge.net/) on Windows.  Then I could change the script so that it would write its output to an _existing_ serial port (rather than create a virtual serial port of its own), which could be one end of a `com0com` looped-back serial port, then you could connect uCenter on Windows to the other end of the same `com0com` looped-back serial port.  I have heard of some people having problems getting `com0com` running, so you'd best confirm that it does something for you before we decide to try it.  FYI, there is also Eltima's [Virtual Serial Port](https://www.eltima.com/products/vspdxp/) if you prefer, but that costs money.

Thoughts?

closed 2024-03-29 by RobMeades