u-blox/ubxlib #211: Problems selecting cipher suites

tobiasadolph, 2024-03-05:
Hello, we are using a SARA-R422S modem.
When connecting to a server we need to modify the supported cipher suites on the modem as our server doesn't support the default ones by the modem. Unfortunately we can't get it to work.

First we created our own default security TLS settings based on `U_SECURITY_TLS_SETTINGS_DEFAULT` from ubxlib:

``` c
#define EXTENDED_U_SECURITY_TLS_SETTINGS_DEFAULT {U_SECURITY_TLS_VERSION_ANY, /* tlsVersion */                          \
                                                NULL, /* Root CA name */                                              \
                                                NULL, /* Client CA name */                                            \
                                                NULL, /* Private key name */                                          \
                                                U_SECURITY_TLS_CERTIFICATE_CHECK_NONE,                                \
                                                NULL, /* Private key PW */                                            \
                                                {/* Cipher suites */                                                  \
                                                 35, /* Number of cipher suites in list */                             \
                                                 {                                                                    \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256, \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384, \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256, \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384, \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_ECDSA_WITH_AES_128_CCM,        \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_ECDSA_WITH_AES_256_CCM,        \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_RSA_WITH_AES_128_CBC_SHA256,   \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_RSA_WITH_AES_256_CBC_SHA384,   \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_RSA_WITH_AES_128_GCM_SHA256,   \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_RSA_WITH_AES_256_GCM_SHA384,   \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_PSK_WITH_AES_128_CBC_SHA256,   \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDHE_PSK_WITH_AES_256_CBC_SHA384,   \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_DSS_WITH_AES_128_CBC_SHA256,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_RSA_WITH_AES_128_CBC_SHA256,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_RSA_WITH_AES_256_CBC_SHA256,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_RSA_WITH_AES_128_GCM_SHA256,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_RSA_WITH_AES_256_GCM_SHA384,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_PSK_WITH_AES_128_CBC_SHA256,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_PSK_WITH_AES_256_CBC_SHA384,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_PSK_WITH_AES_128_GCM_SHA256,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_PSK_WITH_AES_256_GCM_SHA384,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_PSK_WITH_AES_128_CCM,            \
                                                     U_SECURITY_TLS_CIPHER_SUITE_DHE_PSK_WITH_AES_256_CCM,            \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDH_ECDSA_WITH_AES_128_CBC_SHA256,  \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDH_ECDSA_WITH_AES_256_CBC_SHA384,  \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDH_ECDSA_WITH_AES_128_GCM_SHA256,  \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDH_ECDSA_WITH_AES_256_GCM_SHA384,  \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDH_RSA_WITH_AES_128_CBC_SHA256,    \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDH_RSA_WITH_AES_256_CBC_SHA384,    \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDH_RSA_WITH_AES_128_GCM_SHA256,    \
                                                     U_SECURITY_TLS_CIPHER_SUITE_ECDH_RSA_WITH_AES_256_GCM_SHA384,    \
                                                     U_SECURITY_TLS_CIPHER_SUITE_RSA_PSK_WITH_AES_128_CBC_SHA256,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_RSA_PSK_WITH_AES_256_CBC_SHA384,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_RSA_PSK_WITH_AES_128_GCM_SHA256,     \
                                                     U_SECURITY_TLS_CIPHER_SUITE_RSA_PSK_WITH_AES_256_GCM_SHA384,     \
                                                 }},                                                                  \
                                                {NULL, 0}, /* PSK */                                                  \
                                                {NULL, 0}, /* PSK ID */                                               \
                                                false, /* pskGeneratedByRoT */                                        \
                                                NULL, /* Expected server URL */                                       \
                                                NULL, /* SNI */                                                       \
                                                false, /* Session resumption */                                       \
                                                false, /* use device certificate */                                   \
                                                false}; /* include CA certificates */
```

To increase the array of `uSecurityTlsCipherSuiteIana_t` in `uSecurityTlsCipherSuites_t` we increased the define `U_SECURITY_TLS_MAX_NUM_CIPHER_SUITES` accordingly.

Based on our custom default tls settings we now try to connect to a server: (Simplified code snipped)

``` c
int Socket = socket(AF_INET, U_SOCK_TYPE_STREAM, U_SOCK_PROTOCOL_TCP);
if (Socket >= 0)
{
    UBXLIB_fcntlSetFlags(F_SETFL, O_NONBLOCK)
[... 134 lines cut ...]

OK
AT+USECPRF=0,2,99,"00","ad"

OK
AT+USECPRF=0,4,"REMOVED"

OK
AT+USECPRF=0,0,3

OK
AT+USECPRF=0,10,"REMOVED"

OK
AT+USOSEC=0,1,0

OK
```

When sniffing our connection with wireshark we noticed the client hello only includes the last cipher suite in the list `U_SECURITY_TLS_CIPHER_SUITE_RSA_PSK_WITH_AES_256_GCM_SHA384` and not the whole list:

[image]

How do we properly setup the cipher suite list on our modem?
Thank you very much in advance for the help.

tobiasadolph, 2024-03-05:
Thank you very much for the blazing fast response. That is indeed unfortunate news.

So for my understanding, the modem supports a list of cipher suites by default, but doesn't support a custom list. Then only one cipher suite is support at the same time?
This isn't a behavior I expected, but it is what it is. Any chance this will be changed with a future software update of the modem?

Do you see any other way than trying to connect with one cipher suites all by one until one works? Maybe caching the result could help for the next connection.

In addition, I think your proposed change of the API is a good idea 👍

RobMeades, 2024-03-05:
Now you're getting into the sticky details of TLS negotiation, which I can't say I'm 100% sure of. I would have _thought_ that the set of "automatically" supported cipher suites would be sent to the server in the "Client Hello" and, if there were some cross-over between what your server supports and the ones in that list, then it would propose those back again and things would work.  I guess from what you're saying that is not the case.

I can probably run Wireshark on our test echo server and run a TLS test with SARA-R422S to find out what does happen, will take a look.  I will also see if I can't rustle up someone who actually knows what they are talking about...

tobiasadolph, 2024-03-05:
I already did the Wireshark capture. By default our modem sends these 37 cipher suites in the client hello:

[image]

These are also mentioned in the AT commands manual and marked with and (D) for the SARA-R422S modem.

[image]

As these default ones don't overlap with the desired server (server is not in our control) the TLS handshake failes because no shared cipher was found

RobMeades, 2024-03-05:
[wireshark.log](https://github.com/u-blox/ubxlib/files/14497175/wireshark.log)

Aha, yes, just did that same thing and obtained the Wireshark log [attached] from our echo server, which indeed looks the same as yours.  Darn.  Is your server newer or older, i.e. might there be a chance that a later FW revision of SARA-R422S happens to pick up later cipher suites which might work?  What FW version do you have (response to `ATI9`, should be queried and printed-out by `ubxlib` near boot)?

tobiasadolph, 2024-03-05:
Problem is, we have to connect to different servers. I could check whether we have and match between those, but that's not really a future prove option. These cipher suites can change at any time on the server side assuming they might me considered insecure in the future or any other reason... .

RobMeades, 2024-03-05:
Understood.  I've asked the application engineer internally who owns SARA-R422 if he knows of any ways this might be made to work; will get back to you.

tobiasadolph, 2024-03-05:
Perfect, thank you very much.

When there is no other way we might have to implement some kind of mechanism on our side to try out the cipher suites one by one in every connect as mention above.

RobMeades, 2024-03-05:
Oh, and he doesn't believe the set of initial cipher suites have changed but he does recommend using the latest module FW, which can be found here:

https://www.u-blox.com/en/product/sara-r4-series?legacy=Current#Documentation-&-resources

tobiasadolph, 2024-03-06:
Ok, thanks for the update. Then we have to implement the proposed workaround.
Do you have any plans on when the proposed API change will be implemented? Maybe we can additionally update the ubxlib in this step.

Regarding the update. I have an old modem with 00B hardware on my desk which can't be updated to my knowledge.
But a colleague of mine tested it with the 01B hardware and the problem remains.

Thanks for your support!

RobMeades, 2024-03-06:
> But a colleague of mine tested it with the 01B hardware and the problem remains.

Understood.  The API change has been tested and is in review now, should be here very shortly.

RobMeades, 2024-03-06:
Now pushed to `master` here in commit 27c75a8b3ec24762d9dd1bbe402c8de0a73cb505 with a comment-only update to the same code in commit b747695052aad1884196d66d057d51b307dde59e.

tobiasadolph, 2024-03-06:
That was fast 😀
I will close this issue as I know how to proceed and thanks again for your effort

closed 2024-03-06 by tobiasadolph