u-blox/ubxlib #216: SPI interface for short-range devices

CameronMahaffey, 2024-03-19:
Are there any plans to add SPI support for communicating with your short-range devices (e.g. NINA-W156)? Maybe there is another library that uBlox has to already support this?

Thanks :)

RobMeades, 2024-03-19:
Hi and thanks for question.  Tagging product manager for our short-range modules

manemannen, 2024-03-20:
Hi, no other library for this I'm afraid. However, we would consider adding this to ubxlib. See [here](https://content.u-blox.com/sites/default/files/u-connectXpress_SPIPeripheral_ProtocolSpec_UBX-20028725.pdf) for details and let me know if it would be useful. If so, I'll add it to the backlog.

jay-aechh, 2024-06-19:
> Hi, no other library for this I'm afraid. However, we would consider adding this to ubxlib. See [here](https://content.u-blox.com/sites/default/files/u-connectXpress_SPIPeripheral_ProtocolSpec_UBX-20028725.pdf) for details and let me know if it would be useful. If so, I'll add it to the backlog.

Hi this would be very useful. Are there currently plans to add SPI support for short-range devices (e.g. NINA-W156) in ubxlib?

manemannen, 2024-07-04:
We will soon plan the scope for the next release. This is a hot candidate

closed 2024-11-11 by manemannen