u-blox/ubxlib #219: Function rand() do not provide a return status

abattistello-secpat, 2024-03-27:
The int rand() function in zephyr [port]([port/platform/zephyr/src/u_port_clib.c](https://github.com/u-blox/ubxlib/blob/857e2a6641d747302bc0bcdbce9930965e99b65e/port/platform/zephyr/src/u_port_clib.c#L92)) does not provide a return status. Thus a possible problem or malfunction of the hardware generator is not handled. 
So for example if the hardware random number generator is stuck and always produces the same value (thus it is not random), the user would not be aware of it.

RobMeades, 2024-03-29:
Hi and thanks for posting.  This is true but, on the other hand, Zephyr is not our code: you could try posting an issue there: https://github.com/zephyrproject-rtos/zephyr.

RobMeades, 2024-03-29:
I will close this: please feel free to re-open if there is more to discuss.

closed 2024-03-29 by RobMeades