u-blox/ubxlib #220: C2C seems not to have a session counter

abattistello-secpat, 2024-03-27:
Up to tag v1.2.0, the example code [provided](https://github.com/u-blox/ubxlib/blob/v1.2.0/example/security/c2c/c2c_main.c) for the chip 2 chip security, does not mention a session counter, or alternatively some random nonce used to differentiate one session from another. It thus seems possible to replay previous messages exchanged during a legitimate session, to reissue the same commands.

RobMeades, 2024-03-29:
Thanks for the interest but the chip to chip security feature was removed back in commit 26392366cd9723826b915506e28c5a8c046d63ab, mid 2023.

RobMeades, 2024-03-29:
I will close this: please feel free to re-open if there is more to discuss.

closed 2024-03-29 by RobMeades