u-blox/ubxlib #222: Support for NEO-F9P-15 Base & Rover

michaelboeding, 2024-03-28:
It would be very beneficial to have a way to send the messages needed to configure and manage/use the NEO-F9P-15, specifically setting up a NEO-F9P in rover mode and base station mode.. Maybe this is located in a separate repo somewhere but I haven't found it. I also haven't found great documentation on all of the UBX messages for the NEO-F9P-15.  https://content.u-blox.com/sites/default/files/documents/NEO-F9P_IntegrationManual_UBX-22028362.pdf

RobMeades, 2024-03-28:
Hi Michael: all of the NEO-F9P message types _should_ be there, though the link you have pasted above is to the integration manual rather than the interface manual, which can be found here:

https://content.u-blox.com/sites/default/files/documents/u-blox-F9-HPG-1.32_InterfaceDescription_UBX-22008968.pdf

I've had a quick scan through and didn't spot any missing `UBX-CFG-VALXXX` configuration messages, which would be the main source of issues: can you point me at the missing ones?

cturvey, 2024-03-28:
The NEO-F9P would be the HPG 1.40 document for the NEO/ZED L1/L5 platform.
Moving Base is not supported.

RobMeades, 2024-03-28:
Apologies for the mis-direction, the link for the HPG 1.40, for completeness, is:

https://content.u-blox.com/sites/default/files/documents/u-blox-F9-HPG-L1L5-1.40_InterfaceDescription_UBX-23006991.pdf

I _think_ the `UBX-CFG-VALXXX` configuration message IDs there are all supported in `ubxlib`; @michaelboeding do let me know if that is not the case.

michaelboeding, 2024-03-29:
Thanks for that @cturvey and @RobMeades . So basically everything I need to configure can just be sent over UBX messages? Are there any plans to make a more friendly API for maybe the Rover/Base station setup?

RobMeades, 2024-03-29:
> So basically everything I need to configure can just be sent over UBX messages?

Yes, for configuration see the pattern [here](https://github.com/u-blox/ubxlib/blob/050e8edb9cb95e874602f8a52134a5e0812d59f3/example/gnss/cfg_val_main.c#L273).

> Are there any plans to make a more friendly API for maybe the Rover/Base station setup?

It is not something I know anything about, so wouldn't know what shape the API would be: if you had a proposal I'd be happy to create it but, on the other hand, I don't have a test setup for this so it would likely make more sense for you to create something you would actually be able to test yourself.

michaelboeding, 2024-03-29:
Got it - I'll write my own "friendly" APIs for this then using this format. And then I would be happy to share them with you to maybe integrate a version into ubxlib later on. I'm waiting on my second NEO-F9P-15 currently, once I have that I can test the whole system. I will also probably try to integrate some type of post processing via RTKLIB, which @cturvey pointed me to.

RobMeades, 2024-03-29:
>  And then I would be happy to share them with you to maybe integrate a version into ubxlib later on

That would be excellent thank you.  While we're talking, are you done with [preview_cell_ppp_rmea](https://github.com/u-blox/ubxlib/tree/preview_cell_ppp_rmea)?  Would just like to have a tidy-up.

michaelboeding, 2024-03-29:
Yes, sorry I meant to let you know about that. It has been working great!

RobMeades, 2024-03-29:
Phew, one less thing to worry about :-).

RobMeades, 2024-05-10:
Going to close this one as I believe you know how to proceed with it now.  Please feel free to re-open this issue if there is more to discuss on the subject.

closed 2024-05-10 by RobMeades